fwupd/fwupd #1940: P52s update is older than installed & TPM PCR0 differs from reconstruction

vincegio, 2020-04-01:
**Describe the bug**
Attempting to update firmware for SSD, update is available:
```shell
$ fwupdmgr get-updates -v
(fwupdmgr:3814): GLib-DEBUG: 10:10:38.995: setenv()/putenv() are not thread-safe and should not be used after threads are created
• Thunderbolt Controller has the latest available firmware version
(fwupdmgr:3814): FuMain-DEBUG: 10:10:39.005: current version is 20.00: 20.00=same, 18.00=older
• System Firmware has the latest available firmware version
(fwupdmgr:3814): FuMain-DEBUG: 10:10:39.006: current version is 0.1.21: 0.1.21=same, 0.1.20=older, 0.1.19=older, 0.1.18=older, 0.1.16=older
• UEFI Device Firmware has the latest available firmware version
(fwupdmgr:3814): FuMain-DEBUG: 10:10:39.007: current version is 184.70.3626: 184.70.3626=same, 184.65.3590=older, 184.60.3561=older, 184.55.3510=older
• UEFI Device Firmware has no available firmware updates
(fwupdmgr:3814): FuMain-DEBUG: 10:10:39.008: WDC PC SN720 SDAQNTW-512G-1001
  DeviceId:             be29aadda98b637d33b6b740ff7b367f61886a31
  Guid:                 137520ce-3603-53e6-9165-56694ed744e7
  Guid:                 ff2112dc-038c-596d-90ca-d43c5077c6ec
  Guid:                 63af1441-fb24-5368-b450-886fbc98610b
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Protocol:             org.nvmexpress
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot|usable-during-update
  Vendor:               Sandisk Corp
  VendorId:             NVME:0x15B7
  Version:              10160101
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2020-04-01

(fwupdmgr:3814): XbSilo-DEBUG: 10:10:39.008: compiling 8e56617f4ade069ab8be5e3a3a90165be0968efd…
(fwupdmgr:3814): FuMain-DEBUG: 10:10:39.008:     PCSN720 Device Update:
      New version:       10190101
      Remote ID:         lvfs
      Summary:           WD PCSN720 Firmware Update for Lenovo PC
      License:           Proprietary
      Size:              2,3 MB
      Vendor:            Western Digital
      Flags:             is-upgrade
      Description:       Do NOT turn off your computer or remove the AC adapter while update is in progress.
      
      The computer shall be restarted after updating firmware completely. The device may not properly function until you shut down or reboot PC.
      
      Supported devices and firmware version : WDC PCSN720 NVMe SSD Firmware for Lenovo PC
      
      Supported Product Scope : Lenovo ThinkPad, ThinkCentre, ThinkStation, IdeaCentre

  └─PCSN720 Device Update:
        New version:       10190101
        Remote ID:         lvfs
        Summary:           WD PCSN720 Firmware Update for Lenovo PC
        License:           Proprietary
        Size:              2,3 MB
        Vendor:            Western Digital
        Flags:             is-upgrade
        Description:       Do NOT turn off your computer or remove the AC adapter while update is in progress.
        
        The computer shall be restarted after updating firmware completely. The device may not properly function until you shut down or reboot PC.
        
        Supported devices and firmware version : WDC PCSN720 NVMe SSD Firmware for Lenovo PC
        
        Supported Product Scope : Lenovo ThinkPad, ThinkCentre, ThinkStation, IdeaCentre
```

```shell
$ fwupdmgr update -v
(fwupdmgr:3841): GLib-DEBUG: 10:11:03.809: setenv()/putenv() are not thread-safe and should not be used after threads are created
• Thunderbolt Controller has the latest available firmware version
(fwupdmgr:3841): FuMain-DEBUG: 10:11:03.818: current version is 20.00: 20.00=same, 18.00=older
• System Firmware has the latest available firmware version
(fwupdmgr:3841): FuMain-DEBUG: 10:11:03.820: current version is 0.1.21: 0.1.21=same, 0.1.20=older, 0.1.19=older, 0.1.18=older, 0.1.16=older
• UEFI Device Firmware has the latest available firmware version
[... 117 lines cut ...]
│     GUIDs:               fed2efa4-7045-55a1-b4fc-b29283d59fe5
│                          8b72a10c-1279-5f8e-a28a-34fb11a58240
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│   
└─WDC PC SN720 SDAQNTW-512G-1001:
      Device ID:           be29aadda98b637d33b6b740ff7b367f61886a31
      Summary:             NVM Express Solid State Drive
      Current version:     10160101
      Vendor:              Sandisk Corp (NVME:0x15B7)
      GUIDs:               137520ce-3603-53e6-9165-56694ed744e7
                           ff2112dc-038c-596d-90ca-d43c5077c6ec
                           63af1441-fb24-5368-b450-886fbc98610b
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Supported on remote server
                           • Needs a reboot after installation
                           • Device is usable for the duration of the update
```

**Additional questions**
- Operating system and version: **Arch 5.4.28-2-lts**
- Have you tried rebooting? **Yes**
- Is this a regression? **No idea**

hughsie, 2020-04-01:
I've spotted an LVFS bug. Can you clear the LVFS cache ~/.cache/fwupd then refresh the metadata with `fwupdmgr refresh --force` then try updating again? Thanks.

closed 2020-04-01 by hughsie

vincegio, 2020-04-01:
Works, thanks @hughsie

vincegio, 2020-04-01:
Should I send in another issue regarding TPM PCR0 diff?

hughsie, 2020-04-01:
> Should I send in another issue regarding TPM PCR0 diff?

No thanks -- Lenovo are aware of the problem and are prepping new firmware updates for all affected hardware.

vincegio, 2020-04-01:
Great! Thanks for the info.

superm1, 2020-04-01:
> > Should I send in another issue regarding TPM PCR0 diff?
> 
> No thanks -- Lenovo are aware of the problem and are prepping new firmware updates for all affected hardware.

So should we move all those PCR0 bugs outside of fwupd into the lenovo issues repo or close them in fwupd repo?

hughsie, 2020-04-01:
> into the lenovo issues repo

Exactly that I think. Lenovo acknowledge it's a firmware bug. Not sure if I can share details.