fwupd/fwupd #1945: get-updates should report the date the update was released

lovesegfault, 2020-04-01:
**Is your feature request related to a problem? Please describe.**
I've had Lenovo bios updates brick my laptop on a few occasions, and now I don't update my bios for at least a month after the firmware is released, just as a precaution. 

**Describe the solution you'd like**
Following up with [this tweet](https://twitter.com/lovesegfault/status/1245146638688997376?s=20), I'd like for `fwupd get-updates to also report the date an update was made available.

On a secondary note, a config option to only show firmware updates that have been available for a minimum of x days would also solve the issue for me.

**Describe alternatives you've considered**
I currently do `get-updates`, then go to Lenovo's support website and find the same patch, and later add a reminder on my calendar to do the update a few weeks later.

**Additional context**
[image]

hughsie, 2020-04-02:
> I've had Lenovo bios updates brick my laptop on a few occasions

!!! more details required !!!

closed 2020-04-02 by hughsie

lovesegfault, 2020-04-02:
> > I've had Lenovo bios updates brick my laptop on a few occasions
> 
> !!! more details required !!!

This was before I started using LVFS and fwupd. I wrote about the accidents on my blog: https://lovesegfault.com/post/2019-01-24-thinkpad-nightmare/