fwupd/fwupd #1952: weird compatibility error when running fwupdmgr

owlen, 2020-04-04:
I'm trying to install firmware on a T470s thankpad, with a new installation of kubuntu.

```
$ fwupdmgr install 9f5919c3e922568e8e6793bbf1ed4b3e916399ab1076f7dcacf5a7cf6cdfd514-Lenovo-ThinkPad-T470s-SystemFirmware-1.36.cab 
Decompressing…           [***************************************]
Not compatible with firmware version 0.1.4, requires >= 0.1.14
```
The file is taken from [LVFS](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN1WET.firmware).

Note that the message says "requires **0.1.4** or higher" and wile the version mentioned *is* **0.1.4**.

I'm also trying to upgrade the Embedded Controller Firmware with the file from [LVFS](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN1WHT.firmware) an getting this:
```
$ fwupdmgr install fd56cad96f216b4607434a202edd0c0e122d43c3-Lenovo-ThinkPad-T470s-ECFirmware-1.21.cab 
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Installing on UEFI Device Firmware…                       -      ]
Scheduling…              [***************************************]
Secure boot is enabled, but shim isn't installed to the EFI system partition
```

I reached this step after solving some issues by following advice around here, such as tool version, snap tweaks etc. Previously, before these recent tweaks, it was complaining that the current system firmware installed is to old (1.0.4) to be upgraded. I'm not sure if it's still the case, but currently the error message (first one above) is weird. 

My questions are: 
- Any idea why the I get the " version 0.1.4, requires **>=** 0.1.14"? Should this be posted as a bug?
- Any idea how to continue with the installation? 

I have no Microsoft anything, and would really rather avoid installing any.

BTW, I was able to start the fwupd daemon. The discover utility shows no updates for the firmware.

```
Operating System: Kubuntu 10
KDE Plasma Version: 5.16.5
KDE Frameworks Version: 5.62.0
Qt Version: 5.12.4
Kernel Version: 5.3.0-45-generic
OS Type: 64-bit
Processors: 4 × Intel® Core™ i5-7200U CPU @ 2.50GHz
Memory: 7.5 GiB of RAM
```

```shell
$ lsb_release -a
LSB Version:    core-11.0.1ubuntu1-noarch:security-11.0.1ubuntu1-noarch
Distributor ID: Debian
Description:    Debian GNU/Linux 10 (buster)
Release:        10
Codename:       buster

$ fwupdmgr --version
client version: 1.3.9
compile-time dependency versions
        gusb:   0.2.11
        efivar: 34
daemon version: 1.3.9

$ fwupdmgr update
• Thunderbolt Controller has no available firmware updates
• SAMSUNG MZVLW512HMJP-000L7 has no available firmware updates
• System Firmware has no available firmware updates
• UEFI Device Firmware has no available firmware updates
• UEFI Device Firmware has no available firmware updates
• Unifying Receiver has no available firmware updates

```

hughsie, 2020-04-04:
Weird. Can you compile fwupd `1_3_X` from git and add some debugging around fu_engine_check_requirement_firmware please?

superm1, 2020-04-04:
Doesn't the message say you need 0.1.14 or higher not 0.1.4?

owlen, 2020-04-04:
Ah, sorry @hughsie, I think @superm1 is right, and all this digging around made me nauseous... 
Yes, I do have 1.0.4. and I do need 0.1.14 or higher. The message is not weird.

So no I'm left with the big question: How can I upgrade the firmware? The oldest version for Linux is 1.32 and I have a feeling that it wouldn't allow updating as well.

hughsie, 2020-04-04:
Hah, I misread it too. You need to either use winflash or the bootable cd to update to a new enough 'old' version that fwupd can use. It's s Lenovo issue, there's not much fwupd can do. If you force it from fwupd, you might end up with a brick.

owlen, 2020-04-04:
Would that be winflash by Asus? Would it be fair to assume that I'll need to install windows on the machine for that?

hughsie, 2020-04-05:
The windows binary that Lenovo produce, indeed. Try the ISO method first, but please read the README file carefully.

superm1, 2020-04-10:
Closing as invalid for now since
`Not compatible with firmware version 0.1.4, requires >= 0.1.14` is being done correctly.

closed 2020-04-10 by superm1