fwupd/fwupd #1956: Please also install the unit tests

rossburton, 2020-04-06:
fwupd has two test suites: a set of unit tests (as ran by `meson test`) and an integration test (which is installed).

In some situations, such as in cross-compilation, it's useful to be able to also install the unit tests so that they can be executed on the target.  Please consider also installing the unit tests.

superm1, 2020-04-06:
Several projects have started to do this, I don't think there is any problem with it.  If you've got some cycles, definitely happy to review a PR.

FFY00, 2020-04-06:
I think this should be optional, adding a meson option should be pretty easy.

hughsie, 2020-08-18:
Where would we install the local self tests? Would they be called by a new `installed-test` as well?

superm1, 2020-08-18:
For libjcat we did:
`/usr/libexec/installed-tests/libjcat/`

superm1, 2020-08-28:
should be fixed in master now

superm1, 2020-08-28:
Something to note, as part of this change many binaries were removed from tree and won't be installed.  You can fetch them at runtime using `contrib/ci/get_test_firmware.sh`

closed 2020-08-28 by superm1