fwupd/fwupd #1973: Regression from 9e755e2a5

superm1, 2020-04-08:
I noticed that 9e755e2a is causing HID++ peripherals to not enumerate again.  I have two errors in my daemon logs now ending in "no HID++ ID"

hughsie, 2020-04-09:
Seemed to work here:

    devices/logitech-k780.json:
    Running test on Logitech K780
    [INFO]     Current version MPK01.03_B0024
    [INFO]     Installing version MPK01.02_B0021
    [SUCCESS]  Installed MPK01.02_B0021
    [INFO]     Installing version MPK01.03_B0024
    [SUCCESS]  Installed MPK01.03_B0024

Got logs?

superm1, 2020-04-09:
I think it might only be happening when devices are asleep.
[output.txt](https://github.com/fwupd/fwupd/files/4457355/output.txt)

superm1, 2020-04-09:
In which case it's probably too verbose is the issue

superm1, 2020-04-09:
In that sample output, the keyboard is asleep and it's what is showing the error about I think, since it doesn't have a HID++ ID.

hughsie, 2020-04-10:
Is the device enumerated correctly if it goes from sleep -> awake?

superm1, 2020-04-10:
Do you mean while the daemon is running?  As I can tell I'd guess no - we're not listening for any sort of events like that yet from the kernel DJ driver .@FFY00?  

If I wake and call fwupdtool definitely avoid the errors though.

FFY00, 2020-04-10:
If we enumerate the devices in when we start the daemon, if the devices goes from sleep to awake after the daemon started, I would guess they are not enumerated. We would have tried to talk to the device, without luck as it was asleep, and the as far as I can tell, we don't try to ping it at a later state.

superm1, 2020-04-10:
So after that kernel stuff lands we'll probably need to have the daemon just try to enumerate and ping devices again when those events land to improve that.  Otherwise we're looking at polling them from the daemon (yuck).

FFY00, 2020-04-10:
Yes, we will retry pinging the device on udev update events.

PR #1984 (2020-04-14): logitech_hidpp: decrease verbosity of messages that hid++ ID is missing
These are a regression of 9e755e2a5 when devices are asleep.
However due to the current kernel and daemon architecture, logitech devices
are not checked again at any time so if the device isn't awake when
fwupd is started or the unifying dongle is plugged in it won't be present.

This will be changed in the future when the kernel has change events
associated with devices waking up.

Fixes: #1973

Type of pull request:
- [ ] New plugin (Please include [new plugin checklist](https://github.com/hughsie/fwupd/wiki/New-plugin-c [... cut]
commit de60e044 (2020-04-14): logitech_hidpp: decrease verbosity of messages that hid++ ID is missing

These are a regression of 9e755e2a5 when devices are asleep.
However due to the current kernel and daemon architecture, logitech devices
are not checked again at any time so if the device isn't awake when
fwupd is started or the unifying dongle is plugged in it won't be present.

This will be changed in the future when the kernel has change events
associated with devices waking up.

Fixes: #1973
closed 2020-04-14 by superm1