fwupd/fwupd #1975: dell_dock: several devices don't remove on unplug

superm1, 2020-04-08:
With current master I'm finding that the EC device, package level device and VMM5331 are not getting removed when unplugged now.

superm1, 2020-04-08:
And plugging back in it's not enumerating correctly too.

hughsie, 2020-04-08:
Do we call `fu_device_list_emit_device_removed()` on the package level child device? Some logs would be great if you have them.

superm1, 2020-04-08:
Hopefully can do one better.  With such an easy to repro condition bisect is easy.
```
d$  git bisect log
git bisect start
# bad: [6c0e3d4efa8d138540e8c8447e0e87fb3de8b0a8] Do not crash the daemon if a plugin does something dumb
git bisect bad 6c0e3d4efa8d138540e8c8447e0e87fb3de8b0a8
# good: [1e571730d0b39574c3d7f2d0cc38e7c859205dbc] Fix a daemon crash when removing children
git bisect good 1e571730d0b39574c3d7f2d0cc38e7c859205dbc
# good: [49797d7de4a45981cc271fb22474c2b78ffeb0c2] ccgx: Reset the i2c bridge on status failure
git bisect good 49797d7de4a45981cc271fb22474c2b78ffeb0c2
# bad: [2671aaceb26a9b6874468c6569465a0e16f5c6b7] synaptics-cxaudio: Only use the PATCH2 address when the patch level is 2
git bisect bad 2671aaceb26a9b6874468c6569465a0e16f5c6b7
# good: [674478d4b83f91168c8b55f0a29ee513c32d9d0a] bash: add completion for fwupdtool reinstall
git bisect good 674478d4b83f91168c8b55f0a29ee513c32d9d0a
# bad: [60103e22070fa7edffa3b84b4f5e0b8ba4b9ef0f] ccgx: Use Silicon ID and Application type to set flash parameters
git bisect bad 60103e22070fa7edffa3b84b4f5e0b8ba4b9ef0f
# good: [80d9b7450ce7f0172775513e81040a6a6dcab635] logitech_hidpp: set the bootloader protocol
git bisect good 80d9b7450ce7f0172775513e81040a6a6dcab635
# good: [3dbb92e5d5fac147eb5b6163ecd0d9e775024ac3] trivial: fu-config: Fix behavior for UpdateMotd key
git bisect good 3dbb92e5d5fac147eb5b6163ecd0d9e775024ac3
# bad: [b4f14e8f0f242c2367a80f9e1768723ff261e34b] Only set the parent ID when adopting children
git bisect bad b4f14e8f0f242c2367a80f9e1768723ff261e34b
# good: [61f74077edb1c64cc82b7cbb60f0f7c0ed732aee] trivial: dfu: drop quirk to hide DW1820A
git bisect good 61f74077edb1c64cc82b7cbb60f0f7c0ed732aee
# first bad commit: [b4f14e8f0f242c2367a80f9e1768723ff261e34b] Only set the parent ID when adopting children
```

superm1, 2020-04-08:
> Do we call `fu_device_list_emit_device_removed()` on the package level child device? Some logs would be great if you have them.

Dell dock is a little weird here I think.  When we get the remove event, it's the one from the HID bridge we're looking for.  That HID bridge event removal is supposed to be found in the cache and then force a removal of the parent.  The parent removing is supposed to take out the whole chain.

hughsie, 2020-04-08:
>  When we get the remove event, it's the one from the HID bridge

Remind me of the internal topology. Is the HID bridge a child of a virtual device? In this new world order the children an "loosely" defined with the parent ID rather than being explicit children, so I can see why this would break that case.

I think this only used to work by accident if I'm honest. Shouldn't the virtual devices be children of this physical device? I though FuDellDockHub was the parent, and the HID device... If not then fu_plugin_device_removed() in fu-plugin-dell-dock.c probably needs to remove the parent like now, and also the child devices that are not the same as the `@device` passed in.

tbh, I'm a bit confused what all the stuff in fu_plugin_usb_device_added() and fu_plugin_device_removed() in fu-plugin-dell-dock.c is actually doing. If FuDellDockHub is the parent and the other devices are explicit children I think the daemon should do the right thing without all that custom code...

superm1, 2020-04-08:
Yes the HID bridge is a child of a virtual device representation.  
For fwupd the topology is:
```├─WD19TB:
│ │   Device ID:           570867f4ddda9915445da59bd85cd0e0e507270e
│ │   Summary:             High performance dock
│ │   Current version:     01.00.00.04
│ │   Minimum Version:     01.00.00.00
│ │   Vendor:              Dell Inc. (USB:0x413C)
│ │   Install Duration:    1 minute
│ │   GUID:                cd357cf1-40b2-5d87-b8df-bb2dd82774aa
│ │   Device Flags:        • Updatable
│ │                        • Requires AC power
│ │                        • Supported on remote server
│ │                        • Device stages updates
│ │                        • Device can recover flash failures
│ │                        • Device is usable for the duration of the update
│ │
│ ├─Thunderbolt controller in Dell dock:
│ │     Device ID:         8357922f7f208e1f3b7537141ce17db04bc219d9
│ │     Summary:           Thunderbolt controller
│ │     Current version:   43.00
│ │     Vendor:            Dell Inc. (TBT:0x00D4)
│ │     Install Duration:  22 seconds
│ │     GUID:              c94770ca-1773-592c-b20a-e87243bc7cd0
│ │     Device Flags:      • Updatable
│ │                        • Requires AC power
│ │                        • Supported on remote server
│ │                        • Device stages updates
│ │
│ ├─Package level of Dell dock:
│ │     Device ID:         b5a854df85ab30fff48e4365ec7ad2249c90806e
│ │     Summary:           A representation of dock update status
│ │     Current version:   01.00.14.01
│ │     Vendor:            Dell Inc. (USB:0x413C)
│ │     Install Duration:  5 seconds
│ │     GUID:              8ceeeffd-51b6-580c-9b75-69143227aff8
│ │     Device Flags:      • Updatable
│ │                        • Supported on remote server
│ │                        • Device can recover flash failures
│ │                        • Device is usable for the duration of the update
│ │
│ ├─RTS5413 in Dell dock:
│ │     Device ID:         0acceea54e71c4d8002593822afe0f4705616613
│ │     Summary:           USB 3.1 Generation 1 Hub
│ │     Current version:   01.21
│ │     Vendor:            Dell Inc. (USB:0x413C)
│ │     Install Duration:  14 seconds
│ │     GUIDs:             86fb40c0-8bf5-5a8b-a4ad-3156cf6bfaf4
│ │                        b27d25f1-019d-5718-b41a-02ddaefe5577
│ │                        ac5b774c-b49d-566b-9255-85f0f7f8a4ed
│ │     Device Flags:      • Updatable
│ │                        • Requires AC power
│ │                        • Supported on remote server
│ │                        • Device stages updates
│ │                        • Device is usable for the duration of the update
│ │
│ ├─RTS5487 in Dell dock:
│ │     Device ID:         96969b82d977fdcbf607df084ef0dcf10119409b
│ │     Summary:           USB 3.1 Generation 2 Hub
│ │     Current version:   01.47
│ │     Vendor:            Dell Inc. (USB:0x413C)
│ │     Install Duration:  3 seconds
│ │     GUIDs:             707c63d2-e597-5c40-84db-9b1bb4c48d96
│ │                        acfcd89b-105d-55b9-b85b-08bf8508f38c
│ │                        568ffa1e-a0db-5287-9ea3-872b60f7730b
│ │     Device Flags:      • Updatable
│ │                        • Requires AC power
│ │                        • Supported on remote server
│ │                        • Device stages updates
│ │                        • Device is usable for the duration of the update
│ │
│ └─VMM5331 in Dell dock:
│       Device ID:         228ac6846b64d532dd1ea1b7e651e8f5f55fbc34
│       Summary:           Multi Stream Transport controller
│       Current version:   05.04.03
│       Vendor:            Dell Inc. (USB:0x413C)
│       Install Duration:  6 minutes
│       GUID:              89fec0b6-6b76-5008-b82c-5e5c6c164007
│       Device Flags:      • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│                          • Device is usable for the duration of the update
```

>I think this only used to work by accident if I'm honest. Shouldn't the virtual devices be children of this physical device? I though FuDellDockHub was the parent, and the HID device... 

No, this is all very intentional.  The EC is the central controller of the dock, it enables and disables writes for all of the other devices (physical or virtual).  It just so happens the way to get to the EC from the OS is "through" the HID bridge.

>If not then fu_plugin_device_removed() in fu-plugin-dell-dock.c probably needs to remove the parent like now, and also the child devices that are not the same as the @device passed in.

I added some debugging lines to `fu_plugin_device_removed` and noticed that parent is `NULL` on the RTS5487 Hub (which happens to be the HID bridge).  So the part that is missing is setting the parent on that particular child, which is what was happening when adopting previously IIRC.

I think that `fu_engine_adopt_children` might just want to do `fu_device_set_parent` instead of the parent ID like it is right now.

hughsie, 2020-04-08:
I've got some ideas for tomorrow. I don't think setting a parent object without it being a child is a good idea.

hughsie, 2020-04-09:
Okay, new day, new brain. Some nomenclature I just came up with.

 * Strong child = is referenced as a child object, child has parent object set. Child is destroyed with parent. One FuDeviceItem in device list.
 * Weak child = is connected just by the ParentGuid for UI. Child can come and go and parent is unaffected. Multiple FuDeviceItems in device list.

Before we had a mixture where children were sometimes strong, and also where children were also weak.  This is what caused the confusion and the double-removal bugs. The dell dock is unusual in that we want to remove the virtual parent when the physical child is removed. The regression is my fault, I thought the HUB device was the parent. My apologies.

I think for dell dock we want:

 * EC <- HUB = weak
 * EC <- MST = strong
 * EC <- status = strong

What I think is happening now is, on USB add:

 * we add the hub device
 * we add the hub to the cache with the hub ID as the cache key
 * we create the EC virtual device
 * the EC child status and EC MST children get added as strong children
 * we add the EC device (and children get added too)

This means that the EC and hub are only connected by a parent GUID -- which looks right in the UI, but the EC device doesn't get cleaned up when the HUB device goes away. When this is done I think this happens:

 * in `fu_plugin_device_removed` we look in the cache for the hub device, which we find
 * we get the _strong_ parent of the hub device (which isn't set anymore...)
 * if set, we remove the EC device, and its children too

I think this is just a logical mistake; what we want to do is add an entry into the cache as before -- but use the HUB device_id as the key, and point to the newly created **EC** device. This way on removal we get the HUB device id, which we look up in the cache and get the EC device which we remove, which removes the child devices.

I have to switch to daddy mode now, but can work on a PR in an hour or so.

superm1, 2020-04-09:
fixed in https://github.com/fwupd/fwupd/pull/1979

closed 2020-04-09 by superm1