fwupd/fwupd #1992: 1.4.1 shows firmware version in binary

Sean-StarLabs, 2020-04-14:
Shows version: "16973824"
Should be: "1.3.0"

```
└─System Firmware:
      Device ID:           5984fe62b2de5dfa0c7980660e230707d696c8dd
      Current version:     16973824
      Minimum Version:     16973824
      Vendor:              Star Labs (DMI:American Megatrends Inc.)
      GUID:                8265d473-a6c2-42b4-897b-bc220faa2d32
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Needs a reboot after installation
                           • Cryptographic hash verification is available
                           • Device is usable for the duration of the update
    

```
```
client version:	1.4.1
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.4.1

```

superm1, 2020-04-14:
This is intentional.  The ESRT doesn't have information on how to interpret the 32 bit integer.  This is OEM specific and device specific.  In the past quirks were stored in fwupd to set on some common OEMs and fall back to triplet if not set which was incorrect.  Now the version format is stored in the CAB that you pull down from LVFS.

So after you run a refresh cycle for the first time on a supported device the version format from LVFS should be applied to the device.

Sean-StarLabs, 2020-04-14:
Gotcha.

So presumably, we just need to add `<value key="LVFS::VersionFormat">triplet</value>` to the latest cab?

superm1, 2020-04-14:
Well LVFS should be applying it automatically when you upload, but if you are meaning local testing before it's gone to LVFS, yes adding version format to the CAB would work too.

hughsie, 2020-04-14:
@Sean-StarLabs I've set your vendor default on the LVFS to be `triplet` for now, but indeed, you really need to set LVFS::VersionFormat to triplet in future metainfo files. Do you want me to regenerate your embargo remote so that these changes "stick"?