fwupd/fwupd #1995: Add client API to correlate fwupd devices with physical devices

sjoerdsimons, 2020-04-14:
We're doing a frontend for a device using fwupd to update its (usb) input devices (plugin for that should follow later); Typically there will be multiple device attached at the same time in which case we want to be able to indicate to the user which device is getting updated (or let the user pick one). 

The frontend knows which devices are around from udev etc, however there isn't a way to correlate those back to the device id's as listed by fwupd.  A possible solution would be to simply expose the internal physical id (see draft PR #1994 ), which solved my immediate issue.

However the physical id's fwupd uses internally seem rather arbitrary and depends on implementation details. Even for USB it will be different if the plugin uses fu-udev-device (`DEVPATH=<udev path>`) while when using fu-usb-device it's `usb:<bus>:<platform id>`.. So it's not great general API, even if the user of this information will be device sepcific anyway, having something consistent would be good :)

superm1, 2020-04-14:
> We're doing a frontend for a device using fwupd to update its (usb) input devices (plugin for that should follow later); Typically there will be multiple device attached at the same time in which [...]

This exact same problem exists for stuff today too.  The most obvious one I'm aware of is the Unifying Receiver.  If you have two connected, it's impossible to tell which is which, even if you know the device path.  

Do you have a way to influence the device HW architecture still?  Something that would be cool in this instance is if you can try to light up an LED to indicate when something was picked.  We don't have the right APIs in fwupd for this now, but that's just plumbing and could probably be pretty useful.

> However the physical id's fwupd uses internally seem rather arbitrary and depends on implementation details. Even for USB it will be different if the plugin uses fu-udev-device (DEVPATH=<udev path>) [...]

Yup, that was my concern too.  We might need to unify what the plugins are doing to make it more consistent when showing to a user if we're going this route.

In the case of your device, what actually will be put in physical ID?  To my question on the PR, can it nicely correlate to something in sysfs or udev?

sjoerdsimons, 2020-04-14:
It's a standard USB device; so the physical id at the moment refers back to the usb device (through fu-usb-device setting the physical id), which on the client side is enough to correlate it back to the physical device and work out which leds, evdev device etc it belongs to. 

How exactly the frontend will let the end-user identify a device i'm not sure of yet (I'm not doing that part), but the devices do have several leds that could be blinked or otherwise indicate which device is which.

Being able to indicate in a standard way from fwupd could be nice for sure to make it easier for "general" frontends. But it's a bit orthorgonal to my request here as we've got a more specialized UI which already knows which input devices exists (as it looks at udev itself) and would like to pull in the info from fwupd which ones are updatable as well as being able to start the update for a specific device ;)

hughsie, 2020-04-14:
> However the physical id's fwupd uses internally seem rather arbitrary

Right, this is why it wasn't exported. Can you share more details about what kind of hardware your updating, and why we can't just update all of them at the same time to a new version? Thanks.

hughsie, 2020-04-14:
> Being able to indicate in a standard way from fwupd could be nice for sure to make it easier for "general" frontends

I'd be 100% okay with that kind of API addition.

sjoerdsimons, 2020-04-14:
> > However the physical id's fwupd uses internally seem rather arbitrary
> 
> Right, this is why it wasn't exported. Can you share more details about what kind of hardware your updating, and why we can't just update all of them at the same time to a new version? Thanks.

The end-user flow for this isn't fully finalized yet; Not having this information available severely limits the options though; Without this you can't:
* Let the user choose which a device will be updated
* Let the user know which firmware version a given device has (without pulling that 
   seperately from sysfs)
* Let the user know which device is updating (modulo the earlier API suggestion, but that's limited
   to e.g.  blink a led when updating start rather then the UI saying device 3 is updating now) 
* Let the user know which device has an update failure etc.

hughsie, 2020-04-14:
What device is bring updated? Why can't we just update all of them at the same time? Why do we care which device is which?

superm1, 2020-04-14:
I would guess at least one reason to care is so that you don't unplug one in the middle of the update accidentally.  A message saying "Don't unplug your Foo mouse 2.0 and Bar keyboard 5" is clearer than "don't unplug any of these 4 devices".

hughsie, 2020-04-14:
But we know the device name already. How many keyboards do you have plugged in?

superm1, 2020-04-14:
I guess another way to resolve this problem is to let the fwupd plugin itself set more information in the description of the device. It can easily look in udev just like your fatter client would otherwise do to associate the other end points.

Then it becomes a thinner client and can just show description field.

hughsie, 2020-04-15:
> let the fwupd plugin itself set more information in the description of the device

Yes, I'd much prefer this. i.e. they'd be a udev rule that matched some attribute and set some ENV key on the device, which FuUdevDevice read and set the `Name` or `Summary` for instance.

superm1, 2020-06-30:
I'm going to mark this closed since we have a direction that is preferred.  If you think it's worth revisiting, re-open at that time.

closed 2020-06-30 by superm1