fwupd/fwupd #1998: Dynabook (né Toshiba) X30/X40 firmware version quirk entry

yrro, 2020-04-15:
**Describe the bug**
The Dynabook X30 and X40 (and probably other hardware models) combine the version of the "BIOS" and the "EC" into the version field of the `ESRT` entry in BCD format.

After flashing with version 3.80 and 2.90, respectively...

[image]

The following quirk gets `fwupd` to display both versions correctly:

```
[Guid=28108d08-5027-42c2-a5b8-92d6ede9b97b]
VersionFormat = bcd
```

Resulting in:

```
└─System Firmware:
      Device ID:           41c7c0481777610a3aa56cd85affca0935b6207b
      Current version:     3.80.2.90
      Vendor:              TOSHIBA (DMI:TOSHIBA)
      GUID:                28108d08-5027-42c2-a5b8-92d6ede9b97b
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Needs a reboot after installation
                           • Cryptographic hash verification is available
                           • Device is usable for the duration of the update
```

**fwupd version information**
```shell
$ fwupdmgr --version
client version:	1.3.9
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.9
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
`apt` on a Debian system

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
TECRA X40-D
│
├─Thunderbolt Controller:
│     Device ID:           1502b03bcef94ed5877ab7fbc3719ad914d11c7c
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     19.00
│     Vendor:              TOSHIBA (TBT:0x014A)
│     GUIDs:               0180d7eb-8f28-5b69-8d8e-8ba8a7c4da7c
│                          91cc0f76-7da2-57f0-b588-a9d2f1a756e5
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─HD Graphics 620:
│     Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
│     Current version:     02
│     Vendor:              Intel Corporation (PCI:0x8086)
│     GUIDs:               2886e312-afd8-5510-8993-12d568a85e00
│                          38f6c009-a25c-5b50-b3b5-fef4f9c6b846
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
[... 600 lines cut ...]
│       ├── xnu.mod
│       ├── xnu_uuid.mod
│       ├── xnu_uuid_test.mod
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       ├── zfs.mod
│       └── zstd.mod
├── initrd.img-5.4.0-3-amd64
├── initrd.img-5.4.0-4-amd64
├── lost+found
├── System.map-5.4.0-3-amd64
├── System.map-5.4.0-4-amd64
├── vmlinuz-5.4.0-3-amd64
└── vmlinuz-5.4.0-4-amd64

59 directories, 482 files
```

**Additional questions**
- Operating system and version: Debian testing/unstable
- Have you tried rebooting? N/A
- Is this a regression? No
- Are you using an NVMe disk? Yes
- Is secure boot enabled? Yes

hughsie, 2020-04-15:
I guess there's a few ways to handle this. Do you know if TOSHIBA is planning to upload firmware to the LVFS for those models? It might seem a random question but it could affect how we fix this :)

yrro, 2020-04-15:
Sadly I've no reason to think so.

In case it would be helpful, here's a de-obfuscated copy of the vbscript that orchestrates the update process.

https://robots.org.uk/ToshibaX30X40FirmwareUpdate?action=AttachFile&do=view&target=autoExec.vbs.html

I think the same script runs on hardware both ancient and new. It appears to make decisions about which update method to use based on calling methods of some proprietary COM objects bundled with the updater package. Search for `MachineInfo(` to get an idea of the things it checks. I didn't dive in any deeper than that.

hughsie, 2020-04-15:
> Sadly I've no reason to think so.

 :(

> here's a de-obfuscated copy of the vbscript

Well, that was terrifying to read.

I guess we can add your quirk to plugins/uefi/uefi.quirk -- could you create a PR that does that please? TIA.

PR #2000 (2020-04-15): uefi: add quirk for correctly format firmware version of Dynabook (né…
… Toshiba) X30, X40 and probably others

Fixes #1998

Type of pull request:
- [ ] New plugin (Please include [new plugin checklist](https://github.com/hughsie/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit 6a161aa1 (2020-04-15): uefi: correctly format firmware version of Dynabook (né Toshiba) X30, X40 and probably others

Fixes #1998
closed 2020-04-15 by hughsie