fwupd/fwupd #2031: Malformed firmware update description in Lenovo ThinkPad Yoga370 Thunderbolt Firmware

Peetz0r, 2020-04-24:
[This firmware update](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR0HTF19W.firmware) has some `Important Notice[WARNINGS]` in the description. But those seem to be malformed in the XML (missing the `<ul>` around the `<li>`'s), and aren't displayed in any user-visible way. The website, Gnome Software, and `fwupmgr` don't show the list items.

[screenshot](https://i.imgur.com/2FQ6eYj.png)

XML snippet:
```xml
<description>
<p>Lenovo ThinkPad Yoga370 Thunderbolt Firmware</p>
<p>Important Notice[WARNINGS] :
<li>Before update, set the Thunderbolt BIOS Assist Mode to "Disable" in BIOS setup under Config > Thunderbolt 3.</li>
<li>DO NOT FORCE UPDATE Thunderbolt Controller. This may damage the firmware.</li>
</p>
</description>
</release>
```

At this point software versions seem almost irrelevant since the website has the same issue, but anyway: 

```shell
$ fwupdmgr --version 
client version:	1.3.9
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.9
```
This is on Fedora 31, Gnome Software is 3.34.2.

hughsie, 2020-04-24:
I've fixed up the description manually, but it will take another 4 hours for the metadata to regenerate to fix the local issue. I'm debugging why the XML was parsed at all, it should have been rejected at upload time.

hughsie, 2020-04-27:
Server-side fix here, just deployed: https://github.com/fwupd/lvfs-website/pull/593

closed 2020-04-27 by hughsie