fwupd/fwupd #2032: Error on refresh - Ubuntu 20.04

sbmw, 2020-04-26:
**Describe the bug**
On upgrade to Ubuntu 20.04 (clean install) errors are thrown on fwupd refresh

**Steps to Reproduce**

Command:

`fwupd.fwupdmgr refresh --force`

Response:

```
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.jcat

(process:7909): GLib-GObject-WARNING **: 13:58:45.646: ../../../gobject/gsignal.c:2617: signal 'changed::' is invalid for instance '0x5606ac1ce430' of type 'GSettings'

(process:7909): GLib-GObject-WARNING **: 13:58:45.646: ../../../gobject/gsignal.c:2617: signal 'changed::' is invalid for instance '0x5606ac1cec80' of type 'GSettings'

(process:7909): GLib-GObject-WARNING **: 13:58:45.647: ../../../gobject/gsignal.c:2617: signal 'changed::' is invalid for instance '0x5606ac1cedc0' of type 'GSettings'

(process:7909): GLib-GObject-WARNING **: 13:58:45.647: ../../../gobject/gsignal.c:2617: signal 'changed::' is invalid for instance '0x5606ac1ce700' of type 'GSettings'

(process:7909): GLib-GObject-WARNING **: 13:58:45.647: ../../../gobject/gsignal.c:2617: signal 'changed::' is invalid for instance '0x5606ac1ce5c0' of type 'GSettings'
Downloading…             [***************************************]
Fetching metadata https://cdn.fwupd.org/downloads/firmware-00513-stable.xml.gz
Downloading…             [***************************************]
Successfully downloaded new metadata: 6 local devices supported
```

**fwupd version information**
```shell
client version: 1.4.0
compile-time dependency versions
        gusb:   0.2.11
        efivar: 34
daemon version: 1.4.0
```
Snap install, with the default apt install first removed. 

Ubuntu 20.04, kernel 5.6.7

hughsie, 2020-04-26:
Does it happen with the non-snap version?

sbmw, 2020-04-27:
Yes, I'm afraid so.

```
sudo snap remove fwupd

sudo apt-get install fwupd -y

fwupdmgr --version  
client version: 1.3.9
compile-time dependency versions
        gusb:   0.3.4
        efivar: 37
daemon version: 1.3.9

fwupdmgr refresh --force     
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz

(process:8256): GLib-GObject-WARNING **: 12:01:07.222: ../../../gobject/gsignal.c:2617: signal 'changed::' is invalid for instance '0x558932d8d430' of type 'GSettings'

(process:8256): GLib-GObject-WARNING **: 12:01:07.223: ../../../gobject/gsignal.c:2617: signal 'changed::' is invalid for instance '0x558932d8dc30' of type 'GSettings'

(process:8256): GLib-GObject-WARNING **: 12:01:07.223: ../../../gobject/gsignal.c:2617: signal 'changed::' is invalid for instance '0x558932d8d570' of type 'GSettings'

(process:8256): GLib-GObject-WARNING **: 12:01:07.224: ../../../gobject/gsignal.c:2617: signal 'changed::' is invalid for instance '0x558932d8d700' of type 'GSettings'

(process:8256): GLib-GObject-WARNING **: 12:01:07.224: ../../../gobject/gsignal.c:2617: signal 'changed::' is invalid for instance '0x558932d8de10' of type 'GSettings'
Downloading…             [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

Successfully downloaded new metadata: 6 local devices supported

```

hughsie, 2020-04-27:
> signal 'changed::' is invalid for instance

Very strange. Can you try doing:

    G_DEBUG=fatal-criticals gdb --args fwupdmgr refresh --force

and when it stops, issue a `t a a bt` command please. Then paste the output.

sbmw, 2020-04-27:
Hi @hughsie thanks very much for your help on this.  I'm sorry if I misunderstood what you've asked for, but here's what I got:

```
G_DEBUG=fatal-criticals gdb --args fwupdmgr refresh --force
                                                     
GNU gdb (Ubuntu 9.1-0ubuntu1) 9.1
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from fwupdmgr...
(No debugging symbols found in fwupdmgr)
(gdb) t a a bt
(gdb) q

```

hughsie, 2020-04-27:
Gah, I missed a step, sorry. Precoffee. You need to type `r` when you get into the gdb prompt. Then the fwupdmgr will run.

sbmw, 2020-04-27:
Ha, I thought it looked a bit light on detail!

```
G_DEBUG=fatal-criticals gdb --args fwupdmgr refresh --force
GNU gdb (Ubuntu 9.1-0ubuntu1) 9.1
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from fwupdmgr...
(No debugging symbols found in fwupdmgr)
(gdb) r
Starting program: /usr/bin/fwupdmgr refresh --force
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff3feb700 (LWP 33453)]
[New Thread 0x7ffff37ea700 (LWP 33454)]
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
[New Thread 0x7ffff2ef7700 (LWP 33455)]
[Detaching after vfork from child process 33456]

(process:33457): GLib-GObject-WARNING **: 23:51:51.250: ../../../gobject/gsignal.c:2617: signal 'changed::' is invalid for instance '0x55555557e430' of type 'GSettings'

(process:33457): GLib-GObject-WARNING **: 23:51:51.251: ../../../gobject/gsignal.c:2617: signal 'changed::' is invalid for instance '0x55555557ec30' of type 'GSettings'

(process:33457): GLib-GObject-WARNING **: 23:51:51.251: ../../../gobject/gsignal.c:2617: signal 'changed::' is invalid for instance '0x55555557e570' of type 'GSettings'

(process:33457): GLib-GObject-WARNING **: 23:51:51.251: ../../../gobject/gsignal.c:2617: signal 'changed::' is invalid for instance '0x55555557e700' of type 'GSettings'

(process:33457): GLib-GObject-WARNING **: 23:51:51.251: ../../../gobject/gsignal.c:2617: signal 'changed::' is invalid for instance '0x55555557ee10' of type 'GSettings'
[New Thread 0x7ffff2053700 (LWP 33461)]
Downloading…             [***************************************]

(fwupdmgr:33449): GLib-CRITICAL **: 23:51:51.876: g_path_get_basename: assertion 'file_name != NULL' failed

Thread 1 "fwupdmgr" received signal SIGTRAP, Trace/breakpoint trap.
0x00007ffff7c140d5 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
(gdb) t a a bt

Thread 5 (Thread 0x7ffff2053700 (LWP 33461)):
#0  syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
#1  0x00007ffff7c5b5a6 in g_cond_wait_until () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007ffff7bde561 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#3  0x00007ffff7c3832a in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#4  0x00007ffff7c37931 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#5  0x00007ffff76c0609 in start_thread (arg=<optimised out>) at pthread_create.c:477
#6  0x00007ffff78fa103 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Thread 4 (Thread 0x7ffff2ef7700 (LWP 33455)):
#0  0x00007ffff78ed96f in __GI___poll (fds=0x5555555ea310, nfds=1, timeout=-1) at ../sysdeps/unix/sysv/linux/poll.c:29
#1  0x00007ffff7c0e1ae in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007ffff7c0e2e3 in g_main_context_iteration () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#3  0x00007ffff7fbf99d in  () at /usr/lib/x86_64-linux-gnu/gio/modules/libdconfsettings.so
#4  0x00007ffff7c37931 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#5  0x00007ffff76c0609 in start_thread (arg=<optimised out>) at pthread_create.c:477
#6  0x00007ffff78fa103 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Thread 3 (Thread 0x7ffff37ea700 (LWP 33454)):
#0  0x00007ffff78ed96f in __GI___poll (fds=0x5555555b3080, nfds=3, timeout=-1) at ../sysdeps/unix/sysv/linux/poll.c:29
#1  0x00007ffff7c0e1ae in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007ffff7c0e533 in g_main_loop_run () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#3  0x00007ffff7e63e9a in  () at /lib/x86_64-linux-gnu/libgio-2.0.so.0
#4  0x00007ffff7c37931 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#5  0x00007ffff76c0609 in start_thread (arg=<optimised out>) at pthread_create.c:477
#6  0x00007ffff78fa103 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Thread 2 (Thread 0x7ffff3feb700 (LWP 33453)):
#0  0x00007ffff78ed96f in __GI___poll (fds=0x5555555999e0, nfds=1, timeout=-1) at ../sysdeps/unix/sysv/linux/poll.c:29
#1  0x00007ffff7c0e1ae in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007ffff7c0e2e3 in g_main_context_iteration () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#3  0x00007ffff7c0e331 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#4  0x00007ffff7c37931 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#5  0x00007ffff76c0609 in start_thread (arg=<optimised out>) at pthread_create.c:477
#6  0x00007ffff78fa103 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Thread 1 (Thread 0x7ffff4dd6880 (LWP 33449)):
#0  0x00007ffff7c140d5 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#1  0x00007ffff7c153b9 in g_logv () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007ffff7c15583 in g_log () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#3  0x00007ffff7bf938a in g_path_get_basename () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#4  0x000055555555f43d in  ()
#5  0x0000555555562e29 in  ()
#6  0x0000555555567cad in  ()
#7  0x000055555555e478 in  ()
#8  0x00007ffff77ff0b3 in __libc_start_main (main=0x55555555d770, argc=3, argv=0x7fffffffde08, init=<optimised out>, fini=<optimised out>, rtld_fini=<optimised out>, stack_end=0x7fffffffddf8) at ../csu/libc-start.c:308
#9  0x000055555555e74e in  ()
```

hughsie, 2020-04-28:
>  On upgrade to Ubuntu 20.04 (clean install) errors are thrown on fwupd refresh

I can't reproduce at all in a VM. Are you sure this is a clean install?

superm1, 2020-04-28:
Perhaps describing all your steps to get here in more detail will help get to reproduction.  For example if you did:
1. Wipe existing OS
2. Clean install from Ubuntu 20.04 pre-release image
3. Install snap

There might have been some packages that aren't fully updated (run an update manager cycle first).

Or if you installed something that pulled in a PPA, it might have pulled in a library that's not matching what's in the Ubuntu repositories.

sbmw, 2020-04-28:
Thank you so much for your assistance.  I've found the issue and I don't believe that it's fwupd.

> Are you sure this is a clean install?

Pretty much, but I went back to a completely fresh 20.04 install on a new partition and there error didn't occur.  However - one thing I did change originally was to install the Cinnamon desktop...

Install and switch to Cinnamon and you get the error above.  Switch back to Gnome and all is well.

It seems to be a permissions issue as **sudo** fwupd.fwupdmgr refresh --force on Cinnamon works just fine!

For what it's worth I was using Cinnamon 4.2.4 from ppa:embrosyn/cinnamon on Ubuntu 18.04 but on 20.04 I'm just installing 4.4.8 from the default Ubuntu repo.

Thank you again for help and apologies for raising an issue that was down to my own setup.

hughsie, 2020-04-28:
> It seems to be a permissions issue

I think it's worth filing a bug with Cinnamon. No worries.

closed 2020-04-28 by hughsie

uusijani, 2020-05-08:
For future reference, this was probably [Ubuntu bug #1875558](https://bugs.launchpad.net/ubuntu/+source/libproxy/+bug/1875558) ("libproxy1-plugin-gsettings triggers a lot of warnings").