fwupd/fwupd #2033: Thinkpad X1 Carbon Gen 6 - Unable to apply Intel Management Engine updates

belminf, 2020-04-27:
**Describe the bug**

I perform an update and reboot but the update is still seen as available despite `Update State` showing `success` in `fwupdmgr get-updates` output.

This is to upgrade Intel Management Engine from 184.55.3510 to 184.70.3626.

**Steps to Reproduce**

1. Run `fwupdmgr refresh --force && fwupdmgr update`
2. Reboot
3. Check `fwupdmgr get-updates`

**Expected behavior**

I expect the update to apply or see something helpful in the debug output.

**fwupd version information**

```shell
$  fwupdmgr --version
client version: 1.4.0
compile-time dependency versions
        gusb:   0.3.4
        efivar: 37
daemon version: 1.4.0
```

Installed via `pacman`: https://www.archlinux.org/packages/?name=fwupd 

**fwupd device information**

```shell
$ fwupdmgr get-devices --show-all-devices

20KH002KUS
│
├─Thunderbolt Controller:
│     Device ID:           76755e859785d05252ff0ed3740d95e4dea9e1f7
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     43.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               5ff9544e-fce2-5fd9-9cd2-671c95ac91dc
│                          b8360cb5-b152-506a-9990-49d7bbbc3b88
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│   
├─Embedded Controller:
│     Device ID:           9698faabddf0d7b18925cfbbda95f8b0d0dacc53
│     Current version:     0.1.20
│     Minimum Version:     0.1.20
│     Vendor:              DMI:LENOVO
│     GUID:                3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─Intel AMT [unprovisioned]:
│     Device ID:           e2623122c99d58220498aacbfcfdb1baebbae3c5
│     Summary:             Hardware and firmware technology for remote out-of-band management
│     Current version:     11.8.55.3510
│     Bootloader Version:  11.8.55.3510
│     Vendor:              Intel Corporation
│     GUID:                2800f812-b7b4-2d4b-aca8-46e0ff65814c
│     Device Flags:        • Internal device
[... 185 lines cut ...]

/boot/EFI
└── Arch
    ├── fw
    │   ├── fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap
    │   └── fwupd-42a0a96e-c9f3-438f-9687-7826be33e4ce.cap
    ├── fwupdx64.efi
    └── grubx64.efi
```

**Additional questions**
- Operating system and version: Archlinux
- Have you tried rebooting? Yes
- Is this a regression? No
- Are you using an NVMe disk? Yes
- Is secure boot enabled? No

**Additional info**

```
$ sudo fwupdate -L
../fwupd-1.4.0/plugins/uefi/efi/fwupdate.c:158:fwup_populate_update_table(): Found update fwupd-ux-capsule
../fwupd-1.4.0/plugins/uefi/efi/fwupdate.c:158:fwup_populate_update_table(): Found update fwupd-42a0a96e-c9f3-438f-9687-7826be33e4ce-0
../fwupd-1.4.0/plugins/uefi/efi/fwupdate.c:564:efi_main(): No updates to process.  Called in error
```

keithy999, 2020-04-30:
I have the same laptop as above and am experiencing the same issue :(

dubinsky, 2020-05-06:
I see the same on ThinkPad X1 Yoga 3rd Gen

renffah, 2020-05-21:
I'm experiencing the same issue with my X1 Carbon 6th Gen.
Except Version `184.70.3626` is already installed and the new Version is `184.77.3664`.

When updating to `184.77.3664`, the Bootscreen is showing that it is applying the capsule and after another reboot it shows for a second that it is updating the firmware.

But if I reinstall `184.70.3626`, the _updating firmware_ message is shown for about half a minute.

The regular System firmware update from `1.46` to `1.48` worked as expected, but no change to the UEFI Device firmware update process.
```
$ fwupdmgr get-devices 
20KGS2A900
│
├─ […]
│   
├─System Firmware:
│     Device ID:           1c53551e7da69d896138fac1ae131c83ad46d923
│     Current version:     0.1.48
│     Minimum Version:     0.0.1
│     Vendor:              LENOVO (DMI:LENOVO)
│     GUID:                a4b51dca-8f97-4310-8821-3330f83c9135
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           e563ad307df81c99f0de8c26292afd71cf409673
│     Current version:     184.70.3626
│     Minimum Version:     184.70.3626
│     Vendor:              DMI:LENOVO
│     Update State:        success
│     GUID:                42a0a96e-c9f3-438f-9687-7826be33e4ce
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─ […]
```

ozbenh, 2020-05-26:
Same problem here on ThinkPad X1 gen 6

ankostis, 2020-05-27:
@renfahh reports about `184.77.3664`, dupe of #2114.