fwupd/fwupd #2034: Changelog and releases

the-spyke, 2020-04-28:
**Is your feature request related to a problem? Please describe.**

No.

**Describe the solution you'd like**

Please add `CHANGELOG.md` file with history of changes for releases and/or post changes to GutHub releases. The first one is a common practice. The second one is too, but more importantly it is convenient: you can subscribe to release notifications on GitHub, so you want to see the changes in the release notification email.

**Describe alternatives you've considered**

Looking for changes in commit log or mail lists is great, but you can do better.

**Additional context**

An example of good changelog/releases: https://github.com/babel/babel

hughsie, 2020-04-28:
Would a NEWS file be sufficient? Generating it is already part of the release process, but it's just not checked into git...

the-spyke, 2020-04-28:
I think `CHANGELOG.md` is a more established name, but at least `NEWS` will be something. Nowadays some projects discard using a file and just post changes to GitHub Releases. I'm fine with this too.

the-spyke, 2020-05-19:
@hughsie `1.4.2` -- still no notes in GitHub releases and I haven't found the NEWS file.

superm1, 2020-05-19:
It looks like this automation failed somehow.  We'll have to dig into why.

superm1, 2020-05-19:
Maybe because we didn't backport a1fda2a179679a0377035321c03c62151c3b8b0d to 1_4_X

superm1, 2020-05-19:
backport to 1_4_X: 11b7d4dd384a97266c54fa4337014eb741456728.  1.4.3 and 1.5.0 should have it.

closed 2020-05-19 by superm1

superm1, 2021-01-13:
Just FYI; it looks like automatic changelogs are working properly now.  They're showing in the recent 1.5.5 release:
https://github.com/fwupd/fwupd/releases/tag/1.5.5

commit a1fda2a1 (2020-04-28): Generate a body for github releases (Fixes: #2034)
commit 89130342 (2020-04-28): Generate a body for github releases (Fixes: #2034)
commit 11b7d4dd (2020-05-19): Generate a body for github releases (Fixes: #2034)