fwupd/fwupd #2038: Add Clevo N240 seeries in superio.quirk

NaKroTeK, 2020-04-29:
I run a Clevo N240-BU, with an IT8587 chip, is it still possible add it in ?

Best Regards

Thank you

hughsie, 2020-04-29:
> is it still possible add it in

It is, but does it work? The SuperIO updating is really sensitive, and I'd be really careful trying to flash images on new hardware without a hardware programmer handy for recovery.

NaKroTeK, 2020-04-30:
Frankly I don't know, I was looking for some info about that chip, and foud your old post about it, and call for more Hwid with it.
I don't have a  HW-programmer with me. (yet)

superm1, 2020-06-12:
I would say it's not safe to try to add support for this until you have a recovery method.  Furthermore my opinion is it's a lost cause since kernel lockdown for now will prevent this interface from working (port IO is locked down in kernel lockdown).

closed 2020-06-12 by superm1

NaKroTeK, 2020-06-12:
Thanks for the updates !