fwupd/fwupd #2039: TPM PCR0 Error Log

toadzky, 2020-04-29:
As requested, here's the output from the log:

[fwupdtpmevlog.log](https://github.com/fwupd/fwupd/files/4552798/fwupdtpmevlog.log)
```
PCR:                     BIOS (0)
Type:                    0x7
Description:             EV_S_CRTM_CONTENTS
ChecksumSha1:            a6f807d160e7f182f26513f2e50006817abe500a085d8812f44ee839d4433451
BlobStr:                 Boot Guard Measured S-CRTM.

PCR:                     BIOS (0)
Type:                    0x8
Description:             EV_S_CRTM_VERSION
ChecksumSha1:            d4720b4009438213b803568017f903093f6bea8ab47d283db32b6eabedbbf155
BlobStr:                 ..kT..U@..N....:

PCR:                     BIOS (0)
Type:                    0x1
Description:             EV_POST_CODE
ChecksumSha1:            9cffed427db3cc30efb0ea1637069ef7b7256ff722474339097807888d475adb
BlobStr:                 ..........j.....

PCR:                     BIOS (0)
Type:                    0x4
Description:             EV_SEPARATOR
ChecksumSha1:            df3f619804a92fdb4057192dc43dd748ea778adc52bc498ce80524c014b81119

PCR:                     BIOS Configuration (1)
Type:                    0x80000009
Description:             EV_EFI_HANDOFF_TABLES
ChecksumSha1:            0f57271e82d06cab06e58b458ad63bb406d375525694b1ee692fafb341bd7431
BlobStr:                 ...........? boD.....|....UQ....

PCR:                     BIOS Configuration (1)
Type:                    0x80000009
Description:             EV_EFI_HANDOFF_TABLES
ChecksumSha1:            9c844a8fe22e6309ce1b358b0066a1c9cf0608b563b291563c6f388c3a3149ff
BlobStr:                 ...........? boD.....|...V.Q....

PCR:                     BIOS Configuration (1)
Type:                    0xa
Description:             EV_PLATFORM_CONFIG_FLAGS
ChecksumSha1:            f58e6c016150ccdbdec3d3c3fc54a03c03460256eac65583a54fb2bdf687d815
BlobStr:                 ..........0........."....#...................B....J..............a....`...._....D$..................|.........x....7....)....#...."....$....b..............2.........

PCR:                     BIOS Configuration (1)
Type:                    0x4
Description:             EV_SEPARATOR
ChecksumSha1:            df3f619804a92fdb4057192dc43dd748ea778adc52bc498ce80524c014b81119

PCR:                     BIOS Configuration (1)
Type:                    0x80000002
Description:             EV_EFI_VARIABLE_BOOT
ChecksumSha1:            faa9c96bd54ba27f521338c1629ec6e7e329555eb806d2ba66b48172575cc89d
BlobStr:                 a.............+.................B.o.o.t.O.r.d.e.r.....

PCR:                     BIOS Configuration (1)
Type:                    0x80000002
Description:             EV_EFI_VARIABLE_BOOT
ChecksumSha1:            4a0702a7accea5230bbdc9b85834f647c37cd6db317ff3aa03d3ecd44e2a564d
BlobStr:                 a.............+.........v.......B.o.o.t.0.0.0.0.....b.u.b.u.n.t.u.....*.........................................4.\.E.F.I.\.u.b.u.n.t.u.\.s.h.i.m.x.6.4...e.f.i.......

PCR:                     BIOS Configuration (1)
Type:                    0x80000002
Description:             EV_EFI_VARIABLE_BOOT
ChecksumSha1:            f3b9709de4f6b259e3cbdd2f8a17d340b61f9cc39cf055765fcfa93cb096eebf
BlobStr:                 a.............+.................B.o.o.t.0.0.0.1.......U.E.F.I.:. .P.C.4.0.0. .N.V.M.e. .S.K. .h.y.n.i.x. .5.1.2.G.B.,. .P.a.r.t.i.t.i.o.n. .2.....*.........................................0.\.E.F.I.\.b.o.o.t.\.b.o.o.t.x.6.4...e.f.i.........8..Gd-.;.A..MQ..L.M.J.8.7.N.4.3.0.3.1.0.1.0.7.A.2.N.........BO

PCR:                     BIOS Configuration (1)
Type:                    0x80000001
[... 472 lines cut ...]
ChecksumSha1:            668b315729c39ad5f0a0b59f870b607776edeb425481b70a1de01b0a19c12e2d
BlobStr:                 (hd0,msdos5)/boot/grub/fonts/unicode.pf2.

PCR:                     Static operating system (9)
Type:                    0xd
ChecksumSha1:            b32a0876f9b7c8cc4769e890dc2ebd15d0beb65d710f2896e5f852c6d42cddb5
BlobStr:                 /boot/vmlinuz-5.4.0-28-generic.

PCR:                     Static operating system (9)
Type:                    0xd
ChecksumSha1:            da3462454052012305f39e0af41c910f94c3394bb1453bca19108195d18be5e8
BlobStr:                 /boot/initrd.img-5.4.0-28-generic.

PCRs:
  0:                     abb86fc17d7390bdefc60491193022d0db0333a0
  1:                     8f7fd77c2c7df0aa79b550a16b62ed126cb0d8cf
  2:                     3a4285c323c5af121f270900e88ffeefb2a34233
  3:                     3a4285c323c5af121f270900e88ffeefb2a34233
  4:                     8b3422f26c1b96b0d9ba01807f434ea69525b255
  5:                     3a4285c323c5af121f270900e88ffeefb2a34233
  6:                     779c58e337edbdebc708901c952a8c6bfd3a00ea
  7:                     205cf7255dc8264c50f05035c061b9243c0d4beb
  8:                     0eb2644a1f555c7997a2a2c3a7033cad3a790e26
  9:                     fcbcca94fbba12a55c0cb1420833978f3ff3aadb
```

superm1, 2020-04-29:
Can you please add the following:

1. Which system firmware version you have
2. The output from tpm2-tools pcr listing tool (to compare)

toadzky, 2020-04-29:
1. 1.12 (i just downgraded to 1.11 to try and fix some issues i'm having)
2. not sure it's any good since i changed the firmware compared to the last one, but here's the output from `tpm2_pcrread`:
```sha1:
  0 : 0x0C9E09910E8BC968A7B241C34CCA86402C3C3C6D
  1 : 0x6CB40A50BECC8937DF2031E6D885E0F1E39F57A5
  2 : 0xB2A83B0EBF2F8374299A5B2BDFC31EA955AD7236
  3 : 0xB2A83B0EBF2F8374299A5B2BDFC31EA955AD7236
  4 : 0xC315FE11F08C1D91E9E928BC0CAF82EF72DEAB05
  5 : 0xB2A83B0EBF2F8374299A5B2BDFC31EA955AD7236
  6 : 0x3B90C535AE6CEF3ED30932455CAAF071079BCF12
  7 : 0xB36F15F2C49096DAC35F4828DCE876AC23F681CC
  8 : 0xC4561BDEFF41B74F4476792B9AB28A789A7957D6
  9 : 0xDF0B17F84C714199D7CDA2CFBCC4BD80119057DE
  10: 0xC004BBF20895CE0120F053A99D485F05AEE8F407
  11: 0x0000000000000000000000000000000000000000
  12: 0x0000000000000000000000000000000000000000
  13: 0x0000000000000000000000000000000000000000
  14: 0x0000000000000000000000000000000000000000
  15: 0x0000000000000000000000000000000000000000
  16: 0x0000000000000000000000000000000000000000
  17: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  18: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  19: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  20: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  21: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  22: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  23: 0x0000000000000000000000000000000000000000
sha256:
  0 : 0x6E467F947FD0D2E0BB60ED31FE8BCFF1467298C48351301EA3F909C3C9471775
  1 : 0x1BDAD7C5380169485C76A426F86A982D1064EDACA37C606BF2CEF4371598FA32
  2 : 0x3D458CFE55CC03EA1F443F1562BEEC8DF51C75E14A9FCF9A7234A13F198E7969
  3 : 0x3D458CFE55CC03EA1F443F1562BEEC8DF51C75E14A9FCF9A7234A13F198E7969
  4 : 0xB114A39007BDDDCEF843DED084CFDCFF8ACD8901B0DB5985939FD4BD621F843C
  5 : 0x3D458CFE55CC03EA1F443F1562BEEC8DF51C75E14A9FCF9A7234A13F198E7969
  6 : 0xEC2761E98A45161CA6211C446DCD1B7B8AC6FCADFAC3E0FAAFE9EBBEC1F6CD39
  7 : 0x7134C4A1895DCF8E94D7DD41929C96E2300734AC317EDB616742C48DBD6D6B3E
  8 : 0x61372083775292DF9E74B0F0DE204E511383AEEA215419BE84E4F116D8478DFA
  9 : 0x0ECC41051C24BF31F8BB1522727171DE7581FBBF310403603C31EE89FE622FC1
  10: 0xA19ACBEA917CFC1DF35EC54AF588E83BD0C3BF47645BD38AC16F4256958A2F91
  11: 0x0000000000000000000000000000000000000000000000000000000000000000
  12: 0x0000000000000000000000000000000000000000000000000000000000000000
  13: 0x0000000000000000000000000000000000000000000000000000000000000000
  14: 0x0000000000000000000000000000000000000000000000000000000000000000
  15: 0x0000000000000000000000000000000000000000000000000000000000000000
  16: 0x0000000000000000000000000000000000000000000000000000000000000000
  17: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  18: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  19: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  20: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  21: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  22: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  23: 0x0000000000000000000000000000000000000000000000000000000000000000
sha384:
```

superm1, 2020-04-29:
So is this only a problem with one of those firmware versions? If it's a problem with both, final pcr values compared to that log on either is fine, but I would like them both from the same.

toadzky, 2020-04-29:
when i do `fwupdmgr get-devices` it doesn't show the PCR0 error anymore, so i think it's only with the 1.12 firmware. do you still want the output?

superm1, 2020-04-29:
Just to make sure - are you sure that's the only thing that has changed was the 1.12 firmware?  No kernel changes or turning on/off secure boot or anything else in between.

toadzky, 2020-04-29:
yep, no other changes except downgrading to 1.11.0

superm1, 2020-04-29:
OK thanks.  I'll let folks know about this one.