fwupd/fwupd #2042: failed to efi_generate_file_device_path on Dell Precision 7740

hpvb, 2020-04-29:
**Describe the bug**

fwupdmgr fails to install the UEFI update for system firmware

**Steps to Reproduce**
```shell
# fwupdmgr --verbose update
(fwupdmgr:4980): GLib-DEBUG: 20:21:44.340: setenv()/putenv() are not thread-safe and should not be used after threads are created
• Thunderbolt Controller has no available firmware updates
• KXG50PNV2T04 NVMe KIOXIA 2048GB has no available firmware updates
Upgrade available for System Firmware from 1.7.0 to 1.8.2
Precision 7740 must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: 
Downloading 1.8.2 for System Firmware...
(fwupdmgr:4980): FuCommon-DEBUG: 20:21:45.099: creating path /root/.cache/fwupd
(fwupdmgr:4980): FuMain-DEBUG: 20:21:45.137: skpping download as file already exists
(fwupdmgr:4980): Fwupd-DEBUG: 20:21:45.143: Emitting ::status-changed() [decompressing]
Decompressing…           [      \                                ](fwupdmgr:4980): Fwupd-DEBUG: 20:21:45.401: Emitting ::status-changed() [idle]
Decompressing…           [***************************************]
(fwupdmgr:4980): Fwupd-DEBUG: 20:21:45.402: Emitting ::status-changed() [waiting-for-auth]
Authenticating…          [      \                                ](fwupdmgr:4980): Fwupd-DEBUG: 20:21:45.403: Emitting ::status-changed() [idle]
Authenticating…          [***************************************]
(fwupdmgr:4980): Fwupd-DEBUG: 20:21:45.419: Emitting ::status-changed() [scheduling]
Updating System Firmware…[      \                                ]
Scheduling…              [      \                                ](fwupdmgr:4980): Fwupd-DEBUG: 20:21:45.419: Emitting ::device-changed(cac9de2369fa3c52f34448afb98d95c9d18cc6d4)
(fwupdmgr:4980): Fwupd-DEBUG: 20:21:45.439: Emitting ::status-changed() [idle]
Scheduling…              [***************************************]
(fwupdmgr:4980): Fwupd-DEBUG: 20:21:45.439: Emitting ::device-changed(cac9de2369fa3c52f34448afb98d95c9d18cc6d4)
failed to efi_generate_file_device_path(/boot/efi/EFI/fedora/fw/fwupd-74992bad-d49d-4f82-bb77-bbe865bea34e.cap)
```

**Expected behavior**
Successful update

**fwupd version information**
Please provide the version of the daemon and client.
```shell
# fwupdmgr --version
client version:	1.4.1
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.4.1

```
Installed as part of Fedora 32

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
# fwupdmgr get-devices --show-all-devices
Precision 7740
│
├─Thunderbolt Controller:
│     Device ID:           77954416d56a4d84c7830c713a347c3b74d5fa7d
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     41.01
│     Vendor:              Dell (TBT:0x00D4)
│     GUIDs:               81db9688-b112-5e24-9f92-e96f05770934 ← TBT-00d40927-native
│                          1707b992-6eb5-5913-8601-6ac003d0f6c1 ← TBT-00d40927-native-controller0-0
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─Ellesmere [Radeon Pro WX 7100 Mobile]:
│     Device ID:           71b677ca0f1bc2c5b804fa1d59e52064ce589293
│     Vendor:              Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002)
│     GUIDs:               5f5b633a-e5ed-52b5-894f-69fa6aa8d41e ← PCI\VEN_1002&DEV_67C0&REV_00
│                          e0b2ecf7-eead-5fce-8db1-442f79ffeeca ← PCI\VEN_1002&DEV_67C0
[... 95 lines cut ...]
- Are you using an NVMe disk?
Yes
- Is secure boot enabled?
Yes

* Additional information *
My system uses separate mdraid devices for each of the normal paritions. ESP, /boot, and / are all on separate mds. I suspect that this is the problem as a strace of the fwupd process shows this:

```shell
# strace -f -p 4477 -e open,openat,fork
... cut ...
[pid  4477] openat(AT_FDCWD, "/boot/efi/EFI/fedora/fw/fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap", O_WRONLY|O_CREAT|O_EXCL, 0666) = 23
[pid  4477] openat(AT_FDCWD, "/proc/self/mounts", O_RDONLY) = 24
[pid  4477] openat(AT_FDCWD, "/dev/md125", O_RDONLY) = 24
[pid  4477] openat(AT_FDCWD, "/sys/dev/block/../../devices/virtual/block/md125/partition", O_RDONLY) = -1 ENOENT (No such file or directory)
[pid  4477] openat(AT_FDCWD, "/dev/md125", O_RDONLY) = 24
[pid  4477] openat(AT_FDCWD, "/etc/os-release", O_RDONLY) = 23
[pid  4477] openat(AT_FDCWD, "/boot/efi/EFI/fedora/fw/fwupd-74992bad-d49d-4f82-bb77-bbe865bea34e.cap.8K52J0", O_RDWR|O_CREAT|O_EXCL, 0666) = 23
[pid  4477] openat(AT_FDCWD, "/proc/self/mounts", O_RDONLY) = 23
[pid  4477] openat(AT_FDCWD, "/dev/md125", O_RDONLY) = 23
[pid  4477] openat(AT_FDCWD, "/sys/dev/block/../../devices/virtual/block/md125/partition", O_RDONLY) = -1 ENOENT (No such file or directory)
[pid  4477] openat(AT_FDCWD, "/dev/md125", O_RDONLY) = 23
[pid  4477] openat(AT_FDCWD, "/var/lib/fwupd/pending.db-journal", O_RDWR|O_CREAT|O_NOFOLLOW|O_CLOEXEC, 0644) = 22
```
It *seems* that fwupd doesn't like that my ESP isn't on a 'partition' as such.

hughsie, 2020-04-29:
> but I also changed the way my system boots

What was it before?

superm1, 2020-04-29:
I'm a bit surprised you're able to boot in this configuration where there is no FAT32 ESP directly on the partition map.  As this is a Xeon, I should ask is this using VROC (Formally known as RSTe) ?

hpvb, 2020-04-29:
> > but I also changed the way my system boots
> 
> What was it before?

I think this was before I encrypted /boot and moved ESP to an MD partition

hpvb, 2020-04-29:
> I'm a bit surprised you're able to boot in this configuration where there is no FAT32 ESP directly on the partition map. As this is a Xeon, I should ask is this using VROC (Formally known as RSTe) ?

There is still a partition that has a fat32 esp, it's just that the partition type is Linux RAID rather than the normal ESP guid. It appears that the Dell bios doesn't care about that at all though.

I'm not using VROC or any other vendor supplied RAID. This is 100% pure Linux MDRaid only.

hughsie, 2020-04-29:
@vathpela is this something we want/need to support?

superm1, 2020-04-29:
>There is still a partition that has a fat32 esp, it's just that the partition type is Linux RAID rather than the normal ESP guid. It appears that the Dell bios doesn't care about that at all though.

If it's really a FAT32 partition, why did you lie about the GUID in the partition table?  I would think that's asking for problems here and it's better to exclude from the RAID array if possible.

VROC is controlled by `mdadm` and is exclusive to Xeons right now: https://www.intel.com/content/dam/support/us/en/documents/memory-and-storage/ssd-software/5-4_VROC_Linux_User_Guide.pdf.
So what I'm wondering is if actually using `mdadm` on the Xeon did provision it with RSTe metadata for you.

Do you by chance have problems with efibootmgr or efivar when interacting with stuff?  I would think that `Boot####` entries might misbehave too.

@vathpela this one might be up your alley.

vathpela, 2020-04-29:
I would expect it's generating Boot#### through the fallback boot path, i.e. using a device path the firmware generated during boot instead of creating it with efibootmgr.  I wonder if it's using vendor-specific path, or if it's just found one leg of the RAID device and used that.

hpvb, 2020-04-29:
@superm1 The reason why it's a raid partition is that I'd like the system to continue to be able to boot if one of the drives fail. I don't know how else to accomplish that other than having ESP be on a raid partition?

As for efibootmgr: It seems to work fine. I added the fedora boot entry myself using it. 

It's not so much a lie as that I seem to recall that operating systems can generally open md managed raid-1 partitions without support (of course if they write it'll break the raid, but I don't expect the uefi firmware to write to the partition) perhaps I'm wrong, but the fact that UEFI can apparently read it just fine seems to suggest this is correct.

It appears that the issue is that fwupd can't map between an MD device to a ... 'physical'? partition.

I guess my use case is this:
My ESP is mirrored so in case of disk failure I can continue to boot with maybe only poking in the UEFI bios settings to disable the faulty drive, and perhaps having to enter efi shell to manually point to the other drive. Point is I can do this on the go in case I need to. I need my laptop to function :)

hpvb, 2020-04-29:
> I would expect it's generating Boot#### through the fallback boot path, i.e. using a device path the firmware generated during boot instead of creating it with efibootmgr. I wonder if it's using vendor-specific path, or if it's just found one leg of the RAID device and used that.

If this information would you useful to you regardless of whether it's useful for this particular issue I'm happy to provide whatever information you need. If you tell me how.

vathpela, 2020-04-29:
> There is still a partition that has a fat32 esp, it's just that the partition type is Linux RAID rather than the normal ESP guid. It appears that the Dell bios doesn't care about that at all though.

Which RAID metadata format did you use?  The different versions wind storing their metadata at different places on the disk, and if you used 1.1 or 1.2, we can't find the partition by looking at the raw disk.

hpvb, 2020-04-29:
> Which RAID metadata format did you use? The different versions wind storing their metadata at different places on the disk, and if you used 1.1 or 1.2, we can't find the partition by looking at the [...]

I used the 1.0 metadata for ESP and /boot for that reason.

superm1, 2020-04-29:
>LI would expect it's generating Boot#### through the fallback boot path, i.e. using a device path the firmware generated during boot instead of creating it with efibootmgr. I wonder if it's using vendor-specific path, or if it's just found one leg of the RAID device and used that.

Looking at this above logs, it's not using a vendor created one.  He's booted into Boot000 which points to shimx64.efi with abbreviated path.

> I don't know how else to accomplish that other than having ESP be on a raid partition?

You can create an ESP on each drive and a `BootXXXX` entry for each ESP, put them both in `BootOrder`.  Then if one of the drives has failed it should fall back to the other one.

> It's not so much a lie as that I seem to recall that operating systems can generally open md managed raid-1 partitions without support (of course if they write it'll break the raid, but I don't [...]

I'm pretty sure what's happening is the FW scans all the disks for stuff that looks like ESPs.  It probably just gets lucky in what it finds on the first disk and goes with it.
I'll tell you this is going to be problematic though with Dell's firmware when you do updates.  One of the things that Dell's firmware does is makes a copy of the  BIOS in the ESP in case of any failures before/during/after flashing.  This feature is marketed as "BIOS recovery".  You will end up with these files put on one of the disks (whichever was marked as the first ESP) and then they won't be cloned to the second.  Your metadata will go out of sync with the disks.

Now if you _were_ using VROC then there is an RSTe DXE driver in the BIOS that actually does interpret RAID arrays.  I expect it would be able to handle updating both disks and the metadata as necessary.

so TL;DR, I would recommend you make two distinct ESP and two distinct `BootXXXX` for each ESP.  Put both in `BootOrder`

hpvb, 2020-04-29:
> You can create an ESP on each drive and a BootXXXX entry for each ESP, put them both in BootOrder. Then if one of the drives has failed it should fall back to the other one.

I'd have to manually keep the in sync then though, right? And if I forget I may not be able to boot?

> I'll tell you this is going to be problematic though with Dell's firmware when you do updates. One of the things that Dell's firmware does is makes a copy of the BIOS in the ESP in case of any [...]

That does seem like a bit of a fatal flaw in my approach. Maybe I should see if I an use VROC somehow then?

superm1, 2020-04-29:
>I'd have to manually keep the in sync then though, right? And if I forget I may not be able to boot?

Well, yes and no.  In a common situation I expect that your kernel is on `/boot` and your bootloader is the only thing on the ESP.  The bootloader doesn't change very frequently, but the kernel does.  So most times keeping them in sync wouldn't be a problem.

Since you use secure boot the problematic scenario would be if there was a security vulnerability in GRUB or SHIM and a UEFI dbx update is released that revokes hashes.  I expect that distros should coordinate putting the updated binaries in the default ESP, but it's not reasonable to expect it to happen on both of them.  Those are also rare things though.  You can very easily just mount both in your OS and have a script that keeps them in sync on a regular basis.

>That does seem like a bit of a fatal flaw in my approach. Maybe I should see if I an use VROC somehow then?

Probably not without blowing away data I'm afraid.

hpvb, 2020-04-29:
> Probably not without blowing away data I'm afraid.

Since I'm not traveling at all now with the COVID-19 happening that's probably ok. I'll reinstall on top of a vroc mirror and see if that will help. I don't think it will as I suspect that I will still get an MD device instead of a raw partition so I don't think the situation will change for fwupd. But the other feedback you gave seems to make it worth to go anyway. I'll report back

vathpela, 2020-04-29:
Can you show us the entire strace instead of just open,openat,fork?  I think there should be more information that'll help show what's going wrong.

hpvb, 2020-04-29:
> Can you show us the entire strace instead of just open,openat,fork? I think there should be more information that'll help show what's going wrong.

Yes, I'm taking a backup now to reinstall with VROC I'll reboot before reinstall and get a full strace

hpvb, 2020-04-29:
@vathpela https://tmm.cx/nextcloud/s/ocWMRcwWB8zD8KK There's the full strace output (gzipped)

hpvb, 2020-04-30:
@superm1 well, it turns out that nvme drives in raid mode are entirely unsupported in Linux. Apparently the system hides the nvme's pcie access behind the ahci/raid controller and patches to support that monstrosity have been rejected. I'll go with your original suggestion to just manually sync the ESP.

superm1, 2020-04-30:
That is the behavior for PCH port remapping (raid on in BIOS setup). Do you see anything separate for vroc/rste? If not, then unfortunately your cpu might not support it still.

hpvb, 2020-04-30:
> That is the behavior for PCH port remapping (raid on in BIOS setup). Do you see anything separate for vroc/rste? If not, then unfortunately your cpu might not support it still.

I don't see any options besides setting the SATA controller in RAID mode. When I do this the nvme devices disappear. I'm on a Xeon E-2286M maybe that's not new enough, or the wrong series. I think it's basically an i9 with ECC not fused out?

hpvb, 2020-04-30:
At least with ESP on a normal partition fwupd managed to install the update like normal. So that worked. :) I guess the question is then whether or not fwupd wants to support esp on an MD device.

vathpela, 2020-04-30:
> @vathpela https://tmm.cx/nextcloud/s/ocWMRcwWB8zD8KK There's the full strace output (gzipped)

Sorry, any chance you can do that with like "-s 1024"?  The default of 32-characters has thwarted my attempts to make this simple to trace :( , as you see:

3054  write(16, "linux-nvme.c:84 parse_nvme(): cu"..., 87) = 87
3054  lseek(16, 0, SEEK_SET)            = 0
3054  write(16, "linux.c:556 device_get(): dev_pr"..., 79) = 79
3054  lseek(16, 0, SEEK_SET)            = 0
3054  write(16, "linux.c:521 device_get(): trying"..., 36) = 36
3054  lseek(16, 0, SEEK_SET)            = 0
3054  write(16, "linux-ata.c:54 parse_ata(): entr"..., 33) = 33
3054  lseek(16, 0, SEEK_SET)            = 0

hpvb, 2020-04-30:
> (I realize you've re-installed a different way already, so if that's too much trouble, I'll understand.)

The main issue right now is that I don't have any updates to install now :-/ When the next update for my box comes I'll setup an MDRaid for ESP again and make a new strace. I should've realized that the default capture size was too small, sorry.

superm1, 2020-04-30:
@hpvb you can certainly reinstall updates.

```
# fwupdmgr reinstall
```
should exercise the same thing.

superm1, 2020-05-13:
I think for now this should be closed, no plans for fwupd to support ESP on mdraid device at this time.  We can revisit it if there is a need in the future.  And if you end up capturing the stuff that @vathpela was looking for, feel free to re-open and share it too.

closed 2020-05-13 by superm1

jafasbay, 2021-07-09:
I Need bios bin form DELL PRECISION 7740 i lost my backup . jafasbay@gmail.com