fwupd/fwupd #2043: Expose firmware security attributes to the client

hughsie, 2020-04-29:
I’d like to export some platform firmware attributes from the daemon to the client. These attributes would be things like SecureBoot, BIOS_WE, PRx, etc. There are two ways we can export this, either as a method on the daemon e.g. `GetSecurityAttrs()` or a method that takes a DeviceID. I don’t think the attributes are per-device, they’re supposed to represent the *system* as a whole rather than a specific device (which we already have some flags for). I guess we could use the device with the `system-firmware` GUID but I’m not sure it’s helpful if we’re defining it as system state.

The structure I thought we would export for each attribute would be something like this:

    bool success, as in TRUE if this is a good thing
    uint sec_level, as in SEC1-SEC5 as defined roughly in https://blogs.gnome.org/hughsie/2020/01/27/hunting-uefi-implants/
    string appstream_id, for instance org.intel.BiosGuiard → which allows localizing name in the client
    string name, a fallback for the appstream_id localisation, for instance “Intel BIOSGuard”

The idea of including the security level would be to define a single value for the platform, defined by taking the MAX() of each level with at least one `success=True` and without any `success=False`. i.e. even with BootGuard if BIOSWE is unset then you get SEC1 not SEC3. There are a few gotchas, so if you have BIOSGuard the PRx failure isn't a big deal hence you can't expect an attr in all cases.

Implementation wise I guess we’d have a new vfunc that gets called on each plugin which can add things from different places. I’m also working with a vendor who’s contributing a kernel module to get things like PRx via sysfs, so we’d need a new plugin for that longer term. We could either return results with no caching when `GetSecurityAttrs()` is called, or just add these as part of daemon setup `probe()` as I don’t think they can change at runtime. I suppose adding a libfwupd/FwupdSecurityAttr object makes sense, which gets marshalled to and from a `GVariant a{sv}` over the bus to avoid paining ourselves into a corner when we add things in the future.

I think for more information we could provide one wiki/LVFS page explaining the terms in more detail. My long term goal is to increase the SEC level of systems, and to include some of the SEC data as part of of the signed uploaded report so that we can tag systems on the LVFS. Long term plan obviously.

@superm1 Comments welcome!

superm1, 2020-04-30:
So I think an important starting point is to draw a line in the sand on what types of firmware will factor into this security equation.  I'd be tempted to stay "internal" devices only initially but then you have things to contend with around Thunderbolt and USB4.  Perhaps the various mitigations (IOMMU and the like) are enough for them, none of us know until the next vulnerability is disclosed.

> I guess we could use the device with the system-firmware GUID but I’m not sure it’s helpful if we’re defining it as system state.

What do you think about making a top level "System" device?  Then that's the implicit parent for anything without a parent.  Then these can be attributes of that "System" device.  UEFI firmware then becomes one part of this equation.  CSME becomes another, CPU microcode becomes another, etc.

>I’m also working with a vendor who’s contributing a kernel module to get things like PRx via sysfs, so we’d need a new plugin for that longer term. 

I'd actually argue things like this, and checking BIOSWE are more important for starting points and shouldn't be long term.  I don't think that there is a lot of value to the daemon guessing the security state of the system.  You want a real way to attest that these things were set correctly.

> I think for more information we could provide one wiki/LVFS page explaining the terms in more detail. My long term goal is to increase the SEC level of systems, and to include some of the SEC data [...]

In general I don't think a fwupd issue is the right place to discuss all the design details for this.  Probably this should go to a google doc or something interactive and commentable and invite people to make suggestions and comments there.

hughsie, 2020-04-30:
> So I think an important starting point is to draw a line in the sand on what types of firmware will factor into this security equation

My thoughts here would be restricting it to anything controlled by the system firmware, else we get into a whole world of pain when it comes to talking about security of the entire OS. For that reason it probably makes sense to rule out TBT devices and anything like Selinux.

> What do you think about making a top level "System" device

I've not done this so far, as the API provided by fwupd is somewhat 'flat' by design. I guess for UI like fwupdmgr it might make sense, but I'm not sure how useful it is compared to just rooting it on system-firmware for example.

> checking BIOSWE are more important for starting points and shouldn't be long term

BIOSWE is actually the first attribute the sysfs interface would provide. It's WIP now.

> In general I don't think a fwupd issue is the right place to discuss all the design details for this

Yes, you're probably right. I've sent you an invite to the google doc I've started. Feel free to jump in with suggestions and comments.

superm1, 2020-04-30:
> My thoughts here would be restricting it to anything controlled by the system firmware, else we get into a whole world of pain when it comes to talking about security of the entire OS. For that reason it probably makes sense to rule out TBT devices and anything like Selinux.

<cough> potential sideband signals from AMT </cough>

> I've not done this so far, as the API provided by fwupd is somewhat 'flat' by design. I guess for UI like fwupdmgr it might make sense, but I'm not sure how useful it is compared to just rooting it on system-firmware for example.

I guess we can talk more about this on the document.

> BIOSWE is actually the first attribute the sysfs interface would provide. It's WIP now.

Good!

hughsie, 2020-08-18:
I think we can close this now. 🙂

closed 2020-08-18 by hughsie