fwupd/fwupd #2044: Can I downgrade Logitech K375s keyboard firmware? And is there a way to backup the current firmware on the keyboard before I do so?

fondfire, 2020-04-30:
I have a Logitech K375s keyboard that I'm using on Ubuntu 18.04 and Ubuntu 20.04 systems. The following are screenshots from Solaar on my system:

[image]
[image]

The current MPK 03.04.B0014 firmware for my keyboard seems to have a bug.

The keyboard goes into a suspend or sleep mode after its untouched for a short while, which I am sure preserves the battery life. When it is touched again, it "wakes up".

On my Logitech K375s with the MPK 03.04.B0014 firmware, there's a string of gibberish that flies out of the keyboard every time it "wakes up" from this light, and otherwise perfectly functional, sleep. I've tried swapping the keyboard mode -- to iOS mode (by pressing Fn-I for about seven seconds), to MacOS mode (by pressing Fn-O for about seven seconds), and to Windows/Android mode (by pressing Fn-P for about seven seconds) -- and none of those modes is without this issue. In fact, every time a switch modes, a new set of garbage is randomly spewed by the keyboard (albeit, each mode seems to use a slightly different set of garbage characters). I suppose there is a small chance my keyboard was physically damaged in some way, but it's working perfectly in all other ways. And the keyboard has been doing this since I purchased it, IIRC.

I like this keyboard in every other way. I believe it's too late to return it. I really don't WANT to replace it.

Can I try downgrading to the MPK 03.02.B0009 firmware available on LVFS already? Is there a way I can back-up the current firmware in case that goes horribly wrong? I've tried with Ubuntu 18.04 and Ubuntu 20.04 systems, but both seem to be running the same version of fwupd at the time of this writing:

```
user@xubuntu:~$ sudo fwupdmgr --version
client version:	1.2.10
compile-time dependency versions
	gusb:	0.2.11
	efivar:	34
daemon version:	1.2.10
user@xubuntu:~$ 
```

When I run `fwupdmgr get-devices`, the Unifying Receiver shows up, but not the keyboard device.

```
user@xubuntu:~$ sudo fwupdmgr --allow-older get-devices
Unifying Receiver
  DeviceId:             71d6fa6e8873db78a7d81ac01104b67d32c44299
  Guid:                 cc4cbfa9-bf9d-540b-b92b-172ce31013c1
  Guid:                 279ed287-3607-549e-bacc-f873bb9838c4 <- HIDRAW\VEN_046D&DEV_C52B
  Summary:              A miniaturised USB wireless receiver
  Plugin:               unifying
  Flags:                updatable|supported|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR24.10_B0036
  VersionBootloader:    BOT03.01_B0008
  VersionFormat:        plain
  Icon:                 preferences-desktop-keyboard
  InstallDuration:      7
  Created:              2020-04-30

LVM PV iHUzDB-gMpb-YuG7-b2r6-yzqL-6SSV-nqDcCi on /dev/sda
  DeviceId:             f219ac034743df43be49a235e41f128dd47d855f
  Guid:                 4d642611-e9f4-50fe-9034-8aee4b6707f1 <- IDE\LVM_PV_iHUzDB-gMpb-YuG7-b2r6-yzqL-6SSV-nqDcCi_on_/dev/sdaCC26
  Guid:                 8fbdc88a-2f9b-5552-93f7-c0b5cbc67e37 <- IDE\0LVM_PV_iHUzDB-gMpb-YuG7-b2r6-yzqL-6SSV-nqDcCi_on_/dev/sda
  Guid:                 015ff5ef-99ee-59f5-bb47-51d7d9b3d1cf <- LVM PV iHUzDB-gMpb-YuG7-b2r6-yzqL-6SSV-nqDcCi on /dev/sda
  Serial:               Z4Z7EK93
  Summary:              ATA Drive
  Plugin:               ata
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              CC26
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2020-04-30

LVM PV 1s1Sxb-Rtn3-eRAq-aahI-WnPI-mdTn-pexbuF on /dev/sdb
  DeviceId:             7466579651466dd8b56d7a19479bed768440026d
  Guid:                 a5f7bbab-3cd9-500f-a754-bd85b4ade5ad <- IDE\LVM_PV_1s1Sxb-Rtn3-eRAq-aahI-WnPI-mdTn-pexbuF_on_/dev/sdb80.00A80
  Guid:                 8b83217d-ec29-5842-afaf-5e60cf735edc <- IDE\0LVM_PV_1s1Sxb-Rtn3-eRAq-aahI-WnPI-mdTn-pexbuF_on_/dev/sdb
  Guid:                 c6e59e9a-995a-5092-a9fd-ded3f5bb4627 <- LVM PV 1s1Sxb-Rtn3-eRAq-aahI-WnPI-mdTn-pexbuF on /dev/sdb
  Serial:               WD-WCC4M7VUSXYA
  Summary:              ATA Drive
  Plugin:               ata
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              80.00A80
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2020-04-30

Samsung SSD 840 EVO 120GB
  DeviceId:             70079f60e6e7fb777f0cad46adc124f60c371b7b
  Guid:                 cfee1a72-b48e-5d5f-8443-a53d6acd0510 <- IDE\Samsung_SSD_840_EVO_120GB_______________EXT0BB6Q
  Guid:                 7ec0cdc1-a34c-5515-8672-3c1051b8fbec <- IDE\0Samsung_SSD_840_EVO_120GB_______________
  Guid:                 87e7963d-ec79-5bee-a9ff-5c66e487aa92 <- Samsung SSD 840 EVO 120GB
  Serial:               S1D5NSAF716308E
  Summary:              ATA Drive
  Plugin:               ata
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              EXT0BB6Q
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2020-04-30

user@xubuntu:~$ 
```

Adding the `--show-all-devices` switch also shows my graphics card, but still doesn't show the keyboard.

How would I force the downgrade? Or will there be an upgrade? I'm REALLY hoping there's a solution. You have no idea how badly this interferes with writing documents, gaming... virtually everything worthwhile I do on this system. And it's otherwise SUCH a great device...

This seems related to, but unique from, long-expired issue #449. I have also submitted Logitech/fw_updates issue #28 (also on GitHub) in hopes of soliciting new firmware from Logitech themselves into LVFS. Please let me know what can be done! Thanks.

fondfire, 2020-05-02:
I forced my laptop to upgrade to Ubuntu 20.04 and now had access to this version of fwupdmgr:

```
user@xubuntu:~$ sudo fwupdmgr --version
client version:	1.3.9
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.9
user@xubuntu:~$ 
```

I was able to get a bit farther. It would not let me downlevel the firmware on this keyboard, though I was able to see the device:

```
user@xubuntu:~$ sudo fwupdmgr --show-all-devices get-devices
20BE004EUS
│
├─4th Gen Core Processor Integrated Graphics Controller:
│     Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
│     Current version:     06
│     Vendor:              Intel Corporation (PCI:0x8086)
│     GUIDs:               5570d9e8-f98d-54d6-977f-d4a02487bb96 ← PCI\VEN_8086&DEV_0416&REV_06
│                          630813c1-8104-5749-a082-1de4ff3f1497 ← PCI\VEN_8086&DEV_0416
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│   
├─Event Log:
│     Device ID:           58bd405f31c48e6eca290b425f530a94c91e955c
│     GUID:                a25657fe-b5dc-5be0-8b78-8b9dfec678ff ← system-tpm-eventlog
│     Device Flags:        • Internal device
│   
├─Intel AMT [unprovisioned]:
│     Device ID:           e2623122c99d58220498aacbfcfdb1baebbae3c5
│     Summary:             Hardware and firmware technology for remote out-of-band management
│     Current version:     9.1.32.1002
│     Bootloader Version:  9.1.32.1002
│     Vendor:              Intel Corporation
│     GUID:                2800f812-b7b4-2d4b-aca8-46e0ff65814c
│     Device Flags:        • Internal device
│   
├─PNY CS1311 480GB SSD:
│     Device ID:           602b0a6cc821d155208724f0e22f8d111542b74c
│     Summary:             ATA Drive
│     Current version:     CS131122
│     Serial Number:       PNY09162169920311BB8
│     GUIDs:               18838bec-eeb9-5551-a452-54045c409c7a ← IDE\PNY_CS1311_480GB_SSD____________________CS131122
│                          2aa773aa-643f-5835-ab77-6901b0eda7ed ← IDE\0PNY_CS1311_480GB_SSD____________________
│                          eddcaf37-fbc5-5490-a6a3-a00cd7446c54 ← PNY CS1311 480GB SSD
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           977691b910335d0503f9c3dabad3354cb9f79edd
│     Current version:     0.2.20
│     Minimum Version:     0.2.20
│     Vendor:              LENOVO (DMI:LENOVO)
│     GUID:                3eee1af1-9b48-4be5-b236-f8f8716d5a8e
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           b9909f51e03c9659947996d77163d12d2cbba67a
│     Current version:     145.32.1002
│     Minimum Version:     145.32.1002
│     Vendor:              DMI:LENOVO
│     GUID:                e239ac97-5788-4a07-8b24-fe9dca1cf8be
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─Unifying Receiver:
  │   Device ID:           2b91df41f1656f3d332c314e30d5c722de1f56de
  │   Summary:             A miniaturised USB wireless receiver
  │   Current version:     RQR24.10_B0036
  │   Bootloader Version:  BOT03.01_B0008
  │   Vendor:              USB:0x046D
  │   Install Duration:    30 seconds
  │   GUIDs:               cc4cbfa9-bf9d-540b-b92b-172ce31013c1
  │                        fcf55bf5-767b-51ce-9c17-f6f538c4ee9f ← HIDRAW\VEN_046D&DEV_C52B&REV_00
  │                        279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
  │   Device Flags:        • Updatable
  │                        • Supported on remote server
  │ 
  └─Logitech K370s/K375s:
        Device ID:         7d8aa8eabc1c6bd74a385396092abd8d0f7f9f85
        Vendor:            USB:0x046D
        GUIDs:             ccff4b1e-879b-50d5-b2d7-d6c15f52232f ← HIDRAW\VEN_046D&DEV_4061&REV_00
                           3fd1f5c9-4ead-5503-b78f-b3821399e4eb ← HIDRAW\VEN_046D&DEV_4061
                           021565af-852e-50cd-9f67-9f00903367b5 ← UFY\VID_046D&PID_4061
      
user@xubuntu:~$ sudo fwupdmgr downgrade 7d8aa8eabc1c6bd74a385396092abd8d0f7f9f85
No downgrades for Logitech K370s/K375s: no version set
user@xubuntu:~$ sudo fwupdmgr unlock 7d8aa8eabc1c6bd74a385396092abd8d0f7f9f85
Authenticating…          [***************************************]
Device 7d8aa8eabc1c6bd74a385396092abd8d0f7f9f85 is not locked
user@xubuntu:~$ sudo fwupdmgr activate 7d8aa8eabc1c6bd74a385396092abd8d0f7f9f85
No firmware to activate
user@xubuntu:~$ sudo fwupdmgr install ~/Downloads/66924773f5ff849c0e26b2bd4ff4fdbb228a895d-Logitech-K370s_K375s-MPK03.02_B0009.cab 7d8aa8eabc1c6bd74a385396092abd8d0f7f9f85
Decompressing…           [***************************************]
Device Logitech K370s/K375s [7d8aa8eabc1c6bd74a385396092abd8d0f7f9f85] does not currently allow updates
user@xubuntu:~$ 
```

Is there anything I didn't try that I should have tried? Would updating to fwupd 1.4.1 buy me anything here? It seems support for the Logitech K375s was dropped from fwupd... and Logitech doesn't seem interested in releasing further firmware for this keyboard via this channel... but would a much OLDER version of fwupd give me more options to manage this?

I guess I'll just have to get someone to loan me their Windows laptop...

closed 2020-06-08 by stale[bot]

ppc52776, 2020-12-30:
firmware MPK03.02 and MPK03.04 have power issue. MPK03.03 works.