fwupd/fwupd #2047: fwupd fails to/very slow to start with Logitech G610 keyboard

robhancock, 2020-05-01:
**Describe the bug**
fwupd often fails to start within the timeout period allowed under Fedora 32. When it does start I see this in the system log:

```
Apr 30 19:54:29 haswell fwupd[4367]: 01:54:29:0025 FuEngine             failed to add udev device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.0/0003:046D:C338.0004/hidraw/hidraw3: failed to ping Logitech Gaming Keyboard G610: too many messages to ignore
Apr 30 19:54:29 haswell fwupd[4367]: 01:54:29:0035 FuEngine             failed to add udev device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.1/0003:046D:C338.0005/hidraw/hidraw4: failed to ping Logitech Gaming Keyboard G610: generic failure
```
If I run fwupd manually with --verbose I see messages like these whenever I press a key on the keyboard:

```
01:58:15:0474 FuPluginLogitechHidPp HID++1.0 report 0x20 has unknown length, ignoring
01:58:15:0593 FuPluginLogitechHidPp HID++1.0 report 0x00 has unknown length, ignoring
01:58:15:0984 FuPluginLogitechHidPp HID++1.0 report 0x20 has unknown length, ignoring
01:58:16:0085 FuPluginLogitechHidPp HID++1.0 report 0x00 has unknown length, ignoring
01:58:16:0213 FuPluginLogitechHidPp HID++1.0 report 0x20 has unknown length, ignoring
01:58:16:0303 FuPluginLogitechHidPp HID++1.0 report 0x00 has unknown length, ignoring
01:58:16:0381 FuPluginLogitechHidPp HID++1.0 report 0x20 has unknown length, ignoring
01:58:16:0488 FuPluginLogitechHidPp HID++1.0 report 0x00 has unknown length, ignoring
```
It seems like this device is being mishandled somehow.

**Steps to Reproduce**
Boot system

**Expected behavior**
fwupd starts properly

**fwupd version information**
client version:	1.4.1
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.4.1

Fedora package: fwupd-1.4.1-1.fc32.x86_64

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
System Product Name
│
├─BCM920702 Bluetooth 4.0:
│     Device ID:           e1ab76b5c4f8a1b88a37b3636286be83bee8397a
│     Current version:     1.12
│     Vendor:              Broadcom Corp (USB:0x0A5C)
│     Serial Number:       5CF3707DEA8F
│     GUIDs:               6faa2765-8bc2-51d3-bf6b-ffd6312c24d2 ← USB\VID_0A5C&PID_21E8&REV_0112
│                          a1e76ca3-d35b-53b8-a17c-454b700274ea ← USB\VID_0A5C&PID_21E8
│     Device Flags:        • Updatable
│   
├─HD Graphics 630:
│     Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
│     Current version:     04
│     Vendor:              Intel Corporation (PCI:0x8086)
│     GUIDs:               c3e852c7-3e4c-5fc4-8fd3-2f8293de3b00 ← PCI\VEN_8086&DEV_5912&REV_04
│                          83d2b615-717b-5030-bc06-2994db598c7b ← PCI\VEN_8086&DEV_5912
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│   
├─Intel(R) Core™ i5-7500 CPU @ 3.40GHz:
│     Device ID:           4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:     0xca
│     Vendor:              GenuineIntel
│     GUID:                b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│     Device Flags:        • Internal device
│   
├─ST2000DM001-9YN1:
│     Device ID:           948241a24320627284597ec95079cc1341c90518
│     Summary:             ATA Drive
│     Current version:     CC4H
│     Vendor:              Seagate (ATA:0x1BB1)
│     Serial Number:       Z3401QGG
│     GUIDs:               b302a339-bba0-58b9-a9a0-babb151c358e ← IDE\ST2000DM001-9YN164______________________CC4H
│                          5428c853-6f37-5bde-ba87-e609f636e0f7 ← IDE\0ST2000DM001-9YN164______________________
│                          c1b6122a-206a-570a-95cb-2a1103976f1f ← ST2000DM001-9YN164
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─Samsung SSD 970 EVO Plus 1TB:
│     Device ID:           310f45f1f223064b5c16bf6dff31146755a64480
│     Summary:             NVM Express Solid State Drive
│     Current version:     2B2QEXM7
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     Serial Number:       S59ANG0MA14430E
│     GUIDs:               6e54c992-d302-59ab-b454-2d26ddd63e6d ← NVME\VEN_144D&DEV_A808&REV_00
│                          47335265-a509-51f7-841e-1c94911af66b ← NVME\VEN_144D&DEV_A808
│                          f05ecac9-5cc2-5f77-b23b-a08ad8879ada ← Samsung SSD 970 EVO Plus 1TB
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           0d1cd489ca9a66f39caa7c7cb707e16e7d35b53e
│     Current version:     4613
│     Minimum Version:     4613
│     Vendor:              System manufacturer (DMI:American Megatrends Inc.)
│     GUID:                e8477574-dd7b-5339-a835-b449da468be1
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─TOSHIBA DT01ACA3:
│     Device ID:           e421b2fc248391f6fe3e55ddbb3c9043be068bd0
│     Summary:             ATA Drive
│     Current version:     ABB0
│     Vendor:              Toshiba (ATA:0x1179)
│     Serial Number:       95KEVT7GS
│     GUIDs:               9e08c332-b3f2-56ce-a868-45e67008c78f ← IDE\TOSHIBA_DT01ACA300______________________ABB0
│                          db29c086-1366-5842-978f-8bc403684b73 ← IDE\0TOSHIBA_DT01ACA300______________________
│                          022af89b-fba6-5171-8750-e0704f2e13bc ← TOSHIBA DT01ACA300
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─TPM:
│ │   Device ID:           c6a80ac3a22083423992a3cb15018989f37834d6
│ │   Summary:             TPM 2.0 Device
│ │   Current version:     302.12.0.0
│ │   Vendor:              Intel (TPM:INTC)
│ │   GUIDs:               ff71992e-52f7-5eea-94ef-883e56e034c6 ← system-tpm
│ │                        34801700-3a50-5b05-820c-fe14580e4c2d ← TPM\VEN_INTC&DEV_0000
│ │                        8e1cbc5d-5a11-5149-bfea-b6065d5296ba ← TPM\VEN_INTC&MOD_Intel
│ │                        03f304f4-223e-54f4-b2c1-c3cf3b5817c6 ← TPM\VEN_INTC&DEV_0000&VER_2.0
│ │                        52d7b679-db28-5bf7-bd87-41d77aeec600 ← TPM\VEN_INTC&MOD_Intel&VER_2.0
│ │   Device Flags:        • Internal device
│ │ 
│ └─Event Log:
│       Device ID:         58bd405f31c48e6eca290b425f530a94c91e955c
│       GUID:              a25657fe-b5dc-5be0-8b78-8b9dfec678ff ← system-tpm-eventlog
│       Device Flags:      • Internal device
│     
└─Unifying Receiver:
      Device ID:           fb1a5fd6e7d8cbf5d5a2ba7df68aee106ce41027
      Summary:             A miniaturised USB wireless receiver
      Current version:     RQR12.10_B0032
      Bootloader Version:  BOT01.02_B0015
      Vendor:              USB:0x046D
      Install Duration:    30 seconds
      GUIDs:               9d131a0c-a606-580f-8eda-80587250b8d6
                           fcf55bf5-767b-51ce-9c17-f6f538c4ee9f ← HIDRAW\VEN_046D&DEV_C52B&REV_00
                           279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
      Device Flags:        • Updatable
                           • Supported on remote server
    
```

**Additional questions**
- Operating system and version: Fedora 32
- Have you tried rebooting? Yes
- Is this a regression? Yes, worked under Fedora 31 version

FFY00, 2020-05-01:
These reports should be ignored, I don't think it is even worth spitting out a message.

robhancock, 2020-05-01:
Indeed, but it seems that until enough of them are ignored due to pressing keys, the daemon doesn't give up on whatever it's trying to do and continue its startup process (hence the "too many messages to ignore" log entry). This causes GNOME Software to raise a notification on every login of a failed software update.

superm1, 2020-05-01:
This is certainly a side effect of fwupd now looking at logitech devices that don't have metadata.

This looks like a few classes of information.
1. `report 0x20 has unknown length, ignoring` is ignored in the code.  Those don't show any output.  I think this is correct - those are reports coming in related to your key press from your description.
2. `generic failure`: @FFY00 I think we must be missing some error codes.  `generic failure` is the fallback error message.  Are there intentionally some codes that aren't in `the fu-logitech-hidpp-msg.c`
3. `too many messages to ignore`, this i'm not sure.

snugglej, 2020-05-02:
Just wanted to comment that I have the same problem and I reverted back to the [fwupd-1.3.9-2.fc32](<https://koji.fedoraproject.org/koji/buildinfo?buildID=1474421>) and I no longer have the timeout or update failure notification on my system.

hughsie, 2020-05-02:
For RHEL I'll change it to EnumerateAllDevices=false, but I'm wondering what to do for fedora. I was worried this might cause regressions on older hardware... 🙁

FFY00, 2020-05-02:
@superm1 I think we should whitelist the HID++ report IDs, those are the only ones we are certain are valid, and the only ones we care about.

hughsie, 2020-05-02:
Does https://github.com/fwupd/fwupd/pull/2053 fix things for now?

snugglej, 2020-05-03:
@hughsie, 

I'm testing this right now for Fedora 32. I had blacklisted the udpate as the previous version was working but I'm upgrading to 1.4.1-1.fc32.

I edited the /etc/fwupd/daemon.conf and added the lines that you did for #2053. The EnumerateAllDevices did not exist in daemon.conf at all in Fedora 32.

#### This did not fix my issue as I booted into gnome and checked the status of fwupd and it was stuck at activating. I was then greeted again with the "Software Update Failed"

```Detailed errors from the package manager follow:

Error calling StartServiceByName for org.freedesktop.fwupd: Timeout was reached```

snugglej, 2020-05-03:
@hughsie 

So adding `EnumaerteAllDevices=False` did fix the loading of fwupd but it introduced another issue where the Logitech keyboard stopped responding for a while. As I said in the other issue, my keyboard Logitech PRO stopped responding for around 25 seconds and then came back (similar to another persons post).

I went ahead and followed the second step from @jprupp [disable Logitech plugin](https://github.com/fwupd/fwupd/issues/2048#issuecomment-622958514) and then my keyboarding responded fine with no interruptions at all.

So I think the Logitech issue is a two prong issue, should I open a new issue for the logitech keyboard stops responding?

hughsie, 2020-05-03:
> EnumaerteAllDevices=False

Did you use that or `EnumerateAllDevices=true` ?

snugglej, 2020-05-03:
@hughsie I used `EnumerateAllDevices=false (screenshot of the current changes shown)

[image]

smrqdt, 2020-05-18:
I have the same problem with the same keyboard model. (Arch Linux, versions all the same)
If it would be helpful to gather additional logs please tell me.

closed 2020-06-24 by stale[bot]