fwupd/fwupd #2048: Fedora 32: Error calling StartServiceByName for org.freedesktop.fwupd: Timeout was reached

flashkiller999, 2020-05-01:
**Describe the bug**
After the upgrade from F31 to F32 after the login I always get this message:
Error calling StartServiceByName for org.freedesktop.fwupd: Timeout was reached

**Steps to Reproduce**
Upgrade from F31 to F32 and login to user

**Expected behavior**
No error message and upgrade possible from "Software"-application (currently in the tab "Update" it only shows the error message "Error calling StartServiceByName for org.freedesktop.fwupd: Timeout was reached")

**fwupd version information**
```shell
$ fwupdmgr --version
client version:	1.4.1
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.4.1
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): dnf

**fwupd device information**

```shell
$ fwupdmgr get-devices --show-all-devices
To be filled by O.E.M.
│
├─88SE9172 SATA 6Gb/s Controller:
│     Device ID:           e32966e41f54b47823eef28bd8b4bd9e1018b923
│     Current version:     11
│     Vendor:              Marvell Technology Group Ltd. (PCI:0x1B4B)
│     GUIDs:               f20b5994-f3db-5be0-8a88-4a3bd417b4dd
│                          76a209d6-e1dc-598a-b9e9-a8dd911da340
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│   
├─GM204 [GeForce GTX 970]:
│     Device ID:           71b677ca0f1bc2c5b804fa1d59e52064ce589293
│     Current version:     a1
│     Vendor:              NVIDIA Corporation (PCI:0x10DE)
│     GUIDs:               c32e752c-53d9-53dd-a7ab-f7b9a72e9b7d
│                          e6857de4-0602-5690-90d1-df576c354a9b
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│   
├─I210 Gigabit Network Connection (Ethernet I210-T1 GbE NIC):
│     Device ID:           03281da317dccd2b18de2bd1cc70a782df40ed7e
│     Current version:     03
│     Vendor:              Intel Corporation (PCI:0x8086)
│     GUIDs:               54830d50-f47b-57b4-9a79-58bb9aba113c
│                          0d49c464-7ded-57eb-8ca9-dacaa2b9d986
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│   
├─Intel(R) Xeon(R) CPU E3-1230 V2 @ 3.30GHz:
│     Device ID:           4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:     0x21
│     Vendor:              GenuineIntel
│     GUID:                b9a2dd81-159e-5537-a7db-e7101d164d3f
│     Device Flags:        • Internal device
│   
├─Samsung SSD 840 EVO 120GB:
│     Device ID:           0a8c36d4c09c803cd6b5861e443fb7a41a20cbe6
│     Summary:             ATA Drive
│     Current version:     EXT0DB6Q
│     Vendor:              Samsung (ATA:0x144D)
│     GUIDs:               1cf8bf0e-390d-5802-8774-ad5e767cee85
│                          7ec0cdc1-a34c-5515-8672-3c1051b8fbec
│                          87e7963d-ec79-5bee-a9ff-5c66e487aa92
[... 479 lines cut ...]
09:59:54:0139 FuMain               Emitting PropertyChanged('Status'='idle')
09:59:54:0139 XbSilo               ignoring for OR statement: indexed string '1cf8bf0e-390d-5802-8774-ad5e767cee85' was unfound
09:59:54:0139 XbSilo               ignoring for OR statement: indexed string '7ec0cdc1-a34c-5515-8672-3c1051b8fbec' was unfound
09:59:54:0139 XbSilo               ignoring for OR statement: indexed string 'f37354a2-fa11-5539-9039-c5194ac4ea16' was unfound
09:59:54:0139 XbSilo               ignoring for OR statement: indexed string '3831476a-add3-5bc8-88ff-31739e99ad07' was unfound
09:59:54:0139 XbSilo               ignoring for OR statement: indexed string 'd0281ee4-5748-58a0-93fd-2f9a8cefa155' was unfound
09:59:54:0139 XbSilo               ignoring for OR statement: indexed string '1d5e255e-66d3-5c93-ac37-31817ebd0571' was unfound
09:59:54:0139 FuCommon             creating path /var/cache/fwupd/motd.d
09:59:54:0143 GLib-GIO             Failed to initialize portal (GMemoryMonitorPortal) for gio-memory-monitor: Not using portals
09:59:54:0143 GLib-GIO             _g_io_module_get_default: Found default implementation dbus (GMemoryMonitorDBus) for ‘gio-memory-monitor’
09:59:54:0145 FuMain               Started with locale en_US.UTF-8
09:59:54:0145 FuMain               Daemon ready for requests
09:59:54:0146 FuMain               FuMain: acquired name: org.freedesktop.fwupd
10:03:16:0593 FuCommon             not checking 1.00 as no version format set
10:03:16:0594 FuPlugin             performing fu_plugin_device_created() on ata
10:03:16:0665 FuCommon             not checking 1.00 as no version format set
10:03:16:0666 FuPlugin             performing fu_plugin_device_created() on ata
10:08:27:0695 FuMain               Called GetDevices()
10:08:27:0697 FuMain               Called GetHistory()
```

**Additional questions**
- Operating system and version: Fedora 32
- Have you tried rebooting? Yes, multiple times with different kernels
- Is this a regression? Probably, it worked before the upgrade with Fedora 31

snugglej, 2020-05-02:
I just did a recent update to my Fedora 32 system and have the same exact issue. Before my updates today, I was not having this issue on Fedora 32.

```
20:43:17:0659 XbSilo               1.42ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='ff71992e-52f7-5eea-94ef-883e56e034c6']/../..|components/component/provides/firmware[@type='flashed'][text()='34801700-3a50-5b05-820c-fe14580e4c2d']/../..|components/component/provides/firmware[@type='flashed'][text()='8e1cbc5d-5a11-5149-bfea-b6065d5296ba']/../..|components/component/provides/firmware[@type='flashed'][text()='03f304f4-223e-54f4-b2c1-c3cf3b5817c6']/../..|components/component/provides/firmware[@type='flashed'][text()='52d7b679-db28-5bf7-bd87-41d77aeec600']/../..` limit=1 -> 0 results
20:43:17:0660 XbSilo               0.52ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='0a368367-cada-5b6c-a012-379142aed89a']/../..|components/component/provides/firmware[@type='flashed'][text()='d8930dfe-a1ce-5f00-9718-71dca91ac992']/../..` limit=1 -> 0 results
20:43:17:0660 FuEngine             Emitting PropertyChanged('Status'='idle')
20:43:17:0660 FuMain               Emitting PropertyChanged('Status'='idle')
20:43:17:0660 XbSilo               ignoring for OR statement: indexed string 'e2283c99-5aca-5bea-a5da-8e1f77034e30' was unfound
20:43:17:0660 XbSilo               ignoring for OR statement: indexed string '9fc11be0-da74-5add-942b-5b27647f2995' was unfound
20:43:17:0660 XbSilo               ignoring for OR statement: indexed string 'f18f6814-7f0e-53a1-91b8-6969a316ce6c' was unfound
20:43:17:0660 XbSilo               ignoring for OR statement: indexed string '8f72ef92-2740-56c2-8025-bac145539815' was unfound
20:43:17:0660 XbSilo               ignoring for OR statement: indexed string '914b1609-6152-57a3-9cd0-74f347a2de81' was unfound
20:43:17:0660 XbSilo               ignoring for OR statement: indexed string '9f29d21f-f7b0-524d-a6b3-901ce4e0f2fe' was unfound
20:43:17:0660 FuCommon             creating path /var/cache/fwupd/motd.d
20:43:17:0665 GLib-GIO             Failed to initialize portal (GMemoryMonitorPortal) for gio-memory-monitor: Not using portals
20:43:17:0665 GLib-GIO             _g_io_module_get_default: Found default implementation dbus (GMemoryMonitorDBus) for ‘gio-memory-monitor’
20:43:17:0666 FuMain               Started with locale en_US.UTF-8
20:43:17:0666 FuMain               Daemon ready for requests
20:43:17:0667 FuMain               FuMain: acquired name: org.freedesktop.fwupd
22:43:18:0182 FuIdle               status now shutdown
```

snugglej, 2020-05-02:
These are the updates that were applied and then this issue started to appear again for me.

Command run: `rpm -qa --last | head -n 200 | grep "01 May"`                                                                   ```                                 
kmod-nvidia-5.6.7-300.fc32.x86_64-440.82-1.fc32.x86_64 Fri 01 May 2020 12:53:35 AM CDT
vulkan-loader-1.2.135.0-1.fc32.i686           Fri 01 May 2020 12:52:48 AM CDT
mesa-vulkan-drivers-20.0.5-1.fc32.i686        Fri 01 May 2020 12:52:48 AM CDT
mesa-libglapi-20.0.5-1.fc32.i686              Fri 01 May 2020 12:52:48 AM CDT
mesa-libGL-20.0.5-1.fc32.i686                 Fri 01 May 2020 12:52:48 AM CDT
mesa-libgbm-20.0.5-1.fc32.i686                Fri 01 May 2020 12:52:48 AM CDT
mesa-libEGL-20.0.5-1.fc32.i686                Fri 01 May 2020 12:52:48 AM CDT
mesa-dri-drivers-20.0.5-1.fc32.i686           Fri 01 May 2020 12:52:48 AM CDT
crun-0.13-2.fc32.x86_64                       Fri 01 May 2020 12:52:48 AM CDT
kernel-headers-5.6.7-300.fc32.x86_64          Fri 01 May 2020 12:52:47 AM CDT
json-c-0.13.1-11.fc32.x86_64                  Fri 01 May 2020 12:52:47 AM CDT
js-jquery-3.5.0-2.fc32.noarch                 Fri 01 May 2020 12:52:47 AM CDT
ipcalc-0.4.1-1.fc32.x86_64                    Fri 01 May 2020 12:52:47 AM CDT
edk2-ovmf-20200201stable-1.fc32.noarch        Fri 01 May 2020 12:52:47 AM CDT
distribution-gpg-keys-1.38-1.fc32.noarch      Fri 01 May 2020 12:52:47 AM CDT
spirv-tools-libs-2019.5-2.20200421.git67f4838.fc32.x86_64 Fri 01 May 2020 12:52:46 AM CDT
openvpn-2.4.9-1.fc32.x86_64                   Fri 01 May 2020 12:52:46 AM CDT
openssl-pkcs11-0.4.10-6.fc32.x86_64           Fri 01 May 2020 12:52:46 AM CDT
openssh-server-8.2p1-3.fc32.x86_64            Fri 01 May 2020 12:52:46 AM CDT
openssh-clients-8.2p1-3.fc32.x86_64           Fri 01 May 2020 12:52:46 AM CDT
mpfr-4.0.2-4.fc32.x86_64                      Fri 01 May 2020 12:52:46 AM CDT
mesa-libxatracker-20.0.5-1.fc32.x86_64        Fri 01 May 2020 12:52:46 AM CDT
mesa-libGL-20.0.5-1.fc32.x86_64               Fri 01 May 2020 12:52:46 AM CDT
libxcrypt-devel-4.4.16-3.fc32.x86_64          Fri 01 May 2020 12:52:46 AM CDT
libxcrypt-compat-4.4.16-3.fc32.x86_64         Fri 01 May 2020 12:52:46 AM CDT
libuv-1.37.0-1.fc32.x86_64                    Fri 01 May 2020 12:52:46 AM CDT
libtirpc-1.2.6-0.fc32.x86_64                  Fri 01 May 2020 12:52:46 AM CDT
libslirp-4.2.0-2.fc32.x86_64                  Fri 01 May 2020 12:52:46 AM CDT
libibumad-29.0-1.fc32.x86_64                  Fri 01 May 2020 12:52:46 AM CDT
grub2-tools-extra-2.04-14.fc32.x86_64         Fri 01 May 2020 12:52:46 AM CDT
grub2-tools-efi-2.04-14.fc32.x86_64           Fri 01 May 2020 12:52:46 AM CDT
grub2-efi-x64-cdboot-2.04-14.fc32.x86_64      Fri 01 May 2020 12:52:46 AM CDT
grub2-efi-x64-2.04-14.fc32.x86_64             Fri 01 May 2020 12:52:46 AM CDT
grub2-efi-ia32-cdboot-2.04-14.fc32.x86_64     Fri 01 May 2020 12:52:46 AM CDT
grub2-efi-ia32-2.04-14.fc32.x86_64            Fri 01 May 2020 12:52:46 AM CDT
wine-core-5.7-1.fc32.x86_64                   Fri 01 May 2020 12:52:45 AM CDT
librdmacm-29.0-1.fc32.x86_64                  Fri 01 May 2020 12:52:45 AM CDT
grub2-pc-2.04-14.fc32.x86_64                  Fri 01 May 2020 12:52:45 AM CDT
fwupd-1.4.1-1.fc32.x86_64                     Fri 01 May 2020 12:52:45 AM CDT
mesa-libEGL-20.0.5-1.fc32.x86_64              Fri 01 May 2020 12:52:43 AM CDT
curl-7.69.1-3.fc32.x86_64                     Fri 01 May 2020 12:52:43 AM CDT
ostree-2020.3-3.fc32.x86_64                   Fri 01 May 2020 12:52:42 AM CDT
kernel-devel-5.6.7-300.fc32.x86_64            Fri 01 May 2020 12:52:42 AM CDT
hplip-3.20.3-5.fc32.x86_64                    Fri 01 May 2020 12:52:34 AM CDT
f32-backgrounds-gnome-32.1.4-1.fc32.noarch    Fri 01 May 2020 12:52:34 AM CDT
device-mapper-multipath-0.8.2-4.fc32.x86_64   Fri 01 May 2020 12:52:34 AM CDT
webkit2gtk3-2.28.2-1.fc32.x86_64              Fri 01 May 2020 12:52:33 AM CDT
qt5-qtwayland-5.13.2-4.fc32.x86_64            Fri 01 May 2020 12:52:33 AM CDT
keybinder3-0.3.2-9.fc32.x86_64                Fri 01 May 2020 12:52:32 AM CDT
kernel-modules-5.6.7-300.fc32.x86_64          Fri 01 May 2020 12:52:29 AM CDT
tigervnc-server-minimal-1.10.1-5.fc32.x86_64  Fri 01 May 2020 12:52:27 AM CDT
kernel-core-5.6.7-300.fc32.x86_64             Fri 01 May 2020 12:52:27 AM CDT
dbus-x11-1.12.16-4.fc32.x86_64                Fri 01 May 2020 12:52:27 AM CDT
cairo-gobject-1.16.0-8.fc32.x86_64            Fri 01 May 2020 12:52:25 AM CDT
cairo-1.16.0-8.fc32.x86_64                    Fri 01 May 2020 12:52:25 AM CDT
fontconfig-2.13.92-9.fc32.x86_64              Fri 01 May 2020 12:52:24 AM CDT
coreutils-common-8.32-4.fc32.1.x86_64         Fri 01 May 2020 12:52:24 AM CDT
coreutils-8.32-4.fc32.1.x86_64                Fri 01 May 2020 12:52:24 AM CDT
findutils-4.7.0-4.fc32.x86_64                 Fri 01 May 2020 12:52:23 AM CDT
f32-backgrounds-base-32.1.4-1.fc32.noarch     Fri 01 May 2020 12:52:23 AM CDT
device-mapper-multipath-libs-0.8.2-4.fc32.x86_64 Fri 01 May 2020 12:52:23 AM CDT
cups-filesystem-2.3.1-9.fc32.noarch           Fri 01 May 2020 12:52:23 AM CDT
cups-2.3.1-9.fc32.x86_64                      Fri 01 May 2020 12:52:23 AM CDT
wine-filesystem-5.7-1.fc32.noarch             Fri 01 May 2020 12:52:22 AM CDT
tigervnc-license-1.10.1-5.fc32.noarch         Fri 01 May 2020 12:52:22 AM CDT
ostree-libs-2020.3-3.fc32.x86_64              Fri 01 May 2020 12:52:22 AM CDT
[... cut ...]
grub2-tools-minimal-2.04-14.fc32.x86_64       Fri 01 May 2020 12:52:20 AM CDT
grub2-tools-2.04-14.fc32.x86_64               Fri 01 May 2020 12:52:20 AM CDT
grub2-common-2.04-14.fc32.noarch              Fri 01 May 2020 12:52:20 AM CDT
gettext-libs-0.20.2-1.fc32.x86_64             Fri 01 May 2020 12:52:20 AM CDT
gettext-0.20.2-1.fc32.x86_64                  Fri 01 May 2020 12:52:20 AM CDT
cups-libs-2.3.1-9.fc32.x86_64                 Fri 01 May 2020 12:52:20 AM CDT
glib2-2.64.2-2.fc32.x86_64                    Fri 01 May 2020 12:52:19 AM CDT
```

snugglej, 2020-05-02:
Just wanted to comment that I have the same problem and I reverted back to the [fwupd-1.3.9-2.fc32](<https://koji.fedoraproject.org/koji/buildinfo?buildID=1474421>) and I no longer have the timeout or update failure notification on my system. 

I think at least for me, that #2047 is related.

jprupp, 2020-05-02:
On my system the `fwupd.service` systemd service appears not to start quickly enough, and sometimes fail to start. When it is in that state the command `fwupdmgr --version` ends up with a timeout. It is possible that one or more of my USB devices connected through an external USB switch may be preventing `fwupd.service` from starting. I turned the machine off, removed power from my external USB switch, and then switched it all on again, and I'm not getting this problem anymore.

jprupp, 2020-05-02:
Even when not using the USB switch, the daemon can take longer than a minute to start. My USB keyboard seems to have something to do with that:

```
May 02 14:28:21 nebula fwupd[3171]: 13:28:21:0627 FuPluginUefi         failed to add /sys/firmware/efi/esrt/entries/entry0: ESRT GUID '00000000-0000-0000-0000-000000000000' was not valid
May 02 14:28:21 nebula fwupd[3171]: 13:28:21:0691 FuEngine             failed to get releases for Sabrent Rocket Q: No releases found: no HWIDs matched b31aa89d-cfb8-5c67-8502-ce76a9e7b580
May 02 14:29:31 nebula fwupd[3171]: 13:29:31:0759 FuEngine             failed to add udev device /sys/devices/pci0000:00/0000:00:07.1/0000:0b:00.3/usb5/5-3/5-3.2/5-3.2:1.0/0003:046D:C339.0001/hidraw/hidraw0: failed to ping Logitech PRO X Gaming Keyboard: failed to receive: failed to receive: timeout
May 02 14:29:31 nebula fwupd[3171]: 13:29:31:0766 FuEngine             failed to add udev device /sys/devices/pci0000:00/0000:00:07.1/0000:0b:00.3/usb5/5-3/5-3.2/5-3.2:1.1/0003:046D:C339.0002/hidraw/hidraw1: failed to ping Logitech PRO X Gaming Keyboard: generic failure
May 02 14:29:33 nebula fwupd[3171]: 13:29:33:0006 FuEngine             failed to get releases for Sabrent Rocket Q: No releases found: no HWIDs matched b31aa89d-cfb8-5c67-8502-ce76a9e7b580
May 02 14:29:33 nebula systemd[1]: Started Firmware update daemon.
```

jprupp, 2020-05-02:
This workaround works for me because it's the Logitech HID plugin which makes the daemon slow to start:

```sh
sudo perl -pi.bak -e 's/^BlacklistPlugins=/BlacklistPlugins=logitech_hidpp;/' /etc/fwupd/daemon.conf
```

snugglej, 2020-05-02:
@jprupp 
Will that Blacklist all Logitech devices? I have my keyboard plugged directly into my computer (no hub) and was experiencing the timeout issue. I will upgrade fwupd to the latest version and try that blacklist command for my keyboard. 

I also have a amazon basic's USB 3 hub that I believe was causing this issue also when it was plugged into my computer (I had to unplug it).

Thanks

jprupp, 2020-05-02:
It turned out that the USB switch was not causing any problems. It was only the Logitech HID plugin from `fwupd` that was failing to interact with my Logitech keyboard.

superm1, 2020-05-02:
Rather than blacklist the plug-in a better workaround is to set EnumerateAllDevices=false in fwupd.conf

flashkiller999, 2020-05-02:
I can confirm that the workaround from jprupp works.
Also this bug seems to also have been the cause that my keyboard did not work after logging in with my user. I had another keyboard (Generic Dell) laying around and after implementing the workaround the keyboard is working without issue after logging in.
I did not initially report this as I was unsure if the underlying problem was the same.

Zlopez, 2020-05-02:
I can confirm that this is also happening on Fedora 32 Silverblue.

loreloc, 2020-05-02:
Same problem here.
`May 02 21:41:39 fedorapc systemd[1]: Starting Firmware update daemon...
May 02 21:41:40 fedorapc fwupd[2376]: ERROR:esys:src/tss2-esys/esys_context.c:69:Esys_Initialize() Initialize default tcti. ErrorCode (0x000a000a)
May 02 21:42:10 fedorapc fwupd[2376]: 19:42:10:0097 FuEngine             failed to add udev device /sys/devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0/0003:046D:C336.0001/hidraw/hidraw0: failed to ping Logitech Gaming Keyboard G213: failed to receive: failed to receive: timeout
May 02 21:42:10 fedorapc fwupd[2376]: 19:42:10:0102 FuEngine             failed to add udev device /sys/devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.1/0003:046D:C336.0002/hidraw/hidraw1: failed to ping Logitech Gaming Keyboard G213: generic failure
May 02 21:42:10 fedorapc systemd[1]: Started Firmware update daemon.`

hughsie, 2020-05-02:
Does https://github.com/fwupd/fwupd/pull/2053 fix things?

snugglej, 2020-05-03:
@hughsie 

This fixed my issue after upgrading to the new version of fwupd and enabled `EnumerateAllDevices=false` in the daemon.conf.

_**My question:**_ Was the logitech HID detection or enumerating all devices introduced in the latest update because if I go back to the version before (1.3.9-2) of fwupd I do not have this problem.

Thanks

snugglej, 2020-05-03:
I did experience a weird issue where after the second reboot, I was unable to use my Logitech keyboard till I was booted into the system for a couple of minutes. I have an encrypted drive so I have to type a password in to unlock the drive which wouldn't happen without the  I'm going to do another reboot and see if this continues and test by disabling the Logitech plugin if it does happen again.

This issue is similar to @flashkiller999 report of his keyboard not working also.

loreloc, 2020-05-03:
@hughsie Thank you. I resolved with `EnumerateAllDevices=false`.

oaksisfolks, 2020-05-04:
> This fixed my issue after upgrading to the new version of fwupd and enabled `EnumerateAllDevices=false` in the daemon.conf.

This worked for me as well. Thanks.

superm1, 2020-05-13:
closed with https://github.com/fwupd/fwupd/pull/2053

closed 2020-05-13 by superm1

t3kg33k, 2020-06-19:
The EnumerateAllDevices=false is already set and I am still getting the annoying error message.
Any other possible fixes for this?

superm1, 2020-06-19:
open a new issue with all requested items in the template.  needs to be debugged separately.

dailytabs, 2020-09-14:
@t3kg33k Did you ever open that issue? I am getting the same. fwupd and/or firmware-lenovo-thinkpad have been nothing but trouble since I switched to Fedora. Leaving a really bad taste in my mouth on this brand-new "Fedora certified" laptop.

t3kg33k, 2020-09-16:
> @t3kg33k Did you ever open that issue? I am getting the same. fwupd and/or firmware-lenovo-thinkpad have been nothing but trouble since I switched to Fedora. Leaving a really bad taste in my mouth on this brand-new "Fedora certified" laptop.

Yes. I was able to find the fix.

dnf remove tpm2-abrmd

https://github.com/fwupd/fwupd/issues/2194