fwupd/fwupd #2052: fwupd disables touchpad on Dell K12A kbd-dock until reboot

jwrdegoede, 2020-05-02:
As requested I'm filing a big for this issue which I initially reported by email.

After replacing a quite old Fedora 27 install on one of my test machines with a Fedora 32 install I hit an interesting issue. The machine in question is a Dell Venue 11 Pro 7130 tablet / 2-in-1, with a keyboard dock with touchpad, turning it into more or less a
normal laptop.

The touchpad works fine in gdm, but it would stop working when I logged into a user GNOME-session.  It took me a while to get to the bottom of this. The problem is that the user-session ends up dbus activating fwupd and fwupd does some probe of the touchpad which puts it in a mode where it no longer generates any events.

`sudo rpm -e fwupd gnome-software` works around the issue of the touchpad no longer sending any events after I login to the user session.

hughsie, 2020-05-02:
@superm1 is that going to be wacom? @jwrdegoede can you get me a `sudo /usr/libexec/fwupd/fwupd --verbose` log please.

jwrdegoede, 2020-05-02:
Here are answers to the questions asked in your email response:

> Can we work on a less brutal fix (then the `rpm -e`) ?

Certainly that is the whole reason I reached out to you :)

> Is it an RMI4 touchpad by any chance?

No it is an USB touchpad/kbd combo using the HID-multitouch protocol for the touchpad part.

> Setting /etc/fwupd/daemon.conf BlacklistPlugins to include wacom_usb perhaps?

I've tried blacklisting wacom_usb, that does not help.

superm1, 2020-05-02:
A verbose daemon log would be really useful.

Also Can you please try to blacklist "dell" plugin and see if it helps?

superm1, 2020-05-02:
Btw are you sure it's USB? I didn't think that was how it worked. If so I would expect undocking and redocking fixes it.

jwrdegoede, 2020-05-02:
`fwupd --verbose` [log](https://fedorapeople.org/~jwrdegoede/fwupd.log)

Yes I'm sure it is USB. Unplug + replug fixes it, if I kill fwupd first so that it does not mess it with after the replug.

jwrdegoede, 2020-05-02:
Here are some interesting bits from the log:

```
14:59:12:0640 FuEngine             2 devices with subsystem hidraw
14:59:12:0642 FuDevice             using 82f75bac8ca62bf795945db8b38673283dd38e6f for HID_PHYS=usb-0000:00:14.0-4/input1
14:59:14:0645 FuEngine             failed to add udev device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.1/0003:06CB:2819.0002/hidraw/hidraw1: failed to read PDT entry @ 0x00e9: timeout
14:59:14:0654 FuDevice             using b2e4a7c83964e2fb37852b50e402d24961abc436 for HID_PHYS=usb-0000:00:14.0-4/input2
14:59:20:0026 FuEngine             failed to add udev device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.2/0003:06CB:2819.0003/hidraw/hidraw2: failed to set RMA page 0x0failed to write: wrote -1 of 21
```

hughsie, 2020-05-02:
> failed to read PDT entry

This is the synaptics_rmi plugin. Can you try blacklisting that?

jwrdegoede, 2020-05-02:
@superm1 a bit off-topic, bit I spend the last 2 days pinning down issues with the K12A kbd-dock because I have been doing some work on the intel-vbtn driver and I was testing the SW_TABLET_MODE functionality on the Venue 11 7130. It turns out that ATM doing the unplug + replug will result in the dock not working in about 9 out of 10 times, I have (finally) found a fix for that (use the HID_QUIRK_NO_INIT_REPORTS quirk). Now that I have that, and the surprise issue of the touchpad not working once fwupd starts, both solved I can finally post the intel-vbtn series upstream. I will put yoy in the Cc. I would appreciate it if you can test it (and maybe also review?).

jwrdegoede, 2020-05-02:
> > failed to read PDT entry
> 
> This is the synaptics_rmi plugin. Can you try blacklisting that?

Ok, I just tried this and that works around the issue, so we are getting somewhere.

I was under the impression that RMI was SMBus only, but I guess it can be encapsulated over USB too, and so I learn something new every day.

jwrdegoede, 2020-05-02:
Note the touchpad/kbd combo is a Synaptics device (uses Synaptics vendor id), I don't know if it is RMI capable, but the kernel is using the HID-multi-touch driver for it.

hughsie, 2020-05-02:
@jwrdegoede Does that PR help? ^^^

superm1, 2020-05-03:
> Yes I'm sure it is USB. Unplug + replug fixes it, if I kill fwupd first so that it does not mess it with after the replug.

Color me surprised.  These were never meant to sell with Linux so I never saw them or worked with them.

>@superm1 a bit off-topic, I spend the last 2 days pinning down issues with the K12A kbd-dock because I have been doing some work on the intel-vbtn driver and I was testing the SW_TABLET_MODE functionality on the Venue 11 7130. It turns out that ATM doing the unplug + replug will result in the dock not working in about 9 out of 10 times, I have (finally) found a fix for that (use the HID_QUIRK_NO_INIT_REPORTS quirk). Now that I have that, and the surprise issue of the touchpad not working once fwupd starts, both solved I can finally post the intel-vbtn series upstream. I will put you in the Cc. I would appreciate it if you can test it (and maybe also review?).

Well that's great you've worked that out.  I'll look at the patches next week.  As I said I don't have this hardware so I can't test it.

> I was under the impression that RMI was SMBus only, but I guess it can be encapsulated over USB too, and so I learn something new every day.

I2C too.  All Dell notebooks use it via I2C.

jwrdegoede, 2020-05-03:
@superm1 wrote:

> Well that's great you've worked that out. I'll look at the patches next week. As I said I don't have this hardware so I can't test it.

Right, what I meant was is that it would be good if you can test the intel-vbtn patch-series on some devices which use the intel-vbtn driver, to mkae sure that I've not caused any regressions there.

jwrdegoede, 2020-05-03:
> @jwrdegoede Does that PR help? ^^^

Yes I've applied the change directly to /usr/share/fwupd/quirks.d/synaptics-rmi.quirk on the system with the K12A kbd-dock and dropped synaptics_rmi from BlacklistPlugins.

I can confirm that with the /usr/share/fwupd/quirks.d/synaptics-rmi.quirk modification the touchpad keeps working without needing to add synaptics_rmi to BlacklistPlugins.

superm1, 2020-05-03:
> @superm1 wrote:
> 
> > Well that's great you've worked that out. I'll look at the patches next week. As I said I don't have this hardware so I can't test it.
> 
> Right, what I meant was is that it would be good if you can test the intel-vbtn patch-series on some devices which use the intel-vbtn driver, to make sure that I've not caused any regressions there.

With WFH nearly all of my hardware is inaccessible due to office closures.  I don't think i'll be able to check that, but I'll certainly review the code.

hughsie, 2020-05-05:
@jwrdegoede hey, can you attach the whole log of `sudo fwupdtool --plugin-whitelist synaptics_rmi --verbose get-devices` to this ticket please. That needs to be run from master, without the plugin blacklist or the new whitelist -- I want to see how exactly it's failing. Thanks.

jwrdegoede, 2020-05-07:
[Here](https://fedorapeople.org/~jwrdegoede/fwupd.log2) is the requested log.

Off-topic: I had a bit of trouble building fwupd because I have this small meson-configure script which sets the CFLAGS, etc. to the default Fedora ones before calling meson. This includes -Werror=discarded-qualifiers which breaks building the ep963x plugin. I tried fixing this by making the buf argument passed to various functions const *, but I gave up after the third level down and instead just dropped this from my CFLAGS, but I guess this ought to be fixed when you can make some time for this.

hughsie, 2020-05-07:
Continuing the off topic bit, I did `-Werror=discarded-qualifiers` and this:

    @@ -63,6 +63,7 @@ warning_flags = [
       '-Wcast-align',
       '-Wclobbered',
       '-Wdeclaration-after-statement',
    +  '-Wdiscarded-qualifiers',
       '-Wduplicated-branches',
       '-Wduplicated-cond',
       '-Wempty-body',

And saw no additional warning from anywhere. What other flags are you using? F32? gcc?

jwrdegoede, 2020-05-07:
<note still offtopic>

Fedora 32, gcc-10.0.1-0.11.fc32.x86_64

The problem is (starts with) the fu_ep963x_device_write() call at plugins/ep963x/fu-ep963x-device.c line 273, fu_ep963x_device_write() takes a guint8 *buf, and there its get passed a FuChunk.data which is a const guint8 * .

I tried making the  buf argument to fu_ep963x_device_write() a const guint8 *, which I believe is the right thing to do, but that led down a rabid-hole and I was planning on quickly gathering the log you requested so I decided to tweak the CFLAGS, but following the rabid-hole down to its end it probably the right thing to do.

hughsie, 2020-05-07:
> Fedora 32, gcc-10.0.1-0.11.fc32.x86_64

Huh, same. Weird.

I've updated wip/hughsie/rmi-whitelist which should fix things for you in a more backwards compat way. It basically turns on the requirements for updates on the LVFS before the device is added, unless whitelisted when it does it always. This keeps the enumeration working on hardware like the XPS13 which currently has no touchpad updates. @superm1 is that an acceptable compromise?

hughsie, 2020-05-07:
> Huh, same. Weird.

@jwrdegoede  does wip/hughsie/discarded-qualifiers help for the compile error?

closed 2020-05-14 by hughsie