fwupd/fwupd #2055: Regression with new thunderbolt plugin and X1 carbon

hughsie, 2020-05-04:
Trying to deploy the update I get:

    failed to read is-host: location is outside of the given image: attempted to read 0x01 bytes at offset 0x7f014b from buffer of 0x7e000

Will debug later a bit, but wanted to file first.

superm1, 2020-05-04:
On the X1 carbon host controller?  Any idea what is the generation of the controller, AR or TR?

I'd guess that @YehezkelShB got https://github.com/fwupd/fwupd/blob/master/plugins/thunderbolt/fu-thunderbolt-firmware.c#L530 backwards when correcting it on the PR, but I'll need to look at the original code.

superm1, 2020-05-04:
I looked through and I think what's happening is that the old code used the FARB pointer location in the "new" image as it's basis for offset to say where digital section starts.  It then used this plusrather than that of the NVM coming from the kernel.  So for logic of is-host is at digital + 0x10 doesn't make  sense if that farb pointer is completely off.

@YehezkelShB so is it a fair assumption that digital section starts at 0x00, or is that completely off base?  I guess we need this properly parsing the images the kernel spits out, not just brand new genuine images.

YehezkelShB, 2020-05-04:
When reading the "live" image from sysfs, there is no FARB whatsoever. You don't get the header there, it starts right away with the digital section. The original code used the FARB offset only when reading from a file.

Does it answer the question?

superm1, 2020-05-04:
Oh!  OK that makes more sense, I was trying to understand why this looked like garbage and not the same header.  OK I guess I need to re-do the PR some then.

superm1, 2020-05-04:
@YehezkelShB what about the other bytes in that header?  Are any of those a signature that we can be checking for to tell if it's a FARB header or start of a digital section?

superm1, 2020-05-04:
(or similarly are the first few bytes of the digital section always identical and can be triggered the same way)

YehezkelShB, 2020-05-04:
It's simple: if this is an image file, it MUST start with the headers (which include the FRAB pointer). If it's a "live" FW coming from sysfs, there are no headers at all. The code must know where it's reading from.

superm1, 2020-05-04:
I guess for the plugin flow it will know where it's reading from, but for the `firmware-parse` command this enables it won't know.  It will need to look at it and tell what it looks like.

Right now my PR is just checking those first 3 bytes and if they're bigger than the file it's no header.  I don't think that's guaranteed (who knows what digital starts with).

Looking at a real update file it's `00 04 00 00 00 00 08 00 00 00 00 00 00 00 00 00` followed by `FF`s.  So If I can look for those 00's in some mask then it can at least have a guess whether or not it seems like a file or kernel.

YehezkelShB, 2020-05-04:
This is why the original plugin had two "objects", one for reading from the live FW and one for reading from the image. Each one had its own set of sections. IMHO, this seems like the reasonable thing to do.

YehezkelShB, 2020-05-04:
And I don't think there were ever any promise which bytes in the header are all `0xFF`. I agree we don't expect to see so many FFs in the digital section itself.

superm1, 2020-05-04:
Ok then I think the right thing to do is create two FuFirmware types for the thunderbolt update image and the kernel image then.

superm1, 2020-05-05:
I guess given you branched at 748a65f3c788abfc99769b4aa6d3eb25e815e0de the fix for this will need to go to 1_4_X too.

hughsie, 2020-05-05:
Nah, I branched before and cherry picked any fixes.

superm1, 2020-05-06:
Should be closed on master now.  Please re-open if still similar problems pop up.

closed 2020-05-06 by superm1