fwupd/fwupd #2058: fwupd fails to start on first try

lovesegfault, 2020-05-05:
**Describe the bug**
Whenever I want to run fwupdmgr this is what happens:
```
❯ fwupdmgr refresh
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)

❯ fwupdmgr refresh
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.jcat
Downloading…             [***************************************]
Fetching metadata https://cdn.fwupd.org/downloads/firmware-00522-stable.xml.gz
Downloading…             [***************************************]
Successfully downloaded new metadata: 5 local devices supported
```
Here, `refresh` is illustrative, as any other sub-command will present the exact same behavior of not working on the first try.

While inspecting `dmesg` provides no clues, the journal has some pointers:
```
❯ journalctl --since=today -xeu fwupd.service
-- Logs begin at Fri 2020-04-17 20:09:35 PDT, end at Mon 2020-05-04 19:11:59 PDT. --
May 04 19:09:58 foucault systemd[1]: Starting Firmware update daemon...
-- Subject: A start job for unit fwupd.service has begun execution
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
--
-- A start job for unit fwupd.service has begun execution.
--
-- The job identifier is 981.
May 04 19:09:59 foucault fwupd[16565]: 02:09:59:0456 FuEngine             failed to probe device usb:00:03: USB error on device 0bda:0328 : Entity not found [-5]
May 04 19:10:39 foucault fwupd[16565]: 02:10:39:0625 FuEngine             failed to add udev device /sys/devices/virtual/misc/uhid/0005:046D:B01D.0009/hidraw/hidraw7: failed to ping MX Ergo: failed to receive: failed to receive: timeout
May 04 19:10:39 foucault systemd[1]: Started Firmware update daemon.
-- Subject: A start job for unit fwupd.service has finished successfully
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
--
-- A start job for unit fwupd.service has finished successfully.
--
-- The job identifier is 981.
```

**Steps to Reproduce**
1. run `fwupdmgr`

**Expected behavior**
For it to work normally on the first try

**fwupd version information**
```shell
❯ fwupdmgr --version
client version:	1.4.0
compile-time dependency versions
	gusb:	0.3.3
	efivar:	37
daemon version:	1.4.0
```
(NixOS)

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
20MD0028US
│
├─Thunderbolt Controller:
│     Device ID:           0e94de1012ceedd6f315e0e89989abd11337ec4f
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     50.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               138f4141-7a3f-57a3-8e38-f10c038f2d91
│                          eb0ff241-791a-5fcf-bb4a-b84f89f2a25a
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│   
├─Display controller:
│     Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
│     Vendor:              Intel Corporation (PCI:0x8086)
│     GUIDs:               5c257299-d4b9-5b11-8437-7b6735bcc979
│                          ff004cf7-df9b-5aa9-9aaa-c816cb07fe1f
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│   
├─Embedded Controller:
│     Device ID:           6b3fcb33b92ea5cdd94954b276ad0b63ed14f9bb
│     Current version:     0.1.13
│     Minimum Version:     0.0.1
│     Vendor:              DMI:LENOVO
│     GUID:                74997a6b-1adf-4b12-b994-401f06ea8c72
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─GP107GLM [Quadro P2000 Mobile]:
│     Device ID:           71b677ca0f1bc2c5b804fa1d59e52064ce589293
│     Current version:     a1
│     Vendor:              NVIDIA Corporation (PCI:0x10DE)
│     GUIDs:               bc8f6213-8695-5407-be5d-cd03808d40af
│                          07633373-c1be-577f-a080-1f9ae17806eb
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│   
├─Intel AMT [unprovisioned]:
│     Device ID:           e2623122c99d58220498aacbfcfdb1baebbae3c5
│     Summary:             Hardware and firmware technology for remote out-of-band management
│     Current version:     12.0.47.1524
│     Bootloader Version:  12.0.47.1524
│     Vendor:              Intel Corporation
│     GUID:                2800f812-b7b4-2d4b-aca8-46e0ff65814c
│     Device Flags:        • Internal device
│   
├─Intel Management Engine:
│     Device ID:           aca1c6cf5901b23b637fc8a1fdafeac85700799d
│     Current version:     192.47.1524
│     Minimum Version:     0.0.1
│     Vendor:              DMI:LENOVO
│     GUID:                4226a48a-8c1e-4f66-8f54-7e5dc33dcba9
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─Intel(R) Xeon(R) E-2176M  CPU @ 2.70GHz:
│     Device ID:           4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:     0xd2
│     Vendor:              GenuineIntel
│     GUID:                b9a2dd81-159e-5537-a7db-e7101d164d3f
│     Device Flags:        • Internal device
│   
├─SAMSUNG MZVLB1T0HALR-000L7:
│     Device ID:           c6a0cfba7c7d81e253fce571e1d1e9f6003ae1c7
│     Summary:             NVM Express Solid State Drive
│     Current version:     5L2QEXA7
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     GUIDs:               6e54c992-d302-59ab-b454-2d26ddd63e6d
│                          47335265-a509-51f7-841e-1c94911af66b
│                          08575ca8-c22c-540a-8ac9-e784c04f6e5a
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           71519311a643244a2eaf392fbf17c968ec135dd3
│     Current version:     0.1.28
│     Minimum Version:     0.1.28
│     Vendor:              LENOVO (DMI:LENOVO)
│     GUID:                a95d69e7-2fa0-4849-9159-75645bc6e153
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─TPM:
│ │   Device ID:           c6a80ac3a22083423992a3cb15018989f37834d6
│ │   Current version:     73.4.17568.4452
│ │   Vendor:              ST Microelectronics (TPM:STM)
│ │   GUIDs:               ff71992e-52f7-5eea-94ef-883e56e034c6
│ │                        84df3581-f896-54d2-bd1a-372602f04c32
│ │                        bfaed10a-bbc1-525b-a329-35da2f63e918
│ │                        70b7b833-7e1a-550a-a291-b94a12d0f319
│ │                        06f005e9-cb62-5d1a-82d9-13c534c53c48
│ │   Device Flags:        • Internal device
│ │ 
│ └─Event Log:
│       Device ID:         58bd405f31c48e6eca290b425f530a94c91e955c
│       GUID:              a25657fe-b5dc-5be0-8b78-8b9dfec678ff
│       Device Flags:      • Internal device
│     
└─UEFI Device Firmware:
      Device ID:           1a2df614d3488e4d0238fc21460adcbf6d256de2
      Current version:     4784132
      Minimum Version:     4784132
      Vendor:              DMI:LENOVO
      GUID:                cc426b6d-b6f9-496e-8676-82af7596d239
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Needs a reboot after installation
                           • Device is usable for the duration of the update
```

**Additional questions**
- Operating system and version: NixOS 20.09 unstable
- Have you tried rebooting? Yes
- Is this a regression? I've had this for a while

hughsie, 2020-05-05:
Is this a dupe of https://github.com/fwupd/fwupd/issues/2048?

snugglej, 2020-05-05:
MX Ergo is a logitech device, specifically a mouse so I think you are right @hughsie 

@lovesegfault Did you try the fixes in either of #2048 or #2047 (same fix) for this? It should work.

closed 2020-06-11 by stale[bot]