fwupd/fwupd #2078: Firmware Report request always fails.

ardevd, 2020-05-11:
**Describe the bug**
After installing a firmware update on my X1 Carbon I get the question of whether I want to send a report the next time I run `fwupdmgr get-updates`. However, regardless of what option I select I always get "Request Cancelled" in return. I believe this prevents me from installing future updates since I'm always asked to send a report from the previous upgrade.

**Steps to Reproduce**
```
fwupdmgr update
# Wait for install to complete
fwupdmgr refresh
fwupdmgr get-updates
• Thunderbolt Controller has the latest available firmware version
• Embedded Controller has the latest available firmware version
• Intel Management Engine has the latest available firmware version
• SAMSUNG <REDACTED> has the latest available firmware version
• System Firmware has the latest available firmware version
________________________________________________

Devices that have been updated successfully:

 • System Firmware (0.1.44 → 0.1.45)

Uploading firmware reports helps hardware vendors to quickly identify failing and successful updates on real devices.
Upload report now? (Requires internet connection):
0.	Do not upload reports at this time, but prompt again for future updates
1.	Do not upload reports, and never ask to upload reports for future updates
2.	Upload reports just this one time, but prompt again for future updates
3.	Upload reports this time and automatically upload reports after completing future updates
0
Request canceled
```

**Expected behavior**
The report prompt should properly react to my choice.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version:	1.4.1
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.4.1
```
Fedora F32

hughsie, 2020-05-11:
> Request canceled

And you pressed no buttons at all?

> Fedora F32

What desktop and shell are you using?

ardevd, 2020-05-11:
The Request Cancelled message is returned when I select an option from the list, regardless of what option I pick. It does not happen until I select an option.

As a side note, this used to work just fine up until the last firmware update.

Using swaywm + kitty

superm1, 2020-05-11:
Can you please try outside of kitty?  In something more mainstream like xterm or gnome-terminal?  Or even switching to a TTY and doing it there.

My suspicion is that the interactive check is failing in it for some reason.
https://github.com/fwupd/fwupd/blob/master/src/fu-util.c#L2724

ardevd, 2020-05-12:
It worked in gnome-terminal but only if I picked any other option besides 0.

superm1, 2020-05-12:
Right, so then something kitty is doing is causing it to not look like an interactive terminal (IE 	`if (isatty (fileno (stdout)) == 0)` fails)

I think you should report this to them.

hughsie, 2020-05-12:
> I think you should report it to them.

Agreed, sorry -- you need to report it to them instead. There's nothing much we can do.

closed 2020-05-12 by hughsie