fwupd/fwupd #2081: Build for Windows x86 (32-bit) for usability on ARM64 platform

nhivp, 2020-05-12:
**Is your feature request related to a problem? Please describe.**
As you know, an ARM64 platform is able to execute x86 applications with X86_Emulator mode enabled. So, I see the fwupdtool.exe is built by compiler x86_64-w64-mingw32, not i686-w64-mingw32. I already checked the fwupdtool.exe, it did not work for me.

For a similar x86 Application, the ipmiutil (win32) can work well with the ARM64 platform. http://ipmiutil.sourceforge.net/.

**Describe the solution you'd like**
I'm not an expert in cross-compiling for Windows Application. But, I think we should the target build for Windows should be both for Windows 32-bits users (x86_64-w64-mingw32) and Windows 64-bits users (i686-w64-mingw32).

**Describe alternatives you've considered**
None

**Additional context**
None

hughsie, 2020-05-12:
So, you want to run `fwupdatool.exe` on Windows ARM64? The former is possible I guess, although the number of people using the i686 builds would be pretty tiny. I'd worry if updating would actually work when emulating the hardware. If you tell us what hardware and OS you're using (and what you're trying to do) I guess we can be more helpful :)

superm1, 2020-05-12:
Probably the best first question here is what kind of devices you want to update from windows arm64.   Right now the only thing that can work is USB devices.  Do those even work in x86 emulator?

nhivp, 2020-05-13:
Hi,

Thanks for your interest.

> you want to run fwupdatool.exe on Windows ARM64?

Yes, wanna see the fwuptool.exe on Windows ARM64.
I'm using the eMAG Dev System and get Windows booting on it. You can get hardware info here: https://amperecomputing.com/emag

> Probably the best first question here is what kind of devices you want to update from windows arm64.

I'm using this tool for the UEFI Capsule Update. It seems the plugin_uefi is disabled for windows target build. Anyway, the fwupdtool.exe should work with a simple command like `.\fwupdtool.exe`

superm1, 2020-05-13:
For UEFI capsule update it won't work on windows. Issuing a capsule update on windows is different than Linux. On Windows it's using windows bootloader to install capsule and driver PNP API to stage it. On Linux it's staged using fwupd and efivarfs and a dedicated bootloader.

I think if we want UEFI capsules working on windows best bet is to use that windows API from plugin_uefi instead of the fwupd Linux way.

How about we make a deal, if you get the windows API working for UEFI capsules in a pull request and in exchange we start getting CI building for i686?

nhivp, 2020-05-13:
Yeah, I see. Windows can not create a new boot entry for fwupd efi if we don't have a driver for it. And... something like how to install capsule update packages on Windows.

Ok, thanks for the informative responses. It's just a curious question, we may close this issue.

closed 2020-05-13 by nhivp

superm1, 2020-05-13:
Well you know actually thinking of it, windows does offer an API for creating efi variables. I think the other way this can be done is to port efivar to windows to use that API. Once that's done it might be possible to bundle and use Linux bootloader with windows. But yeah it's not a trivial effort.