fwupd/fwupd #2084: Lenovo T470s - "Secure boot is enabled, but shim isn't installed to the EFI system partition"

miabbott, 2020-05-12:
**Describe the bug**
On a Lenovo T470s running Fedora Silverblue 32, when trying to do `fwupdmgr update` the error "Secure boot is enabled, but shim isn't installed to the EFI system partition" is received after attempting to update one device.  This may be user error on my part (or misconfiguration) but I'm out of my element here and looking for help.  I tried following along with #1797 but I think the problem there may be different enough to warrant a new issue.

**Steps to Reproduce**

```shell
$ fwupdmgr update
Upgrade available for Embedded Controller from 0.1.20 to 0.1.21
20HGS22D00 must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: y
Downloading 0.1.21 for Embedded Controller...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating Embedded Controller… \                                  ]
Scheduling…              [***************************************]
Secure boot is enabled, but shim isn't installed to the EFI system partition
```

**Expected behavior**
All supported devices are updated

**fwupd version information**

```shell
$ fwupdmgr --version
client version:	1.4.1
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.4.1
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): `dnf`

**fwupd device information**

```shell
$  fwupdmgr get-devices --show-all-devices
20HGS22D00
│
├─Embedded Controller:
│     Device ID:           03a6930fc4c833443a7502dde379b98b961c8c57
│     Current version:     0.1.20
│     Minimum Version:     0.0.1
│     Vendor:              DMI:LENOVO
│     Update State:        failed-transient
│     Update Error:        Secure boot is enabled, but shim isn't installed to the EFI system partition
│     Last modified:       2020-05-12 13:30
│     GUIDs:               e2acd8e4-a376-47bb-a316-4f1e62a8ca1f
│                          118d4b5a-8ef0-5dc6-b320-060dc79f2e36
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─HD Graphics 620:
│     Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
│     Current version:     02
│     Vendor:              Intel Corporation (PCI:0x8086)
│     GUIDs:               2886e312-afd8-5510-8993-12d568a85e00
│                          38f6c009-a25c-5b50-b3b5-fef4f9c6b846
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│   
├─Intel AMT [unprovisioned]:
│     Device ID:           e2623122c99d58220498aacbfcfdb1baebbae3c5
│     Summary:             Hardware and firmware technology for remote out-of-band management
│     Current version:     11.8.60.3561
│     Bootloader Version:  11.8.60.3561
[... 135 lines cut ...]
```shell
$ cat /etc/os-release 
NAME=Fedora
VERSION="32.20200507.0 (Silverblue)"
ID=fedora
VERSION_ID=32
VERSION_CODENAME=""
PLATFORM_ID="platform:f32"
PRETTY_NAME="Fedora 32.20200507.0 (Silverblue)"
ANSI_COLOR="0;34"
LOGO=fedora-logo-icon
CPE_NAME="cpe:/o:fedoraproject:fedora:32"
HOME_URL="https://fedoraproject.org/"
DOCUMENTATION_URL="https://docs.fedoraproject.org/en-US/fedora-silverblue/"
SUPPORT_URL="https://fedoraproject.org/wiki/Communicating_and_getting_help"
BUG_REPORT_URL="https://bugzilla.redhat.com/"
REDHAT_BUGZILLA_PRODUCT="Fedora"
REDHAT_BUGZILLA_PRODUCT_VERSION=32
REDHAT_SUPPORT_PRODUCT="Fedora"
REDHAT_SUPPORT_PRODUCT_VERSION=32
PRIVACY_POLICY_URL="https://fedoraproject.org/wiki/Legal:PrivacyPolicy"
VARIANT="Silverblue"
VARIANT_ID=silverblue
OSTREE_VERSION='32.20200507.0'
```

superm1, 2020-05-12:
Thanks for sharing all those logs.  Can you share your `efibootmgr -v` output too?  I notice that you don't have a shimx64.efi but instead a `shim.efi` and `shim-fedora.efi` neither of which fwupd actually looks for.

miabbott, 2020-05-12:
```$ efibootmgr -v
BootCurrent: 0000
Timeout: 0 seconds
BootOrder: 0000,0017,0018,0019,001A,001B,001C,001D,001E,0002
Boot0000* Fedora	HD(1,GPT,b20a17e1-dd39-45b9-9d9d-4e6e78944d7c,0x800,0xfa000)/File(\EFI\fedora\shim.efi)
Boot0001* Linux-Firmware-Updater \fwupx64.efi	HD(1,GPT,b20a17e1-dd39-45b9-9d9d-4e6e78944d7c,0x800,0xfa000)/File(\EFI\fedora\shim.efi)\.f.w.u.p.x.6.4...e.f.i...
Boot0002* Linux-Firmware-Updater	HD(1,GPT,b20a17e1-dd39-45b9-9d9d-4e6e78944d7c,0x800,0xfa000)/File(\EFI\fedora\fwupdx64.efi)
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0015  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0016  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0017* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001A* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot001B* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001C* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001D  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot001E  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot001F* IDER BOOT CDROM	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,1)
Boot0020* IDER BOOT Floppy	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,0)
Boot0021* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0022* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
Boot0023* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
```

superm1, 2020-05-12:
Looks like it's `shim.efi` in use.  This is not something that fwupd is looking for so it throws all this off.
@vathpela can you explain?  Why is silverblue deviating here and installing a different bootloader name?  And what is the extra `shim-fedora.efi` from?

superm1, 2020-05-12:
Code link on what we currently do to build the filename to check:
https://github.com/fwupd/fwupd/blob/7b866e5f92e2249c9de52b4fcacaa79839aa5c97/plugins/uefi/fu-uefi-bootmgr.c#L312
https://github.com/fwupd/fwupd/blob/b87d48a042b64c068a2b1c10b91771f733a01d4f/plugins/uefi/fu-uefi-common.c#L73

Definitely an assumption that it's `shim$arch.efi`

miabbott, 2020-05-12:
FWIW

```
$ sudo rpm -qf /boot/efi/EFI/fedora/shim.efi
shim-x64-15-8.x86_64

$ rpm -qi shim-x64 
Name        : shim-x64
Version     : 15
Release     : 8
Architecture: x86_64
Install Date: Wed 06 May 2020 09:48:46 PM EDT
Group       : Unspecified
Size        : 6353742
License     : BSD
Signature   : RSA/SHA256, Thu 15 Aug 2019 12:36:06 PM EDT, Key ID 6c13026d12c944d0
Source RPM  : shim-15-8.src.rpm
Build Date  : Tue 02 Oct 2018 02:35:15 PM EDT
Build Host  : bkernel03.phx2.fedoraproject.org
Packager    : Fedora Project
Vendor      : Fedora Project
URL         : https://github.com/rhboot/shim/
Bug URL     : https://bugz.fedoraproject.org/shim
Summary     : First-stage UEFI bootloader
Description :
Initial UEFI bootloader that handles chaining to a trusted full
bootloader under secure boot environments. This package contains the
version signed by the UEFI signing service.

$ rpm -ql shim-x64-15-8.x86_64
/boot/efi/EFI/BOOT/BOOTX64.EFI
/boot/efi/EFI/BOOT/fbx64.efi
/boot/efi/EFI/fedora/BOOTX64.CSV
/boot/efi/EFI/fedora/mmx64.efi
/boot/efi/EFI/fedora/shim.efi
/boot/efi/EFI/fedora/shimx64-fedora.efi
/boot/efi/EFI/fedora/shimx64.efi
``

vathpela, 2020-05-12:
> Looks like it's `shim.efi` in use.  This is not something that fwupd is looking for so it throws all this off.
> @vathpela can you explain?  Why is silverblue deviating here and installing a different bootloader name?  And what is the extra `shim-fedora.efi` from?

I don't know why they're doing this, it is wrong. shim-fedora.efi is the one that's signed by our key and not MS (because of ancient politics), but nothing should be using it directly.

superm1, 2020-05-12:
@miabbott have you manually made changes to the ESP?  It looks like according to that RPM you should have had both `shim.efi` and `shimx64.efi`.  Maybe some fallback code got used to add `shim.efi` to boot list because `shimx64.efi` was missing?

miabbott, 2020-05-12:
I don't recall doing any modifications to the ESP.  I've been upgrading this host continuously since Fedora 26 days, so maybe things have changed re: Silverblue + EFI shim since the original install?

I'd have to dig through some history to see if anything crops up.

superm1, 2020-05-12:
OK, well I think based on @vathpela's comments above this not a fwupd bug though.  fwupd expects to find it in `shim$arch.efi` and if it's not there it's a bug in something else. I would recommend to file this in Fedora's bugzilla.  Feel free to link here.  If it turns out that it's something fwupd did, please re-open here.

closed 2020-05-12 by superm1

cgwalters, 2020-05-12:
Possibly related to

https://github.com/coreos/rpm-ostree/pull/969

As well as the way the rpmostreepayload for Anaconda installs things https://github.com/rhinstaller/anaconda/blob/f9ce9f3c2e62f303a700f72cc2ef803c369f2b86/pyanaconda/payload/rpmostreepayload.py#L141

See also https://github.com/ostreedev/ostree/pull/1873

jlebon, 2020-05-12:
> I've been upgrading this host continuously since Fedora 26 days

It's entirely possible that's why, because most of `/boot` doesn't get updated right now. On my FSB at least:

```
# efibootmgr -v | grep shim
Boot0001* Fedora        HD(1,GPT,cce1bfdb-f757-46fb-9c68-a1dcd54502d1,0xffff,0x5fffa)/File(\EFI\fedora\shimx64.efi)
Boot001D* anaconda      HD(1,GPT,104579db-b9e6-45c9-b9d6-216f7d6114a1,0xffff,0x5fffa)/File(\EFI\fedora\shimx64.efi)
Boot001E* Fedora        HD(1,GPT,104579db-b9e6-45c9-b9d6-216f7d6114a1,0xffff,0x5fffa)/File(\EFI\fedora\shimx64.efi)
Boot001F* Fedora        HD(2,GPT,0b86300f-cfc7-449b-a007-5658bbe82c1e,0x1f4800,0x82000)/File(\EFI\fedora\shim.efi)
```

IIRC this is from a fresh f30 install.

cgwalters, 2020-05-12:
All of this TL;DR - for ostree-based systems we have a transactional way to update the OS, but that doesn't cover `/boot`. 

And I don't want a silent "Please don't poweroff your computer" phase for updates, so the status quo is we don't update the ESP (or MBR).

Per discussion in the ostree issue, we don't even have a documented manual tool for this yet unfortunately.

But the eventual solution would ideally be a shared tool much like fwupd, and ideally it'd also understand how to at least do "reconciliable" updates where if we get interrupted halfway through we complete the update and reboot again e.g. - see also https://github.com/ostreedev/ostree/pull/1967

superm1, 2020-05-12:
Not having a method for updating the ESP I think is incredibly dangerous.  If secure boot keys are revoked and need to be rotated with new bootloader binaries signed with new keys there is no infrastructure to let the OS keep booting that means.

cgwalters, 2020-05-12:
> Not having a method for updating the ESP I think is incredibly dangerous. 

Even more dangerous is having your root filesystem managed by software which leaves it a corrupted mess if interrupted halfway through, eventually instilling fear in users about applying security updates.

superm1, 2020-05-12:
That's the whole draw with os-tree.  I get that, and for hundred meg updates it's real value.

I think until there is a design that allows A-B ESP partitions and designating to the firmware which one to use there should be something that allows dropping an updated bootloader to the ESP.  We're talking about a few seconds and something that is updated relatively unfrequently.  Maybe even doing it during bootup the 1 or 2 times a year that it gets updated.

superm1, 2020-05-12:
Thanks.

>I have a years old fwupd EFI binary

shim binary, but yeah point is the same for all stuff that's put on the ESP.

to @miabbott for now what you can do is manually clone `shimx64.efi` from within os-tree directories to your ESP.  Fwupd should work properly then.

miabbott, 2020-05-12:
Thanks @superm1 and @cgwalters (and everyone else!) for the help!  :heart:

miabbott, 2020-05-12:
Here's what I ended up doing:

```
# In a container
$ sudo dnf install --downloadonly --downloaddir=. shim-x64
$ rpm2cpio shim-x64-15-8.x86_64.rpm | cpio -idmv
./boot/efi/EFI/BOOT/BOOTX64.EFI
./boot/efi/EFI/BOOT/fbx64.efi
./boot/efi/EFI/fedora/BOOTX64.CSV
./boot/efi/EFI/fedora/mmx64.efi
./boot/efi/EFI/fedora/shim.efi
./boot/efi/EFI/fedora/shimx64-fedora.efi
./boot/efi/EFI/fedora/shimx64.efi
12412 blocks

# On my host
$ sudo cp ./boot/efi/EFI/fedora/shimx64.efi /boot/efi/EFI/fedora/
$ sudo cp ./boot/efi/EFI/fedora/shimx64-fedora.efi /boot/efi/EFI/fedora/

$ fwupdmgr update
Upgrade available for Embedded Controller from 0.1.20 to 0.1.21
20HGS22D00 must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: y
Downloading 0.1.21 for Embedded Controller...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating Embedded Controller…   \                                ]
Scheduling…              [***************************************]
Successfully installed firmware
Upgrade available for Intel Management Engine from 184.60.3561 to 184.70.3626
20HGS22D00 must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: y
Downloading 184.70.3626 for Intel Management Engine...
Fetching firmware https://fwupd.org/downloads/3e0b9142cec8eca0385734c759e4e3f1023e64cf-Lenovo-ThinkPad-T470s-CorporateMEFirmware-11.8.70.3626.cab
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating Intel Management Engine…                                ]
Scheduling…              [***************************************]
Successfully installed firmware
• Logitech M705 has no available firmware updates
• SAMSUNG MZVLW256HEHP-000L7 has no available firmware updates
Upgrade available for System Firmware from 0.1.30 to 0.1.37
20HGS22D00 must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: y
Downloading 0.1.37 for System Firmware...
Fetching firmware https://fwupd.org/downloads/3f2114ae2c410271d990350231e73f784c6bca465f6949136d201fb1485ab2d0-Lenovo-ThinkPad-T470s-SystemFirmware-1.37.cab
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating System Firmware…[          |                            ]
Scheduling…              [***************************************]
Successfully installed firmware: TPM PCR0 differs from reconstruction, please see https://github.com/fwupd/fwupd/wiki/TPM-PCR0-differs-from-reconstruction
• Unifying Receiver has no available firmware updates

An update requires a reboot to complete. Restart now? [y|N]: n
```

miabbott, 2020-05-12:
For that `TCM PCR0 differs from reconstruction` message - #2083

miabbott, 2020-05-12:
Rebooted successfully and watched the firmware updates land, too!  :tada: 

```
$ fwupdmgr update 
• Embedded Controller has the latest available firmware version
• Intel Management Engine has the latest available firmware version
• Logitech M705 has no available firmware updates
• SAMSUNG MZVLW256HEHP-000L7 has no available firmware updates
• System Firmware has the latest available firmware version
• Unifying Receiver has no available firmware updates
```