fwupd/fwupd #2089: Failed to update metadata for lvfs: checksum failure: failed to verify data

dadosch, 2020-05-13:
**Describe the bug**
downloading fails with checksum failure

**Steps to Reproduce**
run fwupmd refresh. I've tried in in different networks, so it's a network issue on my side.

**Expected behavior**
It should download properly

**fwupd version information** (through pacman)
```shell
$ fwupdmgr --version
client version: 1.4.1
compile-time dependency versions
        gusb:   0.3.4
        efivar: 37
daemon version: 1.4.1
```

```
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.jcat
Downloading?             [***************************************]
Fetching metadata https://cdn.fwupd.org/downloads/firmware-00528-stable.xml.gz
Downloading?             [***************************************]
Failed to update metadata for lvfs: checksum failure: failed to verify data, expected 03cf7e2f57cf619e3e3536840fc8e7c155173610
```

superm1, 2020-05-13:
Huh, with 1.4.1 this is happening?  It should have been fixed in the changes for https://github.com/fwupd/fwupd/issues/391

Can you please share the files cached in `~/.cache/fwupd` to this bug?
* firmware.xml.gz.jcat
* firmware-00528-stable.xml.gz

We should double check if they were actually corrupted relative to what LVFS was sending.

hughsie, 2020-05-13:
I've just downloaded them myself and the SHA256 hash is wrong. I have no idea.

hughsie, 2020-05-13:
Server is OOMing:

[6702617.015996] Out of memory: Kill process 21460 (python) score 533 or sacrifice child
[6702617.017705] Killed process 21460 (python) total-vm:2534340kB, anon-rss:2148856kB, file-rss:0kB, shmem-rss:0kB
[6702617.159063] oom_reaper: reaped process 21460 (python), now anon-rss:0kB, file-rss:0kB, shmem-rss:0kB

hughsie, 2020-05-13:
Server rebooted, just debugging.

hughsie, 2020-05-13:
On reboot it's behaving. I'll continue to debug tomorrow. @dadosch it works for you now?

dadosch, 2020-05-13:
@hughsie can confirm that it works now. :+1:

closed 2020-06-12 by superm1