fwupd/fwupd #2105: RequireShimForSecureBoot=false ignored

smrqdt, 2020-05-18:
**Describe the bug**
`fwupdmgr update` fails with the error 
> Secure boot is enabled, but shim isn't installed to the EFI system partition

even though I have the following line in my `uefi.conf`:
> RequireShimForSecureBoot=false

I uses self-signed secure boot, it worked many times in the past. The system is a ThinkPad T480 with Arch Linux. 

**Steps to Reproduce**
1. set `RequireShimForSecureBoot=false` in `/etc/fwupd/uefi.conf`
1.  `fwupdmgr update`

**Expected behavior**
The update is staged and the system either reboots or BootNext is set to the Firmware Updater EFI boot entry. 

**fwupd version information**
Please provide the version of the daemon and client.
```shell
% fwupdmgr --version
client version:	1.4.1
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.4.1
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
`pacman`

**fwupd device information**

```shell
% fwupdmgr get-devices --show-all-devices
20L5000BGE
│
├─ThinkPad Thunderbolt 3 Dock:
│     Device ID:           72e5302683361a0e892a278590124cd475a8a11a
│     Current version:     15.00
│     Vendor:              Lenovo (TBT:0x0108)
│     GUID:                df675f10-53ac-59ca-bddd-0a86ee492920 ← TBT-01081630
│     Device Flags:        • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─Thunderbolt Controller:
│     Device ID:           7ee3d13a12eed17e585b3b0162293de23cb01bc2
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     14.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               7d131734-64ed-5865-a46e-a67cfb9b5a46 ← TBT-01091707
│                          e45313fd-8d5c-54e9-b175-1f15ffe6b05c ← TBT-01091707-controller0-0
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─Embedded Controller:
│     Device ID:           c19580db344e418cf2b23297bcf46d75c06e432a
│     Current version:     0.1.20
│     Minimum Version:     0.1.20
│     Vendor:              DMI:LENOVO
│     GUID:                b87a926d-189c-49a1-b1ff-921099b1de89
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
[... 194 lines cut ...]
│           ├── fwupdx64.efi
│           ├── linux-signed.efi
│           └── linux-signed.efi.bak
├── efikeys [error opening dir]
├── initramfs-linux-fallback.img
├── initramfs-linux.img
├── intel-ucode.img
└── vmlinuz-linux

5 directories, 9 files

```
(I have not boot partition, /boot is part of the LUKS encrypted root)

**Additional questions**
- Operating system and version:
  - Arch Linux
- Have you tried rebooting?
  - yes
- Is this a regression?
  - #669 
- Are you using an NVMe disk?
  - yes
- Is secure boot enabled?
  - yes

hughsie, 2020-05-18:
Is there any chance you could compile the 1_4_X branch and debug a little and try to work out what's happening? Thanks.

superm1, 2020-05-18:
If I was to guess it's a regression from https://github.com/fwupd/fwupd/commit/7b866e5f92e2249c9de52b4fcacaa79839aa5c97#diff-2f5c39c775e4cac7ebbeabed4cefca6a but the logic seems to read out correctly to me, I don't see a functional change.  I would recommend to double check with a verbose daemon log that the correct `/etc/fwupd/uefi.conf` is being parsed (it should tell you during startup).

smrqdt, 2020-05-27:
I found the reason, why it stopped working: In 334ba7994c869ac393ed35d740e7c4fdd5382f8a the name of the config option changed in a subtle way: 

```
RequireShimForSecureBoot=true 
```

became

```
#DisableShimForSecureBoot=true 
```

I did not notice that change when comparing my `uefi.conf` with the `uefi.conf.pacnew` added by pacman…

After replacing the config by the new one and uncommenting the line (is it intended that the value specified in the comment is not the default value?) the error message is now gone and the update happens as expected.

closed 2020-05-27 by smrqdt