fwupd/fwupd #2112: fwupd daemon fails to start: "Initialize default tcti. ErrorCode (0x000a000a)"

PhilippWendler, 2020-05-19:
**Describe the bug**

When starting the fwupd daemon, it immediately terminates itself.
The log output is
```
May 19 07:52:31 t460p systemd[1]: Started Service for snap application fwupd.fwupd.
May 19 07:52:32 t460p fwupd.fwupd[7120]: 05:52:32:0225 FuPluginMm           failed to probe MM device: modem cannot be put in programming mode
May 19 07:52:32 t460p fwupd.fwupd[7120]: ERROR:esys:src/tss2-esys/esys_context.c:69:Esys_Initialize() Initialize default tcti. ErrorCode (0x000a000a)
May 19 07:52:32 t460p fwupd.fwupd[7120]: 05:52:32:0254 FuEngine             failed to get releases for Intel Management Engine: No releases found: no HWIDs matched 90706264-e399-575b-a9fb-077ead03b9b4
May 19 07:52:35 t460p fwupd.fwupd[7120]: 05:52:35:0068 FuEngine             failed to add USB device 17ef:1010: USB error on device 17ef:1010 : Input/output error [-1]
May 19 07:52:35 t460p fwupd.fwupd[7120]: 05:52:35:0992 FuEngine             failed to get releases for Intel Management Engine: No releases found: no HWIDs matched 90706264-e399-575b-a9fb-077ead03b9b4
May 19 07:52:36 t460p systemd[1]: snap.fwupd.fwupd.service: Consumed 309ms CPU time
```
The same is produced everytime I start fwupd.

Of course, I cannot use `fwupdmgr` because of this. `fwupdtool` seems to work, but it produces the same error messages when I use it, for example:
```
$ sudo fwupd.fwupdtool get-updates
Loading…                 [-                                      ]05:57:57:0978 FuPluginMm           failed to probe MM device: modem cannot be put in programming mode
ERROR:esys:src/tss2-esys/esys_context.c:69:Esys_Initialize() Initialize default tcti. ErrorCode (0x000a000a) 
05:57:58:0007 FuEngine             failed to get releases for Intel Management Engine: No releases found: no HWIDs matched 90706264-e399-575b-a9fb-077ead03b9b4
05:57:58:0941 FuEngine             failed to get releases for Intel Management Engine: No releases found: no HWIDs matched 90706264-e399-575b-a9fb-077ead03b9b4
Loading…                 [***************************************]
• Intel Management Engine has no available firmware updates
• Samsung SSD 850 EVO 1TB has no available firmware updates
• System Firmware has the latest available firmware version
```

<details>
<summary>Here is the full log for `sudo fwupdtool get-updates --verbose`:</summary>

```
06:01:15:0687 FuDebug              Verbose debugging enabled (on console 0)
06:01:15:0695 FuConfig             loading config values from /root/snap/fwupd/2084/etc/fwupd/daemon.conf
06:01:15:0695 FuConfig             failed to read UpdateMotd key: Key file does not have key “UpdateMotd” in group “fwupd”
06:01:15:0695 FuConfig             failed to read EnumerateAllDevices key: Key file does not have key “EnumerateAllDevices” in group “fwupd”
06:01:15:0695 FuRemoteList         loading /snap/fwupd/2084/share/fwupd/metainfo
06:01:15:0695 XbSilo               attempting to load /root/snap/fwupd/2084/var/cache/fwupd/metainfo.xmlb
06:01:15:0695 XbSilo               file: 1373318f-3da0-0c8f-9a9d-627eb12ab649, current:1373318f-3da0-0c8f-9a9d-627eb12ab649, cached: (null)
06:01:15:0695 XbSilo               loading silo with file contents
06:01:15:0695 Fwupd                looking for os-release at /etc/os-release
06:01:15:0695 FuRemoteList         loading remotes from /root/snap/fwupd/2084/etc/fwupd/remotes.d/lvfs-testing.conf
06:01:15:0695 Fwupd                converting Keyring value to Jcat
06:01:15:0696 FuRemoteList         loading remotes from /root/snap/fwupd/2084/etc/fwupd/remotes.d/vendor.conf
06:01:15:0696 FuRemoteList         loading remotes from /root/snap/fwupd/2084/etc/fwupd/remotes.d/vendor-directory.conf
06:01:15:0696 FuRemoteList         loading remotes from /root/snap/fwupd/2084/etc/fwupd/remotes.d/dell-esrt.conf
06:01:15:0696 FuRemoteList         loading remotes from /root/snap/fwupd/2084/etc/fwupd/remotes.d/fwupd-tests.conf
06:01:15:0696 FuRemoteList         loading remotes from /root/snap/fwupd/2084/etc/fwupd/remotes.d/lvfs.conf
06:01:15:0696 Fwupd                converting Keyring value to Jcat
06:01:15:0696 FuRemoteList         ordering lvfs-testing=lvfs+1
06:01:15:0696 FuRemoteList         ignoring unfound remote fwupd
06:01:15:0696 FuRemoteList         ignoring unfound remote fwupd
06:01:15:0696 FuRemoteList         ignoring unfound remote fwupd
06:01:15:0696 FuRemoteList         ignoring unfound remote fwupd
06:01:15:0696 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
06:01:15:0696 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
06:01:15:0696 Jcat                 ignoring GPG-KEY-Hughski-Limited as not PKCS-7 certificate
06:01:15:0696 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
06:01:15:0696 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
06:01:15:0696 Jcat                 reading /root/snap/fwupd/2084/var/lib/fwupd/pki/secret.key with 2459 bytes
06:01:15:0696 Jcat                 reading /root/snap/fwupd/2084/var/lib/fwupd/pki/client.pem with 1383 bytes
06:01:15:0703 FuEngine             client certificate exists and working
06:01:15:0703 FuHistory            trying to open database '/root/snap/fwupd/2084/var/lib/fwupd/pending.db'
06:01:15:0703 FuHistory            got schema version of 5
06:01:15:0703 FuHwids              smbios property Manufacturer=LENOVO
06:01:15:0703 FuHwids              smbios property EnclosureKind=a
06:01:15:0703 FuHwids              smbios property Family=ThinkPad T460p
06:01:15:0703 FuHwids              smbios property ProductName=20FXS1B700
06:01:15:0703 FuHwids              smbios property ProductSku=LENOVO_MT_20FX_BU_Think_FM_ThinkPad T460p
06:01:15:0703 FuHwids              smbios property BiosVendor=LENOVO
[... 975 lines cut ...]
│                          • Device is usable for the duration of the update
│   
└─System Firmware:
      Device ID:           444a229da2b21bc0d24faf814d1c58c5bff3315f
      Current version:     0.2.11
      Minimum Version:     0.2.11
      Vendor:              LENOVO (DMI:LENOVO)
      GUID:                42f29620-3d63-4f09-950b-9b7055570f28
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Supported on remote server
                           • Needs a reboot after installation
                           • Cryptographic hash verification is available
                           • Device is usable for the duration of the update
    
```
</details>

The USB device `17ef:1010` it mentions in one of the error message seems to be an internal USB hub named "Lenovo" according to `lsusb` output.

**Additional questions**
- Operating system and version: Ubuntu 18.04, kernel 5.3.0-51-generic
- Have you tried rebooting? Yes, did not change anything.
- Is this a regression? Yes, until at least one of the last 1.3.x versions it did work.

PhilippWendler, 2020-06-09:
Tried with the 1.4.3 snap, still occurs.

superm1, 2020-06-09:
Does your TPM actually work?  Can you use the TPM2 tools to get PCR information for example?

PhilippWendler, 2020-06-09:
Thanks for responding! I had no idea that this is related to my TPM (which I have never tried to use). In any case, I think a broken TPM should not let fwupd fail ;-)

I now installed `tpm2-tools` (Ubuntu package, version 2.1.0) and ran `sudo tpm2_listpcrs`, which gives me
```
Failed to initialize tcti context: 0x1
```
Is this what you wanted me to try? If not, please give more details, because I have no experience with TPMs.

Further info that might be relevant:
- There is no `/dev/tpm*`.
- `journalctl -k | grep -i tpm` shows this:

      kernel: ima: No TPM chip found, activating TPM-bypass!

I will reboot and try to find TPM-related options in my firmware.

superm1, 2020-06-09:
> Thanks for responding! I had no idea that this is related to my TPM (which I have never tried to use). In any case, I think a broken TPM should not let fwupd fail ;-)

Well there are more and more features in fwupd that are using the TPM now.  I do think that at least the user should be better messaged this scenario.  So we have two things to work on:
* fixing your TPM
* fixing the messaging.

> Is this what you wanted me to try? If not, please give more details, because I have no experience with TPMs.

Yeah that confirms that the TPM stack isn't working on your TPM.  I would recommend to start with a mainline kernel to check if it helps:
https://kernel.ubuntu.com/~kernel-ppa/mainline/v5.7.1/

If it helps then this 1 is a kernel bug.  If it doesn't help, that doesn't confirm it's not a kernel bug, but it might not yet be fixed in the latest kernel.

>kernel: ima: No TPM chip found, activating TPM-bypass!

I don't know that I believe this, if it's a modern machine it _should_ have a TPM.

PhilippWendler, 2020-06-09:
Ok, my TPM was simply disabled in the firmware.

I enabled it (it has two options, one for a Intel PTT for TPM2 and one discrete device for TPM1.2, I chose the former).
Now I do get kernel messages about having a TPM, and I do have `/dev/tpm0`, but `tpm2_listpcrs` still fails with the same error message.

The error message when starting `fwupd` is gone now, but it still fails to start.
Log from snap is this:
```
2020-06-09T14:57:50Z systemd[1]: Started Service for snap application fwupd.fwupd.
2020-06-09T14:57:51Z fwupd.fwupd[6412]: 14:57:51:0135 FuPluginMm           failed to probe MM device: modem cannot be put in programming mode
2020-06-09T14:57:51Z fwupd.fwupd[6412]: 14:57:51:0201 FuEngine             failed to get releases for Intel Management Engine: No releases found: no HWIDs matched 90706264-e399-575b-a9fb-077ead03b9b4
2020-06-09T14:57:52Z fwupd.fwupd[6412]: 14:57:52:0300 FuEngine             failed to get releases for Intel Management Engine: No releases found: no HWIDs matched 90706264-e399-575b-a9fb-077ead03b9b4
2020-06-09T14:57:52Z systemd[1]: snap.fwupd.fwupd.service: Consumed 341ms CPU time
```

<details>
<summary>Here is the output for fwupdtool get-devices --show-all-devices:</summary>

```
14:58:32:0175 FuPluginMm           failed to probe MM device: modem cannot be put in programming mode
14:58:32:0248 FuEngine             failed to get releases for Intel Management Engine: No releases found: no HWIDs matched 90706264-e399-575b-a9fb-077ead03b9b4
14:58:33:0348 FuEngine             failed to get releases for Intel Management Engine: No releases found: no HWIDs matched 90706264-e399-575b-a9fb-077ead03b9b4
20FXS1B700
│
├─Intel AMT [unprovisioned]:
│     Device ID:           e2623122c99d58220498aacbfcfdb1baebbae3c5
│     Summary:             Hardware and firmware technology for remote out-of-band management
│     Current version:     11.0.12.1008
│     Bootloader Version:  11.0.12.1008
│     Vendor:              Intel Corporation
│     GUID:                2800f812-b7b4-2d4b-aca8-46e0ff65814c
│     Device Flags:        • Internal device
│   
├─Intel Management Engine:
│     Device ID:           25127f5d4d5dfd6e903cda1a61a67cb092e4bb0f
│     Current version:     176.12.1008
│     Minimum Version:     176.12.1008
│     Vendor:              DMI:LENOVO
│     Update State:        success
│     GUID:                63451986-0c42-42db-9a18-342a8455ac68
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─Intel(R) Core™ i5-6440HQ CPU @ 2.60GHz:
│     Device ID:           4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:     0xd6
│     Vendor:              GenuineIntel
│     GUID:                b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│     Device Flags:        • Internal device
│   
├─Samsung SSD 850 EVO 1TB:
│     Device ID:           5dbeb140337f610d54913c0f43d15fdaa0eafb51
│     Summary:             ATA Drive
│     Current version:     EMT01B6Q
│     Vendor:              Samsung (ATA:0x144D)
│     Serial Number:       S21DNXAG519797D
│     GUIDs:               e7307cf4-8f2f-5162-b7d2-7942747a765b ← IDE\Samsung_SSD_850_EVO_1TB_________________EMT01B6Q
│                          bab8801e-be63-581d-a10e-14074d3d6b6d ← IDE\0Samsung_SSD_850_EVO_1TB_________________
│                          7faa6c7f-3fb9-5e4f-864e-4bf771d91769 ← Samsung SSD 850 EVO 1TB
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           444a229da2b21bc0d24faf814d1c58c5bff3315f
[... 1047 lines cut ...]
15:00:14:0780 FuEngine             ignoring 0.1.46 as older than 0.2.11
15:00:14:0780 FuEngine             ignoring 0.1.45 as older than 0.2.11
15:00:14:0780 FuEngine             ignoring 0.1.44 as older than 0.2.11
15:00:14:0780 FuEngine             ignoring 0.1.43 as older than 0.2.11
• System Firmware has the latest available firmware version
15:00:14:0780 FuMain               current version is 0.2.11: 0.1.47=older, 0.1.46=older, 0.1.45=older, 0.1.44=older, 0.1.43=older
15:00:14:0784 FuPlugin             performing destroy() on modem_manager
15:00:14:0785 FuPlugin             performing destroy() on uefi
15:00:14:0785 FuPlugin             performing destroy() on redfish
15:00:14:0785 FuPlugin             performing destroy() on dell
15:00:14:0786 FuPlugin             performing destroy() on thunderbolt_power
15:00:14:0786 FuPlugin             performing destroy() on logind
15:00:14:0787 FuPlugin             performing destroy() on upower
15:00:14:0787 FuPlugin             performing destroy() on thunderbolt
15:00:14:0787 FuPlugin             performing destroy() on synaptics_mst
15:00:14:0788 FuPlugin             performing destroy() on tpm_eventlog
```

</details>

Maybe the actual reason for fwupd failing to start has always been unrelated to the TPM error message?

Do you need more debug output? Not sure how I can tell the snap to be more verbose, but I hope the above helps.

Btw: I forgot to mention in my original description that I had fwupd running with several of the 1.3.x versions. But I tried running 1.3.7 and 1.3.8 from snap now and these also fail.

PhilippWendler, 2020-06-09:
After switching the TPM option in the firmware to use the discrete TPM1.2, I get different messages when starting fwupd (which still fails):
```
2020-06-09T15:09:43Z systemd[1]: Started Service for snap application fwupd.fwupd.
2020-06-09T15:09:43Z fwupd.fwupd[4319]: 15:09:43:0461 FuPluginMm           failed to probe MM device: modem cannot be put in programming mode
2020-06-09T15:09:43Z fwupd.fwupd[4319]: 15:09:43:0561 FuEngine             failed to get releases for Intel Management Engine: No releases found: no HWIDs matched 90706264-e399-575b-a9fb-077ead03b9b4
2020-06-09T15:09:44Z fwupd.fwupd[4319]: ERROR:sys:src/tss2-sys/api/Tss2_Sys_Execute.c:114:Tss2_Sys_ExecuteFinish() Unsupported device. The device is a TPM 1.2
2020-06-09T15:09:44Z fwupd.fwupd[4319]: ERROR:esys:src/tss2-esys/api/Esys_Startup.c:216:Esys_Startup_Finish() Received a non-TPM Error
2020-06-09T15:09:44Z fwupd.fwupd[4319]: ERROR:esys:src/tss2-esys/api/Esys_Startup.c:78:Esys_Startup() Esys Finish ErrorCode (0x00080001)
2020-06-09T15:09:44Z fwupd.fwupd[4319]: 15:09:44:0494 FuEngine             failed to add udev device /sys/devices/pnp0/00:07/tpm/tpm0: failed to initialize TPM
2020-06-09T15:09:44Z fwupd.fwupd[4319]: 15:09:44:0498 FuEngine             failed to get releases for Intel Management Engine: No releases found: no HWIDs matched 90706264-e399-575b-a9fb-077ead03b9b4
2020-06-09T15:09:44Z systemd[1]: snap.fwupd.fwupd.service: Consumed 272ms CPU time
```

superm1, 2020-06-09:
Well glad that it was just a setting in firmware and that's not the reason for your fwupd failure related to the daemon.  Since it doesn't happen with `fwupdtool` too, what we need is a verbose flag added to the daemon unit launch (`--verbose`) or in `/root/snap/current/etc/fwupd/daemon.conf` by setting `VerboseDomains` to `*`.

If you choose to do the former, then reload your units using `systemctl`.  If you do the latter, restarting the daemon and sharing the journal output is enough.

PhilippWendler, 2020-06-09:
No problem.

I now
- enabled `VerboseDomains` in `/root/snap/fwupd/current/etc/fwupd/daemon.conf`
- switched back to the Intel PTT TPM2 (which was the default in my firmware)
- booted Linux 5.7.1-050701-generic from the Ubuntu mainline kernel PPA

<details>
<summary>The verbose log for fwupd is this:</summary>

```
2020-06-09T15:26:03Z systemd[1]: Started Service for snap application fwupd.fwupd.
2020-06-09T15:26:06Z fwupd.fwupd[1396]: 15:26:06:0572 FuConfig             failed to read UpdateMotd key: Die Schlüsselwertedatei hat keinen Schlüssel »UpdateMotd« in der Gruppe »fwupd«
2020-06-09T15:26:06Z fwupd.fwupd[1396]: 15:26:06:0572 FuConfig             failed to read EnumerateAllDevices key: Die Schlüsselwertedatei hat keinen Schlüssel »EnumerateAllDevices« in der Gruppe »fwupd«
2020-06-09T15:26:06Z fwupd.fwupd[1396]: 15:26:06:0572 FuRemoteList         loading /snap/fwupd/2155/share/fwupd/metainfo
2020-06-09T15:26:06Z fwupd.fwupd[1396]: 15:26:06:0577 XbSilo               attempting to load /root/snap/fwupd/2155/var/cache/fwupd/metainfo.xmlb
2020-06-09T15:26:06Z fwupd.fwupd[1396]: 15:26:06:0578 XbSilo               file: c2d44109-bae8-e753-e3fe-24fcab8c5565, current:c2d44109-bae8-e753-e3fe-24fcab8c5565, cached: (null)
2020-06-09T15:26:06Z fwupd.fwupd[1396]: 15:26:06:0578 XbSilo               loading silo with file contents
2020-06-09T15:26:06Z fwupd.fwupd[1396]: 15:26:06:0578 Fwupd                looking for os-release at /etc/os-release
2020-06-09T15:26:06Z fwupd.fwupd[1396]: 15:26:06:0578 FuRemoteList         loading remotes from /root/snap/fwupd/2155/etc/fwupd/remotes.d/lvfs-testing.conf
2020-06-09T15:26:06Z fwupd.fwupd[1396]: 15:26:06:0579 Fwupd                converting Keyring value to Jcat
2020-06-09T15:26:06Z fwupd.fwupd[1396]: 15:26:06:0580 FuRemoteList         loading remotes from /root/snap/fwupd/2155/etc/fwupd/remotes.d/vendor.conf
2020-06-09T15:26:06Z fwupd.fwupd[1396]: 15:26:06:0580 FuRemoteList         loading remotes from /root/snap/fwupd/2155/etc/fwupd/remotes.d/vendor-directory.conf
2020-06-09T15:26:06Z fwupd.fwupd[1396]: 15:26:06:0580 FuRemoteList         loading remotes from /root/snap/fwupd/2155/etc/fwupd/remotes.d/dell-esrt.conf
2020-06-09T15:26:06Z fwupd.fwupd[1396]: 15:26:06:0581 FuRemoteList         loading remotes from /root/snap/fwupd/2155/etc/fwupd/remotes.d/fwupd-tests.conf
2020-06-09T15:26:06Z fwupd.fwupd[1396]: 15:26:06:0581 FuRemoteList         loading remotes from /root/snap/fwupd/2155/etc/fwupd/remotes.d/lvfs.conf
2020-06-09T15:26:06Z fwupd.fwupd[1396]: 15:26:06:0581 Fwupd                converting Keyring value to Jcat
2020-06-09T15:26:06Z fwupd.fwupd[1396]: 15:26:06:0582 FuRemoteList         ordering lvfs-testing=lvfs+1
2020-06-09T15:26:06Z fwupd.fwupd[1396]: 15:26:06:0582 FuRemoteList         ignoring unfound remote fwupd
2020-06-09T15:26:06Z fwupd.fwupd[1396]: 15:26:06:0588 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
2020-06-09T15:26:06Z fwupd.fwupd[1396]: 15:26:06:0588 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
2020-06-09T15:26:06Z fwupd.fwupd[1396]: 15:26:06:0588 Jcat                 ignoring GPG-KEY-Hughski-Limited as not PKCS-7 certificate
2020-06-09T15:26:06Z fwupd.fwupd[1396]: 15:26:06:0588 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
2020-06-09T15:26:06Z fwupd.fwupd[1396]: 15:26:06:0588 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
2020-06-09T15:26:06Z fwupd.fwupd[1396]: 15:26:06:0589 Jcat                 reading /root/snap/fwupd/2155/var/lib/fwupd/pki/secret.key with 2459 bytes
2020-06-09T15:26:06Z fwupd.fwupd[1396]: 15:26:06:0590 Jcat                 reading /root/snap/fwupd/2155/var/lib/fwupd/pki/client.pem with 1383 bytes
2020-06-09T15:26:06Z fwupd.fwupd[1396]: 15:26:06:0615 FuEngine             client certificate exists and working
2020-06-09T15:26:06Z fwupd.fwupd[1396]: 15:26:06:0615 FuHistory            trying to open database '/root/snap/fwupd/2155/var/lib/fwupd/pending.db'
2020-06-09T15:26:06Z fwupd.fwupd[1396]: 15:26:06:0649 FuHistory            got schema version of 5
2020-06-09T15:26:06Z fwupd.fwupd[1396]: 15:26:06:0649 FuIdle               setting timeout to 7200s
2020-06-09T15:26:06Z fwupd.fwupd[1396]: 15:26:06:0649 FuHwids              smbios property Manufacturer=LENOVO
2020-06-09T15:26:06Z fwupd.fwupd[1396]: 15:26:06:0649 FuHwids              smbios property EnclosureKind=a
2020-06-09T15:26:06Z fwupd.fwupd[1396]: 15:26:06:0649 FuHwids              smbios property Family=ThinkPad T460p
2020-06-09T15:26:06Z fwupd.fwupd[1396]: 15:26:06:0649 FuHwids              smbios property ProductName=20FXS1B700
2020-06-09T15:26:06Z fwupd.fwupd[1396]: 15:26:06:0649 FuHwids              smbios property ProductSku=LENOVO_MT_20FX_BU_Think_FM_ThinkPad T460p
2020-06-09T15:26:06Z fwupd.fwupd[1396]: 15:26:06:0649 FuHwids              smbios property BiosVendor=LENOVO
2020-06-09T15:26:06Z fwupd.fwupd[1396]: 15:26:06:0649 FuHwids              smbios property BiosVersion=R07ET71W (2.11 )
2020-06-09T15:26:06Z fwupd.fwupd[1396]: 15:26:06:0649 FuHwids              smbios property BiosMajorRelease=02
2020-06-09T15:26:06Z fwupd.fwupd[1396]: 15:26:06:0649 FuHwids              smbios property BiosMinorRelease=0b
2020-06-09T15:26:06Z fwupd.fwupd[1396]: 15:26:06:0649 FuHwids              smbios property BaseboardManufacturer=LENOVO
[... cut ...]
[Firmware Bug]: TPM Final Events table missing or invalid
ACPI: SSDT 0x0000000097FFA000 0004B7 (v02 LENOVO Tpm2Tabl 00001000 INTL 20141107)
ACPI: TPM2 0x0000000097FF8000 000034 (v03 LENOVO TP-R07   00002110 PTEC 00000002)
```
Even with this 5.7.1 kernel the tool `tpm2_listpcrs` still gives me only
```
Failed to initialize tcti context: 0x1
```
The UEFI firmware of this system is quite old because of https://github.com/fwupd/firmware-lenovo-thinkpad/issues/19#issuecomment-630034660 unfortunately....

superm1, 2020-06-09:
Here's your problem:
```
2020-06-09T15:26:10Z fwupd.fwupd[1396]: 15:26:10:0354 FuMain               FuMain: lost name: org.freedesktop.fwupd
```

I suspect you still have a distro package installed too that has stomped on the snap's claim to the dbus service.  Please try:

```
sudo apt purge fwupd
sudo snap remove fwupd
sudo snap install fwupd --classic
```

And that should re-register it.  I'm aware it's a big hammer and there is a lighter way you can do it, but this should help prevent it happening in the future.

PhilippWendler, 2020-06-09:
Reinstalling the snap did it!

I did have the distro package installed months ago, but I had definitely already purged it before (the output from running `apt purge fwupd` right now also confirmed that it did not do anything). And I definitely did have fwupd working from the snap at some time between removing the distro package and now.

Big thanks for your help!

I guess that the log message about loosing the dbus name and terminating should be shown on level warning or higher?

Do you want to keep this issue open as reminder about this? (any maybe about the TPM error messages, although they are gone now with TPM2 enabled)

superm1, 2020-06-09:
> I guess that the log message about loosing the dbus name and terminating should be shown on level warning or higher?

I agree.  I filed a PR for it.

>Do you want to keep this issue open as reminder about this? (any maybe about the TPM error messages, although they are gone now with TPM2 enabled)

They weren't fatal, so fwupd was handling properly.  This issue will close when the PR is reviewed and accepted.

closed 2020-06-09 by superm1

LinuxOnTheDesktop, 2020-10-31:
It might be worth my saying the following.

I disabled TPM on my machine, in its BIOS, and subsequently and for the first time received the cryptic error message of which the original reporter originally complained. To wit: `ERROR:esys:src/tss2-esys/esys_context.c:69:Esys_Initialize() Initialize default tcti. ErrorCode (0x000a000a)`.

I do not see the other error message - the 'lost name' one - in my logs.

```
$ fwupdmgr --version
client version:	1.3.11
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.11

```

PR #2165 (2020-06-09): fu-main: Make it clearer what lost name is and bump it to warning
This happens often enough when people switch from snap to distro
package that we should mention it in non-verbose logs.

Fixes: #2112

Type of pull request:
- [ ] New plugin (Please include [new plugin checklist](https://github.com/hughsie/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit 471a0e21 (2020-06-09): fu-main: Make it clearer what lost name is and bump it to warning

This happens often enough when people switch from snap to distro
package that we should mention it in non-verbose logs.

Fixes: #2112