fwupd/fwupd #2114: Pop!_OS 20.04 on Thinkpad T480 - Unable to apply Intel Platform Update

vancanhuit, 2020-05-19:
**Describe the bug**
I updated the UEFI Device Firmware (via command line and firmware update manager UI on POP!_OS) after received a firmware notification update from Pop!_OS and reboot. After that, I still receive the firmware update notification again.

**Steps to Reproduce**
Steps to reproduce the behavior:
- Run `sudo fwupdmgr refresh --force && sudo fwupdmgr update` or click update button in the firmware update manager UI on Pop!_OS.
- Reboot
- The firmware update status is success but not installed.

**Expected behavior**
The firmware should be installed successfully after reboot.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version: 1.3.9
compile-time dependency versions
        gusb:   0.3.4
        efivar: 37
daemon version: 1.3.9
```

Installed via `apt`.

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
20L5CTO1WW
│
├─Thunderbolt Controller:
│     Device ID:           e108dec5a4f65ac3681b831b095cdb2b7e8afe4d
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     20.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               fe328c2d-5300-597b-a21d-991116db8ab6
│                          15a5ae1d-0ec9-5013-9221-cd6dd753fcb0
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│
├─Intel AMT [unprovisioned]:
│     Device ID:           e2623122c99d58220498aacbfcfdb1baebbae3c5
│     Summary:             Hardware and firmware technology for remote out-of-band management
│     Current version:     11.8.77.3664
│     Bootloader Version:  11.8.77.3664
│     Vendor:              Intel Corporation
│     GUID:                2800f812-b7b4-2d4b-aca8-46e0ff65814c
│     Device Flags:        • Internal device
│
├─SanDisk SD9TB8W256G1001:
│     Device ID:           64ec4cd1e1c9565e79b00f4e7221b1c689b33e96
│     Summary:             ATA Drive
│     Current version:     X6107101
│     GUIDs:               4305dc5e-be6d-5913-b43f-880b1a883d70
│                          534f5ba7-6d8d-5083-ad76-58fd8a4f4e78
│                          57738cf2-7534-5205-97ec-7e5dbb99372e
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│
├─System Firmware:
│     Device ID:           48477ba14bd9bd0c97323a07caa54b7766f917c2
│     Current version:     0.1.31
[... 204 lines cut ...]
- Are you using an NVMe disk? No
- Is secure boot enabled? No

**Additional information**
```
$ sudo fwupdate -L
failed: Error opening file /sys/firmware/efi/efivars/FWUPDATE_DEBUG_LOG-0abba7dc-e516-4167-bbf5-4d9d1c739416: No such file or directory
```

```
$ sudo fwupdate -i
failed: Error opening file /sys/firmware/efi/efivars/fwupd-dc7f0308-1ef8-4774-9ba5-89a58c4d731c-0-0abba7dc-e516-4167-bbf5-4d9d1c739416: No such file or directory
Information for the update status entry 1:
  Information Version: 7
  Firmware GUID: {dd31d983-cf3b-4c84-acc9-70caa94e827b}
  Capsule Flags: 0x00032784x
  Hardware Instance: 0
  Update Status: attempted
  Capsule File Path: /EFI/pop/fw/fwupd-dd31d983-cf3b-4c84-acc9-70caa94e827b.cap

failed: Error opening file /sys/firmware/efi/efivars/fwupd-b87a926d-189c-49a1-b1ff-921099b1de89-0-0abba7dc-e516-4167-bbf5-4d9d1c739416: No such file or directory
failed: Error opening file /sys/firmware/efi/efivars/fwupd-89aab014-a71d-4cae-8824-174808266539-0-0abba7dc-e516-4167-bbf5-4d9d1c739416: No such file or directory
failed: Error opening file /sys/firmware/efi/efivars/fwupd-d4583ddb-bf0a-4801-aefd-951930545cb9-0-0abba7dc-e516-4167-bbf5-4d9d1c739416: No such file or directory
failed: Error opening file /sys/firmware/efi/efivars/fwupd-e0049080-09bd-473a-a02c-19aa0c8dad78-0-0abba7dc-e516-4167-bbf5-4d9d1c739416: No such file or directory
```

hughsie, 2020-05-19:
Doe sit happen with Arch, Ubuntu or Fedora? We don't test PopOS! at all.

klemmster, 2020-05-19:
Yes. I'm seeing the same on Archlinux.

heldchen, 2020-05-19:
this also happens in Ubuntu 20.04 which has the same fwupdmgr version. the problematic firmware package seems to be for the `Lenovo ThinkPad X1 Carbon 6th Corporate ME Firmware`, the other packages can be installed successfully when scheduled separatedly.

```
root@d-wing:~# fwupdmgr get-updates
• ThinkPad Thunderbolt 3 Dock has no available firmware updates
• Thunderbolt Controller has the latest available firmware version
• USB2.0 Hub has no available firmware updates
• USB3.1 Hub has no available firmware updates
• MSP430 has no available firmware updates
• MSP430 has no available firmware updates
• SAMSUNG MZVLB1T0HALR-000L7 has the latest available firmware version
• System Firmware has the latest available firmware version
• ThinkPad Thunderbolt 3 Dock USB Audio has no available firmware updates
• UEFI Device Firmware has the latest available firmware version
• UEFI Device Firmware has no available firmware updates
• USB2.0 Hub has no available firmware updates
• USB3.1 Hub has no available firmware updates
• Unifying Receiver has the latest available firmware version
• VMM5322 has no available firmware updates
20KGSBAH00
│
└─UEFI Device Firmware:
  │   Device ID:           e563ad307df81c99f0de8c26292afd71cf409673
  │   Current version:     184.70.3626
  │   Minimum Version:     0.0.1
  │   Vendor:              DMI:LENOVO
  │   Update State:        success
  │   GUID:                42a0a96e-c9f3-438f-9687-7826be33e4ce
  │   Device Flags:        • Internal device
  │                        • Updatable
  │                        • Requires AC power
  │                        • Supported on remote server
  │                        • Needs a reboot after installation
  │                        • Device is usable for the duration of the update
  │ 
  └─ThinkPad X1 Carbon 6th Corporate ME Update:
        New version:       184.77.3664
        Remote ID:         lvfs
        Summary:           Lenovo ThinkPad X1 Carbon 6th Corporate ME Firmware
        License:           Proprietary
        Size:              7.5 MB
        Vendor:            Lenovo Ltd.
        Flags:             is-upgrade
        Description:       Version 11.8.77.3664 (LVFS: 184.77.3664)
        
        2020.01 Q1 Intel Platform Update (IPU), formerly known as Quarterly security release (QSR)
        
        Several security fixes and enhancements are on this release.
        
        Security issues fixed:
        
         • CVE-2020-0531
         • CVE-2020-0532
         • CVE-2020-0533
         • CVE-2020-0535
         • CVE-2020-0536
         • CVE-2020-0537
         • CVE-2020-0538
         • CVE-2020-0539
         • CVE-2020-0540
         • CVE-2020-0545
         • CVE-2020-0594
         • CVE-2020-0595
         • CVE-2020-0596
         • CVE-2020-8674
```

```
root@d-wing:~# fwupdate -L
failed: Error opening file /sys/firmware/efi/efivars/FWUPDATE_DEBUG_LOG-0abba7dc-e516-4167-bbf5-4d9d1c739416: No such file or directory
```

```      
root@d-wing:~# fwupdate -i
failed: Error opening file /sys/firmware/efi/efivars/fwupd-a4b51dca-8f97-4310-8821-3330f83c9135-0-0abba7dc-e516-4167-bbf5-4d9d1c739416: No such file or directory
Information for the update status entry 1:
  Information Version: 7
  Firmware GUID: {42a0a96e-c9f3-438f-9687-7826be33e4ce}
  Capsule Flags: 0x00032784x
  Hardware Instance: 0
  Update Status: attempted
  Capsule File Path: /EFI/ubuntu/fw/fwupd-42a0a96e-c9f3-438f-9687-7826be33e4ce.cap

failed: Error opening file /sys/firmware/efi/efivars/fwupd-3babca5f-b2bf-4f4b-a72e-2bdc84eb4019-0-0abba7dc-e516-4167-bbf5-4d9d1c739416: No such file or directory
failed: Error opening file /sys/firmware/efi/efivars/fwupd-cea87551-1701-43fb-afbc-6e8ce9728345-0-0abba7dc-e516-4167-bbf5-4d9d1c739416: No such file or directory
```

BriansCode, 2020-05-20:
Same exact problem on Fedora 32.

sudomateo, 2020-05-20:
I should have posted on this issue instead of #2118. Sorry about that.

I'm also unable to apply the ThinkPad T480 Corporate ME Update `184.77.3664` to my ThinkPad T480. I've tried via GNOME Software and `fwupdmgr` multiple times with no success. When rebooting I see the "Adding capsule" message but the update does not go through. I'm running Fedora 32 as well with the following `fwupdmgr` version.

```
$ fwupdmgr --version
client version:	1.4.1
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.4.1
```

VorpalBlade, 2020-05-22:
Same issue with Ubuntu 18.04 on a Thinkpad T480 for the "ThinkPad T480 Consumer ME Update Update".

When rebooting the computer when trying to apply the upgrade, something (grub? shim? I don't know) says it is applying two upgrade capsules: one called something like fwupd-ux and the other with a name indicating it is the firmware upgrade. Then the computer reboots, and shows a splash indicating that upgrades are being installed. It then very quickly (just a couple of seconds) reboots again and boots back into grub.

```
# fwupdmgr --version
client version:	1.2.10
compile-time dependency versions
	gusb:	0.2.11
	efivar:	34
daemon version:	1.2.10
```

tseward, 2020-05-25:
I can also reproduce on my Lenovo ThinkPad X1 Carbon 6th on Fedora 32. Likewise, the issue appears to be due to `Lenovo ThinkPad X1 Carbon 6th Consumer ME Firmware`.

```
PS /home/trevorseward> fwupdmgr get-updates
• Thunderbolt Controller has the latest available firmware version
• Embedded Controller has the latest available firmware version
• SAMSUNG MZVLB512HAJQ-000L7 has the latest available firmware version
• System Firmware has the latest available firmware version
• UEFI Device Firmware has no available firmware updates
20KG0022US
│
└─Intel Management Engine:
  │   Device ID:           29749712e93b58730ecd1b8537a5503ff5b7fb27
  │   Current version:     184.70.3626
  │   Minimum Version:     0.0.1
  │   Vendor:              DMI:LENOVO
  │   GUID:                9c9d9769-32fa-4841-b550-ea998e754e99
  │   Device Flags:        • Internal device
  │                        • Updatable
  │                        • Requires AC power
  │                        • Supported on remote server
  │                        • Needs a reboot after installation
  │                        • Device is usable for the duration of the update
  │ 
  └─ThinkPad X1 Carbon 6th Consumer ME Update:
        New version:       184.77.3664
        Remote ID:         lvfs
        Summary:           Lenovo ThinkPad X1 Carbon 6th Consumer ME Firmware
        License:           Proprietary
        Size:              2.2 MB
        Created:           2016-07-08
        Urgency:           High
        Vendor:            Lenovo Ltd.
        Flags:             is-upgrade
        Description:       Version 11.8.77.3664 (LVFS: 184.77.3664)
        
        2020.01 Q1 Intel Platform Update (IPU), formerly known as Quarterly security release (QSR)
        
        Several security fixes and enhancements are on this release.
        
        Security issues fixed:
        
         • CVE-2020-0531
         • CVE-2020-0532
         • CVE-2020-0533
         • CVE-2020-0535
         • CVE-2020-0536
         • CVE-2020-0537
         • CVE-2020-0538
         • CVE-2020-0539
         • CVE-2020-0540
         • CVE-2020-0545
         • CVE-2020-0594
         • CVE-2020-0595
         • CVE-2020-0596
         • CVE-2020-8674
      
________________________________________________

Devices that have been updated successfully:

 • UEFI Device Firmware (0.1.17 → 0.1.20)
 • System Firmware (0.1.46 → 0.1.48)
```

ankostis, 2020-05-27:
> Doe sit happen with Arch, Ubuntu or Fedora? We don't test PopOS! at all.

Yes, with Debian-SID on T480 (Intel+NVidia GPUs) 
i have the same problem for the exact same version.

```bash
$ fwupdmgr --version
client version:	1.3.9
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.9

$ fwupdmgr get-updates
• Thunderbolt Controller has no available firmware updates
• SAMSUNG MZVLB1T0HALR-000L7 has the latest available firmware version
• System Firmware has the latest available firmware version
• UEFI Device Firmware has the latest available firmware version
• UEFI Device Firmware has no available firmware updates
• UEFI Device Firmware has no available firmware updates
20L5CTO1WW
│
└─UEFI Device Firmware:
  │   Device ID:           ea0fd60df22477668e549841f6a22a8c31cae587
  │   Current version:     184.70.3626
  │   Minimum Version:     184.70.3626
  │   Vendor:              DMI:LENOVO
  │   Update State:        success
  │   GUID:                9e21f98b-fe98-455c-b388-da5450ab6979
  │   Device Flags:        • Internal device
  │                        • Updatable
  │                        • Requires AC power
  │                        • Supported on remote server
  │                        • Needs a reboot after installation
  │                        • Device is usable for the duration of the update
  │ 
  └─ThinkPad T480 Consumer ME Update:
        New version:       184.77.3664
        Remote ID:         lvfs
        Summary:           Lenovo ThinkPad T480 Consumer ME Firmware
        License:           Proprietary
        Size:              2.2 MB
        Vendor:            Lenovo Ltd.
        Flags:             is-upgrade
        Description:       Version 11.8.77.3664 (LVFS: 184.77.3664)
        
        Problem Fixes
        
         • Mitigated security vulnerabilities
        
        Security issues fixed:
        
         • CVE-2020-0531
         • CVE-2020-0532
         • CVE-2020-0533
         • CVE-2020-0535
         • CVE-2020-0536
         • CVE-2020-0537
         • CVE-2020-0538
         • CVE-2020-0539
         • CVE-2020-0540
         • CVE-2020-0545
         • CVE-2020-0594
         • CVE-2020-0595
         • CVE-2020-0596
         • CVE-2020-8674
      
________________________________________________

Devices that have been updated successfully:

 • UEFI Device Firmware (0.1.19 → 0.1.20)
 • System Firmware (0.1.30 → 0.1.31)

Uploading firmware reports helps hardware vendors to quickly identify failing and successful updates on real devices.
Upload report now? (Requires internet connection):
0.	Do not upload reports at this time, but prompt again for future updates
1.	Do not upload reports, and never ask to upload reports for future updates
2.	Upload reports just this one time, but prompt again for future updates
3.	Upload reports this time and automatically upload reports after completing future updates
3
Successfully uploaded 1 report
```

VorpalBlade, 2020-05-29:
So, I rebooted to windows to apply the update from there (as Ubuntu was nagging me constantly about the available update). I noticed something interesting with this update, it did not seem to use an UEFI update capsule:

Previous times when I updated firmware on this Thinkpad T480 (either from Linux or Windows), it rebooted, applied the update from UEFI, then rebooted again. This time however, it seemed to apply the update directly from inside Windows, then just performed a normal reboot. Maybe this has something to do with the update not working from fwupd?

mloiterman, 2020-05-29:
Just to add another data point, I'm seeing the same thing on T480 and Pop!_OS 20.04:

`$ fwupdmgr --version
client version:	1.3.9
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.9

$ fwupdmgr get-devices --show-all-devices
20L5S35200
│
├─Thunderbolt Controller:
│     Device ID:           9d28ca32dff437cf1aea2e3b9170dba6109c0c86
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     20.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               fe328c2d-5300-597b-a21d-991116db8ab6
│                          15a5ae1d-0ec9-5013-9221-cd6dd753fcb0
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│
├─HD Graphics 620:
│     Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
│     Current version:     02
│     Vendor:              Intel Corporation (PCI:0x8086)
│     GUIDs:               2886e312-afd8-5510-8993-12d568a85e00
│                          38f6c009-a25c-5b50-b3b5-fef4f9c6b846
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│
├─Intel AMT [unprovisioned]:
│     Device ID:           e2623122c99d58220498aacbfcfdb1baebbae3c5
│     Summary:             Hardware and firmware technology for remote out-of-band management
│     Current version:     11.8.70.3626
│     Bootloader Version:  11.8.70.3626
│     Vendor:              Intel Corporation
│     GUID:                2800f812-b7b4-2d4b-aca8-46e0ff65814c
│     Device Flags:        • Internal device
│
├─LENSE30256GMSP34MEAT3TA:
│     Device ID:           f2759da7fe8e0388c5f3601cb072f837b1070b03
│     Summary:             NVM Express Solid State Drive
│     Current version:     2.5.0412
│     Vendor:              Lenovo (NVME:0x17AA)
│     GUIDs:               31008855-c4a6-5feb-8d6f-45554dd52a6f
│                          2e7e091e-c779-5751-b5f9-1ad626f3f123
│                          d6a2653a-5464-5b59-bfcb-a05effb8cfe1
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│
├─System Firmware:
│     Device ID:           48477ba14bd9bd0c97323a07caa54b7766f917c2
│     Current version:     0.1.31
│     Minimum Version:     0.0.1
│     Vendor:              LENOVO (DMI:LENOVO)
│     GUID:                dc7f0308-1ef8-4774-9ba5-89a58c4d731c
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
[... 88 lines cut ...]
Boot001C* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001D* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001E  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot001F  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot0020* USBR BOOT CDROM	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,1)
Boot0021* USBR BOOT Floppy	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,0)
Boot0022* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0023* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)

$ sudo fwupdate -L
failed: Error opening file /sys/firmware/efi/efivars/FWUPDATE_DEBUG_LOG-0abba7dc-e516-4167-bbf5-4d9d1c739416: No such file or directory
┌─[✗]─[mike@mike-thinkpad]─[~]
└──╼$ sudo fwupdate -i
failed: Error opening file /sys/firmware/efi/efivars/fwupd-dc7f0308-1ef8-4774-9ba5-89a58c4d731c-0-0abba7dc-e516-4167-bbf5-4d9d1c739416: No such file or directory
Information for the update status entry 1:
  Information Version: 7
  Firmware GUID: {dd31d983-cf3b-4c84-acc9-70caa94e827b}
  Capsule Flags: 0x00032784x
  Hardware Instance: 0
  Update Status: attempted
  Capsule File Path: /EFI/pop/fw/fwupd-dd31d983-cf3b-4c84-acc9-70caa94e827b.cap

failed: Error opening file /sys/firmware/efi/efivars/fwupd-b87a926d-189c-49a1-b1ff-921099b1de89-0-0abba7dc-e516-4167-bbf5-4d9d1c739416: No such file or directory
failed: Error opening file /sys/firmware/efi/efivars/fwupd-2296f4c0-13a0-48f0-b5cb-83e7a1b868a2-0-0abba7dc-e516-4167-bbf5-4d9d1c739416: No such file or directory
failed: Error opening file /sys/firmware/efi/efivars/fwupd-1a30f69d-fd8f-45bc-ac9d-26cbe1c4b875-0-0abba7dc-e516-4167-bbf5-4d9d1c739416: No such file or directory`