fwupd/fwupd #2117: Add support for Google Chrome OS EC's version format

bleungatchromium, 2020-05-19:
**Is your feature request related to a problem? Please describe.**
The Chrome OS EC project's version numbers are of the following format:

servo_micro_v2.4.17-df61092c3
or
kohaku_v2.0.2767-1a1b1e970

A 4 part version number, 5 if you count the board name that's prepended at the front, with the last part being a short hash corresponding to the sha1 of the repository at the point the firmware was built.

This doesn't really fit into any of fwupd's existing version formats.

**Describe the solution you'd like**
We should add a new format to FwupdVersionFormat that can parse this, called FWUPD_VERSION_FORMAT_CROS_EC.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
A triplet could capture the first three, but have to trim off the hash. 

**Additional context**
Add any other context or screenshots about the feature request here.
Version number format is described in the EC's readme here:
https://chromium.googlesource.com/chromiumos/platform/ec/+/master/README.md#EC-Version-Strings

The script that generates the version numbers is here:
https://chromium.googlesource.com/chromiumos/platform/ec/+/master/util/getversion.sh

superm1, 2020-05-19:
Is the hash there because the firmware version doesn't necessarily get bumped up on individual commits to the repository?

bleungatchromium, 2020-05-19:
The first number is a major number.
The second is a "branch" or tag number.
The third is the # of commits since the tag was cut.

So, the firmware version does get bumped on every individual commit to the repository. I believe the hash is there for convenience.

superm1, 2020-05-19:
If the hash is mostly for convenience would it be better place in metadata but not actually used in version comparisons?

Then I think you can get by using triplet and then encoding the board codename as a guid and the hash as a metadata (that you would see in the plugin "to string" function.

bleungatchromium, 2020-05-19:
Hi Mario,

Ok, I can explore that. Thanks for the guidance!

bleungatchromium, 2020-05-20:
Hey Mario,

It looks good to me. Thanks for the advice!

Servo Micro
  DeviceId:             84d0e3f2a0f8b2328f7995767b23ebb40494723f
  Guid:                 8e2f7625-a164-55d7-8f09-f193c8ec33f1 <- USB\VID_18D1&PID_501A&REV_0100
  Guid:                 13564257-c649-586d-b4e4-4f048d480f36 <- USB\VID_18D1&PID_501A
  Guid:                 98f707d0-8a65-52e9-b216-dc60f91ae2cd <- servo_micro
  Serial:               CMO653-00166-040491U00771
  Summary:              Servo Micro (aka "uServo") Debug Board
  Plugin:               cros_ec_usb
  Flags:                registered
  Vendor:               Google Inc.
  VendorId:             USB:0x18D1
  Version:              2.4.17
  VersionFormat:        triplet
  Created:              2020-05-20
  PhysicalId:            usb:03:00:02
  GitHash:               df61092c3

closed 2020-05-20 by bleungatchromium

hughsie, 2020-05-20:
What Mario said, thanks! When you get something that works, it might be worth sending me some hardware for the regression hardware tests. These are the tests we do before doing tarball releases: https://github.com/fwupd/fwupd/tree/master/data/device-tests/devices

To make them work we also need two firmwares on the LVFS (old firmware, new firmware) so we're probably some way off this goal.