fwupd/fwupd #2118: My lenovo X1 reboots and updates are not applied.

lhirlimann, 2020-05-20:
**Describe the bug**
I have firmware updates offered when I run fwupmgr :
```
[ludovic@saraan]~% sudo fwupdmgr update 
• Thunderbolt Controller has the latest available firmware version
• Embedded Controller has the latest available firmware version
Upgrade available for Intel Management Engine from 184.70.3626 to 184.77.3664
20KHCTO1WW must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: y
Downloading 184.77.3664 for Intel Management Engine...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating Intel Management Engine…                                ]
Scheduling…              [***************************************]
Successfully installed firmware
• SAMSUNG MZVLB512HAJQ-000L7 has the latest available firmware version
Upgrade available for System Firmware from 0.1.46 to 0.1.48
20KHCTO1WW must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: y
Downloading 0.1.48 for System Firmware...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating System Firmware…[            /                          ]
Scheduling…              [***************************************]
Successfully installed firmware

An update requires a reboot to complete. Restart now? [y|N]: n
```
When I reboot, I see messages about new capsule being found - it takes a while and then boot continues. firmware update don't seem to have been applied.

**Steps to Reproduce**
sudo fwupmgr update
reboot

**Expected behavior**

After the reboot running the update command returns that I'm up to date.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
[ludovic@saraan]~% fwupdmgr --version
client version:	1.4.2
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.4.2
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

dnf with enable-testing

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
[ludovic@saraan]~% fwupdmgr get-devices --show-all-devices
20KHCTO1WW
│
├─Thunderbolt Controller:
│     Device ID:           199c280418de0aa1347fc161a9df3c68dd5b0848
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     43.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               2da42a33-cd30-5ef5-a8fb-2c800a4b760f
│                          96411853-9f56-53d2-b05c-a54fcad5ee62
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
[... 232 lines cut ...]
├── vmlinuz-0-rescue-881a595ec75941a3b85457f68d10cd38
├── vmlinuz-5.6.11-300.fc32.x86_64
├── vmlinuz-5.6.12-300.fc32.x86_64
└── vmlinuz-5.6.13-300.fc32.x86_64

6 directories, 77 files
```

**Additional questions**
- Operating system and version:
Fedora 32 + some updates-testing
- Have you tried rebooting?

twice

- Is this a regression?

I think so.

- Are you using an NVMe disk?

I believe so.

- Is secure boot enabled?
I think so.

sudomateo, 2020-05-20:
I'm also unable to apply the ThinkPad T480 Corporate ME Update `184.77.3664` to my ThinkPad T480. I've tried via GNOME Software and `fwupdmgr` multiple times with no success. When rebooting I see the "Adding capsule" message but the update does not go through.

```
$ fwupdmgr --version
client version:	1.4.1
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.4.1
```

heldchen, 2020-05-20:
same on my X1C6 - https://github.com/fwupd/fwupd/issues/2114#issuecomment-631078883 - but with different fwupdmgr version:
```
fwupdmgr --version
client version:	1.3.9
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.9
```

mweinelt, 2020-05-21:
Same issue with X1C6 and fwupd 1.4.2.

Details: https://github.com/NixOS/nixpkgs/pull/88506#issuecomment-632218227

damentz, 2020-05-21:
I was able to update my firmware by rolling back to the last fwupd version in Arch that would run at all (1.3.9-2) [1].

However, despite fwupd updating the EFI folder, my X1 Carbon still wasn't automatically picking up the capsules.  I rebooted again and used F12 to boot to a temporary device and found something that looked like *Linux-Firmware-Updater* (might be different, I forgot to take a picture).  Picking this option immediately added the two capsules for the most recent update.  Now I'm running the latest firmware for the X1 Carbon 7th.

Here's some data after the output:

```
$ inxi -M
Machine:   Type: Laptop System: LENOVO product: 20QD001VUS v: ThinkPad X1 Carbon 7th 
           serial: <superuser/root required> 
           Mobo: LENOVO model: 20QD001VUS v: SDK0J40697 WIN 
           serial: <superuser/root required> UEFI: LENOVO v: N2HET50W (1.33 ) 
           date: 05/13/2020
```

And after running `fwupdmgr get-devices` and being prompted whether to report:
```
Devices that have been updated successfully:

 • System Firmware (0.1.30 → 0.1.33)
 • UEFI Device Firmware (0.1.15 → 0.1.17)
```

[1] https://git.archlinux.org/svntogit/community.git/plain/trunk/PKGBUILD?h=packages/fwupd&id=4830366284630fac9fe33273adc1d435bef7f427

heldchen, 2020-05-21:
>  • System Firmware (0.1.30 → 0.1.33)
> • UEFI Device Firmware (0.1.15 → 0.1.17)

those updated both for me without problems in ubuntu 20.04 (fwupd 1.3.9) - `Intel Management Engine` seems to be the update that causes problems for my system, even with 1.3.9.

LinuxOnTheDesktop, 2020-05-22:
Me too, on an X1CG6, and not for the first time (see: https://github.com/fwupd/fwupd/issues/1192). Here are the updates concerned:

```
 $ fwupdmgr get-updates
• Thunderbolt Controller has the latest available firmware version
• Embedded Controller has the latest available firmware version
• LENSE20512GMSP34MEAT2TA has the latest available firmware version
• UEFI Device Firmware has no available firmware updates
• Unifying Receiver has the latest available firmware version
20KHCTO1WW
│
├─Intel Management Engine:
│ │   Device ID:           29749712e93b58730ecd1b8537a5503ff5b7fb27
│ │   Current version:     184.70.3626
│ │   Minimum Version:     0.0.1
│ │   Vendor:              DMI:LENOVO
│ │   Update State:        success
│ │   GUID:                9c9d9769-32fa-4841-b550-ea998e754e99
│ │   Device Flags:        • Internal device
│ │                        • Updatable
│ │                        • Requires AC power
│ │                        • Supported on remote server
│ │                        • Needs a reboot after installation
│ │                        • Device is usable for the duration of the update
│ │ 
│ └─ThinkPad X1 Carbon 6th Consumer ME Update:
│       New version:       184.77.3664
│       Remote ID:         lvfs
│       Summary:           Lenovo ThinkPad X1 Carbon 6th Consumer ME Firmware
│       License:           Proprietary
│       Size:              2.2 MB
│       Created:           2016-07-08
│       Urgency:           High
│       Vendor:            Lenovo Ltd.
│       Flags:             is-upgrade
│       Description:       Version 11.8.77.3664 (LVFS: 184.77.3664)
│       
│       2020.01 Q1 Intel Platform Update (IPU), formerly known as Quarterly security release (QSR)
│       
│       Several security fixes and enhancements are on this release.
│       
│       Security issues fixed:
│       
│        • CVE-2020-0531
│        • CVE-2020-0532
│        • CVE-2020-0533
│        • CVE-2020-0535
│        • CVE-2020-0536
│        • CVE-2020-0537
│        • CVE-2020-0538
│        • CVE-2020-0539
│        • CVE-2020-0540
│        • CVE-2020-0545
│        • CVE-2020-0594
│        • CVE-2020-0595
│        • CVE-2020-0596
│        • CVE-2020-8674
│     
└─System Firmware:
  │   Device ID:           1c53551e7da69d896138fac1ae131c83ad46d923
  │   Current version:     0.1.46
  │   Minimum Version:     0.0.1
  │   Vendor:              LENOVO (DMI:LENOVO)
  │   Update State:        success
  │   GUID:                a4b51dca-8f97-4310-8821-3330f83c9135
  │   Device Flags:        • Internal device
  │                        • Updatable
  │                        • Requires AC power
  │                        • Supported on remote server
  │                        • Needs a reboot after installation
  │                        • Cryptographic hash verification is available
  │                        • Device is usable for the duration of the update
  │ 
  └─ThinkPad X1 Carbon 6th System Update:
        New version:       0.1.48
        Remote ID:         lvfs
        Summary:           Lenovo ThinkPad X1 Carbon 6th System Firmware
        License:           Proprietary
        Size:              9.5 MB
        Created:           2016-07-08
        Urgency:           High
        Vendor:            Lenovo Ltd.
        Flags:             is-upgrade
        Description:       Lenovo ThinkPad X1 Carbon 6th System Firmware
        
        Update includes a security fix.
```

daenney, 2020-05-22:
I'm facing the same issue with an X1C6, the updates just don't seem to want to get installed at all:

```
[daenney@yunikoni ~]$ fwupdmgr --version
client version:	1.4.2
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.4.2
```

I've even tried making the Linux Firmware Updater my first boot device, but no joy.

```
[daenney@yunikoni ~]$ fwupdmgr get-updates
• TS3 Plus has no available firmware updates
• Thunderbolt Controller has the latest available firmware version
• Embedded Controller has the latest available firmware version
• UEFI Device Firmware has no available firmware updates
• Unifying Receiver has no available firmware updates
• WDS100T3X0C-00SJG0 has no available firmware updates
20KHCTO1WW
│
├─Intel Management Engine:
│ │   Device ID:           29749712e93b58730ecd1b8537a5503ff5b7fb27
│ │   Current version:     184.70.3626
│ │   Minimum Version:     184.70.3626
│ │   Vendor:              DMI:LENOVO
│ │   Update State:        success
│ │   GUID:                9c9d9769-32fa-4841-b550-ea998e754e99
│ │   Device Flags:        • Internal device
│ │                        • Updatable
│ │                        • Requires AC power
│ │                        • Supported on remote server
│ │                        • Needs a reboot after installation
│ │                        • Device is usable for the duration of the update
│ │
│ └─ThinkPad X1 Carbon 6th Consumer ME Update:
│       New version:       184.77.3664
│       Remote ID:         lvfs
│       Summary:           Lenovo ThinkPad X1 Carbon 6th Consumer ME Firmware
│       Licence:           Proprietary
│       Size:              2.2 MB
│       Created:           2016-07-08
│       Urgency:           High
│       Vendor:            Lenovo Ltd.
│       Flags:             is-upgrade
│       Description:       Version 11.8.77.3664 (LVFS: 184.77.3664)
│
│       2020.01 Q1 Intel Platform Update (IPU), formerly known as Quarterly security release (QSR)
│
│       Several security fixes and enhancements are on this release.
│
│       Security issues fixed:
│
│        • CVE-2020-0531
│        • CVE-2020-0532
│        • CVE-2020-0533
│        • CVE-2020-0535
│        • CVE-2020-0536
│        • CVE-2020-0537
│        • CVE-2020-0538
│        • CVE-2020-0539
│        • CVE-2020-0540
│        • CVE-2020-0545
│        • CVE-2020-0594
│        • CVE-2020-0595
│        • CVE-2020-0596
│        • CVE-2020-8674
│
└─System Firmware:
  │   Device ID:           1c53551e7da69d896138fac1ae131c83ad46d923
  │   Current version:     0.1.46
  │   Minimum Version:     0.0.1
  │   Vendor:              LENOVO (DMI:LENOVO)
  │   Update State:        success
  │   GUID:                a4b51dca-8f97-4310-8821-3330f83c9135
  │   Device Flags:        • Internal device
  │                        • Updatable
  │                        • Requires AC power
  │                        • Supported on remote server
  │                        • Needs a reboot after installation
  │                        • Cryptographic hash verification is available
  │                        • Device is usable for the duration of the update
  │
  └─ThinkPad X1 Carbon 6th System Update:
        New version:       0.1.48
        Remote ID:         lvfs
        Summary:           Lenovo ThinkPad X1 Carbon 6th System Firmware
        Licence:           Proprietary
        Size:              9.5 MB
        Created:           2016-07-08
        Urgency:           High
        Vendor:            Lenovo Ltd.
        Flags:             is-upgrade
        Description:       Lenovo ThinkPad X1 Carbon 6th System Firmware

        Update includes a security fix.
```

LinuxOnTheDesktop, 2020-05-22:
Perhaps it will be useful to collate the following information from the material people have posted on this page.

**Machines affected**

- T480
- X1CG6
- X1CG7

**OSs affected**

- Arch
- Fedora
- Mint
- NixOS
- Pop
- Ubuntu

**Has this problem occurred before, with Lenovo computers:** yes

hughsie, 2020-05-22:
> Perhaps it will be useful to collate the following information from the material people have posted on this page.

That's helpful, thanks. Some people have reported success by resetting all the firmware options back the factory defaults.

LinuxOnTheDesktop, 2020-05-22:
@hughsie 

'Some people have reported success by resetting all the firmware options back the factory defaults.' That may be a helpful work-around, as is booting (twice if there are two updates) to the firmware boot option thing. Yet, it is, only, a work-around. I trust that you and/or Intel and/or Lenovo will actually fix the problem. I note that my [previous report](https://github.com/fwupd/fwupd/issues/1192) is now a year old. I note also that the current report seems to falsify a hypothesis made in the previous report, namely, that the problem is specific to Mint or to SELinux.

hughsie, 2020-05-22:
The key BIOS option that seems to cause the main problem is the "boot order lock" that's sometimes set on Lenovo hardware. I'm working with Lenovo on a way to detect this without using WMI so we can inform the user before the update is deployed.

bmeira, 2020-05-22:
@hughsie 
Pretty sure that option is not causing it. I have that option disabled (always had) and I'm also affected by this issue (Lenovo T480 with Pop)

LinuxOnTheDesktop, 2020-05-22:
@hughsie: thank you for your work. Yet - _data point_ - like @bmeira, I have 'boot order lock' _disabled_ (on my X1CG6).

One can only hope that the forthcoming Lenovo-Fedora partnership will make Lenovo less dreadful in its Linux support. (So, yes, I am inclined to blame Lenovo for the current problem. That company has form in this regard.)

sudomateo, 2020-05-22:
I also have "boot order lock" disabled and I cannot install any firmware updates. I also tried again with default BIOS settings and no luck. I'm running Fedora 32 on a T480.

vancanhuit, 2020-05-23:
My BIOS has "boot order lock" disabled and I also cannot install firmware updates. And no luck with resetting BIOS to factory defaults. I'm running Pop!_OS 20.04 on T480.

heldchen, 2020-05-23:
have "boot order lock" disabled as well.

additionally, resetting bios options did not help. problem occurs when the "Lenovo ThinkPad X1 Carbon 6th Consumer ME Firmware" update is scheduled, so I'm not surprised that resetting the bios values isn't helping as it is my understanding that the ME Firmware is not controlled by the bios.

renffah, 2020-05-23:
X1CG6, Ubuntu 20.04

System firmware was successfully updated from `1.46` to `1.48`:

```
$ inxi -M
Machine:   Type: Laptop System: LENOVO product: 20KGS2A900 v: ThinkPad X1 Carbon 6th serial: <superuser/root required> 
           Mobo: LENOVO model: 20KGS2A900 v: SDK0J40697 WIN serial: <superuser/root required> UEFI: LENOVO v: N23ET73W (1.48 ) 
           date: 04/30/2020
```

```
$ fwupdmgr get-devices 
20KGS2A900
│
├─[…]
│
├─System Firmware:
│     Device ID:           1c53551e7da69d896138fac1ae131c83ad46d923
│     Current version:     0.1.48
│     Minimum Version:     0.0.1
│     Vendor:              LENOVO (DMI:LENOVO)
│     GUID:                a4b51dca-8f97-4310-8821-3330f83c9135
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│
├─UEFI Device Firmware:
│     Device ID:           e563ad307df81c99f0de8c26292afd71cf409673
│     Current version:     184.70.3626
│     Minimum Version:     184.70.3626
│     Vendor:              DMI:LENOVO
│     Update State:        success
│     Last modified:       2020-05-23 16:06
│     GUIDs:               42a0a96e-c9f3-438f-9687-7826be33e4ce
│                          375afb87-ea51-5a6e-96d5-c35cc342dc65
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│
└─[…]
```

UEFI Device firmware update from `184.70.3626` to `184.77.3664`

After executing `fwupdmgr update` and reboot, the boot screen is showing in the top left corner, that it has found two capsules (`0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-42a0a96e-c9f3-438f-9687-7826be33e4ce-0` and `0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-ux-capsule`), is adding those capsules and resetting the system.

After that — it looks like a second reboot — it shows for a very short time (< 1 sec), right below the Lenovo logo, that it is updating the firmware (translated message, so it should come from ubuntu).

Than I tried to reinstall `184.70.3626`, the _Updating firmware_ message is displayed for about 1 minute and a spinner is rotating.

The cab archive for `184.77.3664` contains an additional file.
At a quick look, the metadata of `184.70.3626` and `184.77.3664` are differently structured — no checksums in `184.70.3626`, I than replaced the .77-metadata with the .70-metadata, removed the additional file and repacked the files. No difference, if I try to install the newly created update file, but no error message either.

Booting of `Linux-Firmware-Updater` is definitely working. When I reboot later and select another boot device and than _Linux-Firmware-Updater_ (the boot entry isn't removed after an update) it displays in the top left corner (same as in _adding capsule_) that it can't find any updates and returns to the boot menu.

Manually installing the downloaded cab archive in `~/.cache/fwupd,` I got different results with `fwupdtool` and `fwupdmgr`:
```
$ sudo fwupdtool install dccab103af545a66e515b415d4d88b13b6db130842f11362dc9db85ffc63fbe4-Lenovo-ThinkPad-X1Carbon6th-CorporateMEFirmware-11.8.77.3664.cab e563ad307df81c99f0de8c26292afd71cf409673
Loading…                 [***************************************]
Decompressing…           [***************************************]
Installing on UEFI Device Firmware…                              ]
Scheduling…              [  -                                    ]16:06:02:0662 FuPluginUefi         missing or invalid embedded capsule header
Scheduling…              [***************************************]

An update requires a reboot to complete. Restart now? [y|N]: n
```

```
$ fwupdmgr install dccab103af545a66e515b415d4d88b13b6db130842f11362dc9db85ffc63fbe4-Lenovo-ThinkPad-X1Carbon6th-CorporateMEFirmware-11.8.77.3664.cab e563ad307df81c99f0de8c26292afd71cf409673
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Installing on UEFI Device Firmware…                              ]
Scheduling…              [***************************************]
Successfully installed firmware

An update requires a reboot to complete. Restart now? [y|N]: n
```

Reinstalling .70 with `fwupdtool` displays no error message:
```
$ sudo fwupdtool install --allow-reinstall 08bd61f562c26ca1b8cdbd5a0ede1bb9818cfa00-Lenovo-ThinkPad-X1Carbon6th-CorporateMEFirmware-11.8.70.3626.cab 
Loading…                 [***************************************]
Decompressing…           [***************************************]
Installing on UEFI Device Firmware…                              ]
Scheduling…              [***************************************]

An update requires a reboot to complete. Restart now? [y|N]:
```
So, it really looks like a broken update.

kleinph, 2020-05-25:
I have the same issue on my T480s (it says _Adding new capsule_ but after reboot the update is shown again). [Someone on reddit said that it may have to do with deactivated IME](https://www.reddit.com/r/LinuxOnThinkpad/comments/gpbuiz/cannot_update_intel_management_engine_on_t480s/frndiav?utm_source=share&utm_medium=web2x).

sdroege, 2020-05-25:
I have the same problem on my X1 Carbon 6th generation. fwupd updated some components, but it always wants to update the ME firmware from 184.70.3626 to 184.77.3664. Doing so causes it to shortly show the "updating firmware" screen on reboot after showing it added the capsules, but it immediately reboots again and is stuck with the old version then.

dbasinge, 2020-05-25:
I did get a bios upgrade on my X1 Carbon 6th gen to work by downloading the bios cabs file Lenovo and running fwupdmgr against the files.
```
fwupdmgr install N23ET73W.cab
fwupdmgr install N23HT35W.cab
```
It would fail if only running fwupdmgr update or via Gnome Software. This method only works for firmware upgrades that are in a cab format.

vancanhuit, 2020-05-26:
@dbasinge How can I download those cabs?

heldchen, 2020-05-26:
you can get them from https://pcsupport.lenovo.com/ - but the currently broken update is not available in this format.

renffah, 2020-05-26:
Those Updates got installed with fwupdmgr, but the ME update did not.

Go to the Lenovo support site of your Model, Downloads, BIOS. There should be a "BIOS update utility (Linux)" — a Zip file. This zip contains those cab archives.

The update for my X1 6th contains 3 cab files. Two ending in W, one ending in P.

One is the BIOS update, the other one the embedded controller update. The P-File is also a BIOS update file, which removes the Computrace option completely from the BIOS settings — and probably breaks TPM.

daenney, 2020-05-26:
Dowloading and installing the cab files directly does indeed work, and the update completes successfully.

The ME firmware keeps showing, and failing. The metadata also seems kinda weird on that one:
```
└─ThinkPad X1 Carbon 6th Consumer ME Update:
        New version:       184.77.3664
        Remote ID:         lvfs
        Summary:           Lenovo ThinkPad X1 Carbon 6th Consumer ME Firmware
        Licence:           Proprietary
        Size:              2.2 MB
        Created:           2016-07-08
        Urgency:           High
        Vendor:            Lenovo Ltd.
        Flags:             is-upgrade
        Description:       Version 11.8.77.3664 (LVFS: 184.77.3664)
```

`Created:           2016-07-08` ... that can't be right?

vancanhuit, 2020-05-27:
Unfortunately, the new firmware version is not available for T480 on the Lenovo Support website.

cpt-kernel-afk, 2020-05-28:
Boot manually into "Linux Firmware Updater" on my X1CG6 get this:

```
Found update fwupd-...
Found update fwupd-...
WARNING: No updates to process. Called in error?
```

The message is for half an secound viewable.

ankostis, 2020-05-29:
@cpt-kernel-afk the message you captured corroborates just reported @vorpalBlade's comment @ [#2114](https://github.com/fwupd/fwupd/issues/2114#issuecomment-635814099), that Windows do not use UEFI at all to apply this capsule, and install it directly from within the OS instead.

TylerBrock, 2020-06-05:
Any ideas what you would do to resolve this if you don't have a windows partition?

renffah, 2020-06-05:
As I had written, System firmware update is working using the capsule.

Reinstalling `184.70.3626` seems to work too, because the System is showing _Applying Firmware Update_ for about one Minute.

Installing `184.77.3664` through `fwupdtool` instead of `fwupdmgr` produces an Error message:
`FuPluginUefi missing or invalid embedded capsule header`. In either case, after a reboot _Applying Firmware_ is just flickering for about one second on the screen.

From my point of view, the capsule is broken and if there's any bug in fwupd, its a broken error checking.

fcelda, 2020-06-10:
I got to the same conclusion as @renffah. The Intel ME 11.8.77.3664 update is started, the firmware update message just flickers on the screen, and the machine reboots immediately. The current Intel ME firmware can be reinstalled without any issues.

fcelda, 2020-06-11:
> > INSERT INTO history VALUES('e563ad307df81c99f0de8c26292afd71cf409673',4,'failed to update to 3091598890: unsuccessful',
> 
> `4` means `FWUPD_UPDATE_STATE_NEEDS_REBOOT` which is correct. Can you share the same thing when you've scheduled the update, rebooted, and then booted back into Linux please.

What do you mean by scheduling the update? I just run `fwupdmgr update`. Is that OK or should I try something else?

```
# fwupdmgr update 
• Thunderbolt Controller has the latest available firmware version
• Embedded Controller has the latest available firmware version
Upgrade available for Intel Management Engine from 184.70.3626 to 184.77.3664
20KH002RUS must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: y
Downloading 184.77.3664 for Intel Management Engine...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating Intel Management Engine…                                ]
Scheduling…              [***************************************]
Successfully installed firmware
• System Firmware has the latest available firmware version
• UEFI Device Firmware has no available firmware updates
• WDC PC SN720 SDAQNTW-512G-1001 has the latest available firmware version

An update requires a reboot to complete. Restart now? [y|N]: y
```

The laptop rebooted into the `fwupdx64.efi`, the capsule was found, then the machine restarted again, the Lenovo firmware update screen just flashed, and finally the machine rebooted back to Linux.

The firmware is still at the old version and I don't see any new entry in the history table. The last entry is the one I shared [previously](https://github.com/fwupd/fwupd/issues/2118#issuecomment-642140013).

> I'd also appreciate if anyone hitting this could install the wip/hughsie/fu_engine_update_history_device branch and see if that correctly detects the case where the firmware does not set LastAttemptStatus.

I'm going to try if I manage to build an RPM with the updated sources. I will report back.

pschichtel, 2020-06-11:
I'm also affected. Archlinux with both fwupd 1.3.9 and and 1.4.2 and boot order lock disabled.

@hughsie I also rebuilt the Arch package using the branch you mentioned and it detected the failure now. I reported the history.

fcelda, 2020-06-11:
> > I'd also appreciate if anyone hitting this could install the wip/hughsie/fu_engine_update_history_device branch and see if that correctly detects the case where the firmware does not set LastAttemptStatus.
> 
> I'm going to try if I manage to build an RPM with the updated sources. I will report back.

I can confirm that the failure is correctly recorded with this branch. I've submitted the report as well.

Also, RPM for Fedora 32 built from `wip/hughsie/fu_engine_update_history_device` is in my [COPR repo](https://copr.fedorainfracloud.org/coprs/jvcelak/fwupd/). It will be deleted in 60 days.