fwupd/fwupd #2119: del-self-test FAIL with SIGTRAP (GLib-GObject-FATAL-WARNING: cannot register existing type 'FuUefiBgrt')

RiderALT, 2020-05-20:
**Describe the bug**
When building a new version for ALT, I got the following error:
```
# FuPlugin-DEBUG: performing init() on /usr/src/RPM/BUILD/fwupd-1.4.2/x86_64-alt-linux/plugins/dell/../uefi/libfu_plugin_uefi.so
Bail out! GLib-GObject-FATAL-WARNING: cannot register existing type 'FuUefiBgrt'
--- stderr ---

(/usr/src/RPM/BUILD/fwupd-1.4.2/x86_64-alt-linux/plugins/dell/dell-self-test:1539020): GLib-GObject-WARNING **: 12:33:34.220: cannot register existing type 'FuUefiBgrt'
```
bt from gdb:
```

#0  0x00007ffff7c4a2f5 in ?? () from /lib64/libglib-2.0.so.0
#1  0x00007ffff7c4b5c9 in g_logv () from /lib64/libglib-2.0.so.0
#2  0x00007ffff7c4b78f in g_log () from /lib64/libglib-2.0.so.0
#3  0x00007ffff7d53b98 in ?? () from /usr/lib64/libgobject-2.0.so.0
#4  0x00007ffff7d53e9b in g_type_register_static () from /usr/lib64/libgobject-2.0.so.0
#5  0x00007ffff7d540e5 in g_type_register_static_simple () from /usr/lib64/libgobject-2.0.so.0
#6  0x00007ffff63c110c in fu_uefi_bgrt_get_type_once () at ../plugins/uefi/fu-uefi-bgrt.c:21
#7  0x00007ffff63c1195 in fu_uefi_bgrt_get_type () at ../plugins/uefi/fu-uefi-bgrt.c:21
#8  0x00007ffff63c15d6 in fu_uefi_bgrt_new () at ../plugins/uefi/fu-uefi-bgrt.c:131
#9  0x00007ffff63bf48b in fu_plugin_init (plugin=0x43b320) at ../plugins/uefi/fu-plugin-uefi.c:38
#10 0x00007ffff7f848d4 in fu_plugin_open (self=self@entry=0x43b320, 
    filename=filename@entry=0x430870 "/usr/src/RPM/BUILD/fwupd-1.4.2/x86_64-alt-linux/plugins/dell/../uefi/libfu_plugin_uefi.so", error=error@entry=0x7fffffffe198)
    at ../libfwupdplugin/fu-plugin.c:435
#11 0x0000000000403a60 in fu_plugin_dell_dock_func () at ../plugins/dell/fu-self-test.c:297
#12 0x00007ffff7c6bc8e in ?? () from /lib64/libglib-2.0.so.0
#13 0x00007ffff7c6ba34 in ?? () from /lib64/libglib-2.0.so.0
#14 0x00007ffff7c6c17a in g_test_run_suite () from /lib64/libglib-2.0.so.0
#15 0x00007ffff7c6c191 in g_test_run () from /lib64/libglib-2.0.so.0
#16 0x000000000040383a in main (argc=<optimized out>, argv=<optimized out>) at ../plugins/dell/fu-self-test.c:526
```

-------

**Steps to Reproduce**
build, run tests

**Expected behavior**
tests work without errors

**fwupd version information**
fwupd 1.4.0-1.4.2

**Additional questions**
- Operating system and version:
ALT Linux Sisyphus, gcc-9.3.1, glib2-2.64.2
- Is this a regression?
yes, version 1.3.x was built without these errors

superm1, 2020-05-20:
I suspect this is caused by compile flags. What are your cflags?

RiderALT, 2020-05-20:
configure flags:

/usr/bin/meson . x86_64-alt-linux --buildtype=plain --prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --libdir=/usr/lib64 --localstatedir=/var --sharedstatedir=/var/lib --includedir=/usr/include --datadir=/usr/share --sysconfdir=/etc --mandir=/usr/share/man --infodir=/usr/share/info --localedir=/usr/share/locale --wrap-mode=nodownload -Dgtkdoc=true  -Dfirmware-packager=true -Dman=false -Dlvfs=true -Dplugin_flashrom=false -Dtests=true -Dplugin_dummy=true -Dplugin_thunderbolt=true -Dplugin_uefi=true -Dplugin_redfish=true -Dplugin_nvme=true -Dplugin_dell=true -Dplugin_synaptics=true

build flags:

[1/2] cc -Iplugins/dell/bdd12ec@@dell-self-test@exe -Iplugins/dell -I../plugins/dell -I. -I.. -Ilibfwupd -I../libfwupd -Ilibfwupdplugin -I../libfwupdplugin -I/usr/include/libxmlb-1 -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/gio-unix-2.0 -I/usr/include/gusb-1 -I/usr/include/libusb-1.0 -I/usr/include/libsoup-2.4 -I/usr/include/gudev-1.0 -I/usr/include/efivar -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -std=c99 -Waggregate-return -Wunused -Warray-bounds -Wcast-align -Wclobbered -Wdeclaration-after-statement -Wdiscarded-qualifiers -Wduplicated-branches -Wduplicated-cond -Wempty-body -Wformat=2 -Wformat-nonliteral -Wformat-security -Wformat-signedness -Wignored-qualifiers -Wimplicit-function-declaration -Winit-self -Wlogical-op -Wmaybe-uninitialized -Wmissing-declarations -Wmissing-format-attribute -Wmissing-include-dirs -Wmissing-noreturn -Wmissing-parameter-type -Wmissing-prototypes -Wnested-externs -Wno-cast-function-type -Wno-address-of-packed-member -Wno-unknown-pragmas -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-strict-aliasing -Wno-suggest-attribute=format -Wno-unused-parameter -Wnull-dereference -Wold-style-definition -Woverride-init -Wpointer-arith -Wredundant-decls -Wreturn-type -Wshadow -Wsign-compare -Wstrict-aliasing -Wstrict-prototypes -Wswitch-default -Wtype-limits -Wundef -Wuninitialized -Wunused-but-set-variable -Wunused-variable -Wvla -Wwrite-strings -fstack-protector-strong -D_DEFAULT_SOURCE -DFWUPD_DISABLE_DEPRECATED -D_BSD_SOURCE -D_XOPEN_SOURCE=700 -frecord-gcc-switches -Wall -g -O2 -pthread '-DG_LOG_DOMAIN="FuPluginDell"' '-DTESTDATADIR="/usr/src/RPM/BUILD/fwupd-1.4.2/plugins/dell/tests"' '-DPLUGINBUILDDIR="/usr/src/RPM/BUILD/fwupd-1.4.2/x86_64-alt-linux/plugins/dell"' -MD -MQ 'plugins/dell/bdd12ec@@dell-self-test@exe/fu-self-test.c.o' -MF 'plugins/dell/bdd12ec@@dell-self-test@exe/fu-self-test.c.o.d' -o 'plugins/dell/bdd12ec@@dell-self-test@exe/fu-self-test.c.o' -c ../plugins/dell/fu-self-test.c
[2/2] cc  -o plugins/dell/dell-self-test 'plugins/dell/bdd12ec@@dell-self-test@exe/fu-self-test.c.o' 'plugins/dell/bdd12ec@@dell-self-test@exe/fu-dell-smi.c.o' 'plugins/dell/bdd12ec@@dell-self-test@exe/fu-plugin-dell.c.o' -Wl,--as-needed -Wl,--no-undefined -Wl,-z,relro -Wl,-z,defs -Wl,-z,now -pipe -frecord-gcc-switches -Wall -g -O2 -Wl,--start-group libfwupd/libfwupd.so.2.0.0 libfwupdplugin/libfwupdplugin.so.1.0.0 -lxmlb -lgio-2.0 -lgobject-2.0 -lglib-2.0 -lgio-2.0 -lgobject-2.0 -lglib-2.0 -lgio-2.0 -lgobject-2.0 -lglib-2.0 -Wl,--export-dynamic -pthread -lgmodule-2.0 -lglib-2.0 -lgusb -lgio-2.0 -lgobject-2.0 -lglib-2.0 -lusb-1.0 -lsoup-2.4 -lgio-2.0 -lgobject-2.0 -lglib-2.0 -larchive -lgudev-1.0 -lgobject-2.0 -lglib-2.0 -lefivar -lsqlite3 -lsmbios_c -ltss2-esys -Wl,--end-group '-Wl,-rpath,$ORIGIN/../../libfwupd:$ORIGIN/../../libfwupdplugin' -Wl,-rpath-link,/usr/src/RPM/BUILD/fwupd-1.4.2/x86_64-alt-linux/libfwupd -Wl,-rpath-link,/usr/src/RPM/BUILD/fwupd-1.4.2/x86_64-alt-linux/libfwupdplugin

superm1, 2020-05-20:
This sounds like the behavior of `-Wl,-Bsymbolic-functions` which we don't support but I don't see it in your linker command.  Can you check that it's not mentioned somewhere?

glebfm, 2020-05-22:
@RiderALT asked me to take a look at this issue.

Long story short, this test fails if you do not have valgrind in your build environment.  Looks like Debian maintainers probably knew ([debian/rules#L93](https://salsa.debian.org/efi-team/fwupd/-/blob/ce914d6f1225f3f620f1b7852e3681b16b073630/debian/rules#L93)) something about this problem, but it is actually a bug introduced in commit 576c012.

I posted pull request #2126 to bug related to valgrind, but it just makes the issue more visible.

glebfm, 2020-05-22:
Just to be clear: this issue is about fwupd not being able to unload module and then load it again.

superm1, 2020-05-28:
should be fixed in master now, thanks.

closed 2020-05-28 by superm1