fwupd/fwupd #2131: fwupd master does not build for arm targets

bleungatchromium, 2020-05-24:
**Describe the bug**
A clear and concise description of what the bug is.
I'm crosscompiling for an arm target (chromebook scarlet, using a rockchip ARM processor). Recent 1_5_X or master branch do not build. Here are the errors:

In file included from ../fwupd-9999/libfwupdplugin/fu-common.c:23:
/usr/lib64/clang/11.0.0/include/cpuid.h:11:2: error: this header is for x86 only
#error this header is for x86 only
 ^
/usr/lib64/clang/11.0.0/include/cpuid.h:271:5: error: invalid output constraint '=a' in asm
    __cpuid(__leaf, __eax, __ebx, __ecx, __edx);
    ^
/usr/lib64/clang/11.0.0/include/cpuid.h:236:11: note: expanded from macro '__cpuid'
        : "=a"(__eax), "=r" (__ebx), "=c"(__ecx), "=d"(__edx) \
          ^
/usr/lib64/clang/11.0.0/include/cpuid.h:286:5: error: invalid output constraint '=a' in asm
    __cpuid(__leaf, *__eax, *__ebx, *__ecx, *__edx);
    ^
/usr/lib64/clang/11.0.0/include/cpuid.h:236:11: note: expanded from macro '__cpuid'
        : "=a"(__eax), "=r" (__ebx), "=c"(__ecx), "=d"(__edx) \
          ^
/usr/lib64/clang/11.0.0/include/cpuid.h:300:5: error: invalid output constraint '=a' in asm
    __cpuid_count(__leaf, __subleaf, *__eax, *__ebx, *__ecx, *__edx);
    ^
/usr/lib64/clang/11.0.0/include/cpuid.h:243:11: note: expanded from macro '__cpuid_count'
        : "=a"(__eax), "=r" (__ebx), "=c"(__ecx), "=d"(__edx) \
          ^
4 errors generated.

Bisecting from 1.4.1 points to this commit:

9223c89019824d2d0c72d63eb29f977691d01f89 is the first bad commit
commit 9223c89019824d2d0c72d63eb29f977691d01f89
Author: Richard Hughes <richard@hughsie.com>
Date:   Sat May 9 20:32:08 2020 +0100

    trivial: Add a helper method to check for an Intel CPU

    Some plugins will be Intel specific.

 libfwupdplugin/fu-common.c     | 30 ++++++++++++++++++++++++++++++
 libfwupdplugin/fu-common.h     |  1 +
 libfwupdplugin/fwupdplugin.map |  1 +
 3 files changed, 32 insertions(+)

**Steps to Reproduce**
Steps to reproduce the behavior.
emerge-scarlet fwupd

build log attached:
[sys-apps:fwupd-9999:20200524-234824.log](https://github.com/fwupd/fwupd/files/4674634/sys-apps.fwupd-9999.20200524-234824.log)

**Expected behavior**
A clear and concise description of what you expected to happen.
Successful build.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
Built it myself. :)

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
```

**Additional questions**
- Operating system and version: Chrome OS
- Have you tried rebooting? NA
- Is this a regression? Yes, build regression. Build of 1.4.1 was fine.

bleungatchromium, 2020-05-24:
@hughsie

superm1, 2020-05-26:
Was this not fixed with https://github.com/fwupd/fwupd/commit/bd44432240278c5453bb34e723e4274b4638d317?

This reminds me we probably need to check if s390x can be restored (it was turned off for CI in https://github.com/fwupd/fwupd/commit/7513750f653401f23e097eb4eae080647a33bad8).
And this also makes wonder if we should add another cross compile target for arm into CI.

hughsie, 2020-05-26:
> cross compile target for arm

I do compile for ARM64 for Fedora at release time, but I have no problems with an ARM CI test.

superm1, 2020-05-26:
I tried to enable s390x again in Debian cross compile and it's still got some library installation problems. Perhaps yes we should add some Fedora based cross compile targets instead.
S390x was specifically because of endian bugs.

bleungatchromium, 2020-05-26:
Hi @superm1 
> Was this not fixed with [bd44432](https://github.com/fwupd/fwupd/commit/bd44432240278c5453bb34e723e4274b4638d317)?

The fix in  [bd44432] doesn't help in my environment, as cpuid.h is present.

`Has header "cpuid.h" : YES`

Is this some quirk in the way that my chromeos/gentoo environment builds the package?

hughsie, 2020-05-26:
I guess it's unexpected that the cpuid.h header was found when cross compiling; how are you building fwupd?

bleungatchromium, 2020-05-26:
Chromiumos is using a lightly modified version of Gentoo's ebuild for this:
https://gitweb.gentoo.org/repo/gentoo.git/tree/sys-apps/fwupd/fwupd-1.4.1.ebuild

Here's the ebuild that's being used for the cross compile:
https://chromium.googlesource.com/chromiumos/overlays/chromiumos-overlay/+/master/sys-apps/fwupd/fwupd-9999.ebuild

bleungatchromium, 2020-05-26:
@djcampello as an FYI.

superm1, 2020-05-26:
It does look like clang's `cpuid.h` specifically has an error clause for !x86 but GCC's doesn't.  On gcc it seems that it's just not included for some architectures.  So we probably need an extra check for the header existing and the architecture both.

@djcampello 
It would be ideal if we could run the chromeos ebuild directly as a CI target, maybe even a cross compile CI target.  Any chance you guys can try to put that together?

PR #2139 (2020-05-26): trivial: detect `cpuid.h` and also look for host architecture
Fixes cross compilation for arm with clang which provides cpuid.h
but helpfully has this error:
```
 #if !(__x86_64__ || __i386__)
 #error this header is for x86 only
 #endif
```

Fixes: #2131

Type of pull request:
- [ ] New plugin (Please include [new plugin checklist](https://github.com/hughsie/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit d39bcee2 (2020-05-26): trivial: detect `cpuid.h` and also look for host architecture

Fixes cross compilation for arm with clang which provides cpuid.h
but helpfully has this error:
```
 #if !(__x86_64__ || __i386__)
 #error this header is for x86 only
 #endif
```

Fixes: #2131
closed 2020-05-26 by superm1