fwupd/fwupd #2132: Failed to update metadata for lvfs: Keyring kind jcat not supported

DanielSiepmann, 2020-05-25:
**Describe the bug**
Since my Ubuntu Update, I get the following error when running `fwupdmgr refresh`:
```
Failed to update metadata for lvfs: Keyring kind jcat not supported
```

**Steps to Reproduce**
Installed fwupd from Ubuntu repositories. Do an Upgrade from 18.04 to 20.04 and execute.

**Expected behavior**
I would expect to extract and parse the downloaded metadata.

**fwupd version information**
```
client version: 1.3.9
compile-time dependency versions
        gusb:   0.3.4
        efivar: 37
daemon version: 1.3.9
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
Via `apt`.

**fwupd device information**
```
XPS 13 9370
│
├─Thunderbolt Controller:
│     Device ID:           c5e19d13d335ce1531cac28908afdce24ec06a39
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     33.00
│     Vendor:              Dell (TBT:0x00D4)
│     GUIDs:               4eeb9d07-a96c-56d6-92d3-4a23ee7a6e4a
│                          b8381cb5-073f-5171-a46d-6b618c4d740f
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─Intel AMT [unprovisioned]:
│     Device ID:           e2623122c99d58220498aacbfcfdb1baebbae3c5
│     Summary:             Hardware and firmware technology for remote out-of-band management
│     Current version:     11.8.70.3626
│     Bootloader Version:  11.8.70.3626
│     Vendor:              Intel Corporation
│     GUID:                2800f812-b7b4-2d4b-aca8-46e0ff65814c
│     Device Flags:        • Internal device
│   
├─KXG50ZNV512G NVMe TOSHIBA 512GB:
│     Device ID:           f954c7acdf5fab61aeaca1cd71d29ea5ade6992f
│     Summary:             NVM Express Solid State Drive
│     Current version:     AADA4104
│     Vendor:              Toshiba Corporation (NVME:0x1179)
│     GUIDs:               af8d42da-167c-546a-9294-c1502f6ccbab
│                          4d0aed03-a30c-52c6-99e7-a8977797c3d9
│                          98d34383-a939-5b5a-9806-e05e0b288b1a
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─PTN5002:
│     Device ID:           983c3cffc6fd36d32b00b62928d30721eaeb93db
│     Current version:     1.0
│     Vendor:              NXP (USB:0x1FC9)
│     GUIDs:               d5ec8dcb-f9b1-59c2-959b-f9bd882d0927
│                          a01d9cb7-dc1c-52dc-88ad-ba94f473681a
│     Device Flags:        • Updatable
│   
├─System Firmware:
│     Device ID:           8a21cacfb0a8d2b30c5ee9290eb71db021619f8b
│     Current version:     1.12.1
│     Minimum Version:     1.12.1
│     Vendor:              Dell Inc. (DMI:Dell Inc.)
│     GUID:                7ceaf7a8-0611-4480-9e30-64d8de420c7c
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─Touchpad:
│     Device ID:           88861d85ec873ffdf484ac0b981c640781bd6937
│     Current version:     1.3.2375007
│     Bootloader Version:  54.0
│     Vendor:              Synaptics (HIDRAW:0x06CB)
│     GUIDs:               f4384034-9243-5334-8075-a534be913e46
│                          424bd00e-9789-5cdf-a12a-3c81bc4676d6
│                          140f4458-951b-5bb9-85e2-879bd5b02615
│                          b29d3c85-cd0e-503e-9c7e-f6731c1eaf2d
│     Device Flags:        • Internal device
│                          • Updatable
│   
├─UHD Graphics 620:
│     Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
│     Current version:     07
│     Vendor:              Intel Corporation (PCI:0x8086)
│     GUIDs:               fed2efa4-7045-55a1-b4fc-b29283d59fe5
│                          8b72a10c-1279-5f8e-a28a-34fb11a58240
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│   
└─USB Keyboard:
      Device ID:           c0170efdf195b8590000fa21474253c0d97e7335
      Summary:             USB PD
      Current version:     149.1.4.3
      Vendor:              VIA Labs, Inc. (USB:0x2109)
      GUIDs:               9b630f16-62b2-57a2-a4d3-36df60ac08ad
                           56a509c6-b4c3-5cf3-bf9d-8c98bc8562dc
                           446d3992-2a03-5143-a99c-6189ec313d86
      Device Flags:        • Updatable
                           • Cryptographic hash verification is available
```

**Additional questions**
- Operating system and version: Ubuntu 20.04 LTS
- Have you tried rebooting? Multiple times
- Is this a regression? Don't think so.

I'm not an expert and can provide further information if necessary and with explanation on how to. Any help to solve the issue are appreciated.

hughsie, 2020-05-25:
Jcat is only supported in 1.4.0. Have you perhaps installed 1.4.x and 1.3.x at the same time? e.g. is 1.4.x snap installed?

superm1, 2020-05-26:
Or did you hand install at some point and your `/etc/fwupd/` has stuff from 1.4.0?

DanielSiepmann, 2020-05-28:
I don't think I've installed anything myself in a specific version or via Snap.

```
jcat -V       
The Sleuth Kit ver 4.6.7
```

But looks like another tool. Just ran `apt install jcat` which installed `jcat` in version `0.1.0-2`. Maybe something broke during the upgrade from Ubuntu 18.04 to 20.04. I should have checked whether the package is still installed. :facepalm: 

Still I get:
```
Fetching file https://cdn.fwupd.org/downloads/firmware.xml.gz.jcat

Failed to update metadata for lvfs: Keyring kind jcat not supported
```

I've checked that it is installed:
```
apt list --installed | grep jcat

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

jcat/focal,now 0.1.0-2 amd64 [installed]
libjcat1/now 0.1.0-2ubuntu1 amd64 [installed,local]
```
Which is an unsupported version? So how can I make fwupd work on Ubuntu 20.04? Preferable using APT, not Snap.

```
apt list --installed | grep fwup

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

fwupd-signed/focal,now 1.27+1.3.9-4 amd64 [installed,automatic]
fwupd/focal,now 1.3.9-4 amd64 [installed]
fwupdate/focal,now 12-7 amd64 [installed]
libfwupd2/now 1.4.0+r1+ga51f0e11 amd64 [installed,local]
libfwupdplugin1/now 1.4.0+r1+ga51f0e11 amd64 [installed,local]
```

When I remove fwupd together with jcat and try to install again, I don't get jcat listed as dependency that will be installed:
```apt install fwupd
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  fwupd-signed libfwupdplugin1
The following NEW packages will be installed:
  fwupd fwupd-signed libfwupdplugin1
0 upgraded, 3 newly installed, 0 to remove and 17 not upgraded.
Need to get 2,194 kB of archives.
After this operation, 6,159 kB of additional disk space will be used.
```

And that nothing is installed via snap:
```
snap list
Name                 Version                     Rev   Tracking         Publisher   Notes
canonical-livepatch  9.5.5                       95    latest/stable    canonical✓  -
chromium             83.0.4103.61                1165  latest/stable    canonical✓  -
core                 16-2.44.3                   9066  latest/stable    canonical✓  core
core18               20200427                    1754  latest/stable    canonical✓  base
gnome-3-26-1604      3.26.0.20191114             98    latest/stable/…  canonical✓  -
gnome-3-28-1804      3.28.0-16-g27c9498.27c9498  116   latest/stable    canonical✓  -
gtk-common-themes    0.1-36-gc75f853             1506  latest/stable    canonical✓  -
slack                4.4.3                       24    latest/stable    slack✓      classic
```

For now I've installed the snap to get updates again.

hughsie, 2020-05-28:
What does /etc/fwupd/remotes.d/lvfs.conf have?

DanielSiepmann, 2020-05-28:
```
cat /etc/fwupd/remotes.d/lvfs.conf 
[fwupd Remote]

# this remote provides metadata and firmware marked as 'stable' from the LVFS
Enabled=true
Title=Linux Vendor Firmware Service
Keyring=gpg
MetadataURI=https://cdn.fwupd.org/downloads/firmware.xml.gz
ReportURI=https://fwupd.org/lvfs/firmware/report
OrderBefore=fwupd
AutomaticReports=false
ApprovalRequired=false
```

superm1, 2020-05-29:
> libfwupd2/now 1.4.0+r1+ga51f0e11 amd64 [installed,local]
libfwupdplugin1/now 1.4.0+r1+ga51f0e11 amd64 [installed,local]

Your problem is that you have a hand build installed here for the library.  That is a git snapshot, not the version in Ubuntu repositories.  Please switch to that and your issue will be sorted.

closed 2020-05-29 by superm1

DanielSiepmann, 2020-06-02:
Thank you very much. Removed those libraries and installed fwupd again (which pulls the depended libs) worked.

Sorry for bothering you and thanks for the kindfull help.