fwupd/fwupd #2138: cros-ec: cros-ec-usb-device's flush transfer causes failures on subsequent bulk transfers

bleungatchromium, 2020-05-26:
/* Flush all data from endpoint to recover in case of error. */
	/* FIXME(bleung): b/157109635. This flush causes problems with the next
	 * xfer, it times out. Remove it for now, but this should be restored
	 * once we figure out what's going on.

outside of chrome OS development I don't think b/#### is very well understood.  Assuming this is a public bug number, could you expand this to something resolvable?

That being said having no idea on the underlying architecture, is this just a case that the flush command returns before the content has _actually_ propagated to the other parts in the black box?  If so, a reasonable thing to do would be to throw a loop here that will try some reasonable number of times until the transfers stop timing out.

I do something similar in the Dell dock setup routine where there is a corner race condition that you could potentially try to probe some composite devices before they've finished booting if you plug everything in at once.
https://github.com/fwupd/fwupd/blob/master/plugins/dell-dock/fu-dell-dock-i2c-ec.c#L916

This is also the reason we have the helper function `fu_device_retry` for sending some commands.

_Originally posted by @superm1 in https://github.com/fwupd/fwupd/pull/2130_

bleungatchromium, 2020-05-26:
I adapted this block of code from usb_updater2:
https://chromium.googlesource.com/chromiumos/platform/ec/+/master/extra/usb_updater/usb_updater2.c#774

```
        /* Flush all data from endpoint to recover in case of error. */
        while (!libusb_bulk_transfer(td->uep.devh,
                                        td->uep.ep_num | 0x80,
                                        (void *)&inbuf, td->uep.chunk_len,
                                        &actual, 10)) {
                printf("flush\n");
        }

```

Here's the fwupd plugin version adapted for cros_ec-usb:

```
        /* Flush all data from endpoint to recover in case of error. */        
        while (g_usb_device_bulk_transfer (usb_device,                          
                                           self->ep_num | 0x80,                
                                           (guint8 *)&inbuf, self->chunk_len,  
                                           &actual, 10, NULL, error)) {        
                                                                               
                g_print ("flush!\n");                                          
        }

```

I don't know what it is about it when I ported it into fwupd (maybe because libgusb's wrapper for libusb_bulk_transfer isn't exactly the same.. it's async), but it causes transfer timeouts for all future transactions.

This is a task to investigate why this happens.

```
06:23:46:0145 libgusb              g_usb_device_bulk_transfer_finish: assertion 'error == NULL || *error == NULL' failed
Crap. there's an error
06:23:46:0145 FuEngine             failed to add USB device 18d1:501a: transfer timed out

```

For now, I've just commented out this code, but we should figure out why this causes errors.

bleungatchromium, 2020-05-26:
Here's a comment from my colleague (and original developer of usb_updater2), Nicolas Boichat:

> This is flushing code to reset the state machine on the EC side, and the transfers are expected to fail in most normal cases.
> 
> "Crap. there's an error": Looks like there's an assertion in the core, we should add a flag to make errors forgivable.

superm1, 2020-05-26:
So I think part of this might be that you're misusing the error in this context.  If you really don't care about the error, then instead pass `NULL`, but typically this is done by using a local error (IE `g_autoptr(GError) error_local = NULL`).  If you use a local error, you can output it with `g_debug ("%s", error_local->message);`

bleungatchromium, 2020-05-26:
Thanks. I'll give this a try.

bleungatchromium, 2020-05-26:
That was it. Next version will clean it up. Thanks @superm1

closed 2020-05-26 by bleungatchromium