fwupd/fwupd #2152: Add ability to filter updates by severity

michel-slm, 2020-05-29:
**Is your feature request related to a problem? Please describe.**
We manage a large and heterogenous fleet of Linux client devices, spanning multiple models and several hardware generations; we would like to automate applying firmware updates but as the process is quite disruptive (needs rebooting) and potentially dangerous (potential bricking), we want to start by applying only the most critical updates - similar to how security updates can be automated via dnf-automatic.

**Describe the solution you'd like**
Comparing to dnf, which has the following toggles:
```
  --bugfix              Include bugfix relevant packages, in updates
  --enhancement         Include enhancement relevant packages, in updates
  --newpackage          Include newpackage relevant packages, in updates
  --security            Include security relevant packages, in updates
  --advisory ADVISORY, --advisories ADVISORY
                        Include packages needed to fix the given advisory, in updates
  --bz BUGZILLA, --bzs BUGZILLA
                        Include packages needed to fix the given BZ, in updates
  --cve CVES, --cves CVES
                        Include packages needed to fix the given CVE, in updates
  --sec-severity {Critical,Important,Moderate,Low}, --secseverity {Critical,Important,Moderate,Low}
                        Include security relevant packages matching the severity, in updates
```

LVFS currently has the Urgency field, and fwupdmgr has the `--filter` directive, but I don't think it can be used to select updates by urgency at the moment.

Also - if we can make it possible to filter by specific CVEs that would be great, e.g. for this update:

```
  └─XPS 15 7590/Precision 5540 System Update:
        New version:       1.6.0
        Remote ID:         lvfs
        Summary:           Firmware for the Dell XPS 15 7590/Precision 5540
        License:           Proprietary
        Size:              21.7 MB
        Created:           2020-02-06
        Urgency:           Critical
        Vendor:            Dell Inc.
        Description:       This stable release fixes the following issues:
        
         • Firmware updates to address security advisory
         • Firmware updates to address the Intel Security Advisories   and  .
         • Fixed the issue where the keyboard and touchpad lags when the system is connected to a Dell U2419HC monitor using the USB Type-C port.
         • Fixed the issue where the system updates the BIOS continuously when the embedded controller firmware is upgraded or downgraded.
        
        Some new functionality has also been added:
        
         • Updated the BIOS warning message that is displayed when an AC adapter with low wattage is connected to the system.
         • Updated the Enhanced Pre-Boot System Assessment (ePSA) firmware.
         • Updated the Power Delivery firmware.
        
        Security issues fixed:
        
         • CVE-2019-14598
         • CVE-2019-11157
         • CVE-2019-14607
         • INTEL-SA-00307
         • INTEL-SA-00289
         • INTEL-SA-00317
```

**Describe alternatives you've considered**
It's already possible to use `set-approved-firmware`, but this puts us back in the situation where we have to check every possible update for every possible hardware in our fleet, and programmatically configure individual machines.

**Additional context**
Add any other context or screenshots about the feature request here.

superm1, 2020-05-29:
Could you describe some of the larger picture that you want this to fit into?  Existing tooling, consoles etc?

The last time that this was discussed was in https://github.com/fwupd/fwupd/issues/933 (which you may consider looking at for more context of the conversation).

For example something like `fwupdagent` might make sense to fit into something bigger that is part of a feedback loop.

michel-slm, 2020-05-31:
So that feels a bit different - it describes the need to canary a given update, and if the signals are positive, roll it out to the wider fleet.

Our use case is that we are not in a position to vet every available update given the fragmentation of our fleet, but would like to automatically apply any update that match some criteria (eg criticalness, or whether it addresses specific CVEs).

We manage the fleet using Chef so we can already invoke fwupdmgr remotely on the clients - if user interaction is needed we can autogenerate a prompt on our users' desktop that will then run fwupdmgr in a terminal

superm1, 2020-05-31:
Yeah that is related but you're right, a different type of request.  So you still want to rely upon the updates coming from LVFS, but want to only apply updates that have a reduced criteria.

Have you already looked at what you get from `fwupdagent get-updates` output?  It passes output in JSON.  

I'm wondering if it's reasonable for you to use that and then do the filtering with json parsing tools on the criteria you're looking for.  You can come up with some more advanced filtering in a higher level language too then.  For example if you want to only apply updates that included a particular CVE you could use the output of `fwupdagent get-updates`, find all device ID with updates that mention that CVE in release notes.  Then either call `fwupdmgr update $DEVICE_ID`.

Another idea for you is to instead have a simple python based client that is called and you can directly parse daemon output to make your own decisions. https://github.com/fwupd/fwupd/blob/master/contrib/firmware_packager/simple_client.py

> We manage the fleet using Chef so we can already invoke fwupdmgr remotely on the clients - if user interaction is needed we can autogenerate a prompt on our users' desktop that will then run fwupdmgr in a terminal

`fwupdmgr` can absolutely be run without user interaction.  If it doesn't detect it's running on an interactive tty it won't prompt anything.  You can control the time you want to reboot to do the update separately from staging it on the EFI system partition.