fwupd/fwupd #2153: Dell Precision M5520 UEFI update fails - 'The update image is not for this system'

1t5j0y, 2020-05-31:
**Describe the bug**
I'm trying to update the bios on my Dell Precision M5520. The currently installed version is 0.0.5. The latest bios update is [1.18.0](https://fwupd.org/lvfs/devices/com.dell.uefi34578c72.firmware). However, fwupdmgr shows the latest version as 1.22.3 (which seems to be the latest firmware version for the Dell Latitude E5X70 based on the ```fwupdmgr --get-updates``` output). Attempting to install this version throws the error 'The update image is not for this system'.

**Steps to Reproduce**

1. Click the 'Update' button within Ubuntu Software for Dell Inc System Firmware 0.0.5 -> 1.22.3

**Expected behavior**
1. Expected to see bios version 1.18.0 and have it installed successfully.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version:	1.3.9
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.9
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): snap (pre-installed)

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
Precision M5520
│
├─Thunderbolt Controller:
│     Device ID:           d26481a97230d6295224caaf56edb68a671929fd
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     18.05
│     Vendor:              Dell (TBT:0x00D4)
│     GUIDs:               c5364422-2ec3-5092-b30e-03bdb1a53e4a
│                          20d1ca48-b5d7-5144-b162-85258d466e57
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─BC501 NVMe SK hynix 256GB:
│     Device ID:           04e17fcf7d3de91da49a163ffe4907855c3648be
│     Summary:             NVM Express Solid State Drive
│     Current version:     80001C00
│     Vendor:              SK hynix (NVME:0x1C5C)
│     GUIDs:               b637d847-48b0-519f-a4f1-ed2b8535d300
│                          da85e71f-d425-e811-b467-0ed5f89f718b
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─GM107GLM [Quadro M1200 Mobile]:
│     Device ID:           71b677ca0f1bc2c5b804fa1d59e52064ce589293
│     Current version:     a2
│     Vendor:              NVIDIA Corporation (PCI:0x10DE)
│     GUIDs:               5a37b326-73f6-5b5c-b5dc-14cd16efef2f
│                          adf52255-8e5e-50f3-86ff-3efc1210840a
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│   
├─HD Graphics 630:
│     Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
│     Current version:     01
│     Vendor:              Intel Corporation (PCI:0x8086)
│     GUIDs:               b2ef0d75-0a73-5e00-b43d-660bcd0f4004
│                          f744dc01-a247-5543-b9db-2adba3091233
[... 556 lines cut ...]
│       └── zstd.mod
├── initrd.img -> initrd.img-5.4.0-33-generic
├── initrd.img-5.4.0-31-generic
├── initrd.img-5.4.0-33-generic
├── initrd.img.old -> initrd.img-5.4.0-31-generic
├── memtest86+.bin
├── memtest86+.elf
├── memtest86+_multiboot.bin
├── System.map-5.4.0-31-generic
├── System.map-5.4.0-33-generic
├── vmlinuz -> vmlinuz-5.4.0-33-generic
├── vmlinuz-5.4.0-31-generic
├── vmlinuz-5.4.0-33-generic
└── vmlinuz.old -> vmlinuz-5.4.0-31-generic

4 directories, 297 files

```

**Additional questions**
- Operating system and version: Ubuntu 20.04.
- Have you tried rebooting? Yes.
- Is this a regression? Not sure.
- Are you using an NVMe disk? Yes.
- Is secure boot enabled? No.

superm1, 2020-05-31:
Is this a production sample?  You shouldn't be having any firmware that is less than < 1.0.0.  Can you please confirm what is in `/sys/class/dmi/id/bios_version`?  Is it also 0.0.5?

1t5j0y, 2020-06-01:
Picked up this machine second-hand so I'm not 100% sure about the provenance.

```shell
$ cat /sys/class/dmi/id/bios_version
0.0.5
```

superm1, 2020-06-01:
Well it sure sounds like someone sold you a non production sample. Depending upon how old it is it quite possibly can't take a production BIOS. If you still have the opportunity to return it wherever you bought it I would recommend doing so.

If not, does it have a service tag set, serial number anywhere or yellow sticker on it it with any text?

You can check for service tag and serial in /sys/class/dmi/id. You can also look on bottom of the chassis if that doesn't have anything.

If you have a yellow sticker please email me a picture of that sticker (mario.limonciello@dell.com)

1t5j0y, 2020-06-01:
Thanks for the quick feedback and I understand I might be SOL.

There is no yellow sticker but there are some tags and serial numbers set.

```shell
sys/class/dmi/id$ sudo grep ^ /dev/null *_*
bios_date:07/14/2016
bios_vendor:Dell Inc.
bios_version:0.0.5
board_asset_tag:
board_name:01679D
board_serial:/2NMCKC2/CN129636810004/
board_vendor:Dell Inc.
board_version:X01
chassis_asset_tag:
chassis_serial:2NMCKC2
chassis_type:9
chassis_vendor:Dell Inc.
chassis_version:
product_family:
product_name:Precision M5520
product_serial:2NMCKC2
product_sku:07BF
product_uuid:4c4c4544-004e-4d10-8043-b2c04f4b4332
product_version:
sys_vendor:Dell Inc.
```

superm1, 2020-06-01:
Thanks for sharing all that.  I would say at that board level it is _very_ risky to actually flash a production firmware.  You might end up with a non-bootable system that requires a hardware tool to recover or worse; a CPU stepping with incompatible with production firmware.

In that regard you're fortunate that there is happens to be a bug in the preproduction firmware that it had the wrong UEFI ESRT GUID so the firmware distributed from fwupd wouldn't update.  This is confirmed by your above logs referring to ESRT GUID `6180aaaa-5529-4bbf-b4fd-65b4f788de5b`. It's a trade off to be stuck at this pre-production BIOS because you won't have the ability to get firmware security updates either.

If you still intend to continue to use it I would recommend for you to blacklist the `uefi` plugin in `/etc/fwupd/daemon.conf` so that fwupd doesn't continue to try to install the wrong firmware update for it.

closed 2020-06-01 by superm1