fwupd/fwupd #2156: This is perhaps a feature request: document somewhere prominent how to install the Flatpak version

LinuxOnTheDesktop, 2020-06-02:
**Describe the question**
I think it is insufficiently clear how to install (and run) the Flatpak version. Admittedly many distributions have Software Managers that make it somewhat evident how to do that, but why not, on github.com/fwupd/fwupd, provide the terminal commands?

**fwupd version information**
Not applicable.

superm1, 2020-06-02:
I think this is perfect content to be contributed to the wiki: https://github.com/fwupd/fwupd/wiki

Please feel free to create a new wiki page there and link back here.

> EDIT: This is now a support request too. For, I see this - which I did not before, when I used the Snap version of fwupd. (Note: Linux Mint 20 is going to drop Snap.)
>03:04:37:0963 FuEngine             device fb1a5fd6e7d8cbf5d5a2ba7df68aee106ce41027 [Unifying Receiver] does not define an update protocol

I suspect this was happening in snap too, but you didn't notice it because it was in the daemon log.
When you run the snap you're running `fwupdtool` not `fwupdmgr` . `fwupdtool` includes both the daemon (engine) output.

Could you run:
```
sudo flatpak run org.freedesktop.fwupd get-devices --verbose
```

And share the resultant output?

>EDIT: even flatpak run org.freedesktop.fwupd help took over twenty seconds (and quite a few seconds to type!).

That's my experience with the flatpak too.

LinuxOnTheDesktop, 2020-06-02:
Writing a wiki page: I would, but I have little understanding of Flatpak and I am confused by the various names that (to use one of the names) fwupdmgr goes by.

[Here is the output](https://gist.github.com/LinuxOnTheDesktop/71d8bd4af83ff3b381c11c6c369aa85a) you requested (from a different computer to the one I was writing about before, but that computer has pretty much the same setup, including an instance of the same USB receiver; and it has the same problems).

That twenty second delay really is not good enough.

Let me add that the problem with the Logitech receiver means that, given the problems I have with Lenovo firmware updates (which I have reported to the bugtracker), fwupd does not work properly for any of my devices on either of my two machines.

superm1, 2020-06-02:
>Writing a wiki page: I would, but I have little understanding of Flatpak and I am confused by the various names that (to use one of the names) fwupdmgr goes by.

If you can take a stab from whatever you understand that would be appreciated.  Someone who better understands how flatpaks work can certainly fixup anything you get wrong later.

>Here is the output you requested (from a different computer to the one I was writing about before, but that computer has pretty much the same setup, including an instance of the same USB receiver; and it has the same problems).

I think I identified the regression from your debugging log.  I've submitted a pull request.

>That twenty second delay really is not good enough.

I'm sorry; what does this mean?  Good enough for what?

closed 2020-06-02 by superm1

LinuxOnTheDesktop, 2020-06-02:
Wiki: writing a guide on a subject of which I am ignorant would involve much work for me and runs the risk that the result will contain mistakes that people will act upon - or be frustrated by - before those mistakes are fixed.

Not good enough: I suppose I mean, at least, the following. The current situation is lamentable and so should be improved.

Regressions and pull request: good stuff; thank you.

superm1, 2020-06-02:
>Wiki: writing a guide on a subject of which I am ignorant would involve much work for me and runs the risk that the result will contain mistakes that people will act upon - or be frustrated by - before those mistakes are fixed.

FWIW I'm also ignorant in flatpak.  I know just enough to launch it and that it's available for users to try a newer version of the software stack that hasn't been packaged by the distro or in a place that they can't use a snap.
But some documentation on the wiki with your experience is better than no documentation (which is your original point in lack of documentation).

> Not good enough: I suppose I mean, at least, the following. The current situation is lamentable and so should be improved.

This happens to me with all flatpaks I've experimented with, I wouldn't say it's a fwupd bug.

LinuxOnTheDesktop, 2020-06-02:
superm1: thanks for the information. Could you tell me what distros you have encountered the flatpak delay on? If the problem is in flatpak itself, and not some particular distros, then that seems an especially severe - and a rather surprising - problem. I myself use Mint and I have used flatpak, so far as I recall, only for fwupd.

superm1, 2020-06-02:
I encountered it on Ubuntu at some point, but I don't recall what release. I think you can report this long delay behavior to the flatpak repo.

LinuxOnTheDesktop, 2020-06-03:
I am about to file a (new) issue, against flatpak. In order to do that I need to know two things.

**1**) _What is the actual terminal command to install the fwupd/fwupmgr flatpak?_ (In the 'steps to reproduce' section of the bug report I do not wish to tell the developers to use Mint's Software Manager even though that is the method that I had to take.)

**2**) _What is the syntax for running the fwupd flatpak?_ Is it, as I take it to be, `sudo flatpak run org.freedesktop.fwupd <command>`?

superm1, 2020-06-03:
I think it's `flatpak install org.freedesktop.fwupd`?  I don't know off hand if that's all that is needed, like I said i'm not well versed in flatpak.

I think that's the right set of commands to run it, that's what you were using right?

LinuxOnTheDesktop, 2020-06-03:
Thanks, but `flatpak install org.freedesktop.fwupd` is wrong (though the other command/syntax is, I think, right). This is getting frustrating now. I _cannot file a bug report, against flatpak but to do also with fwup, because fwupd does not actually tell anyone anywhere how to use the flatpak version of fwupd_.

While I wait for that basic information, I will add more something, here, from the draft of my bug report for flatpak. For, I have found out more. To wit: _the problem involves authentication_. There is no thirty-second stall when (1) running the help command without sudo (for that fwupd command does not need sudo) or (2) when running the help command, or the 'refresh' command, as root. My authentication log shows this warning from sudo: `pam_unix(sudo:auth) conversation failed`.

superm1, 2020-06-03:
As I just looked up from a search engine myself ("flatpak flathub howto") here's how to use flatpatk's from flathub (start from `Flathub is the Flatpak App Store`)
https://www.omgubuntu.co.uk/2019/02/how-to-install-flatpak-on-ubuntu-flathub

From that part onwards it's identical no matter what OS you're using.

LinuxOnTheDesktop, 2020-06-03:
Thanks, but I saw that page already and it does not help. For I was after a terminal command and the page does not give one. It does direct one to the Flathub website. But using the search on that website for 'fwupd' yields nothing.

superm1, 2020-06-03:
I just followed it myself.

```
$ flatpak remote-add --if-not-exists flathub https://flathub.org/repo/flathub.flatpakrepo
$ flatpak install org.freedesktop.fwupd
Looking for matches…
Found similar ref(s) for ‘org.freedesktop.fwupd’ in remote ‘flathub’ (system).
Use this remote? [Y/n]: y
Required runtime for org.freedesktop.fwupd/x86_64/stable (runtime/org.gnome.Platform/x86_64/3.34) found in remote flathub
Do you want to install it? [Y/n]: y

org.freedesktop.fwupd permissions:
    network     devices     file access [1]     system dbus access [2]

    [1] /boot, /sys, xdg-download
    [2] org.freedesktop.UPower, org.freedesktop.fwupd

        ID                                             Branch            Op           Remote            Download
 1. [|] org.freedesktop.Platform.GL.default            19.08             i            flathub              84.6 MB / 85.1 MB
 2. [ ] org.freedesktop.Platform.openh264              2.0               i            flathub           < 916.8 kB
 3. [ ] org.freedesktop.fwupd.Locale                   stable            i            flathub             < 1.8 MB (partial)
 4. [ ] org.gnome.Platform.Locale                      3.34              i            flathub           < 322.3 MB (partial)
 5. [ ] org.gtk.Gtk3theme.Ambiance                     3.22              i            flathub             < 1.5 MB
 6. [ ] org.gnome.Platform                             3.34              i            flathub           < 320.7 MB
 7. [ ] org.freedesktop.fwupd                          stable            i            flathub             < 3.1 MB

Installing 1/7… ████████████████████ 100%  28.2 MB/s
```

LinuxOnTheDesktop, 2020-06-03:
Thanks very much. I did not see `flatpak install org.freedesktop.fwupd` (or similar) on that page - and on the system I tested it on, _it fails_, thusly:
```
$flatpak remote-add --if-not-exists flathub https://flathub.org/repo/flathub.flatpakrepo
$  flatpak install org.freedesktop.fwupd
Usage:
  flatpak install [OPTION…] LOCATION/REMOTE [REF...] - Install applications or runtimes

Help Options:
  -h, --help              Show help options

Application Options:
  --user                  Work on user installations
  --system                Work on system-wide installations (default)
  --installation=NAME     Work on specific system-wide installation(s)
  --arch=ARCH             Arch to install for
  --no-pull               Don't pull, only install from local cache
  --no-deploy             Don't deploy, only download to local cache
  --no-related            Don't install related refs
  --no-deps               Don't verify/install runtime dependencies
  --no-static-deltas      Don't use static deltas
  --runtime               Look for runtime with the specified name
  --app                   Look for app with the specified name
  --bundle                Assume LOCATION is a .flatpak single-file bundle
  --from                  Assume LOCATION is a .flatpakref application description
  --gpg-file=FILE         Check bundle signatures with GPG key from FILE (- for stdin)
  --subpath=PATH          Only install this subpath
  -y, --assumeyes         Automatically answer yes for all questions
  --reinstall             Uninstall first if already installed
  -v, --verbose           Print debug information during command processing, -vv for more detail
  --ostree-verbose        Print OSTree debug information during command processing

error: REMOTE and REF must be specified
```

superm1, 2020-06-03:
I don't know what is wrong with your remotes.  Based on that help text alone I guess you can try

`flatpak install flathub org.freedesktop.fwupd`

LinuxOnTheDesktop, 2020-06-03:
Thank you. `flatpak install flathub org.freedesktop.fwupd` works. I've filed the bug report [here](https://github.com/flatpak/flatpak/issues/3653).

LinuxOnTheDesktop, 2020-06-14:
The commit does squash the 'does not define an update protocol', which is good.

It remains the case that there is no documentation - other than what is in this issue report - on using the flatpak version of fwupd. (I was asked to write such documentation myself but as I said I am not qualified.)

It remains the case also that - as mentioned above - the flatpak version takes over twenty seconds to start on my computers. Should I open a new issue for that? (I have already, as said above, filed a bug for it against flatpak itself, [here](https://github.com/flatpak/flatpak/issues/3653). There has been no response to that bug report as yet.)

hughsie, 2020-06-15:
> the flatpak version takes over twenty seconds to start on my computers

This isn't expected. Can you attach a `--verbose` log from fwupdtool when using flatpak please? Thanks.

LinuxOnTheDesktop, 2020-06-15:
[Here is a log, generated via the verbose switch,](https://gist.github.com/LinuxOnTheDesktop/eadd9f19649890fd766cd01565191b8a) from one of the computers - call it 'Computer A'.

superm1, 2020-06-15:
https://www.reddit.com/r/linuxmint/comments/cqo6fh/bug_flatpak_on_linux_mint_mate_192_starts_really/ sounds like this

LinuxOnTheDesktop, 2020-06-15:
@superm1 

I think what you've found might allow us to fix this problem. Let me explain.

That bug report - call it [report 1](https://www.reddit.com/r/linuxmint/comments/cqo6fh/bug_flatpak_on_linux_mint_mate_192_starts_really/), itself points to another bug report, call it [report 2](https://github.com/flatpak/flatpak/issues/2843). And reports 1 and 2 alike do sound like my problem. Now, combining those two reports with my situation (I have Mint Cinnamon) reveals that the problem occurs on the following configurations.

| Distribution  | Desktop Environment |
| ------------- | ------------- |
| Arch  | Plasma  |
| Mint  | Mate  |
| Mint  | Cinnamon  |

Given that range of configurations, together with the fact that Plasma fixed the problem by doing what was reported at the end of [here](https://bugs.kde.org/show_bug.cgi?id=406672, I infer that the problem that I have on Cinnamon, and the problem that someone else had on Mate, can be fixed by doing something similar to what the Plasma folks did.

So, should I file a bug against Mint Cinnamon - and Mate, but Mint Mate doesn't have a separate bug tracker - pointing the Mint folks at the fix that Plasma implemented? (I'd summarise here the nature of the fix, but I do not understand the fix well enough to do that.)

superm1, 2020-06-15:
All I know is it's not fwupd's fault. Sorry beyond that I'm not sure.

LinuxOnTheDesktop, 2020-06-15:
I've filed a bug against Mint Cinnamon - as you can see from the 'mention' immediately above this present comment of mine.

LinuxOnTheDesktop, 2020-06-25:
Here is some further information on the long delay before the flatpak version of fwupd starts. As I wrote above, I could open a new issue on this tracker if that would be desirable.

I have gathered the extra information by running the flatpak command - as against the fwupd command that _it_ runs - with a 'verbose' switch.

```
$ /usr/bin/flatpak run --verbose org.freedesktop.fwupd/x86_64/stable
F: No installations directory in /etc/flatpak/installations.d. Skipping
F: Opening user flatpak installation at path /home/nicholas/.local/share/flatpak
F: Opening system flatpak installation at path /var/lib/flatpak
F: Opening user flatpak installation at path /home/nicholas/.local/share/flatpak
F: Opening system flatpak installation at path /var/lib/flatpak
F: Opening user flatpak installation at path /home/nicholas/.local/share/flatpak
F: Opening system flatpak installation at path /var/lib/flatpak
F: Regenerating ld.so.cache /home/nicholas/.var/app/org.freedesktop.fwupd/.ld.so/2991f51303dc4b676bdf3944566183f4ef737152b2738f29a40223ebf1c109a6
F: Running: 'bwrap --args 14 ldconfig -X -C /run/ld-so-cache-dir/2991f51303dc4b676bdf3944566183f4ef737152b2738f29a40223ebf1c109a6'
ldconfig: Can't create temporary cache file /run/ld-so-cache-dir/2991f51303dc4b676bdf3944566183f4ef737152b2738f29a40223ebf1c109a6~: Permission denied
error: ldconfig failed, exit status 256
nicholas@x1 ~ $ sudo /usr/bin/flatpak run --verbose org.freedesktop.fwupd/x86_64/stable
F: No installations directory in /etc/flatpak/installations.d. Skipping
F: Opening user flatpak installation at path /home/nicholas/.local/share/flatpak
F: Opening system flatpak installation at path /var/lib/flatpak
F: Opening user flatpak installation at path /home/nicholas/.local/share/flatpak
F: Opening system flatpak installation at path /var/lib/flatpak
F: Opening user flatpak installation at path /home/nicholas/.local/share/flatpak
F: Opening system flatpak installation at path /var/lib/flatpak
F: Regenerating ld.so.cache /home/nicholas/.var/app/org.freedesktop.fwupd/.ld.so/2991f51303dc4b676bdf3944566183f4ef737152b2738f29a40223ebf1c109a6
F: Running: 'bwrap --args 15 ldconfig -X -C /run/ld-so-cache-dir/2991f51303dc4b676bdf3944566183f4ef737152b2738f29a40223ebf1c109a6'
F: Cleaning up unused container id 3845347325
F: Allocated instance id 4265470028
F: Disallowing ipc access
F: Failed to run in transient scope: No systemd user session available, cgroups not available
F: Running 'bwrap --args 31 /usr/lib/flatpak/flatpak-dbus-proxy --args=33'
F: Running 'bwrap --args 29 /app/bin/fwupdtool'
Command not found
```

PR #2157 (2020-06-02): trivial: logitech_hidpp: set the protocol properly when bootloader un…
…known

This appears to be a regression from c6ae0d998b7e5577b3de4763c25d9d90056bebf1
where the case of bootloader version 0 didn't get protocol set.

Fixes: #2156

Type of pull request:
- [ ] New plugin (Please include [new plugin checklist](https://github.com/hughsie/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit c0a2798f (2020-06-02): trivial: logitech_hidpp: set the protocol properly when bootloader unknown

This appears to be a regression from c6ae0d998b7e5577b3de4763c25d9d90056bebf1
where the case of bootloader version 0 didn't get protocol set.

Fixes: #2156
commit 80fc7e04 (2020-06-02): trivial: logitech_hidpp: set the protocol properly when bootloader unknown

This appears to be a regression from c6ae0d998b7e5577b3de4763c25d9d90056bebf1
where the case of bootloader version 0 didn't get protocol set.

Fixes: #2156
commit 5ad74b25 (2020-06-09): trivial: logitech_hidpp: set the protocol properly when bootloader unknown

This appears to be a regression from c6ae0d998b7e5577b3de4763c25d9d90056bebf1
where the case of bootloader version 0 didn't get protocol set.

Fixes: #2156