fwupd/fwupd #2160: Wiki for "Low host security level" specifies unavailable kernel version

dnoliver, 2020-06-03:
**Describe the bug**
The Wiki page in https://github.com/fwupd/fwupd/wiki/Low-host-security-level recommends updating to the Kernel 5.9 for SPI errors. Kernel 5.8 is currently under development. Specify correct kernel version

```text
SPI: Kernel support not present
The platform SPI chip contains firmware used for initialization and some runtime services. In order to inspect the security configuration of the chip a kernel module is needed that is not present on your system. To fix this, upgrade to kernel 5.9 or later.
```

**Steps to Reproduce**
Steps to reproduce the behavior.

1. Open https://github.com/fwupd/fwupd/wiki/Low-host-security-level 
2. Look at SPI documentation

**Expected behavior**
A clear and concise description of what you expected to happen.

A valid Kernel version should be specified

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
```

N/A

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

N/A

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
```

N/A

**Additional questions**
- Operating system and version: N/A
- Have you tried rebooting? N/A
- Is this a regression? N/A

superm1, 2020-06-03:
The module is expected to be accepted into 5.9 for the next merge window.  That's why it's worded as such.  Feel free to propose another wording for the wiki page (or modify it yourself).

superm1, 2020-06-03:
However... that being said the pci-bcr plugin should have overridden that message coming up.
https://github.com/fwupd/fwupd/blob/master/plugins/pci-bcr/fu-plugin-pci-bcr.c#L74

Is this possibly pre-production hardware that might not meet this criteria?
https://github.com/fwupd/fwupd/blob/master/plugins/pci-bcr/pci-bcr.quirk

superm1, 2020-06-03:
> $ fwupdmgr --version

And yes; I think this is relevant in case you have a checkout that is missing pci-bcr or specifically if this is pre-production hardware and you're missing f6b48edebff00743ce7d6247241fd1a42851a813

dnoliver, 2020-06-03:
OK! My bad! 

I got into that url following hughsie tweet https://twitter.com/hughsient/status/1263429681182912512

I am using the version in https://copr.fedorainfracloud.org/coprs/rhughes/fwupd/

This url redirected me to this repo https://github.com/hughsie/fwupd/issues

```sh
$ rpm -qa fwupd
fwupd-1.5.0-0.585.20200528git.fc32.x86_64

$ fwupdmgr --version
client version: 1.5.0
compile-time dependency versions
        gusb:   0.3.4
        efivar: 37
daemon version: 1.5.0
```

Is there a newer version I can run?

dnoliver, 2020-06-03:
No pre-production hardware, this is a CompuLab Fitlet2 

```sh
$ sudo dmidecode -t system
# dmidecode 3.2
Getting SMBIOS data from sysfs.
SMBIOS 3.0.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: Compulab
        Product Name: fitlet2
        Version: N/A
        Serial Number: 1181004-04904
        UUID: a3129680-daaa-11e8-9b5f-ae83317c3b00
        Wake-up Type: Power Switch
        SKU Number: N/A
        Family: N/A

Handle 0x0022, DMI type 12, 5 bytes
System Configuration Options
        Option 1: N/A

Handle 0x0025, DMI type 32, 20 bytes
System Boot Information
        Status: No errors detected
```

superm1, 2020-06-03:
@hughsie it seems that `fwupdmgr --version` lost the git hash in your RPM build, so it's hard to tell what that contains..  If it's really a build from 05/28 then it _probably_ contains that commit.

> CompuLab Fitlet2

OK so Apollo lake I guess.  Can you share output from 
```shell
fwupdtool security --force --verbose
```

Hopefully it's obvious where this is failing.

dnoliver, 2020-06-03:
Sure!

```sh
$ sudo fwupdtool security --force --verbose
21:17:27:0059 FuDebug              Verbose debugging enabled (on console 1)
21:17:27:0067 FuMain               Failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
21:17:27:0068 FuConfig             loading config values from /etc/fwupd/daemon.conf
21:17:27:0069 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
21:17:27:0070 FuRemoteList         loading /usr/share/fwupd/metainfo
21:17:27:0070 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
21:17:27:0070 XbSilo               file: d8cfcb2a-46a9-dc1f-7ce2-4b884fbf5ae5, current:d8cfcb2a-46a9-dc1f-7ce2-4b884fbf5ae5, cached: (null)
21:17:27:0071 XbSilo               loading silo with file contents
21:17:27:0071 Fwupd                looking for os-release at /etc/os-release
21:17:27:0072 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/dell-esrt.conf
21:17:27:0072 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/lvfs-testing.conf
21:17:27:0073 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/lvfs.conf
21:17:27:0073 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/vendor-directory.conf
21:17:27:0074 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/vendor.conf
21:17:27:0074 FuRemoteList         ordering lvfs-testing=lvfs+1
21:17:27:0075 FuRemoteList         ignoring unfound remote fwupd
21:17:27:0075 FuRemoteList         ignoring unfound remote fwupd
21:17:27:0076 FuRemoteList         ignoring unfound remote fwupd
21:17:27:0076 FuRemoteList         ignoring unfound remote fwupd
21:17:27:0076 Jcat                 ignoring GPG-KEY-Hughski-Limited as not PKCS-7 certificate
21:17:27:0077 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
21:17:27:0077 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
21:17:27:0077 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
21:17:27:0078 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
21:17:27:0079 Jcat                 loaded 1 certificates
21:17:27:0079 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
21:17:27:0079 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
21:17:27:0079 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
21:17:27:0079 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
21:17:27:0079 Jcat                 loaded 1 certificates
21:17:27:0080 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2459 bytes
21:17:27:0080 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
21:17:27:0101 FuEngine             client certificate exists and working
21:17:27:0101 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
21:17:27:0102 FuHistory            got schema version of 5
21:17:27:0102 FuHwids              smbios property Manufacturer=Compulab
21:17:27:0102 FuHwids              smbios property EnclosureKind=3
21:17:27:0102 FuHwids              smbios property Family=N/A
21:17:27:0102 FuHwids              smbios property ProductName=fitlet2
21:17:27:0102 FuHwids              smbios property ProductSku=N/A
21:17:27:0102 FuHwids              smbios property BiosVendor=American Megatrends Inc.
21:17:27:0102 FuHwids              smbios property BiosVersion=FLT2.0.46.01.00
21:17:27:0102 FuHwids              smbios property BiosMajorRelease=05
21:17:27:0102 FuHwids              smbios property BiosMinorRelease=0c
21:17:27:0102 FuHwids              smbios property BaseboardManufacturer=Compulab
21:17:27:0102 FuHwids              smbios property BaseboardProduct=fitlet2
21:17:27:0109 FuQuirks             no /var/lib/fwupd/quirks.d, skipping
21:17:27:0109 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
21:17:27:0109 XbSilo               file: fe9d679c-730c-7b16-305d-398bbda5f8b9, current:fe9d679c-730c-7b16-305d-398bbda5f8b9, cached: (null)
21:17:27:0109 XbSilo               loading silo with file contents
21:17:27:0110 FuEngine             remote lvfs-testing not enabled, so skipping
21:17:27:0110 FuEngine             no /var/lib/fwupd/remotes.d/lvfs/metadata.xml.gz, so skipping
21:17:27:0110 FuEngine             remote vendor not enabled, so skipping
21:17:27:0110 FuEngine             remote vendor-directory not enabled, so skipping
21:17:27:0111 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
21:17:27:0111 XbSilo               0.02ms      parse blob
21:17:27:0111 XbSilo               0.08ms      loaded file
21:17:27:0111 XbSilo               file: 1a6eb95f-85e1-931a-0d92-0f10fe5dacd5, current:1a6eb95f-85e1-931a-0d92-0f10fe5dacd5, cached: (null)
21:17:27:0111 XbSilo               loading silo with file contents
21:17:27:0111 XbSilo               0.01ms      parse blob
21:17:27:0111 XbSilo               0.01ms      query on / with `components/component` limit=0 -> 1 results
21:17:27:0111 FuEngine             1 components now in silo
21:17:27:0111 XbSilo               0.01ms      query on / with `components/component/provides/firmware` limit=0 -> 1 results
[... cut ...]
21:17:28:0677 FuPlugin             performing destroy() on linux_lockdown
21:17:28:0678 FuPlugin             performing destroy() on uefi_dbx
21:17:28:0678 FuPlugin             performing destroy() on dell
21:17:28:0678 FuPlugin             performing destroy() on synaptics_mst
21:17:28:0679 FuPlugin             performing destroy() on redfish
21:17:28:0679 FuPlugin             performing destroy() on logind
21:17:28:0679 FuPlugin             performing destroy() on uefi
21:17:28:0680 FuPlugin             performing destroy() on linux_tainted
```

My HSI is 0 :(

superm1, 2020-06-03:
> ✘ SPI BIOS region:               Unlocked
> ✘ SPI lock:                      Disabled

Ah so it is working properly, it's reading it from pci_bcr, but we haven't updated the wiki page for the fact that we have the pci_bcr plugin to override the kernel lpc lookup method.

But yes, your SPI doesn't appear locked down so malware could overwrite your BIOS.

dnoliver, 2020-06-03:
At least now I know! 

Thank you!

superm1, 2020-06-04:
adjusted wiki for those two errors and dropped the 5.9 reference (since LPC plugin not relevant for now)

closed 2020-06-04 by superm1

dnoliver, 2020-06-15:
A little comment on this,

The BIOS option in the Compulab Fitlet 2 (Vendor American Megatrend Inc) to fix this is located in `Chipset -> South Cluster Configuration -> Miscellaneous Configuration -> BIOS Lock`, set it to `[Enabled]`, and you get HSI-1! :)

With BIOS Lock Enabled, a couple of kernel error logs appears:
a.	intel_telemetry intel_telemetry: TELEMETRY Setup Failed.
b.	intel_telemetry intel_telemetry: IOSS TELEM_INFO Read Failed
c.	pmc-ipc-plat INT34D2:00: IPC timed out, TS=0x24, CMD=0x20eb

But the system seems functional.

As noted before, there is a `✘ TPM PCR0 reconstruction:       Invalid` statement. 
Based on https://github.com/fwupd/fwupd/issues/2020, I should file a bug right? 

Thank you!

superm1, 2020-06-15:
Yes, I think so.