fwupd/fwupd #2164: 1.4.3: install target fails

kloczek, 2020-06-09:
```
+ cd fwupd-1.4.3
+ DESTDIR=/home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64
+ /usr/bin/ninja install -v -j48 -C x86_64-redhat-linux-gnu
ninja: Entering directory `x86_64-redhat-linux-gnu'
[0/1] /usr/bin/meson install --no-rebuild
Installing libfwupd/libfwupd.so.2.0.0 to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64
Installing libfwupd/Fwupd-2.0.gir to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/share/gir-1.0
Installing libfwupd/Fwupd-2.0.typelib to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64/girepository-1.0
Installing libfwupd/fwupd.vapi to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/share/vala/vapi
Installing policy/org.freedesktop.fwupd.policy to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/share/polkit-1/actions
Installing data/remotes.d/org.freedesktop.fwupd.remotes.lvfs.metainfo.xml to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/share/fwupd/metainfo
Installing data/remotes.d/org.freedesktop.fwupd.remotes.lvfs-testing.metainfo.xml to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/share/fwupd/metainfo
Installing data/installed-tests/fakedevice123.cab to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/share/installed-tests/fwupd
Installing data/installed-tests/fakedevice124.cab to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/share/installed-tests/fwupd
Installing libfwupdplugin/libfwupdplugin.so.1.0.0 to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64
Installing libfwupdplugin/FwupdPlugin-1.0.gir to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/share/gir-1.0
Installing libfwupdplugin/FwupdPlugin-1.0.typelib to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64/girepository-1.0
Installing libfwupdplugin/fwupdplugin.vapi to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/share/vala/vapi
Installing src/fwupdmgr to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/bin
Installing src/fwupdagent to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/bin
Installing src/fwupdoffline to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/libexec/fwupd
Installing src/fwupdtool to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/bin
Installing src/fwupdmgr.1 to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/share/man/man1
Installing src/fwupdagent.1 to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/share/man/man1
Installing src/fwupdtool.1 to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/share/man/man1
Installing src/fwupd to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/libexec/fwupd
Installing plugins/ccgx/libfu_plugin_ccgx.so to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64/fwupd-plugins-3
Installing plugins/cpu/libfu_plugin_cpu.so to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64/fwupd-plugins-3
Installing plugins/dfu/libfu_plugin_dfu.so to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64/fwupd-plugins-3
Installing plugins/dfu/dfu-tool to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/bin
Installing plugins/dfu/dfu-tool.1 to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/share/man/man1
Installing plugins/colorhug/libfu_plugin_colorhug.so to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64/fwupd-plugins-3
Installing plugins/ebitdo/libfu_plugin_ebitdo.so to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64/fwupd-plugins-3
Installing plugins/ep963x/libfu_plugin_ep963x.so to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64/fwupd-plugins-3
Installing plugins/fastboot/libfu_plugin_fastboot.so to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64/fwupd-plugins-3
Installing plugins/fresco-pd/libfu_plugin_fresco_pd.so to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64/fwupd-plugins-3
Installing plugins/jabra/libfu_plugin_jabra.so to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64/fwupd-plugins-3
Installing plugins/steelseries/libfu_plugin_steelseries.so to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64/fwupd-plugins-3
Installing plugins/dell-dock/libfu_plugin_dell_dock.so to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64/fwupd-plugins-3
Installing plugins/nitrokey/libfu_plugin_nitrokey.so to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64/fwupd-plugins-3
Installing plugins/rts54hid/libfu_plugin_rts54hid.so to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64/fwupd-plugins-3
Installing plugins/rts54hub/libfu_plugin_rts54hub.so to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64/fwupd-plugins-3
Installing plugins/solokey/libfu_plugin_solokey.so to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64/fwupd-plugins-3
Installing plugins/synaptics-cxaudio/libfu_plugin_synaptics_cxaudio.so to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64/fwupd-plugins-3
Installing plugins/synaptics-prometheus/libfu_plugin_synaptics_prometheus.so to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64/fwupd-plugins-3
Installing plugins/test/libfu_plugin_test.so to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64/fwupd-plugins-3
Installing plugins/test/libfu_plugin_invalid.so to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64/fwupd-plugins-3
Installing plugins/upower/libfu_plugin_upower.so to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64/fwupd-plugins-3
Installing plugins/wacom-usb/libfu_plugin_wacom_usb.so to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64/fwupd-plugins-3
Installing plugins/vli/libfu_plugin_vli.so to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64/fwupd-plugins-3
Installing plugins/ata/libfu_plugin_ata.so to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64/fwupd-plugins-3
Installing plugins/logitech-hidpp/libfu_plugin_logitech_hidpp.so to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64/fwupd-plugins-3
Installing plugins/optionrom/libfu_plugin_optionrom.so to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64/fwupd-plugins-3
Installing plugins/superio/libfu_plugin_superio.so to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64/fwupd-plugins-3
Installing plugins/synaptics-rmi/libfu_plugin_synaptics_rmi.so to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64/fwupd-plugins-3
Installing plugins/thelio-io/libfu_plugin_thelio_io.so to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64/fwupd-plugins-3
Installing plugins/wacom-raw/libfu_plugin_wacom_raw.so to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64/fwupd-plugins-3
Installing plugins/logind/libfu_plugin_logind.so to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64/fwupd-plugins-3
Installing plugins/csr/libfu_plugin_csr.so to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64/fwupd-plugins-3
Installing plugins/tpm/libfu_plugin_tpm.so to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64/fwupd-plugins-3
Installing plugins/tpm-eventlog/libfu_plugin_tpm_eventlog.so to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64/fwupd-plugins-3
Installing plugins/tpm-eventlog/fwupdtpmevlog to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/bin
Installing plugins/tpm-eventlog/fwupdtpmevlog.1 to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/share/man/man1
Installing plugins/emmc/libfu_plugin_emmc.so to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64/fwupd-plugins-3
Installing plugins/nvme/libfu_plugin_nvme.so to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64/fwupd-plugins-3
Installing plugins/modem-manager/libfu_plugin_modem_manager.so to /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/lib64/fwupd-plugins-3
[... cut ...]
    layout = PangoCairo.create_layout(cctx)
  File "/usr/lib64/python3.8/site-packages/gi/module.py", line 131, in __getattr__
    raise AttributeError("%r object has no attribute %r" % (
AttributeError: 'gi.repository.PangoCairo' object has no attribute 'create_layout'
gzip: /home/tkloczko/rpmbuild/BUILDROOT/fwupd-1.4.3-2.fc33.x86_64/usr/share/locale/*/LC_IMAGES/*.bmp: No such file or directory
FAILED: meson-install
/usr/bin/meson install --no-rebuild
ninja: build stopped: subcommand failed.
```

kloczek, 2020-06-09:
set of meson parametrs which I'm using:
```
%build
%meson \
        -D agent=true \
        -D build=all \
        -D consolekit=true \
        -D firmware-packager=true \
        -D gpg=true \
        -D gtkdoc=true \
        -D gudev=true \
        -D introspection=true \
        -D lvfs=true \
        -D man=true \
        -D pkcs7=true \
        -D plugin_altos=true \
        -D plugin_amt=true \
        -D plugin_coreboot=true \
        -D plugin_dell=true \
        -D plugin_dummy=true \
        -D plugin_emmc=true \
        -D plugin_flashrom=true \
        -D plugin_modem_manager=true \
        -D plugin_nvme=false \
        -D plugin_nvme=true \
        -D plugin_nvme=true \
        -D plugin_redfish=true \
        -D plugin_synaptics=true \
        -D plugin_thunderbolt=true \
        -D plugin_tpm=true \
        -D plugin_uefi=true \
        -D systemd=true \
        -D tests=true \
        %{nil}
```

hughsie, 2020-06-09:
> PangoCairo.create_layout(cctx)

This requires cairo 1.10 -- what version do you have installed?

kloczek, 2020-06-09:
cairo 1.16

kloczek, 2020-06-09:
```
+ /usr/bin/meson --buildtype=plain --prefix=/usr --libdir=/usr/lib64 --libexecdir=/usr/libexec --bindir=/usr/bin --sbindir=/usr/sbin --includedir=/usr/include --datadir=/usr/share --mandir=/usr/share/man --infodir=/usr/share/info --localedir=/usr/share/locale --sysconfdir=/etc --localstatedir=/var --sharedstatedir=/var/lib --wrap-mode=nodownload --auto-features=enabled . x86_64-redhat-linux-gnu -D agent=true -D build=all -D consolekit=true -D firmware-packager=true -D gpg=true -D gtkdoc=true -D gudev=true -D introspection=true -D lvfs=true -D man=true -D pkcs7=true -D plugin_altos=true -D plugin_amt=true -D plugin_coreboot=true -D plugin_dell=true -D plugin_dummy=true -D plugin_emmc=true -D plugin_flashrom=true -D plugin_modem_manager=true -D plugin_nvme=false -D plugin_nvme=true -D plugin_nvme=true -D plugin_redfish=true -D plugin_synaptics=true -D plugin_thunderbolt=true -D plugin_tpm=true -D plugin_uefi=true -D systemd=true -D tests=true
The Meson build system
Version: 0.54.2
Source dir: /home/tkloczko/rpmbuild/BUILD/fwupd-1.4.3
Build dir: /home/tkloczko/rpmbuild/BUILD/fwupd-1.4.3/x86_64-redhat-linux-gnu
Build type: native build
Using 'PKG_CONFIG_PATH' from environment with value: ':/usr/lib64/pkgconfig:/usr/share/pkgconfig'
Using 'PKG_CONFIG_PATH' from environment with value: ':/usr/lib64/pkgconfig:/usr/share/pkgconfig'
WARNING: Unknown options: "gpg, pkcs7"
The value of new options can be set with:
meson setup <builddir> --reconfigure -Dnew_option=new_value ...
Project name: fwupd
Project version: 1.4.3
Using 'CC' from environment with value: 'gcc'
Using 'CFLAGS' from environment with value: '-O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -flto=auto -flto-partition=none -Os'
Using 'LDFLAGS' from environment with value: '-Wl,-z,relro -Wl,--as-needed  -Wl,-z,now -specs=/usr/lib/rpm/redhat/redhat-hardened-ld -flto=auto -flto-partition=none -fuse-linker-plugin'
Using 'AR' from environment with value: '/usr/bin/gcc-ar'
Using 'CC' from environment with value: 'gcc'
Using 'CFLAGS' from environment with value: '-O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -flto=auto -flto-partition=none -Os'
Using 'LDFLAGS' from environment with value: '-Wl,-z,relro -Wl,--as-needed  -Wl,-z,now -specs=/usr/lib/rpm/redhat/redhat-hardened-ld -flto=auto -flto-partition=none -fuse-linker-plugin'
C compiler for the host machine: gcc (gcc 10.1.1 "gcc (GCC) 10.1.1 20200507 (Red Hat 10.1.1-1)")
C linker for the host machine: gcc ld.bfd 2.34.0.20200522
Using 'AR' from environment with value: '/usr/bin/gcc-ar'
Host machine cpu family: x86_64
Host machine cpu: x86_64
Program git found: YES (/usr/bin/git)
Compiler for C supports arguments -Waggregate-return: YES
Compiler for C supports arguments -Wunused: YES
Compiler for C supports arguments -Warray-bounds: YES
Compiler for C supports arguments -Wcast-align: YES
Compiler for C supports arguments -Wclobbered: YES
Compiler for C supports arguments -Wdeclaration-after-statement: YES
Compiler for C supports arguments -Wdiscarded-qualifiers: YES
Compiler for C supports arguments -Wduplicated-branches: YES
Compiler for C supports arguments -Wduplicated-cond: YES
Compiler for C supports arguments -Wempty-body: YES
Compiler for C supports arguments -Wformat=2: YES
Compiler for C supports arguments -Wformat-nonliteral: YES
Compiler for C supports arguments -Wformat-security: YES
Compiler for C supports arguments -Wformat-signedness: YES
Compiler for C supports arguments -Wignored-qualifiers: YES
Compiler for C supports arguments -Wimplicit-function-declaration: YES
Compiler for C supports arguments -Wincompatible-pointer-types-discards-qualifiers: NO
Compiler for C supports arguments -Winit-self: YES
Compiler for C supports arguments -Wlogical-op: YES
Compiler for C supports arguments -Wmaybe-uninitialized: YES
Compiler for C supports arguments -Wmissing-declarations: YES
Compiler for C supports arguments -Wmissing-format-attribute: YES
Compiler for C supports arguments -Wmissing-include-dirs: YES
Compiler for C supports arguments -Wmissing-noreturn: YES
Compiler for C supports arguments -Wmissing-parameter-type: YES
Compiler for C supports arguments -Wmissing-prototypes: YES
Compiler for C supports arguments -Wnested-externs: YES
Compiler for C supports arguments -Wno-cast-function-type -Wcast-function-type: YES
Compiler for C supports arguments -Wno-address-of-packed-member -Waddress-of-packed-member: YES
Compiler for C supports arguments -Wno-unknown-pragmas -Wunknown-pragmas: YES
Compiler for C supports arguments -Wno-deprecated-declarations -Wdeprecated-declarations: YES
Compiler for C supports arguments -Wno-missing-field-initializers -Wmissing-field-initializers: YES
Compiler for C supports arguments -Wno-strict-aliasing -Wstrict-aliasing: YES
Compiler for C supports arguments -Wno-suggest-attribute=format -Wsuggest-attribute=format: YES
Compiler for C supports arguments -Wno-unused-parameter -Wunused-parameter: YES
[... cut ...]
Has header "/usr/include/efi/x64/efibind.h" : YES
Message: efi-libdir: "/usr/lib64"
Message: efi-ldsdir: "/usr/lib64/gnuefi"
Message: efi-includedir: "/usr/include/efi"
Program meson_post_install.sh found: YES (/home/tkloczko/rpmbuild/BUILD/fwupd-1.4.3/meson_post_install.sh)
Program makensis found: NO
Build targets in project: 128
WARNING: Project specifies a minimum meson_version '>=0.47.0' but uses features which were added in newer versions:
 * 0.50.0: {'install arg in configure_file'}

Found ninja-1.10.0 at /usr/bin/ninja
```

kloczek, 2020-06-09:
As you see in meson output is:
```
Run-time dependency cairo found: YES 1.16.0
```

hughsie, 2020-06-09:
Hmm, F33 is rawhide -- I just build the actual tarball for f32 just fine.

kloczek, 2020-06-09:
`po/meson.build` has only fragment about installing .bmp files.
```
if get_option('plugin_uefi')
meson.add_install_script('make-images.sh', localedir, python3.path())
endif
```
I don't what  may generate those files

BTW. For what all those .bmp files are used?

hughsie, 2020-06-09:
The BMP are the translated ux capsules the firmware uses. Does make-images also fail on your local machine?

hughsie, 2020-06-11:
> something like this might help

I think we should just fix the broken rawhide build.

hughsie, 2020-06-11:
I've just kicked off pango-1.45.1-2.fc33 into f33.

closed 2020-06-11 by hughsie

kloczek, 2020-06-11:
Issue is that I've reported that issue with pango 1.45.1 installed.

superm1, 2020-06-11:
Richard specifically added a -2 upload of that in F33 according to above comment.  Upgrade to that -2 upload.

kloczek, 2020-06-11:
negative.

Just updated my pango to add thet upstream patch and still during fwupd .bmp files are not generated.

hughsie, 2020-06-11:
Ya, upstream said the same: https://gitlab.gnome.org/GNOME/pango/-/merge_requests/197#note_835459

I'm out of time for this week, but I can try building that updated version in Fedora on Monday if that helps.

kloczek, 2020-06-11:
Looking on fwupd build log I cannot find any traces of running 

As well I cannot find any traces of running make-images script.
```
[tkloczko@barrel x86_64-redhat-linux-gnu]$ grep make-images compile_commands.json .ninja_log build.ninja
[tkloczko@barrel x86_64-redhat-linux-gnu]$
```
And definitelly I'm passing `plugin_uefi=true` in meson params
```
+ /usr/bin/meson --buildtype=plain --prefix=/usr --libdir=/usr/lib64 --libexecdir=/usr/libexec --bindir=/usr/bin --sbindir=/usr/sbin --includedir=/usr/include --datadir=/usr/share --mandir=/usr/share/man --infodir=/usr/share/info --localedir=/usr/share/locale --sysconfdir=/etc --localstatedir=/var --sharedstatedir=/var/lib --wrap-mode=nodownload --auto-features=enabled . x86_64-redhat-linux-gnu -D agent=true -D build=all -D consolekit=true -D firmware-packager=true -D gpg=true -D gtkdoc=true -D gudev=true -D introspection=true -D lvfs=true -D man=true -D pkcs7=true -D plugin_altos=true -D plugin_amt=true -D plugin_coreboot=true -D plugin_dell=true -D plugin_dummy=true -D plugin_emmc=true -D plugin_flashrom=true -D plugin_modem_manager=true -D plugin_nvme=false -D plugin_nvme=true -D plugin_redfish=true -D plugin_synaptics=true -D plugin_thunderbolt=true -D plugin_tpm=true -D plugin_uefi=true -D systemd=true -D tests=true
```

In po/meson.build is only:
```
if get_option('plugin_uefi')
meson.add_install_script('make-images.sh', localedir, python3.path())
endif
```
which is responsible for  installing .bmpfiles but (again) I don't see anything in meson which could fire `po/make-images` on build.

superm1, 2020-06-11:
>which is responsible for installing .bmp files but (again) I don't see anything in meson which could fire po/make-images on build.

As I said above, it (confusingly) only runs during the install step.  That was how @vathpela originally submitted it and how it works today.

kloczek, 2020-06-11:
OK looks like with new pango 1.45.2 now there is no problems.