fwupd/fwupd #2169: fwupd creates duplitate boot entries for Linux-Firmware-Updater on ThinkPad Carbon g7

crosser, 2020-06-11:
**Describe the bug**
On Lenovo ThinkPad Carbon gen 7 running Ubuntu focal, fwupd downloads and prepares installation of updates, but after reboot fwupdx64 does not run.

Ubuntu bug is [here](https://bugs.launchpad.net/ubuntu/+source/fwupd/+bug/1882357)

**Steps to Reproduce**
After fwupd run but before reboot, efi boot table contains two entries for the updater, boot entry #0 being added by fwupdmgr. After reboot there is one entry #2, that was always there. When the existing entry is selected by hand (`sudo efibootmgr -n 2`) then after reboot updater is run and installs updates successfully.

**Expected behavior**
efi boot table should be such as to load fwupdx64 on next boot.

**EFI boot table after running fwupdmgr**
```
BootNext: 0000
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0019,001A,001B,001C,001D,001E,001F,0020,0021,0022,0023,0024,0002,0000
Boot0000* Linux-Firmware-Updater HD(1,GPT,6ccce482-e2c2-48ca-991e-608bee5d38af,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)\.f.w.u.p.d.x.6.4...e.f.i...
Boot0001* ubuntu HD(1,GPT,6ccce482-e2c2-48ca-991e-608bee5d38af,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)
Boot0002* Linux-Firmware-Updater HD(1,GPT,6ccce482-e2c2-48ca-991e-608bee5d38af,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)\.f.w.u.p.d.x.6.4...e.f.i...
Boot0010 Setup FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011 Boot Menu FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012 Diagnostic Splash Screen FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
...
```

**"Normal" EFI boot table**
```
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0019,001A,001B,001C,001D,001E,001F,0020,0021,0022,0023,0024,0002
Boot0001* ubuntu HD(1,GPT,6ccce482-e2c2-48ca-991e-608bee5d38af,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)
Boot0002* Linux-Firmware-Updater HD(1,GPT,6ccce482-e2c2-48ca-991e-608bee5d38af,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)\.f.w.u.p.d.x.6.4...e.f.i...
Boot0010 Setup FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011 Boot Menu FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012 Diagnostic Splash Screen FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
...
```

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
$ fwupdmgr --version
client version:	1.3.9
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.9
```
The version comes standard with ubuntu focal.

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
20QDCTO1WW
│
├─Thunderbolt Controller:
│     Device ID:           f49704bb15e05c3b4e72dc2f0716f3b772c11c33
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     44.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               c7920601-0cda-507f-851e-92229eb1d370
│                          96c6512b-0138-51da-b02f-f44f2c4c7744
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
[... 118 lines cut ...]
│                          • Cryptographic hash verification is available
│   
└─WDC PC SN730 SDBQNTY-512G-1001:
      Device ID:           3743975ad7f64f8d6575a9ae49fb3a8856fe186f
      Summary:             NVM Express Solid State Drive
      Current version:     11130101
      Vendor:              Sandisk Corp (NVME:0x15B7)
      GUIDs:               a39943dd-3afb-54f8-b110-c5a21f071200
                           fccbb6ea-e20e-58ad-bf8a-7fb7d43ff4c2
                           06b4e2aa-91af-508b-b06e-65e3b3189e97
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Supported on remote server
                           • Needs a reboot after installation
                           • Device is usable for the duration of the update
```

**Additional questions**
- Operating system and version:
  ubuntu focal
- Have you tried rebooting?
  you bet!
- Is this a regression?
  I think so. I believe that it worked in eoan, though I am not 100% sure.

hughsie, 2020-06-11:
Could you try with something a little newer, 1.4.x perhaps? We certainly shouldn't have two entries with the same name...

superm1, 2020-06-11:
I did review this in depth as part of a shim bug. 

My conclusions were an intersection of:
* Firmware bug not handling duplicate entries well
* fwupd not discovering the existing entry when there is a hole in the ordering (0000 is empty).

The existing shim bug was:
https://bugs.launchpad.net/ubuntu/+source/shim/+bug/1864223
My particular conclusion:
https://bugs.launchpad.net/ubuntu/+source/shim/+bug/1864223/comments/33

hughsie, 2020-06-12:
> fwupd not discovering the existing entry when there is a hole in the ordering

hughsie, 2020-06-12:
>  fwupd not discovering the existing entry when there is a hole in the ordering

But we could fix this, right?

superm1, 2020-06-12:
I think we probably need a verbose daemon log to show why the 0000 entry is skipped.  I glanced through the code but it wasn't obvious.  Perhaps 0002 is a corrupted entry, or non-obviously something else.

superm1, 2020-06-12:
@crosser can you please repro after you turn on verbose daemon logging:
```shell
fwupdmgr modify-config VerboseDomains '*'
```
and reproduce again and share your journal output for fwupd?

crosser, 2020-06-12:
> and reproduce again and share your journal output for fwupd?
Is there a way to make it modify efi boot table in the absence of new firmwares? If I just run the command efiboot is not modified.
[fwupd-log.txt](https://github.com/fwupd/fwupd/files/4773324/fwupd-log.txt)

superm1, 2020-06-12:
You can use fwupdmgr reinstall

crosser, 2020-06-15:
Situation with duplicate boot entries did happen several times in the past months, every time when new firmware updates arrived. But after I ran `fwupdmgr modify-config VerboseDomains '*'` and after that `fwupdmgr reinstall` behaviour was different. Of several candidate updates thar were presented, some, when selected, did not resulted in any changes in the boot table (i.e. there was no "BootNext" and no new entry for the updater), and one did result in "correct" behaviour: no new boot entry appeared, and BootNext was set to 2 (existing updater entry). After reboot updater did run as it should.

It might be that fwupd was upgraded with regular focal upgrades between the last faulty run and these days. (I see that it is going to be upgraded right now to version 1.3.9-4ubuntu0.1) 

I don't know what to make of it. And I will certainly document proceedings next time when a "real" update arrives.

Will it be useful to check if behaviour changes if `VerboseDomains '*'` is reveted?

superm1, 2020-06-16:
I don't think changing verbose domains will impact this at all.  That just turns on additional logging, it behaves the same.

>I don't know what to make of it. And I will certainly document proceedings next time when a "real" update arrives.

Doesn't make sense to me either.  Reinstall runs the same code as update.

Is it possible that this was with two updates being performed simultaneously?  Like a BIOS update and ME update in the same go?  And that's the only time you reproduced it previously?

crosser, 2020-06-16:
> Is it possible that this was with two updates being performed simultaneously? Like a BIOS update and ME update in the same go? And that's the only time you reproduced it previously?

Yes, to the best of my remembrance, _every_ time I was prompted to install updates since I got this notebook, there were more than one item. Conversely, when I run `fwupdmgr reinstall` now, it prompts me to select one of possible items:

```
crosser@journex:~$ sudo fwupdmgr update
• Thunderbolt Controller has the latest available firmware version
• Prometheus has no available firmware updates
• Prometheus IOTA Config has no available firmware updates
• WDC PC SN730 SDBQNTY-512G-1001 has the latest available firmware version
crosser@journex:~$ sudo fwupdmgr reinstall
Choose a device:
0.	Cancel
1.	f49704bb15e05c3b4e72dc2f0716f3b772c11c33 (Thunderbolt Controller)
2.	f09b748a98b31bb5c7c64d83f24ddbc80dff7c5f (System Firmware)
3.	f5275562e72a2f6cb4b09e559ff4e2860eff55d8 (UEFI Device Firmware)
4.	c085a3cbf4dc1d60d110b7616816e2ffd279b64e (UEFI Device Firmware)
5.	3743975ad7f64f8d6575a9ae49fb3a8856fe186f (WDC PC SN730 SDBQNTY-512G-1001)
```
and when I select most of them, that does not even lead to modification of the boot table! Before run:

```
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0019,001A,001B,001C,001D,001E,001F,0020,0021,0022,0023,0024,0002
Boot0001* ubuntu	HD(1,GPT,6ccce482-e2c2-48ca-991e-608bee5d38af,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)
Boot0002* Linux-Firmware-Updater	HD(1,GPT,6ccce482-e2c2-48ca-991e-608bee5d38af,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)\.f.w.u.p.d.x.6.4...e.f.i...
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
```
* Selecting 5: successful update, no change in efiboot
* Selecting 4: "is not updatable"
* Selecting 3: "is not updatable"
* Selecting 1: successful update, no change in efiboot
* Selecting 2: "is not updatable" now, but it agreed to update yesterday and _did result in successful reflash_, with BootNext pointing to entry 2.

Note that the version of efiboot did change since yesterday, as I said.

superm1, 2020-06-16:
I think considering this, we'll probably need to wait if and when you can reproduce it again when a few updates come out simultaneously for more details.  I would suggest leaving the verbose logging on so that you're prepared to provide the relevant logging.

crosser, 2020-10-12:
It reproduced again.
I am running groovy,
crosser@journex:~$ dpkg --list|grep fwupd
ii  fwupd                                        1.4.5-1                                    amd64        Firmware update daemon
ii  fwupd-signed                                 1.30+1.4.5-1                               amd64        Linux Firmware Updater EFI signed binary
ii  fwupdate                                     12-7                                       amd64        Transitional package for fwupd
ii  libfwupd2:amd64                              1.4.5-1                                    amd64        Firmware update daemon library
ii  libfwupdplugin1:amd64                        1.4.5-1                                    amd64        Firmware update daemon plugin library

Today new firmware updates arrived:

    3 devices have a firmware upgrade available.
    Run `fwupdmgr get-upgrades` for more information.

Boot table before running fwupdmgr:

crosser@journex:~$ efibootmgr -v|head
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0019,001A,001B,001C,001D,001E,001F,0020,0021,0022,0023,0024,0002
Boot0001* ubuntu        HD(1,GPT,6ccce482-e2c2-48ca-991e-608bee5d38af,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)
Boot0002* Linux-Firmware-Updater        HD(1,GPT,6ccce482-e2c2-48ca-991e-608bee5d38af,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)\.f.w.u.p.d.x.6.4...e.f.i...
Boot0010  Setup FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu     FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen      FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics    FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Regulatory Information        FvFile(478c92a0-2622-42b7-a65d-5894169e4d24)

Running fwupdmgr:

crosser@journex:~$ sudo fwupdmgr update
• Thunderbolt Controller has the latest available firmware version
Upgrade available for Embedded Controller from 0.1.17 to 0.1.20
20QDCTO1WW must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: 
X11 connection rejected because of wrong authentication.
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating Embedded Controller…************************************]
Scheduling…              [***************************************]
Successfully installed firmware
Upgrade available for Intel Management Engine from 192.47.1524 to 192.64.1551
20QDCTO1WW must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: 
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating Intel Management Engine…********************************]
Scheduling…              [***************************************]
Successfully installed firmware
• Prometheus has the latest available firmware version
• Prometheus IOTA Config has the latest available firmware version
Upgrade available for System Firmware from 0.1.33 to 0.1.38
20QDCTO1WW must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: 
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating System Firmware…[***************************************]
Scheduling…              [***************************************]
Successfully installed firmware
• UEFI Device Firmware has no available firmware updates
• UEFI Device Firmware has no available firmware updates
• WDC PC SN730 SDBQNTY-512G-1001 has the latest available firmware version

An update requires a reboot to complete. Restart now? [y|N]: 

Boot table before answering to this last question above:

crosser@journex:~$ efibootmgr -v|head
BootNext: 0000
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0019,001A,001B,001C,001D,001E,001F,0020,0021,0022,0023,0024,0002,0000
Boot0000* Linux-Firmware-Updater	HD(1,GPT,6ccce482-e2c2-48ca-991e-608bee5d38af,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)\.f.w.u.p.d.x.6.4...e.f.i...
Boot0001* ubuntu	HD(1,GPT,6ccce482-e2c2-48ca-991e-608bee5d38af,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)
Boot0002* Linux-Firmware-Updater	HD(1,GPT,6ccce482-e2c2-48ca-991e-608bee5d38af,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)\.f.w.u.p.d.x.6.4...e.f.i...
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)

You can see that there are two entries with the updater, and BootNext points to the entry 0.
After reboot, updater was not called, and boot table reverted to default state (entry 0000 disappeared).
After manually selecting NextBoot 2 (`sudo efibootmgr -n 2`), I have a "good" state of the boot table:

crosser@journex:~$ efibootmgr -v|head
BootNext: 0002
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0019,001A,001B,001C,001D,001E,001F,0020,0021,0022,0023,0024,0002
Boot0001* ubuntu	HD(1,GPT,6ccce482-e2c2-48ca-991e-608bee5d38af,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)
Boot0002* Linux-Firmware-Updater	HD(1,GPT,6ccce482-e2c2-48ca-991e-608bee5d38af,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)\.f.w.u.p.d.x.6.4...e.f.i...
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)

and after reboot updater ran and updates were installed.

crosser, 2020-10-12:
> I don't think changing verbose domains will impact this at all. That just turns on additional logging, it behaves the same.
>
>
> Doesn't make sense to me either. Reinstall runs the same code as [...]
Looks like indeed there was more than one update this time.

crosser, 2020-10-12:
Verbose log as requested:
[fwupd.log.txt](https://github.com/fwupd/fwupd/files/5367626/fwupd.log.txt)

hughsie, 2020-10-14:
@superm1 I still wonder whether we should avoid ever writing to `Boot0000`  -- it's seems special in some way...

superm1, 2020-10-14:
To me it sounds like a firmware bug on duplicate handling that we stroke by having two entries.  Why did we insert at 0000 when there was a 0002 already?  If we get that sorted we will avoid the firmware duplicate handling bug.

superm1, 2020-10-15:
@crosser can you please create an archive of your `/sys/firmware/efi/efivars/Boot*` variables from before an update and share it?  I expect that we can reproduce the behavior of this bug by doing something like this:

```
 sudo FU_PATH_KIND_SYSFSDIR_FW=/home/supermario/foo/ EFIVARFS_PATH=/home/supermario/foo/efi/efivars/ fwupdtool  install ~/Downloads/xps_13_7390_1_6_0.cab --allow-reinstall  --verbose
```

crosser, 2020-10-15:
@superm1 do you mean when I see two entries for the updater? I guess that we'd need to wait for the next occasion when I get more than one update?..  Then I can do that.

superm1, 2020-10-15:
No I mean when there is only one entry and the missing Boot0000. I think that's what the situation is right now after you done your most recent update right?

crosser, 2020-10-15:
There you are:
[efibootvars.cpio.gz](https://github.com/fwupd/fwupd/files/5387290/efibootvars.cpio.gz)
(tar does not work complaining that it cannot stat() the files)

superm1, 2020-10-15:
I tested with my side and that set of files and unfortunately it didn't reproduce.

```
sudo FU_PATH_KIND_SYSFSDIR_FW=/home/supermario/efivar-issue EFIVARFS_PATH=/home/supermario/efivar-issue/efi/efivars/ fwupdtool  install ~/Downloads/xps_13_7390_1_6_0.cab --allow-reinstall  --verbose
```
This re-wrote `Boot0002` (to match my HDD path) and set `BootNext` for me.

```
$ git status
On branch master
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
	modified:   efi/efivars/Boot0002-8be4df61-93ca-11d2-aa0d-00e098032b8c

Untracked files:
  (use "git add <file>..." to include in what will be committed)
	efi/efivars/BootNext-8be4df61-93ca-11d2-aa0d-00e098032b8c
```

So I tried to overwrite `Boot0002` with a valid `Linux-Firmware-Updater` entry from my system.
This still didn't put something into `Boot0000`.  It just updated `BootNext` for me.

```
$ git status
On branch master
Untracked files:
  (use "git add <file>..." to include in what will be committed)
	efi/efivars/BootNext-8be4df61-93ca-11d2-aa0d-00e098032b8c

nothing added to commit but untracked files present (use "git add" to track)
```

superm1, 2020-10-15:
And i'm also running Ubuntu focal, so it should be same efivar (37-2ubuntu2).

@hughsie any more ideas?  I'm really not sure why a new entry is made for @crosser.  I would expect the [code](https://github.com/fwupd/fwupd/blob/master/plugins/uefi/fu-uefi-bootmgr.c#L105) for  `efi_get_next_variable_name` to be deterministic, especially with the same set of data on both our sides.

superm1, 2020-10-15:
Could this maybe be some funny handling related to `Linux-Firmware-Updater` and `Linux Firmware Updater`?

hughsie, 2020-10-15:
> I tested with my side and that set of files and unfortunately it didn't reproduce.

Can you get a `--verbose` -- e.g. I was expecting to see stuff like `description does not match`

superm1, 2020-10-15:
With my local one (with Linux Firmware Updater and my HDD) dropped in place I see:
```
21:04:02:0216 FuPluginUefi         no /usr/local/share/locale/en_US/LC_IMAGES/fwupd-1920-1080.bmp.gz found
21:04:02:0216 FuCommon             reading /usr/local/share/locale/en/LC_IMAGES/fwupd-1920-1080.bmp.gz with 4147 bytes
21:04:02:0216 FuPluginUefi         decompressed image to 56kb
21:04:02:0217 FuPluginUefi         DP type:0x04 subtype:0x01 size:0x002a
21:04:02:0217 FuPluginUefi         DP type:0x04 subtype:0x04 size:0x0080
21:04:02:0217 FuPluginUefi         DP type:0x7f subtype:0xff size:0x0004
21:04:02:0230 FuDevice             installing onto b6c08fb9e5384d9d101853cc1ca20cf0ce2df2e2:
FuFirmware:
  FuFirmwareImage:
  Data:                 0xbb25c2

21:04:02:0230 FuPluginUefi         ESRT matches payload GUID
21:04:02:0230 FuCommon             writing /boot/efi/EFI/ubuntu/fw/fwupd-cbe49cca-492b-93e8-b0f4-f693501f271b.cap with 12264898 bytes
21:04:02:0246 FuPluginUefi         DP type:0x04 subtype:0x01 size:0x002a
21:04:02:0246 FuPluginUefi         DP type:0x04 subtype:0x04 size:0x0080
21:04:02:0246 FuPluginUefi         DP type:0x7f subtype:0xff size:0x0004
21:04:02:0263 FuPluginUefi         description does not match
21:04:02:0263 FuPluginUefi         description does not match
21:04:02:0263 FuPluginUefi         description does not match
21:04:02:0263 FuPluginUefi         description does not match
21:04:02:0264 FuPluginUefi         description does not match
21:04:02:0264 FuPluginUefi         description does not match
21:04:02:0264 FuPluginUefi         description does not match
21:04:02:0264 FuPluginUefi         description does not match
21:04:02:0264 FuPluginUefi         description does not match
21:04:02:0264 FuPluginUefi         description does not match
21:04:02:0264 FuPluginUefi         description does not match
21:04:02:0264 FuPluginUefi         description does not match
21:04:02:0265 FuPluginUefi         description does not match
21:04:02:0265 FuPluginUefi         description does not match
21:04:02:0265 FuPluginUefi         description does not match
21:04:02:0265 FuPluginUefi         description does not match
21:04:02:0265 FuPluginUefi         description does not match
21:04:02:0265 FuPluginUefi         description does not match
21:04:02:0265 FuPluginUefi         description does not match
21:04:02:0265 FuPluginUefi         description does not match
21:04:02:0265 FuPluginUefi         description does not match
21:04:02:0266 FuPluginUefi         description does not match
21:04:02:0266 FuPluginUefi         description does not match
21:04:02:0266 FuPluginUefi         description does not match
21:04:02:0266 FuHistory            modifying device System Firmware [b6c08fb9e5384d9d101853cc1ca20cf0ce2df2e2]
21:04:02:0268 FuEngine             attach -> FuUefiDevice:
```

superm1, 2020-10-15:
That debugging is terrible, I added some more clear debugging.  With @crosser's content and this patch applied to force the legacy path like he uses:
```
$ git diff
diff --git a/plugins/uefi/fu-plugin-uefi.c b/plugins/uefi/fu-plugin-uefi.c
index 0be6e161..77e1a8d3 100644
--- a/plugins/uefi/fu-plugin-uefi.c
+++ b/plugins/uefi/fu-plugin-uefi.c
@@ -513,7 +513,7 @@ fu_plugin_uefi_coldplug_device (FuPlugin *plugin, FuUefiDevice *dev, GError **er
        /* if not already set by quirks */
        if (fu_device_get_custom_flags (FU_DEVICE (dev)) == NULL) {
                /* for all Lenovo hardware */
-               if (fu_plugin_check_hwid (plugin, "6de5d951-d755-576b-bd09-c5cf66b27234")) {
+               if (fu_plugin_check_hwid (plugin, "65faf55d-732d-51b1-ba28-c6b863aab9af")) {
                        fu_device_set_custom_flags (FU_DEVICE (dev), "use-legacy-bootmgr-desc");
                        fu_plugin_add_report_metadata (plugin, "BootMgrDesc", "legacy");
                }
```

I get this, which is still not pointing at the root cause but I think will be helpful to solving it.
```
21:33:01:0068 FuPluginUefi         Boot001C -> 'NVMe1' : does not match
21:33:01:0068 FuPluginUefi         Boot001F -> 'USB HDD' : does not match
21:33:01:0068 FuPluginUefi         Boot0026 -> 'IDER BOOT Floppy' : does not match
21:33:01:0068 FuPluginUefi         Boot0011 -> 'Boot Menu' : does not match
21:33:01:0068 FuPluginUefi         Boot0018 -> 'MEBx Hot Key' : does not match
21:33:01:0069 FuPluginUefi         Boot0020 -> 'PXE BOOT' : does not match
21:33:01:0069 FuPluginUefi         Boot001B -> 'NVMe0' : does not match
21:33:01:0069 FuPluginUefi         Boot0010 -> 'Setup' : does not match
21:33:01:0069 FuPluginUefi         Boot0023 -> 'Other CD' : does not match
21:33:01:0069 FuPluginUefi         Boot0012 -> 'Diagnostic Splash Screen' : does not match
21:33:01:0069 FuPluginUefi         Boot0016 -> 'Startup Interrupt Menu' : does not match
21:33:01:0069 FuPluginUefi         Boot0015 -> 'ThinkShield secure wipe' : does not match
21:33:01:0069 FuPluginUefi         Boot0024 -> 'Other HDD' : does not match
21:33:01:0070 FuPluginUefi         Boot001A -> 'USB FDD' : does not match
21:33:01:0070 FuPluginUefi         Boot0017 -> 'Rescue and Recovery' : does not match
21:33:01:0070 FuPluginUefi         Boot001D -> 'ATA HDD0' : does not match
21:33:01:0070 FuPluginUefi         Boot0014 -> 'Regulatory Information' : does not match
21:33:01:0070 FuPluginUefi         Boot0022 -> 'LENOVO CLOUD' : does not match
21:33:01:0070 FuPluginUefi         Boot0019 -> 'USB CD' : does not match
21:33:01:0070 FuPluginUefi         Boot0028 -> 'ATAPI CD' : does not match
21:33:01:0070 FuPluginUefi         Boot0027 -> 'ATA HDD' : does not match
21:33:01:0070 FuPluginUefi         Boot0021 -> 'HTTPS BOOT' : does not match
21:33:01:0071 FuPluginUefi         Boot0001 -> 'ubuntu' : does not match
21:33:01:0071 FuPluginUefi         Boot0013 -> 'Lenovo Diagnostics' : does not match
21:33:01:0071 FuPluginUefi         Boot0002 -> 'Linux-Firmware-Updater' : updating existing boot entry
```

superm1, 2020-10-15:
@crosser after that's merged and the snap builds, would you be able to download/try the edge snap with it integrated?  I think we can get a lot more detail if so.

The steps to do so will be:
1. `sudo apt purge fwupd`
2. `sudo snap install fwupd --edge`
3. `sudo fwupd.fwupdtool reinstall f09b748a98b31bb5c7c64d83f24ddbc80dff7c5f  --daemon-verbose=FuPluginUefi`

And share the output from that last one.  That should get us the output from a reproduce with extra debugging in place.

hughsie, 2020-10-16:
> Boot0002 -> 'Linux-Firmware-Updater' : updating existing boot entry

If @crosser gets the same output as that, then the only thing that this can be is a firmware bug, surely?!

superm1, 2020-10-16:
I don't expect that same output, in all his examples a new Boot0000 was getting created. So I'm hoping we have a reason why.

crosser, 2020-10-16:
> > Boot0002 -> 'Linux-Firmware-Updater' : updating existing boot entry
> 
> If @crosser gets the same output as that, then the only thing that this can be is a firmware bug, surely?!

The problem is, for me, a _second_ entry for Linux-Firmware-Updater is created by by fwupdmgr, but _only_ when more than one update is in the queue to be installed (apparently).

I will try to run what @superm1 asks, perhaps tomorrow. (However my systems are de-snap'ed, I'll need to either rebuild fwupd or temporarily install snapd)

superm1, 2020-10-16:
Hmm, if it's only when more than one update in the queue to install, I don't have a way to simulate that myself and the proposal I had with reinstall wouldn't be able to either.

crosser, 2020-10-16:
Maybe a temporary change can be made in fwupdmgr that would allow to reinstall multiple updates in one go? Then I could try to run it and hopefully reproduce the situation.

superm1, 2020-10-16:
@hughsie considering it only happens when two updates are being run my suspicion is that it's because the second update calls `efi_get_next_variable_name` but that's actually iterating on the same GUID.  So it already "passed" `0002` and thinks that it needs to make a new one - `0000`.

I think we need to not call `fu_uefi_bootmgr_bootnext` for both devices, just once.

superm1, 2020-10-16:
Yup, looking through the efivar code, this makes sense to me now.  There is a global `dir` variable that only gets reset when the `close_dir` "destructor" is called.
https://github.com/rhboot/efivar/blob/07fbba7acf9c757a3ae8192a64cb3a1f631c5f74/src/generics.h#L19

`close_dir` is a static function that we don't get a chance to call from fwupd side.
https://github.com/rhboot/efivar/blob/07fbba7acf9c757a3ae8192a64cb3a1f631c5f74/src/generics.h#L110

And so with this patch (which simulates calling to create boot entries twice)
```
diff --git a/plugins/uefi/fu-uefi-device.c b/plugins/uefi/fu-uefi-device.c
index f35964b6..a79c0aa9 100644
--- a/plugins/uefi/fu-uefi-device.c
+++ b/plugins/uefi/fu-uefi-device.c
@@ -598,6 +598,8 @@ fu_uefi_device_write_firmware (FuDevice *device,
                bootmgr_desc = "Linux-Firmware-Updater";
        if (!fu_uefi_bootmgr_bootnext (esp_path, bootmgr_desc, flags, error))
                return FALSE;
+       if (!fu_uefi_bootmgr_bootnext (esp_path, bootmgr_desc, flags, error))
+               return FALSE;

        /* success! */
        return TRUE;
```

I finally reproduce the bug.
```
15:14:59:0725 FuPluginUefi         Boot001C -> 'NVMe1' : does not match
15:14:59:0725 FuPluginUefi         Boot001F -> 'USB HDD' : does not match
15:14:59:0725 FuPluginUefi         Boot0026 -> 'IDER BOOT Floppy' : does not match
15:14:59:0725 FuPluginUefi         Boot0011 -> 'Boot Menu' : does not match
15:14:59:0725 FuPluginUefi         Boot0018 -> 'MEBx Hot Key' : does not match
15:14:59:0726 FuPluginUefi         Boot0020 -> 'PXE BOOT' : does not match
15:14:59:0726 FuPluginUefi         Boot001B -> 'NVMe0' : does not match
15:14:59:0726 FuPluginUefi         Boot0010 -> 'Setup' : does not match
15:14:59:0726 FuPluginUefi         Boot0023 -> 'Other CD' : does not match
15:14:59:0727 FuPluginUefi         Boot0012 -> 'Diagnostic Splash Screen' : does not match
15:14:59:0727 FuPluginUefi         Boot0016 -> 'Startup Interrupt Menu' : does not match
15:14:59:0727 FuPluginUefi         Boot0015 -> 'ThinkShield secure wipe' : does not match
15:14:59:0727 FuPluginUefi         Boot0024 -> 'Other HDD' : does not match
15:14:59:0727 FuPluginUefi         Boot001A -> 'USB FDD' : does not match
15:14:59:0728 FuPluginUefi         Boot0017 -> 'Rescue and Recovery' : does not match
15:14:59:0728 FuPluginUefi         Boot001D -> 'ATA HDD0' : does not match
15:14:59:0728 FuPluginUefi         Boot0014 -> 'Regulatory Information' : does not match
15:14:59:0728 FuPluginUefi         Boot0022 -> 'LENOVO CLOUD' : does not match
15:14:59:0728 FuPluginUefi         Boot0019 -> 'USB CD' : does not match
15:14:59:0729 FuPluginUefi         Boot0028 -> 'ATAPI CD' : does not match
15:14:59:0729 FuPluginUefi         Boot0027 -> 'ATA HDD' : does not match
15:14:59:0729 FuPluginUefi         Boot0021 -> 'HTTPS BOOT' : does not match
15:14:59:0729 FuPluginUefi         Boot0001 -> 'ubuntu' : does not match
15:14:59:0729 FuPluginUefi         Boot0013 -> 'Lenovo Diagnostics' : does not match
15:14:59:0730 FuPluginUefi         Boot0002 -> 'Linux-Firmware-Updater' : updating existing boot entry
15:14:59:0733 FuPluginUefi         Boot001E -> 'ATA HDD1' : does not match
15:14:59:0734 FuPluginUefi         Boot0025 -> 'IDER BOOT CDROM' : does not match
15:14:59:0734 FuPluginUefi         Boot0000 -> creating new entry
```

hughsie, 2020-10-16:
>  There is a global dir variable that only gets reset when the close_dir "destructor" is called.

I **did not** think of that... Is there any way we can use our efivar parsing code rather that the one in efivars?

superm1, 2020-10-16:
It would be new code to walk the directory basically.  That's certainly one way.  I just sent up a PR for another way to approach this.

hughsie, 2020-10-16:
> It would be new code to walk the directory basically

Would you mind if I tried this on Monday? I don't trust efi_get_next_variable_name() and would rather put an `interator in fu-efivar.c` that we can control the lifetime for.

superm1, 2020-10-16:
> > It would be new code to walk the directory basically
> 
> Would you mind if I tried this on Monday? I don't trust efi_get_next_variable_name() and would rather put an `interator in fu-efivar.c` that we can control the lifetime for.

Sounds fine be me.

hughsie, 2020-10-16:
@superm1 I've pushed something like what I meant to https://github.com/fwupd/fwupd/pull/2485 -- 100% untested.

commit ac53f60a (2020-10-16): uefi: Don't overwrite BootNext and create new Boot#### when installing 2 FW

Fixes: #2169
closed 2020-10-19 by hughsie