fwupd/fwupd #2173: Vendor object wrong for NVM Express Solid State Drive on X1 carbon ( 6th Gen )

johannbg, 2020-06-12:
**Describe the bug**
See  Summary and below

**Steps to Reproduce**
Step 1 obtain Lenovo ThinkPad X1 Carbon 6th with UMIS NVME SSD 
Step 2 run fwupdmgr get-devices
Observe output for string  "LENSE30512GMSP34MEAT3TA" 
...
Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Vendor:               Lenovo <--- Wrong 
  VendorId:             NVME:0x17AA
  Version:              2.5.0412 
  VersionFormat:        plain
  Icon:                 drive-harddisk

**Expected behavior**
Expected to see UMIS/Ramaxel as vendor for string "LENSE30512GMSP34MEAT3TA" and I would also expect the version being matched to the one on Lenovo's site and in general it would be helpful to see the Issue Date and or Issue Date from Lenovo and from FWUP to see when Lenovo releases updates and FWUP actually offered them ( fwupd does not appear to have the last NVME firmware update "No upgrades for LENSE30512GMSP34MEAT3TA, current is 2.5.0412: 2.5.0412=same" ) . 

Release  Rev   Issue Date
-------  ---   ----------
FWNVA39w 01   2020/06/08 <--- ( critical fix for nvme issued on June 8th ) 

fwup shows 
Version:              2.5.0412
Obviously 2.5.0412 != FWNVA39w so where does the "Version: 2.5.0412" string come from? 

Expected output...
Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Vendor:               UMIS/Ramaxel <--- What I expected 
  VendorId:             NVME:0x17AA
  Version:              2.5.0412 <--- Should be FWNVA39w right? 
  VersionFormat:        plain
  Icon:                 drive-harddisk

fwupd version
client version:	1.2.12
compile-time dependency versions
	gusb:	0.3.1
	efivar:	37
daemon version:	1.2.12

Add info...
Here are the vendor strings for  Lenovo ThinkPad X1 Carbon 6th ( obtained from [1] ) and presumably fwupd wants to deal with these per real vendors as in the actual company that manufactures it since there is high probably that those NVME are being shipped with other Laptops as well ( Dell,HP, Asus etc. ) and the same firmware bugs and updates existing in all of them right.
 
 Toshiba
           THNSN5512GPU7 TOSHIBA
           THNSF5256GCJ7 TOSHIBA
           THNSF5256GPU7 TOSHIBA
           THNSF5256GPUK TOSHIBA
           THNSF5512GPUK TOSHIBA
           KBG30ZMT128G TOSHIBA
           KBG30ZMT256G TOSHIBA
           KBG30ZMT512G TOSHIBA
           KXG5AZNV256G TOSHIBA
           KXG5AZNV512G TOSHIBA
           KXG5AZNV1T02 TOSHIBA
           KXG5APNV2T04 TOSHIBA
           KXG6AZNV256G TOSHIBA
           KXG6AZNV512G TOSHIBA
           KXG6AZNV1T02 TOSHIBA
           KXG6APNV2T04 TOSHIBA
           KBG40ZMT128G TOSHIBA
           KBG40ZNT256G TOSHIBA
           KBG40ZNT512G TOSHIBA
           KBG40ZNT1T02 TOSHIBA

  Samsung
           SAMSUNG MZVKV512HAJH-000L1
           SAMSUNG MZVPV256HDGL-000L7
           SAMSUNG MZVLW256HEHP-000L7
           SAMSUNG MZVLW512HMJP-000L7
           SAMSUNG MZVLW1T0HMLH-000L7
           SAMSUNG MZVLB256HAHQ-000L7
           SAMSUNG MZVLB512HAJQ-000L7
           SAMSUNG MZVLB1T0HALR-000L7
           SAMSUNG MZVLB2T0HMLB-000L7
           SAMSUNG MZVLB256HBHQ-000L7
           SAMSUNG MZVLB512HBHQ-000L7
           SAMSUNG MZVLB1T0HBLR-000L7
           SAMSUNG MZVLB2T0HBLB-000L7
           SAMSUNG MZVPW256HEGLL-000L7
           SAMSUNG MZVKW512HMJPL-000L7
           SAMSUNG MZALQ128HBHQ-000L1
           SAMSUNG MZALQ256HAJD-000L1
           SAMSUNG MZALQ512HALU-000L1
           SAMSUNG MZALQ1T0HALB-000L1

  Intel
           Intel SSDPEKKF512G7L
           Intel SSDPEKKF360G7L
           Intel SSDPEKKF256G7L
           Intel SSDPEKKF256G8L
           Intel SSDPEKKF512G8L
           Intel SSDPEKKF010T8L
           Intel HBRPEKNX0203AL
           Intel HBRPEKNX0203AL0

  UMIS
           LENSE20512GMSP34MEAT2TA
           LENSE20256GMSP34MEAT2TA
           UMIS RPJTJ256MED1OWX
           UMIS RPJTJ128MED1MWX
           LENSE30256GMSP34MEAT3TA
           LENSE30512GMSP34MEAT3TA
           LENSE40256GMSP34MESTB3A
           LENSE40512GMSP34MESTB3A
           UMIS LENSE30256GMSP34MEAT3TA
           UMIS LENSE30512GMSP34MEAT3TA
           UMIS LENSE40256GMSP34MESTB3A
           UMIS LENSE40512GMSP34MESTB3A

  WD
           WDC PC SN720 SDAQNTW-256G-1001
           WDC PC SN720 SDAQNTW-512G-1001
           WDC PC SN720 SDAQNTW-1T00-1001
           WDC PC SN520 SDAPMUW-128G-1001
           WDC PC SN520 SDAPMUW-256G-1001
           WDC PC SN520 SDAPMUW-512G-1001
           WDC PC SN730 SDBQNTY-256G-1001
           WDC PC SN730 SDBQNTY-512G-1001
           WDC PC SN730 SDBQNTY-1T00-1001

  Lite-On
           LITEON CA3-8D256
           LITEON CA3-8D512
|          LITEON CA5-8D256
|          LITEON CA5-8D512

  SSSTC
|          SSSTC CA5-8D256-Q79
|          SSSTC CA5-8D512-Q79

  SK hynix
           HFM128GDHTNG-8310B
           HFM256GDHTNG-8310B
           HFM128GDHTNG-8510B
           HFM256GDHTNG-8510B
           HFM512GDHTNG-8510B
           SKHynix_HFS256GD9TNG-L5B0B
           SKHynix_HFS512GD9TNG-L5B0B
           SKHynix_HFS001TD9TNG-L5B0B

1. https://download.lenovo.com/pccbbs/mobiles/fwnva39w.txt

hughsie, 2020-06-12:
>  Vendor: Lenovo <--- Wrong

Nah, it is actually seemingly Lenovo branded / custom firmware.

> it would be helpful to see the Issue Date and or Issue Date from Lenovo and from FWUP to see when Lenovo releases updates and FWUP actually offered them

This is something Lenovo has access to on the LVFS, but isn't public for commercial reasons I'm afraid.

> current is 2.5.0412: 2.5.0412=same

I'm afraid I can't comment on what updates are not yet available on the LVFS. If there's a newer version on their homepage then I guess someone has a red checkbox on their mdsync page...

johannbg, 2020-06-12:
Given the recent love Lenovo has shown RH and Ubuntu I would have expected them doing a better job supporting what matters ;) You can close this but there are two other issues I have noticed one being minor cosmetic one the Vendor string being displayed as Lenovo and LENOVO <-- all caps( in System Firmware ) and lack of firmware update support for the Synaptics Touchpad and the TrackPoint Firmware which I could file as a separate bugs for or skipped if those are WONTFIX as well

hughsie, 2020-06-12:
> the Vendor string being displayed as Lenovo and LENOVO

There's "Lenovo Inc.` too IIRC. @superm1 do you think we should try to sanitize things like this?

> and lack of firmware update support for the Synaptics Touchpad

I can't comment on future firmware, but I will say there's been a lot of interest in the synaptics RMI4 plugin in fwupd recently. I don't know if that covers TrackPoint too, I don't know how that stuff works under the covers.

superm1, 2020-06-12:
>There's "Lenovo Inc.` too IIRC. @superm1 do you think we should try to sanitize things like this?

I think we should show whatever is in SMBIOS, not try to adjust it because we don't like how it looks when we render it.  There must be a reason that a vendor put it in all caps no?

johannbg, 2020-06-12:
Probably human error or legacy relic. these types of input mismatch usually are. If fwupd is just fetching that string then the bug is not in fwupd and thus should not be workaround in fwupd. Does the fwupd team have some magical powers within  Lenovo which manages to have them fix it on their side and or explain why it is like it is or is that a dead end?

superm1, 2020-06-12:
>If fwupd is just fetching that string then the bug is not in fwupd and thus should not be workaround in fwupd

Yes my point exactly.

johannbg, 2020-06-12:
Yeps closing this issue.

closed 2020-06-12 by johannbg