fwupd/fwupd #2176: Error opening directory “/sys/firmware/efi/esrt/entries”: No such file or directory

morfyum, 2020-06-12:
**Describe the bug**
"esrt" directory not exist!
Error opening directory “/sys/firmware/efi/esrt/entries”: No such file or directory
Fedora 32 - Updated from Fedora 31

**Steps to Reproduce**
I dont know, i try to reinstall, reinstall ok, but:
Failed to try-restart pesign.service: Unit pesign.service not found.
this service is doesnt exist too - Real service is: "fwupd.service"

**Expected behavior**
Work fwupd.service, need esrt directory

**fwupd version information**
```
client version:	1.4.2
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.4.2
```
Install:
``` 
dnf install fwupd
```
Reinstall:
```
dnf reinstall fwupd
```

**fwupd device information**
Dell Inspiron 5000
 - i5 5200U
 - Nvidia 920M

```shell
Inspiron 5558
│
├─GK208BM [GeForce 920M]:
│     Device ID:           8e92eda1ab953dd9f27d11cd320c36e5d1127030
│     Current version:     a1
│     Vendor:              NVIDIA Corporation (PCI:0x10DE)
│     GUIDs:               485ad3be-a50c-5b74-b1fb-e82632dba0ce ← PCI\VEN_10DE&DEV_1299&REV_A1
│                          6e57f226-8eab-53b0-9d19-32ff0ff13119 ← PCI\VEN_10DE&DEV_1299
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│   
├─HD Graphics 5500:
│     Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
│     Current version:     09
│     Vendor:              Intel Corporation (PCI:0x8086)
│     GUIDs:               19e1f63e-bef0-5582-b236-364069bcd0d5 ← PCI\VEN_8086&DEV_1616&REV_09
│                          64c0ea55-d861-5dd0-aa6b-bcc307f57ed4 ← PCI\VEN_8086&DEV_1616
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│   
├─Intel(R) Core™ i5-5200U CPU @ 2.20GHz:
│     Device ID:           4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:     0x2e
│     Vendor:              GenuineIntel
│     GUID:                b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│     Device Flags:        • Internal device
│   
├─SA400S37480G:
│     Device ID:           602b0a6cc821d155208724f0e22f8d111542b74c
│     Summary:             ATA Drive
│     Current version:     SBFKK1B3
│     Vendor:              Kingston (ATA:0x2646)
│     Serial Number:       50026B7380694679
│     GUIDs:               d8e02563-67f0-5da4-b388-e3b18a5f0dd3 ← IDE\KINGSTON_SA400S37480G___________________SBFKK1B3
│                          1ab2d344-65ac-5bc5-adbd-9aab1dfc2be3 ← IDE\0KINGSTON_SA400S37480G___________________
│                          b4b189f9-ed32-5438-a5e4-d3313483647a ← KINGSTON SA400S37480G
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           123fd4143619569d8ddb6ea47d1d3911eb5ef07a
│     Current version:     A18
│     Vendor:              Dell Inc. (DMI:Dell Inc.)
│     Update Error:        UEFI Capsule updates not available or enabled
│     GUID:                230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│     Device Flags:        • Internal device
│                          • Requires AC power
│                          • Needs a reboot after installation
│   
└─Touchpad:
      Device ID:           6fa19a8e88ffa967ed25bded1be1ad977bd7a197
      Current version:     1.3.1832337
      Bootloader Version:  54.0.0
      Vendor:              Synaptics (HIDRAW:0x06CB)
      GUIDs:               3b044239-5bdf-532e-8d4c-15dafd842fcb ← HIDRAW\VEN_06CB&DEV_75BF&REV_00
                           30a712f5-60e1-5ea3-b222-343a99141590 ← HIDRAW\VEN_06CB&DEV_75BF
                           df4f83f6-859f-5bbf-830c-d0bb1a13948a ← SYNAPTICS_RMI\TM3014-003
                           296ceaa3-24b0-56dc-8a7e-04ca67bae54a ← SYNAPTICS_RMI\TM3014
      Device Flags:        • Internal device
                           • Updatable
```

**Additional questions**
- Operating system and version:
Fedora 32 - updated from Fedora 31
- Have you tried rebooting?
Yes
- Is this a regression?
I dont understand the question, maybe Nothing.
But before F31 or F30 fwupdate is work for me :/

xvitaly, 2020-06-12:
>  Failed to try-restart pesign.service: Unit pesign.service not found.

This is a known issue: https://bugzilla.redhat.com/show_bug.cgi?id=1776680

superm1, 2020-06-12:
>"esrt" directory not exist!
> Error opening directory “/sys/firmware/efi/esrt/entries”: No such file or directory

You aren't booted into UEFI mode or your system doesn't support UEFI capsule updates.

Also mentioned in "Update Error":
```
│     Update Error:        UEFI Capsule updates not available or enabled
```

closed 2020-06-12 by superm1

LinuxOnTheDesktop, 2020-07-02:
I think this issue needs re-opening and fixing. Here is why.

@superm1: are you saying that seeing the 'error [. .] no such file of directory' message is not a bug? If you are, I do not see why (and I too see that message). For, if the message is reporting that something is inapplicable to the user's system, then, while I can just about understand reporting that to the user, why report it as an *error*? So doing looks bad, seems to misrepresent as an error something that is not an error, and generally causes confusion.

superm1, 2020-07-02:
We've actually gotten that feedback recently and made changes in the 1.5.x branch as such.  In 1.5.x if you're not in UEFI mode the uefi plugin won't run, but instead a "bios" plugin will run indicating what's going on:
https://github.com/fwupd/fwupd/blob/master/plugins/bios/fu-plugin-bios.c#L81