fwupd/fwupd #2178: GTKDoc Helper Script Error During Installation (Master)

moriel5, 2020-06-15:
Distro: Solus

**Describe the bug**
While compiling FWUpD from master, I had encountered no errors (I did make sure that there all required dependencies are installed, some of which I had to compile myself), though oddly enough, Meson did not recognize the installed `gcab` development files (gcab-devel is the package here), and just compiled `gcab` as a submodule. Ninja did not spit any errors either.

However, when attempting to install via "ninja -C build install", the following error is spit out (Sensitive information was replaced by "***"):
"Running custom install script '/usr/bin/meson --internal gtkdoc --sourcedir=/***/moriel/.build/fwupd --builddir=/home/***/.build/fwupd/build --subdir=subprojects/gcab/docs/reference --headerdirs=/home/***/.build/fwupd/libgcab@@/home/***/.build/fwupd/build/libgcab --mainfile=gcab-docs.sgml --modulename=gcab --moduleversion= --mode=auto --gtkdoc-scan=/usr/bin/gtkdoc-scan --gtkdoc-scangobj=/usr/bin/gtkdoc-scangobj --gtkdoc-mkdb=/usr/bin/gtkdoc-mkdb --gtkdoc-mkhtml=/usr/bin/gtkdoc-mkhtml --gtkdoc-fixxref=/usr/bin/gtkdoc-fixxref --content-files= --cc=cc --ld=cc --cflags=-Waggregate-return -Wunused -Warray-bounds -Wcast-align -Wclobbered -Wduplicated-branches -Wduplicated-cond -Wempty-body -Wformat=2 -Wformat-nonliteral -Wformat-security -Wformat-signedness -Wignored-qualifiers -Wimplicit-function-declaration -Winit-self -Wlogical-op -Wmissing-declarations -Wmissing-format-attribute -Wmissing-include-dirs -Wmissing-noreturn -Wmissing-parameter-type -Wmissing-prototypes -Wnested-externs -Wno-discarded-qualifiers -Wno-missing-field-initializers -Wno-strict-aliasing -Wno-suggest-attribute=format -Wno-unused-parameter -Wnull-dereference -Wold-style-definition -Woverride-init -Wpointer-arith -Wredundant-decls -Wreturn-type -Wshadow -Wsign-compare -Wstrict-aliasing -Wstrict-prototypes -Wswitch-default -Wtype-limits -Wundef -Wuninitialized -Wunused-but-set-variable -Wunused-variable -Wwrite-strings -fstack-protector-strong'
Building documentation for gcab
ERROR: Error in gtkdoc helper script:

ERROR: ['/usr/bin/gtkdoc-scan', '--module=gcab', '--source-dir=/home/***/.build/fwupd/libgcab', '--source-dir=/home/***/.build/fwupd/build/libgcab'] failed with status 1
Traceback (most recent call last):
  File "/usr/bin/gtkdoc-scan", line 59, in <module>
    scan.Run(options)
  File "/usr/share/gtk-doc/python/gtkdoc/scan.py", line 200, in Run
    ScanHeaders(dir, section_list, decl_list, get_types, seen_headers, options)
  File "/usr/share/gtk-doc/python/gtkdoc/scan.py", line 343, in ScanHeaders
    for file in sorted(os.listdir(source_dir)):
FileNotFoundError: [Errno 2] No such file or directory: '/home/***/.build/fwupd/libgcab'"

The funny thing is, FWUpD seems to have installed, as I have no issues invoking it to see my devices (my laptop is a Lenovo IdeaPad 300-15ISK, so no support for LVFS, even if it were still supported by Lenovo (it was literally dropped after slightly over half a year), and also Gnome-Firmware (also compiled by me) seems to see it, and work properly (I have nothing to test it out for firmware flashing, so could be placebo), except for one thing (though that may be due to my laptop not being on LVFS).

Whenever I attempt to verify the installed motherboard firmware's cryptographic signature, I always get the following error:
failed to verify System Firmware: no results for XPath query 'components/component/provides/firmware[@type='flashed'][text()='12b99262-648c-4365-bafd-eb869fb7eb47']/../../releases/release'
This happens from both Gnome-Firmware and the terminal.

superm1, 2020-06-23:
I hit this too.  The same error happens with gusb.  It's related to subprojecting and hardcoding paths.  Although this happens in fwupd, it is a bug in those projects.  You should report it to those projects.

Something like this is needed to fix it (untested):
```
/src/fwupd/subprojects/gusb$ git diff
diff --git a/docs/api/meson.build b/docs/api/meson.build
index 1bdcb1d..95e94a3 100644
--- a/docs/api/meson.build
+++ b/docs/api/meson.build
@@ -7,7 +7,7 @@ gusb_version_xml = configure_file(
 gnome.gtkdoc(
   'gusb',
   src_dir : [
-    join_paths(meson.source_root(), 'gusb'),
+    join_paths(meson.current_source_dir(), '..', '..', '..', 'gusb'),
     join_paths(meson.build_root(), 'gusb'),
   ],
   main_sgml : 'gusb-docs.sgml',
```

superm1, 2020-06-23:
To work around it for now you can turn off the subproject's gtkdoc strings.
`meson build -Dgusb:docs=false -Dgcab:docs=false`

closed 2020-06-23 by superm1

hughsie, 2020-06-24:
-    join_paths(meson.source_root(), 'gusb'),
    +    join_paths(meson.current_source_dir(), '..', '..', '..', 'gusb'),

Using `..`. in meson is discouraged; usually you just create a dir() and then reference that. I'll talk to the maintainer, I hear he's a really nice guy. :)

moriel5, 2020-06-28:
Thanks, I'll test this later this week, when I have time.

I guess the cryptographic signature issue is a separate issue?