fwupd/fwupd #2179: fwupd does not cleanup files in /boot/efi

xclaesse, 2020-06-15:
My /boot/efi partition is only 50M because before fwupd existed that partition only contained a few text files. Fortunately a firmware update seems to take only 17M in `/boot/efi/EFI/ubuntu/fw` so it can fit and the update worked. But when I got a 2nd update it seems previous files where not deleted and downloading a 2nd firmware exceeds the space available on that partition. gnome-software gives an error but with no way or clue how to fix it. Manually deleting `/boot/efi/EFI/ubuntu/fw` folder fixed the error and the firware update succeeded.

I'm not sure who should be responsible for deleting old files, fwupd or gnome-software?

superm1, 2020-06-15:
What version of fwupd do you have installed?

The old files are supposed to be cleaned up when a new update is run:
https://github.com/fwupd/fwupd/blob/master/plugins/uefi/fu-uefi-device.c#L449

xclaesse, 2020-06-15:
1.3.9-4ubuntu0.1 on Ubuntu 20.04.

xclaesse, 2020-06-15:
When is that function supposed to be called? Before downloading an update, or after having installed the update? Because those files are still there on my system:
```
ls /boot/efi/EFI/ubuntu/fw
fwupd-39f17d37-5d18-4e72-934c-226ba948743e.cap
fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap
```

superm1, 2020-06-15:
It's _supposed_ to get called next time you try to run an update.  The general flow is like this:

1. Download CAB update to home directory
2. Extract update to temporary directory.
3. Clean up old updates on ESP
4. Copy CAP file to the ESP.
5. Schedule for next reboot
6. Reboot machine
7. Install update
8. Boot back into OS

Are you saying that you're having a problem with that flow because of the small ESP? If so we might have a bug still.
If you can please turn on verbose logging in the daemon and show the output when trying to install the second update that would be appreciated.

xclaesse, 2020-06-15:
Those steps make sense, as long as steps 3 and 4 does not get inverted by mistake? 

Unfortunately the update is done now because I deleted files manually. I'll only be able to give logs when/if lenovo release a new firmware for my x395.

xclaesse, 2020-06-15:
@superm1 is there a debug tool or something I can use to simulare running step 3?

xclaesse, 2020-06-15:
ooohhhh, reading fu_uefi_device_prepare() it checks free space before cleanup.

superm1, 2020-06-15:
You can use `fwupdmgr reinstall` to re-run the whole process.

superm1, 2020-06-15:
> ooohhhh, reading fu_uefi_device_prepare() it checks free space before cleanup.

Oh the implication being that because of the small ESP, there will be a low amount of free space and that fails before cleanup?  So swapping those two in prepare should fix your problem?

xclaesse, 2020-06-15:
That  would be my guess, yes.

xclaesse, 2020-06-15:
Yes, confirmed:

> `fwupdmgr reinstall`

/boot/efi does not have sufficient space, required 33,6 MB, got 22,8 MB

superm1, 2020-06-15:
OK, just sent up a PR for this.

superm1, 2020-06-15:
This fix won't land until fwupd 1.5.0, so until then I would recommend you manually configure `/etc/fwupd/uefi.conf` to change `RequireESPFreeSpace` to something small to hopefully override this check.

closed 2020-06-15 by superm1

reilg, 2022-03-17:
Hello, has this been confirmed to work? I'm running fwupd 1.5.11 and I'm getting a similar error.

```bash
$ sudo fwupdmgr update
... snip
/boot/efi does not have sufficient space, required 33.6 MB, got 32.0 MB
```

xclaesse, 2022-03-17:
Pretty sure it's fixed, yes. You can delete manually `/boot/efi/EFI/ubuntu/fw` (or whatever directory your firmware gets stored) and see if you have enough free space. But if your partition is too small for a single fw then there is nothing fwupd can do unfortunately.

reilg, 2022-03-17:
Oh I think I get it now. 

Apologies, this isn't a fwupd issue, it's more of a dual booting Windows on a Dell laptop issue >.<.
Both occupy, 60+MB on `/boot/efi/EFI` which, along with the ubuntu folder, leaves 32MB on the partition. 

Thanks for the swift reply.

PR #2180 (2020-06-15): uefi: check for free space after cleaning up ESP
In a very small ESP situation it's possible that the amount of free
space is insufficient until it's actually been cleaned.

Fixes: #2179

Type of pull request:
- [ ] New plugin (Please include [new plugin checklist](https://github.com/hughsie/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit 2caea542 (2020-06-15): uefi: check for free space after cleaning up ESP

In a very small ESP situation it's possible that the amount of free
space is insufficient until it's actually been cleaned.

Fixes: #2179
commit 5a675cb2 (2020-06-18): uefi: check for free space after cleaning up ESP

In a very small ESP situation it's possible that the amount of free
space is insufficient until it's actually been cleaned.

Fixes: #2179
commit 8c7d7c4f (2020-06-18): uefi: check for free space after cleaning up ESP

In a very small ESP situation it's possible that the amount of free
space is insufficient until it's actually been cleaned.

Fixes: #2179