fwupd/fwupd #2181: TPM PCR0 reconstruction invalid on CompuLab Fitlet2

dnoliver, 2020-06-15:
**Describe the bug**
A clear and concise description of what the bug is.

After running `fwupdmgr security --force`, on a CompuLab Fitlet2, a wild `✘ TPM PCR0 reconstruction: Invalid` appears!

**Steps to Reproduce**
Steps to reproduce the behavior.

1. Run fwupdmgr security --force on a CompuLab Fitlet2
2. check the HIS-2 tests, PCR0 reconstruction is invalid

**Expected behavior**
A clear and concise description of what you expected to happen.

I have been testing TPM based attestation systems for a while on this devices.
I believe I was always able to reconstruct the TPM PCR 0 value from the TPM Event Log in `/sys/kernel/security/tpm0/binary_bios_measurements`, (but I may be wrong of course).

When I run this tool, it tells me that my PCR0 Reconstruction is bad, which impacts negatively in my level of trust in the TPM PCR 0, TPM Event Log, fwupdmgr, and my previous assessments. One of those may be wrong! 

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
```

```shell
$ fwupdmgr --version
client version: 1.5.0
compile-time dependency versions
        gusb:   0.3.4
        efivar: 37
daemon version: 1.5.0
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

```shell
dnf copr enable rhughes/fwupd
dnf update
dnf update fwupd
```

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
```

```shell
$ fwupdmgr get-devices --show-all-devices
fitlet2
│
├─Celeron N3350/Pentium N4200/Atom E3900 Series Integrated Graphics Controller:
│     Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
│     Current version:     0b
│     Vendor:              Intel Corporation (PCI:0x8086)
│     GUIDs:               7ee0c805-ca7f-51ea-8bbd-e20e623b9323
│                          8257b546-4d2d-5c32-adc5-fc54b8fdfe53
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│   
├─Intel(R) Atom™ Processor E3950 @ 1.60GHz:
│     Device ID:           4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:     0x38
│     Vendor:              GenuineIntel
│     GUID:                b9a2dd81-159e-5537-a7db-e7101d164d3f
│     Device Flags:        • Internal device
│   
├─TPM:
│ │   Device ID:           c6a80ac3a22083423992a3cb15018989f37834d6
│ │   Summary:             TPM 2.0 Device
│ │   Current version:     301.9.0.0
│ │   Vendor:              Intel (TPM:INTC)
│ │   GUIDs:               ff71992e-52f7-5eea-94ef-883e56e034c6
│ │                        34801700-3a50-5b05-820c-fe14580e4c2d
│ │                        8e1cbc5d-5a11-5149-bfea-b6065d5296ba
│ │                        03f304f4-223e-54f4-b2c1-c3cf3b5817c6
│ │                        52d7b679-db28-5bf7-bd87-41d77aeec600
│ │   Device Flags:        • Internal device
│ │ 
│ └─Event Log:
│       Device ID:         58bd405f31c48e6eca290b425f530a94c91e955c
│       GUID:              a25657fe-b5dc-5be0-8b78-8b9dfec678ff
│       Device Flags:      • Internal device
│     
├─TS128GMTS600:
│     Device ID:           c46944fe84f2b057692258cc87a812cc86364881
│     Summary:             ATA Drive
│     Current version:     P1225CH1
│     Vendor:              Transcend (ATA:0x8564)
│     GUIDs:               6001e5d6-33a9-5b6c-9884-b461cdc2cf32
│                          920180a6-f62c-564a-b6d2-1fb6d7decd3e
│                          e25079dc-149b-5eb7-aee9-c7ebe27e4fd3
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─Unifying Receiver:
      Device ID:           5a3b142f3f6762434b567a7321f9db3846c1997a
      Summary:             A miniaturised USB wireless receiver
      Current version:     RQR12.03_B0025
      Bootloader Version:  BOT01.02_B0015
      Vendor:              USB:0x046D
      Install Duration:    30 seconds
      GUIDs:               9d131a0c-a606-580f-8eda-80587250b8d6
                           fcf55bf5-767b-51ce-9c17-f6f538c4ee9f
                           279ed287-3607-549e-bacc-f873bb9838c4
      Device Flags:        • Updatable
```

**Additional questions**
- Operating system and version: Fedora Server 32
- Have you tried rebooting?: Yes
- Is this a regression? Unknown

superm1, 2020-06-15:
Can you please share the output of the tool `fwupdtpmevlog`?

dnoliver, 2020-06-15:
```shell
$ sudo fwupdtpmevlog
PCR:                     BIOS (0)
Type:                    0x8
Description:             EV_S_CRTM_VERSION
ChecksumSha1:            d4720b4009438213b803568017f903093f6bea8ab47d283db32b6eabedbbf155
BlobStr:                 ..kT..U@..N....:

PCR:                     BIOS (0)
Type:                    0x1
Description:             EV_POST_CODE
ChecksumSha1:            405e3f51e650cee3194c57cbfc3fed6ac0ef8d00199f5e96468f226109c3bc97
BlobStr:                 .........p@.....

PCR:                     BIOS (0)
Type:                    0x4
Description:             EV_SEPARATOR
ChecksumSha1:            df3f619804a92fdb4057192dc43dd748ea778adc52bc498ce80524c014b81119

PCR:                     BIOS Configuration (1)
Type:                    0x4
Description:             EV_SEPARATOR
ChecksumSha1:            df3f619804a92fdb4057192dc43dd748ea778adc52bc498ce80524c014b81119

PCR:                     BIOS Configuration (1)
Type:                    0x80000002
Description:             EV_EFI_VARIABLE_BOOT
ChecksumSha1:            4913464729685ea14c61b2a2ed2c8e4f0e9a377d4083d85a657edcd04d4c2e6a
BlobStr:                 a.............+.................B.o.o.t.O.r.d.e.r.....

PCR:                     BIOS Configuration (1)
Type:                    0x80000002
Description:             EV_EFI_VARIABLE_BOOT
ChecksumSha1:            3f95768698c72d865ebb10b839d12e458661e0a516f50e5e9ba694abc6843a9b
BlobStr:                 a.............+.........t.......B.o.o.t.0.0.0.1.....\.F.e.d.o.r.a.....*.....................1.-.y..F.....f........\.E.F.I.\.F.E.D.O.R.A.\.S.H.I.M...E.F.I.........BO

PCR:                     BIOS Configuration (1)
Type:                    0x80000002
Description:             EV_EFI_VARIABLE_BOOT
ChecksumSha1:            0e15a20294e0fb29b42ce11852327856f3cd51b11b311f83e02b7f14e410fa47
BlobStr:                 a.............+.........v.......B.o.o.t.0.0.0.0.....b.F.e.d.o.r.a.....*.....................1.-.y..F.....f......4.\.E.F.I.\.F.E.D.O.R.A.\.S.H.I.M.X.6.4...E.F.I.......

PCR:                     Option ROMs (2)
Type:                    0x4
Description:             EV_SEPARATOR
ChecksumSha1:            df3f619804a92fdb4057192dc43dd748ea778adc52bc498ce80524c014b81119

PCR:                     Option ROM configuration (3)
Type:                    0x4
Description:             EV_SEPARATOR
ChecksumSha1:            df3f619804a92fdb4057192dc43dd748ea778adc52bc498ce80524c014b81119

PCR:                     Initial program loader code (4)
Type:                    0x4
Description:             EV_SEPARATOR
ChecksumSha1:            df3f619804a92fdb4057192dc43dd748ea778adc52bc498ce80524c014b81119

PCR:                     Initial program loader code (4)
Type:                    0x80000003
Description:             EV_BOOT_SERVICES_APPLICATION
ChecksumSha1:            0ce02100f67c7ef85f4eed368f02bf7092380a3c23ca91fd7f19430d94b00c19
BlobStr:                 .p.s.....y..............x............A........................*.....................1.-.y..F.....f........\.E.F.I.\.F.E.D.O.R.A.\.S.H.I.M...E.F.I.......

PCR:                     Initial program loader code (4)
Type:                    0x80000003
Description:             EV_BOOT_SERVICES_APPLICATION
ChecksumSha1:            083259d7862f1555a15cd762fe18eec7559868fc8b4ac10b8a18fc5a8400ff16
BlobStr:                 .................................@&..P&.<

PCR:                     Initial program loader code (4)
[... 346 lines cut ...]
ChecksumSha1:            93f85bb90def39349edbaf383556416edd76354b44eee297081426bca4a448f1
BlobStr:                 (hd0,gpt2)/loader/entries//72ef17eb92b94306bbafba79dafa7bf9-5.6.15-300.fc32.x86_64.conf.

PCR:                     Static operating system (9)
Type:                    0xd
ChecksumSha1:            0ea9b3c694824934bc566883e81c5b72bfc7c709a980d02579a7f399360c8a06
BlobStr:                 (hd0,gpt2)/vmlinuz-5.6.15-300.fc32.x86_64.

PCR:                     Static operating system (9)
Type:                    0xd
ChecksumSha1:            8197c2418fac14d357d6b1b13be4cb826da16f0187ef2d29d4a1331707bf48c0
BlobStr:                 (hd0,gpt2)/initramfs-5.6.15-300.fc32.x86_64.img.

Reconstructed PCRs:
  0:                     a3723cf09d8a9ef4578375ddea45d6394a5968aa
  1:                     e7150eb8c3c96b3d66bef8a2f8bb37dd41f1c362
  2:                     3a4285c323c5af121f270900e88ffeefb2a34233
  3:                     3a4285c323c5af121f270900e88ffeefb2a34233
  4:                     c2fb76d175813e9142227b1a83811248bcb7e6db
  5:                     3bf14dc42126c2c6b90566b73847c5e00e10d021
  6:                     3a4285c323c5af121f270900e88ffeefb2a34233
  7:                     9af4db3ea6321f463cb9b3d9a7b6299206c33009
  8:                     21ed08e2c88193823c0d7f3528fc3677ecb37c71
  9:                     3f77928e59256513a5fbb58368a027494775e58a
```

superm1, 2020-06-15:
> I believe I was always able to reconstruct the TPM PCR 0 value from the TPM Event Log in /sys/kernel/security/tpm0/binary_bios_measurements, (but I may be wrong of course).

Can you show your work in trying to reconstruct?  It's of course possible there is a fwupd bug in the implementation too.

Since you're missing the UEFI device, also can you share what the actual PCR0 measurement was on the TPM?  If we do indeed have a bug here with the reconstruction code, that would be needed to confirm.

dnoliver, 2020-06-15:
Sharing the TPM values reported with tpm2_pcrread, and the reconstructed pcrs reported by fwupdmgr

I wanted to share all the PCR values, because none of the matches :)

```shell
Reconstructed PCRs:
  0:                     a3723cf09d8a9ef4578375ddea45d6394a5968aa
  1:                     e7150eb8c3c96b3d66bef8a2f8bb37dd41f1c362
  2:                     3a4285c323c5af121f270900e88ffeefb2a34233
  3:                     3a4285c323c5af121f270900e88ffeefb2a34233
  4:                     c2fb76d175813e9142227b1a83811248bcb7e6db
  5:                     3bf14dc42126c2c6b90566b73847c5e00e10d021
  6:                     3a4285c323c5af121f270900e88ffeefb2a34233
  7:                     9af4db3ea6321f463cb9b3d9a7b6299206c33009
  8:                     21ed08e2c88193823c0d7f3528fc3677ecb37c71
  9:                     3f77928e59256513a5fbb58368a027494775e58a

$ tpm2_pcrread sha1

sha1:
  0 : 0x38F8FFE07F2099A35A23159EF3793B87F88BBB2F
  1 : 0xBC96CAE524217F397E98C9CC03BD152A87E9A996
  2 : 0xB2A83B0EBF2F8374299A5B2BDFC31EA955AD7236
  3 : 0xB2A83B0EBF2F8374299A5B2BDFC31EA955AD7236
  4 : 0x1096D44B9A24781CB6E26D826E8EF42598AA519B
  5 : 0x0F1424FA0DB70F44C963A949BBF0F51BB42D2FBB
  6 : 0xB2A83B0EBF2F8374299A5B2BDFC31EA955AD7236
  7 : 0x7CDC4B8E595C531F58B93055EAC4F167CDC0207C
  8 : 0x1DA51D1167508D7BE26DD1FBC3FC3549DFB863CE
  9 : 0x4F0B322CB4149541A28155736AEB2B31E061445A
  10: 0xE31FAF71F1AFF619D3D38696B28CDE613CC1F2AA
  11: 0x0000000000000000000000000000000000000000
  12: 0x0000000000000000000000000000000000000000
  13: 0x0000000000000000000000000000000000000000
  14: 0x0000000000000000000000000000000000000000
  15: 0x0000000000000000000000000000000000000000
  16: 0x0000000000000000000000000000000000000000
  17: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  18: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  19: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  20: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  21: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  22: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  23: 0x0000000000000000000000000000000000000000

$ tpm2_pcrread sha256
sha256:
  0 : 0x85749DAD791A4125477BF1454958D4647A95FC41A08219E9387F6546C4121E19
  1 : 0x222475AFA46BC6132DEF7D6B5C07385A1090B4003F8D9AF07553E302C9C3D6AE
  2 : 0x3D458CFE55CC03EA1F443F1562BEEC8DF51C75E14A9FCF9A7234A13F198E7969
  3 : 0x3D458CFE55CC03EA1F443F1562BEEC8DF51C75E14A9FCF9A7234A13F198E7969
  4 : 0xFBFBC1B92747965AABC22E3AF4CA434281CE21E938469C758EE592614BFD0FF5
  5 : 0xCDDEAC8327E6A487B58AC39F8E94941D77F26017C11C43226F3186011C9FA265
  6 : 0x3D458CFE55CC03EA1F443F1562BEEC8DF51C75E14A9FCF9A7234A13F198E7969
  7 : 0x481733ABFDD0FDC38B0ED086EDFDF5FD46ED262053A982505BD2B42824DFF4C4
  8 : 0xE2EE3BDB0445FAB1B006AFD8F57F5DA4FA97B5F6EF1D18332CF5FF6351770CC8
  9 : 0x2C8F38D7834153B7E3E65F545644126050B930302D2EFE1B4BE5027DD0217B4A
  10: 0xA9F128521EE45DE89F5047E7D1019343D1DACFA839AD6D2F141FF0F118C90703
  11: 0x0000000000000000000000000000000000000000000000000000000000000000
  12: 0x0000000000000000000000000000000000000000000000000000000000000000
  13: 0x0000000000000000000000000000000000000000000000000000000000000000
  14: 0x0000000000000000000000000000000000000000000000000000000000000000
  15: 0x0000000000000000000000000000000000000000000000000000000000000000
  16: 0x0000000000000000000000000000000000000000000000000000000000000000
  17: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  18: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  19: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  20: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  21: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  22: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
[... 45 lines cut ...]
 PCR 04: fb fb c1 b9 27 47 96 5a ab c2 2e 3a f4 ca 43 42 81 ce 21 e9 38 46 9c 75 8e e5 92 61 4b fd 0f f5 
 PCR 05: cd de ac 83 27 e6 a4 87 b5 8a c3 9f 8e 94 94 1d 77 f2 60 17 c1 1c 43 22 6f 31 86 01 1c 9f a2 65 
 PCR 06: 3d 45 8c fe 55 cc 03 ea 1f 44 3f 15 62 be ec 8d f5 1c 75 e1 4a 9f cf 9a 72 34 a1 3f 19 8e 79 69 
 PCR 07: 48 17 33 ab fd d0 fd c3 8b 0e d0 86 ed fd f5 fd 46 ed 26 20 53 a9 82 50 5b d2 b4 28 24 df f4 c4 
 PCR 08: e2 ee 3b db 04 45 fa b1 b0 06 af d8 f5 7f 5d a4 fa 97 b5 f6 ef 1d 18 33 2c f5 ff 63 51 77 0c c8 
 PCR 09: 2c 8f 38 d7 83 41 53 b7 e3 e6 5f 54 56 44 12 60 50 b9 30 30 2d 2e fe 1b 4b e5 02 7d d0 21 7b 4a 
 PCR 10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 PCR 11: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 PCR 12: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 PCR 13: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 PCR 14: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 PCR 15: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 PCR 16: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 PCR 17: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 PCR 18: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 PCR 19: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[... cut ...]  00 00 00 
 PCR 21: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 PCR 22: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 PCR 23: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 
boot aggregate: 77 f7 be 4a b5 a2 fd ba 05 39 d7 e1 96 70 59 a8 e3 de 45 05 12 95 fe be 7f 8d 72 ed d6 88 52 a6
```

The SHA1 reconstruction matches my actual PCR values

superm1, 2020-06-15:
Thanks for those pointers to those other tools.  I 

On my XPS 7390:
```
$ sudo tpm2_pcrread
...
sha1:
  0 : 0x5E161271110E18B60B7102C06976A8A256B5D66F
...

$ sudo tpm2_eventlog /sys/kernel/security/tpm0/binary_bios_measurements
...
pcrs:
  sha1:
    0  : 0x5e161271110e18b60b7102c06976a8a256b5d66f
...

$ sudo fwupdtpmevlog
...
Reconstructed PCRs:
  0:                     5223a96a2441937b5e051296b53dc91b45f3a439
...
```

But looking at my system, I don't have the match error.  So that tells me that something is wrong fwupdtpmevlog, but not with fwupd... quite confusing as they are supposed to run the same code.

```
19:39:07:0342 FuMain               System Firmware
  DeviceId:             b6c08fb9e5384d9d101853cc1ca20cf0ce2df2e2
  Guid:                 cbe49cca-492b-93e8-b0f4-f693501f271b
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a <- main-system-firmware
  Guid:                 40d34c18-f6d5-54db-b81d-0aaa600c0540 <- UEFI\RES_{CBE49CCA-492B-93E8-B0F4-F693501F271B}
  Plugin:               uefi
  Protocol:             org.uefi.capsule
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot|can-verify|usable-during-update|md-set-verfmt
  Checksum:             SHA1(5e161271110e18b60b7102c06976a8a256b5d66f)
  Checksum:             SHA256(42116e9a1dea4b6f285584e5f1e2a63bd80a3deb156b0a03d4c46227059402f6)
  Vendor:               Dell Inc.
  VendorId:             DMI:Dell Inc.
  Version:              1.5.1
  VersionLowest:        1.5.1
  VersionFormat:        dell-bios
  VersionRaw:           0x00010501
  VersionLowestRaw:     0x00010501
  Icon:                 computer
  Created:              2020-06-15
```

dnoliver, 2020-06-15:
I have updated my package to the latest version

```shell
$ rpm -qa fwupd
fwupd-1.5.0-0.587.20200605git.fc32.x86_64
```

The PCRs still does not match. I have also updated my OS to the latest version available. Still failing. 

Any hints for further debug?

superm1, 2020-06-15:
That above PR helps the problem for me and possibly other systems with the problem, not sure it will for you though.

superm1, 2020-06-17:
@dnoliver It would be great if you can re-try again with master if possible to confirm there isn't a secondary issue.

hughsie, 2020-06-17:
I'm building a new f32 fwupd in the copr now if that helps. Give it ~5 min to sync.

dnoliver, 2020-06-17:
```shell
[root@compulab-fitlet2-1181004-04904 test]# rpm -qa fwupd
fwupd-1.5.0-0.590.20200617git.fc32.x86_64

[root@compulab-fitlet2-1181004-04904 test]# fwupdmgr --version
client version: 1.5.0
compile-time dependency versions
        gusb:   0.3.4
        efivar: 37
daemon version: 1.5.0

[root@compulab-fitlet2-1181004-04904 test]# fwupdmgr security --force
Host Security ID: HSI:1 (v1.5.0)

HSI-1
✔ MEI manufacturing mode:        Locked
✔ MEI override strap:            Locked
✔ MEI version:                   OK
✔ SPI BIOS region:               Locked
✔ SPI lock:                      Enabled
✔ SPI write:                     Disabled
✔ TPM v2.0:                      Found
✔ UEFI dbx:                      Found

HSI-2
✔ IOMMU:                         Enabled
✘ TPM PCR0 reconstruction:       Invalid

HSI-3
✔ Suspend-to-ram:                Disabled
✘ Intel CET:                     Not supported
✘ Pre-boot DMA protection:       Disabled
✘ Suspend-to-idle:               Disabled

HSI-4
✔ Intel SMAP:                    Enabled
✘ Encrypted RAM:                 Not supported

Runtime Suffix -U
✘ Firmware updates:              Not found

Runtime Suffix -A
✘ Firmware attestation:          Not supported

Runtime Suffix -!
✔ Linux kernel:                  Untainted
✔ Linux kernel lockdown:         Enabled
✔ Linux swap:                    Encrypted
✔ UEFI secure boot:              Enabled
✔ fwupd plugins:                 Untainted

Failed to upload to https://fwupd.org/lvfs/hsireports/upload: Bad Request
```

I still see the PCR0 Reconstruction error with that version.
I have updated the BIOS with a new TXE, so now I have MEI Version OK :)
And I started seeing those `Failed to upload` errors a few days ago, and also the UI Software Manager complained a couple of times that he could not update the `lvfs` repo. 

Running `fwupdtpmevlog` I see this: 

```shell
Reconstructed PCRs:
  0:                     7aaad8a30adc6e5bb85270581637f743778439ad
  0:                     e59d39c535937e03eb6a9713f602651fa5dfebcb80a6b6ffb83c711f44adffa5
  1:                     fd009238c7ea899562cf7fd017fa1ce817f90672
  1:                     ed2587f237577c91d2ff0f1475dca8a106876c34125fcd630f1977c78044eee8
  2:                     b2a83b0ebf2f8374299a5b2bdfc31ea955ad7236
  2:                     3d458cfe55cc03ea1f443f1562beec8df51c75e14a9fcf9a7234a13f198e7969
  3:                     b2a83b0ebf2f8374299a5b2bdfc31ea955ad7236
  3:                     3d458cfe55cc03ea1f443f1562beec8df51c75e14a9fcf9a7234a13f198e7969
[... 48 lines cut ...]
 PCR 04: 84 4a 9f 48 f2 c8 59 96 93 a0 d1 85 42 04 2f d5 77 47 78 b4 cb ab 9c 81 e1 64 57 8f 1d 59 bc 64 
 PCR 05: cd de ac 83 27 e6 a4 87 b5 8a c3 9f 8e 94 94 1d 77 f2 60 17 c1 1c 43 22 6f 31 86 01 1c 9f a2 65 
 PCR 06: 3d 45 8c fe 55 cc 03 ea 1f 44 3f 15 62 be ec 8d f5 1c 75 e1 4a 9f cf 9a 72 34 a1 3f 19 8e 79 69 
 PCR 07: 48 17 33 ab fd d0 fd c3 8b 0e d0 86 ed fd f5 fd 46 ed 26 20 53 a9 82 50 5b d2 b4 28 24 df f4 c4 
 PCR 08: 3e 8e 33 88 15 d4 fe d5 27 d5 f9 e1 6a 05 6f b3 99 cd ab 1b 33 3e 8a 5e 5b db f7 3e d3 94 a0 55 
 PCR 09: 4a e0 d5 5e 8e 8c 51 1a 10 3b ab ce 68 1c ef d4 da 2a 20 dc b9 25 96 b0 17 94 99 74 d3 66 59 89 
 PCR 10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 PCR 11: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 PCR 12: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 PCR 13: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 PCR 14: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 PCR 15: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 PCR 16: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 PCR 17: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 PCR 18: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 PCR 19: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 PCR 20: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 PCR 21: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 PCR 22: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 PCR 23: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 
boot aggregate: 0c 6f d4 59 79 57 c2 8b 38 c3 58 30 f0 8d d9 1f 5d 28 b2 97 13 83 03 7f 04 90 c2 7f cf 54 59 8b
```

The tool is now calculating the PCRs correctly! but still telling me that it cannot reconstruct 0

superm1, 2020-06-17:
Can you show `fwupdmgr get-devices --verbose` now?

dnoliver, 2020-06-17:
```shell
[root@compulab-fitlet2-1181004-04904 test]# fwupdmgr get-devices --verbose
(fwupdmgr:8997): GLib-DEBUG: 10:28:08.624: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:8997): FuMain-DEBUG: 10:28:08.643: TS128GMTS600
  DeviceId:             c46944fe84f2b057692258cc87a812cc86364881
  Guid:                 6001e5d6-33a9-5b6c-9884-b461cdc2cf32 <- IDE\TS128GMTS600____________________________P1225CH1
  Guid:                 920180a6-f62c-564a-b6d2-1fb6d7decd3e <- IDE\0TS128GMTS600____________________________
  Guid:                 e25079dc-149b-5eb7-aee9-c7ebe27e4fd3 <- TS128GMTS600
  Serial:               E473140061
  Summary:              ATA Drive
  Plugin:               ata
  Protocol:             org.t13.ata
  Flags:                internal|updatable|require-ac|registered|needs-reboot|usable-during-update
  Vendor:               Transcend
  VendorId:             ATA:0x8564
  Version:              P1225CH1
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2020-06-17
```

superm1, 2020-06-17:
You're missing the system firmware device (I noticed in this in the original post too). Are you in UEFI mode?

dnoliver, 2020-06-17:
Yes, I am in UEFI mode with Secure Boot enabled (hope the output below confirms it)

```shell
[root@compulab-fitlet2-1181004-04904 test]# dmesg | grep UEFI
[    0.008866] ACPI: UEFI 0x000000007992B0B0 000042 (v01 ALASKA A M I    00000000      00000000)
[    2.032892] integrity: Loading X.509 certificate: UEFI:db
[    2.032970] integrity: Loaded X.509 cert 'Microsoft Corporation UEFI CA 2011: 13adbf4309bd82709c8cd54f316ed522988a1bd4'
[    2.032971] integrity: Loading X.509 certificate: UEFI:db

[root@compulab-fitlet2-1181004-04904 test]# dmesg | grep Secure
[    0.000000] secureboot: Secure boot enabled
[    0.000000] Kernel is locked down from EFI Secure Boot mode; see man kernel_lockdown.7
[    0.008728] secureboot: Secure boot enabled
[    4.379165] sdhci: Secure Digital Host Controller Interface drive
```

hughsie, 2020-06-17:
> Failed to upload to https://fwupd.org/lvfs/hsireports/upload: Bad Request

My bug, sorry. I've fixed that in https://github.com/fwupd/fwupd/commit/87143298cd212e6bf298a488e0eff359b90c1a9d and am rebuilding the COPR. 1.5.0-0.591.20200617git.fc32  should allow you to upload the report now. Please give it 5 mins to build :)

dnoliver, 2020-06-17:
Updated. Sharing the log with you so you can see the `pesign.service` restart failure. Not sure if relevant to this issue (or at all)

```shell
root@compulab-fitlet2-1181004-04904 test]# dnf update fwupd --refresh
Copr repo for fwupd owned by rhughes                                                                                                                    1.6 kB/s | 3.3 kB     00:02    
Copr repo for fwupd owned by rhughes                                                                                                                     17 kB/s |  66 kB     00:03    
Fedora 32 openh264 (From Cisco) - x86_64                                                                                                                1.2 kB/s | 986  B     00:00    
Fedora Modular 32 - x86_64                                                                                                                              8.9 kB/s |  16 kB     00:01    
Fedora Modular 32 - x86_64 - Updates                                                                                                                    7.1 kB/s |  17 kB     00:02    
Fedora 32 - x86_64 - Updates                                                                                                                             26 kB/s |  17 kB     00:00    
Fedora 32 - x86_64                                                                                                                                       25 kB/s |  16 kB     00:00    
Dependencies resolved.
========================================================================================================================================================================================
 Package                     Architecture                 Version                                              Repository                                                          Size
========================================================================================================================================================================================
Upgrading:
 fwupd                       x86_64                       1.5.0-0.591.20200617git.fc32                         copr:copr.fedorainfracloud.org:rhughes:fwupd                       2.8 M

Transaction Summary
========================================================================================================================================================================================
Upgrade  1 Package

Total download size: 2.8 M
Is this ok [y/N]: y
Downloading Packages:
fwupd-1.5.0-0.591.20200617git.fc32.x86_64.rpm                                                                                                           143 kB/s | 2.8 MB     00:20    
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Total                                                                                                                                                   143 kB/s | 2.8 MB     00:20     
Running transaction check
Transaction check succeeded.
Running transaction test
Transaction test succeeded.
Running transaction
  Preparing        :                                                                                                                                                                1/1 
  Running scriptlet: fwupd-1.5.0-0.591.20200617git.fc32.x86_64                                                                                                                      1/1 
  Upgrading        : fwupd-1.5.0-0.591.20200617git.fc32.x86_64                                                                                                                      1/2 
  Running scriptlet: fwupd-1.5.0-0.591.20200617git.fc32.x86_64                                                                                                                      1/2 
  Running scriptlet: fwupd-1.5.0-0.590.20200617git.fc32.x86_64                                                                                                                      2/2 
  Cleanup          : fwupd-1.5.0-0.590.20200617git.fc32.x86_64                                                                                                                      2/2 
  Running scriptlet: fwupd-1.5.0-0.590.20200617git.fc32.x86_64                                                                                                                      2/2 
Failed to try-restart pesign.service: Unit pesign.service not found.

  Verifying        : fwupd-1.5.0-0.591.20200617git.fc32.x86_64                                                                                                                      1/2 
  Verifying        : fwupd-1.5.0-0.590.20200617git.fc32.x86_64                                                                                                                      2/2 

Upgraded:
  fwupd-1.5.0-0.591.20200617git.fc32.x86_64                                                                                                                                             

Complete!
```

Running the tool

```shell
[root@compulab-fitlet2-1181004-04904 test]# fwupdmgr security --force
Host Security ID: HSI:1 (v1.5.0)

HSI-1
✔ MEI manufacturing mode:        Locked
✔ MEI override strap:            Locked
✔ MEI version:                   Valid
✔ SPI BIOS region:               Locked
[... cut ...]
✔ Intel SMAP:                    Enabled
✘ Encrypted RAM:                 Not supported

Runtime Suffix -U
✘ Firmware updates:              Not found

Runtime Suffix -A
✘ Firmware attestation:          Not supported

Runtime Suffix -!
✔ Linux kernel:                  Untainted
✔ Linux kernel lockdown:         Enabled
✔ Linux swap:                    Encrypted
✔ UEFI secure boot:              Enabled
✔ fwupd plugins:                 Untainted

Host Security ID attributes uploaded successfully, thanks!
```

hughsie, 2020-06-17:
> so you can see the pesign.service restart failure

You can ignore this; the COPR doesn't have any signing hardware and so the fwupd.efi file doesn't get signed. On actual Fedora/Red Hat infra the binaries get signed.

hughsie, 2020-06-17:
> [root@compulab-fitlet2

Would it be possible to get access to one of these machines so we can do some local debugging? Or be sent loaner hardware? There's something fishy going on with the fwupd reconstruction still and without comparing the hashes one by one it might be hard going to do remotely.

superm1, 2020-06-17:
I think I found it, see the last commit in the PR.

dnoliver, 2020-06-17:
`The system must support UEFI capsule updates in order to measure this.`

I believe this system supports capsule update, but not yet through fwupdmgr. Is that what you mean?

[image]

superm1, 2020-06-17:
> `The system must support UEFI capsule updates in order to measure this.`
> 
> I believe this system supports capsule update, but not yet through fwupdmgr. Is that what you mean?
> 
> [image]

Well it's definitely not showing up in your devices list.  So it must either be turned off in BIOS setup, or the firmware you're running doesn't have it.

If you share your `fwupdtool --verbose` log we can see what the uefi plugin says, but based on the above symptoms I would say UEFI capsule updates is disabled.

closed 2020-06-17 by superm1

dnoliver, 2020-06-25:
I am having this same problem in an Intel NUC8i7HVK.

```shell
[test@dnoliver-nuc1 ~]$ fwupdmgr --version
client version:	1.5.0
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.5.0
[test@dnoliver-nuc1 ~]$ sudo fwupdmgr security --force
Host Security ID: HSI:0 (v1.5.0)

HSI-1
✔ MEI manufacturing mode:        Locked
✔ MEI override strap:            Locked
✔ SPI BIOS region:               Locked
✔ SPI lock:                      Enabled
✔ SPI write:                     Disabled
✔ TPM v2.0:                      Found
✔ UEFI dbx:                      Found
✘ MEI version:                   Invalid

HSI-2
✘ IOMMU:                         Not found
✘ TPM PCR0 reconstruction:       Invalid

HSI-3
✘ Intel CET:                     Not supported
✘ Pre-boot DMA protection:       Disabled
✘ Suspend-to-idle:               Disabled
✘ Suspend-to-ram:                Enabled

HSI-4
✔ Intel SMAP:                    Enabled
✘ Encrypted RAM:                 Not supported

Runtime Suffix -U
✘ Firmware updates:              Not supported

Runtime Suffix -A
✘ Firmware attestation:          Not supported

Runtime Suffix -!
✔ Linux kernel:                  Untainted
✔ Linux kernel lockdown:         Enabled
✔ Linux swap:                    Encrypted
✔ UEFI secure boot:              Enabled
✔ fwupd plugins:                 Untainted

This system has a low HSI security level.
 » https://github.com/fwupd/fwupd/wiki/Low-host-security-level

Host Security ID attributes uploaded successfully, thanks!
```

Reconstructed PCRs from `fwupdtpmevlog` and `tpm2_pcrread` matches for both sha1 and sha256

```shell
Reconstructed PCRs:
  PCR 0:                 SHA1(c9919f0bd5ff50392c6436b717afbfde258efcdd)
  PCR 0:                 SHA256(1cc90eec440d4c89479455d329f88610513faff2996a39b722156b23e100687d)
  PCR 1:                 SHA1(c943898dfa73f9b34d6b8fb57f92776ba768d491)
  PCR 1:                 SHA256(07a280050ff03409752ecc86d09311607ccd592b78228c0836c24a4fe7dbd051)
  PCR 2:                 SHA1(e1c1e7a9758a3eb8cf70758083e02d4b28fbc31c)
  PCR 2:                 SHA256(85f63d955975fbd1ca4e934b32b93bf93f4d685fcbb43fe580055169b757fd6e)
  PCR 3:                 SHA1(b2a83b0ebf2f8374299a5b2bdfc31ea955ad7236)
  PCR 3:                 SHA256(3d458cfe55cc03ea1f443f1562beec8df51c75e14a9fcf9a7234a13f198e7969)
  PCR 4:                 SHA1(d50027759782370300404e8e43a085fd3c06152b)
  PCR 4:                 SHA256(5889d86ceb7e3cabe86eb1f427f039fc0baf2d3e6e91ba1846c90ae71e7e835b)
  PCR 5:                 SHA1(ea7ac4fe5eb58e2aefab831a566da225ef381b2d)
  PCR 5:                 SHA256(2fdd58246a80b5917aa4e514555c6c53b1c123c5a1d6f440e68cf617dd59ad3a)
  PCR 6:                 SHA1(b2a83b0ebf2f8374299a5b2bdfc31ea955ad7236)
  PCR 6:                 SHA256(3d458cfe55cc03ea1f443f1562beec8df51c75e14a9fcf9a7234a13f198e7969)
  PCR 7:                 SHA1(f1c0600ab4096ee9152e7c926e62ccb9d83e2fed)
  PCR 7:                 SHA256(4d994af4e5962e262650ca24afda7318bf65b9ea13f56a273161a836b48e30ad)
  PCR 8:                 SHA1(b4943abbce96907a3d51928a58ae9d6d28cfca4d)
  PCR 8:                 SHA256(01ec3e955ea11f53cc1c19ca2be1498054f8e477228fb68fa6814e1976a60356)
  PCR 9:                 SHA1(ce02fb8169e96dac3c89ca179d6de9160ac27cd4)
  PCR 9:                 SHA256(5ad7d0d8b9cb5e979af8a7cbad6645df752c90573e23edd15601233ae88b9a67)
  PCR e:                 SHA1(73fe5f7c50d62033642e728efa19e9ba9eadd8e8)
  PCR e:                 SHA256(4ea89fb0cfb6f8c70742b28f635cb014faea722e039d23dcc404ea8fe7b032fa)

[test@dnoliver-nuc1 ~]$ sudo tpm2_pcrread sha1:0+sha256:0
sha1:
  0 : 0xC9919F0BD5FF50392C6436B717AFBFDE258EFCDD
sha256:
  0 : 0x1CC90EEC440D4C89479455D329F88610513FAFF2996A39B722156B23E100687D
```

Hope https://github.com/fwupd/fwupd/commit/f41222741538e936a372c2341431a62b9927ff21 also fixes this. 

And again, I have a vulnerable CSME :(

```shell
[test@dnoliver-nuc1 ~]$ cat /sys/class/mei/mei0/fw_ver 
0:11.8.50.3460
0:11.8.50.3460
0:11.8.50.3434
```

PR #2183 (2020-06-15): tpm-eventlog: fix PCR0 calculation
One of the core problems is that systems with both sha1 and sha256
were miscalculating.

Fixes: #2181

Type of pull request:
- [ ] New plugin (Please include [new plugin checklist](https://github.com/hughsie/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit 5b63015c (2020-06-15): tpm-eventlog: fix PCR0 calculation

One of the core problems is that systems with both sha1 and sha256
were miscalculating.

Fixes: #2181
commit 67b2dca0 (2020-06-15): tpm-eventlog: fix PCR0 calculation

One of the core problems is that systems with both sha1 and sha256
were miscalculating.

Fixes: #2181
commit 57b8b7fe (2020-06-15): tpm-eventlog: fix PCR0 calculation

One of the core problems is that systems with both sha1 and sha256
were miscalculating.

Fixes: #2181