fwupd/fwupd #2184: MEI Version Invalid reported for device without AMT support

dnoliver, 2020-06-15:
**Describe the bug**
A clear and concise description of what the bug is.

My device does not have Intel Management Engine (ME), hence it does not support Intel Active Management Technology (AMT).
Running the `fwupdmgr security --force` command tells me that my HSI is 0 because the MEI Version is invalid.

```shell
$ fwupdmgr security --force
Host Security ID: HSI:0 (v1.5.0)

HSI-1
✔ MEI manufacturing mode:        Locked
✔ MEI override strap:            Locked
✔ SPI BIOS region:               Locked
✔ SPI lock:                      Enabled
✔ SPI write:                     Disabled
✔ TPM v2.0:                      Found
✔ UEFI dbx:                      Found
✘ MEI version:                   Invalid
```

I have compiled and executed the https://github.com/mjg59/mei-amt-check tool to do some debugging, and this is what it tells me:

```shell
$ ./mei-amt-check 
Error: Management Engine refused connection. This probably means you don't have AMT
```

My questions are:
1. Does it make sense to report MEI version on non-AMT enabled devices?
2. Should this check for the Intel Trusted Execution Engine (TXE) version, or Intel Converged Security and Manageability Engine (CSME) version if that was present?
3. I have a `/dev/mei0` device present for some reason. Is that what is causing the MEI version to be inspected?

**Steps to Reproduce**
Steps to reproduce the behavior.

On a device using Intel TXT (or Intel CSME), run the `fwupdmgr security --force` command.
MEI version will be displayed as invalid.

**Expected behavior**
A clear and concise description of what you expected to happen.

I would expect not seeing MEI Version.
Or I would expect seeing the Intel TXT version (which also have some CVEs there to solve)

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
```

```shell
$ fwupdmgr --version
client version: 1.5.0
compile-time dependency versions
        gusb:   0.3.4
        efivar: 37
daemon version: 1.5.0

$ rpm -qa fwupd
fwupd-1.5.0-0.587.20200605git.fc32.x86_64
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

```shell
dnf copr enable rhughes/fwupd
dnf update
dnf update fwupd
```

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
```

```shell
$ fwupdmgr get-devices --show-all-devices
fitlet2
│
├─Celeron N3350/Pentium N4200/Atom E3900 Series Integrated Graphics Controller:
│     Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
│     Current version:     0b
│     Vendor:              Intel Corporation (PCI:0x8086)
│     GUIDs:               7ee0c805-ca7f-51ea-8bbd-e20e623b9323 ← PCI\VEN_8086&DEV_5A84&REV_0B
│                          8257b546-4d2d-5c32-adc5-fc54b8fdfe53 ← PCI\VEN_8086&DEV_5A84
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│   
├─Intel(R) Atom™ Processor E3950 @ 1.60GHz:
│     Device ID:           4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:     0x38
│     Vendor:              GenuineIntel
│     GUID:                b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│     Device Flags:        • Internal device
│   
├─TPM:
│ │   Device ID:           c6a80ac3a22083423992a3cb15018989f37834d6
│ │   Summary:             TPM 2.0 Device
│ │   Current version:     301.9.0.0
│ │   Vendor:              Intel (TPM:INTC)
│ │   GUIDs:               ff71992e-52f7-5eea-94ef-883e56e034c6 ← system-tpm
│ │                        34801700-3a50-5b05-820c-fe14580e4c2d ← TPM\VEN_INTC&DEV_0000
│ │                        8e1cbc5d-5a11-5149-bfea-b6065d5296ba ← TPM\VEN_INTC&MOD_Intel
│ │                        03f304f4-223e-54f4-b2c1-c3cf3b5817c6 ← TPM\VEN_INTC&DEV_0000&VER_2.0
│ │                        52d7b679-db28-5bf7-bd87-41d77aeec600 ← TPM\VEN_INTC&MOD_Intel&VER_2.0
│ │   Device Flags:        • Internal device
│ │ 
│ └─Event Log:
│       Device ID:         58bd405f31c48e6eca290b425f530a94c91e955c
│       GUID:              a25657fe-b5dc-5be0-8b78-8b9dfec678ff ← system-tpm-eventlog
│       Device Flags:      • Internal device
│     
├─TS128GMTS600:
│     Device ID:           c46944fe84f2b057692258cc87a812cc86364881
│     Summary:             ATA Drive
│     Current version:     P1225CH1
│     Vendor:              Transcend (ATA:0x8564)
│     Serial Number:       E473140061
│     GUIDs:               6001e5d6-33a9-5b6c-9884-b461cdc2cf32 ← IDE\TS128GMTS600____________________________P1225CH1
│                          920180a6-f62c-564a-b6d2-1fb6d7decd3e ← IDE\0TS128GMTS600____________________________
│                          e25079dc-149b-5eb7-aee9-c7ebe27e4fd3 ← TS128GMTS600
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─Unifying Receiver:
      Device ID:           5a3b142f3f6762434b567a7321f9db3846c1997a
      Summary:             A miniaturised USB wireless receiver
      Current version:     RQR12.03_B0025
      Bootloader Version:  BOT01.02_B0015
      Vendor:              USB:0x046D
      Install Duration:    30 seconds
      GUIDs:               9d131a0c-a606-580f-8eda-80587250b8d6
                           fcf55bf5-767b-51ce-9c17-f6f538c4ee9f ← HIDRAW\VEN_046D&DEV_C52B&REV_00
                           279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
      Device Flags:        • Updatable
```

**Additional questions**
- Operating system and version: Fedora Server 32
- Have you tried rebooting? Yes
- Is this a regression? Unknown

hughsie, 2020-06-16:
What's the contents of `cat /sys/class/mei/mei0/fw_ver` please?

dnoliver, 2020-06-16:
```shell
$ cat /sys/class/mei/mei0/fw_ver
0:3.1.55.2269
0:3.1.55.2269
0:3.1.55.2270
```

hughsie, 2020-06-16:
> My device does not have Intel Management Engine

Are you sure? That looks like CSME to me. And a vulnerable CSME at that...

dnoliver, 2020-06-16:
The manufactured confirmed that I have a TXE, an insecure one :)

They confirmed that the test is correctly checking for the TXE version, and reporting it as the MEI version. 

On their opinion, for this platform, the MEI test should be named as TXEI test.

hughsie, 2020-06-16:
Ahh, TXE indeed. Never do bitshifts in your head :) Isn't MEI the name for the superset group of (SPS,TXE,ME and CSME)?

dnoliver, 2020-06-16:
According to the expert I could find in this topic, the MEI is the name of the linux interface to communicate with the the engine (which can be, TXE, CSME, ME, etc).

It is true that the MEI is used to communicate with the TXE then.
But I don't think it is the name of the engines superset.

But I don't find any name that could group them all :)

So, if it is not going to say "TXE version" or "CSME version" depending on what that platform is using, I believe that MEI version could be acceptable. The only problem that I have is people thinking that they have a vulnerable ME, when they actually have a vulnerable TXE.

hughsie, 2020-06-16:
@superm1 got any ideas? I guess we could split up `FWUPD_SECURITY_ATTR_ID_MEI_VERSION` into `FWUPD_SECURITY_ATTR_ID_TXT_VERSION`, `FWUPD_SECURITY_ATTR_ID_CSME_VERSION` and all the others but it seems like a lot of work. Got any other ideas about a term for the superset?

superm1, 2020-06-17:
How about MEI reported version?

hughsie, 2020-06-17:
> How about MEI reported version

superm1, 2020-06-17:
How about `Intel platform security core` to get a superset?

dnoliver, 2020-06-17:
Intel Security Coprocessor Version?
Intel Platform Security Coprocessor Version?

closed 2020-07-01 by hughsie

dnoliver, 2020-07-02:
Looks great now!

With TXE:

```shell
[test@compulab-fitlet2-1181004-04904 ~]$ rpm -qa fwupd
fwupd-1.5.0-0.601.20200701git.fc32.x86_64

[test@compulab-fitlet2-1181004-04904 ~]$ fwupdmgr security --force
Host Security ID: HSI:1 (v1.5.0)

HSI-1
✔ SPI BIOS region:               Locked
✔ SPI lock:                      Enabled
✔ SPI write:                     Disabled
✔ TPM v2.0:                      Found
✔ TXE manufacturing mode:        Locked
✔ TXE override strap:            Locked
✔ TXE v0:3.1.75.2351:            Valid
✔ UEFI dbx:                      Found
```

With CSME:

```shell
[test@dnoliver-nuc1 ~]$ fwupdmgr security --force
Host Security ID: HSI:0 (v1.5.0)

HSI-1
✔ CSME manufacturing mode:       Locked
✔ CSME override strap:           Locked
✔ SPI BIOS region:               Locked
✔ SPI lock:                      Enabled
✔ SPI write:                     Disabled
✔ TPM v2.0:                      Found
✔ UEFI dbx:                      Found
✘ CSME v0:11.8.60.3561:          Invalid
```