fwupd/fwupd #2185: Dell WD19 upgrade issue

sitnik, 2020-06-16:
**Describe the bug**
I've tried to install 01.00.14.01 update on WD19TB but it reverts after powering off the dock.

**Steps to Reproduce**
1. Enable lvfs-testing:
```shell
$ sudo fwupdmgr enable-remote lvfs-testing
╔══════════════════════════════════════════════════════════════════════════════╗
║ The LVFS is a free service that operates as an independent legal entity and  ║
║ has no connection with Debian GNU/Linux. Your distributor may not have       ║
║ verified any of the firmware updates for compatibility with your system or   ║
║ connected devices. All firmware is provided only by the original equipment   ║
║ manufacturer.                                                                ║
║                                                                              ║
║ This remote contains firmware which is not embargoed, but is still being     ║
║ tested by the hardware vendor. You should ensure you have a way to manually  ║
║ downgrade the firmware if the firmware update fails.                         ║
║                                                                              ║
║ Enabling this functionality is done at your own risk, which means you have   ║
║ to contact your original equipment manufacturer regarding any problems       ║
║ caused by these updates. Only problems with the update process itself        ║
║ should be filed at https://bugs.debian.org/.                                 ║
║                                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝

Agree and enable the remote? [Y|n]: y
Successfully enabled remote
```
2. Check for available updates:
```shell
$ sudo fwupdmgr get-updates
• Thunderbolt Controller has no available firmware updates
• Thunderbolt controller in Dell dock has the latest available firmware version
• RTS5413 in Dell dock has the latest available firmware version
• RTS5487 in Dell dock has the latest available firmware version
• VMM5331 in Dell dock has the latest available firmware version
• SSDPEKKF512G8 NVMe INTEL 512GB has no available firmware updates
• System Firmware has the latest available firmware version
Latitude 5491
│
├─Package level of Dell dock:
│ │   Device ID:           b5a854df85ab30fff48e4365ec7ad2249c90806e
│ │   Summary:             A representation of dock update status
│ │   Current version:     01.00.14.00
│ │   Vendor:              Dell Inc. (USB:0x413C)
│ │   Install Duration:    5 seconds
│ │   GUID:                8ceeeffd-51b6-580c-9b75-69143227aff8 ← USB\VID_413C&PID_B06E&hub&status
│ │   Device Flags:        • Updatable
│ │                        • Supported on remote server
│ │                        • Device can recover flash failures
│ │                        • Device is usable for the duration of the update
│ │
│ └─WD19/WD19DC/WDTB Device Update:
│       New version:       01.00.14.01
│       Remote ID:         lvfs-testing
│       Summary:           Update package for the WD19 family
│       License:           Proprietary
│       Size:              24 bytes
│       Details:           https://www.dell.com/support/article/sln320770/
│       Vendor:            Dell Inc.
│       Duration:          5 seconds
│       Flags:             is-upgrade
│       Description:       -Addressed rare video/display issues related to isolated combinations of host PCs, discrete GPUs and monitors.
│
│       -Improved the compatibility of the Dock firmware update utility.
│
│       -Updated firmware to comply with Apple certification requirements, which is applicable to the Dell Thunderbolt Dock - WD19TB only. Note that WD19TB is not fully compatible with Apple USB-C hosts.
│
└─WD19TB:
  │   Device ID:           570867f4ddda9915445da59bd85cd0e0e507270e
[... 395 lines cut ...]
Idle…                    [***************************************]
Successfully installed firmware
Upload message: 843b87bf2ce45589a8b00b0799a650a92072bafb did not match any known firmware archive; 843b87bf2ce45589a8b00b0799a650a92072bafb did not match any known firmware archive
Successfully uploaded 1 report
• RTS5413 in Dell dock has the latest available firmware version
• RTS5487 in Dell dock has the latest available firmware version
• VMM5331 in Dell dock has the latest available firmware version
• WD19TB has the latest available firmware version
• SSDPEKKF512G8 NVMe INTEL 512GB has no available firmware updates
• System Firmware has the latest available firmware version

```

Then try to reproduce the issue.  Even if it doesn't reproduce, please attach the
daemon verbose logs collected from the system journal.
```shell
$ journalctl -b -u fwupd.service
```
Log is too long for an issue so I've posted it here: https://gist.github.com/sitnik/34298496fecadb3cb71c19b1ffc589d5

**Additional questions**
- Operating system and version: -- Debian Testing, updated today
- Have you tried unplugging the dock or any peripherals from your machine? -- No
- Have you tried to power cycle the dock from the AC adapter? -- Yes
- Is this a regression? -- Don't know

superm1, 2020-06-16:
When you say you power cycled the dock - did you unplug it from the system or from the wall ac adapter? It needs to be kept plugged into wall ac adapter to complete the update, but unplugged from system.

sitnik, 2020-06-16:
I unplugged it from the ac adapter, but it was connected to laptop whole time.

sitnik, 2020-06-16:
I'll try that.

sitnik, 2020-06-16:
Ok, after unplugging/plugging back laptop to the dock updates don't show up anymore in `fwupdmgr get-updates` output. I even tried to unplug the dock from both laptop and ac adapter and plug it again.

It looks like it's finally updated. All components are showing right versions for the FW update in testing. Thanks.

superm1, 2020-06-16:
Well glad it got updated for you.

I'm curious; what led you to think you should unplug the AC adapter though on the dock?  Was it bad wording somewhere?

superm1, 2020-06-17:
Oh, I think I see why you could have drawn that conclusion...

```
			fu_device_set_update_error (device, "An update is pending "
							    "next time the dock is "
							    "unplugged");
```
https://github.com/fwupd/fwupd/blob/master/plugins/dell-dock/fu-dell-dock-i2c-ec.c#L526

There is nothing specifying unplugged from **host** or **AC adapter**.  It definitely needs to be unplugged from host.  I'll send up a PR to clarify this.

sitnik, 2020-06-17:
I've seen message `Upload message: 843b87bf2ce45589a8b00b0799a650a92072bafb did not match any known firmware archive; 843b87bf2ce45589a8b00b0799a650a92072bafb did not match any known firmware archive` in `fwupdmgr upgrade` output and this worried me. So I've tried the usual 'have you tried turning it off and on again?' thing by unplugging ac adapter from switch.

Nothing in `fwupdmgr upgrade` output states that I need to unplug Thunderbolt cable from the laptop to finish the upgrade. So I decided this is bug and wrote this report.

Thank you and Richard for fwupd. Being able to upgrade firmware on the laptop and dock from Linux still a very good experience comparing to what we had a few years ago.

sitnik, 2020-06-17:
> Oh, I think I see why you could have drawn that conclusion...
>
> ```
> 			fu_device_set_update_error (device, "An update is pending "
> 							    "next time the dock is "
> 							    "unplugged");
> ``` [...]

This doesn't exists in version 1.3.10. Is 1.4 branch is considered stable?

I think we can close the issue now.

superm1, 2020-06-17:
> I've seen message Upload message: 843b87bf2ce45589a8b00b0799a650a92072bafb did not match any known firmware archive; 843b87bf2ce45589a8b00b0799a650a92072bafb did not match any known firmware archive [...]

Ah thanks for sharing this.  It was specifically related to uploading the report that failed.  @hughsie this seems unexpected to me, anything jump out at you?

>Nothing in fwupdmgr upgrade output states that I need to unplug Thunderbolt cable from the laptop to finish the upgrade. So I decided this is bug and wrote this report.

Huh, that's actually not expected either.  You were supposed to see an `UpdateMessage` through this code flow.  That should have told you about the update finishing on unplug.

https://github.com/fwupd/fwupd/blob/master/src/fu-util.c#L1831
https://github.com/fwupd/fwupd/blob/master/src/fu-util.c#L585
https://github.com/fwupd/fwupd/blob/master/src/fu-util.c#L147

>This doesn't exist in version 1.3.10. Is 1.4 branch considered stable?

It should:
https://github.com/fwupd/fwupd/blob/1_3_X/plugins/dell-dock/fu-dell-dock-i2c-ec.c#L537
And yes 1.4.x is stable.

superm1, 2020-06-17:
>Thank you and Richard for fwupd. Being able to upgrade firmware on the laptop and dock from Linux still a very good experience comparing to what we had a few years ago.

Thanks!

superm1, 2020-06-17:
>Huh, that's actually not expected either. You were supposed to see an UpdateMessage through this code flow. That should have told you about the update finishing on unplug.

I think LVFS might have lost the `LVFS::UpdateMessage` key when the XML got rewritten... I checked a 1.0.14 CAB file from LVFS and compared it to one that wasn't yet uploaded to LVFS and notice the key is missing.  This looks like an LVFS bug to me, opened an issue here: https://github.com/fwupd/lvfs-website/issues/662

superm1, 2020-06-17:
> 843b87bf2ce45589a8b00b0799a650a92072bafb did not match any known firmware archive;

So this is coming from LVFS back to the client.  From here:
https://github.com/fwupd/lvfs-website/blob/c358f0d42690c1edcda54f05af9940826d631231/lvfs/reports/routes.py#L200

My suspicion is that it's actually caused by the mucking with metadata to make it "WD19TB" (which we did as a result of https://github.com/fwupd/firmware-dell/issues/28#issuecomment-644178727).  This caused the CAB to be rebuilt, and then the checksum on LVFS to change too.  So you have old file downloaded, try to upload a report for it and LVFS says uhhh what's that, I don't recognize it because it didn't keep a copy of the old checksum around.

hughsie, 2020-06-18:
>  This caused the CAB to be rebuilt, and then the checksum on LVFS to change too

superm1, 2020-06-18:
I think the re-signing thing is pretty rare while in testing/stable, no?

hughsie, 2020-06-18:
> I think the re-signing thing is pretty rare while in testing/stable, no?

Yes, it's something only the admin team can do. I've only done it 3 times every I think. What is more common is:

 * Upload to embargo (->sign)
 * Push to stable
 * eeek, forgot something
 * Push to embargo
 * Fix the update description or add CVEs (->sign)
 * Push to stable

superm1, 2020-06-18:
> an obsolete fwupd version is that really something we want to hold against the newly modified firmware?

Considering that the going back to embargo could mean you add CHIDs or change protocols, I suppose that actually could have a big impact on reports being run on invalid SKUs.

I would rather see the failures for that but not hold it against automatic downgrades.  So maybe a 4th class to `success`/`definite fail`/`known issues`/`possible fail`  Or feed it back into known issues?

bulletmark, 2020-06-23:
Apart from disconnecting the dock from the laptop, is there anything else one needs to do? I have waited 15 mins, tried a power cycle of the dock, but it always shows the following "updates will complete" message:
```
─WD19TB:
│ │   Device ID:           570867f4ddda9915445da59bd85cd0e0e507270e
│ │   Summary:             High performance dock
│ │   Current version:     01.00.00.04
│ │   Minimum Version:     01.00.00.00
│ │   Vendor:              Dell Inc. (USB:0x413C)
│ │   Install Duration:    1 minute
│ │   Serial Number:       4DN5523/3159128927644594
│ │   Update Message:      Updates will complete after the dock is unplugged. The dock will be unavailable for approximately 1 minute.
│ │   GUID:                cd357cf1-40b2-5d87-b8df-bb2dd82774aa ← USB\VID_413C&PID_B06E&hub&embedded
│ │   Device Flags:        • Updatable
│ │                        • Requires AC power
│ │                        • Supported on remote server
│ │                        • Device stages updates
│ │                        • Device can recover flash failures
│ │                        • Device is usable for the duration of the update
```
Did the update again and seem to download ok but same problem no matter how long I wait.

superm1, 2020-06-23:
@bulletmark The dock needs to be unplugged from the machine, not the AC adapter..
If that is what you were doing and it's having problems, please file a separate issue so we can get all your logs and relevant details.

bulletmark, 2020-06-23:
As first per sentence I am unplugging from the laptop. I will raise another issue.

superm1, 2020-06-23:
The several issues in here got resolved, I'm going to close this one.  If you find this issue, and also have a WD19TB upgrade problem, please file a new issue using the template so that we can get all your relevant logs.

closed 2020-06-23 by superm1