fwupd/fwupd #2194: Fedora 32: "Error calling StartServiceByName for org.freedesktop.fwupd: Timeout was reached"

t3kg33k, 2020-06-19:
**Describe the bug**
A clear and concise description of what the bug is.

When logging into Fedora 32 receive a pop-up that states that it failed to update with the following details:
"Error calling StartServiceByName for org.freedesktop.fwupd: Timeout was reached"

**Steps to Reproduce**
Steps to reproduce the behavior.

Log into Fedora

**Expected behavior**
A clear and concise description of what you expected to happen.
Not receive the pop-up error.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
```
client version:	1.4.4
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.4.4

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

dnf

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
```
All Series
│
├─GM204 [GeForce GTX 970]:
│     Device ID:           03281da317dccd2b18de2bd1cc70a782df40ed7e
│     Current version:     a1
│     Vendor:              NVIDIA Corporation (PCI:0x10DE)
│     GUIDs:               c32e752c-53d9-53dd-a7ab-f7b9a72e9b7d
│                          e6857de4-0602-5690-90d1-df576c354a9b
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│   
├─Intel(R) Core™ i5-4690K CPU @ 3.50GHz:
│     Device ID:           4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:     0x28
│     Vendor:              GenuineIntel
│     GUID:                b9a2dd81-159e-5537-a7db-e7101d164d3f
│     Device Flags:        • Internal device
│   
├─Samsung SSD 860 QVO 1TB:
│     Device ID:           602b0a6cc821d155208724f0e22f8d111542b74c
│     Summary:             ATA Drive
│     Current version:     RVQ01B6Q
│     Vendor:              Samsung (ATA:0x144D)
│     GUIDs:               7474bfdf-404b-5a4c-bd38-5c3708a2b027
│                          78b69e45-01fb-566d-aaae-e7aa6ff0431c
│                          8f2db83f-1fb1-59bb-966a-88718689210c
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─System Firmware:
      Device ID:           1d70a31dd3ff0941761550a38436fc990c6c3ff0
      Current version:     11
      Minimum Version:     11
      Vendor:              ASUS (DMI:American Megatrends Inc.)
      GUID:                7039436b-6acf-433b-86a1-368ec2ef7e1f
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Needs a reboot after installation
                           • Cryptographic hash verification is available
                           • Device is usable for the duration of the update

**Additional questions**
- Operating system and version:

uname -a
Linux 5.6.19-300.fc32.x86_64 #1 SMP Wed Jun 17 16:10:48 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

Fedora 32

- Have you tried rebooting?

Yes

- Is this a regression?

No

superm1, 2020-06-19:
Can you share your journal output?

`journalctl -u fwupd.service -b`

t3kg33k, 2020-06-19:
Jun 19 09:21:38 edfed systemd[1]: Starting Firmware update daemon...
Jun 19 09:22:03 edfed fwupd[10208]: 13:22:03:0643 FIXME                failed to allocate dbus proxy object: Error calling StartServiceByName for com.inte>
Jun 19 09:22:03 edfed fwupd[10208]: ERROR:esys:src/tss2-esys/esys_context.c:69:Esys_Initialize() Initialize default tcti. ErrorCode (0x000a000a)
Jun 19 09:22:03 edfed systemd[1]: Started Firmware update daemon.

superm1, 2020-06-19:
Do you have tpm2-abrmd installed?

t3kg33k, 2020-06-19:
Yes

superm1, 2020-06-19:
Try removing it (see https://github.com/fwupd/fwupd/issues/1857)

t3kg33k, 2020-06-19:
I think I tried that in Fedora 31. But I will try again. Thanks.

t3kg33k, 2020-06-19:
So that looks like it worked. I was receiving that error on my desktop and laptop. Removed it from both, rebooted and am no longer seeing the pop-up when I login.
Kudos

superm1, 2020-06-19:
anyone else who encounters this reported issue; please try to remove `tpm2-abrmd`, and if it doesn't help open a new bug to dig into.

closed 2020-06-19 by superm1